openwrt/mt76 #379: MT7628: Clients keep disconnecting. #169

brentonv, 2020-03-27:
Issue #169 remains unresolved.

> I have a HC5661A using MT7628. I've just test the latest commit e2eedc9 and it's very unstable.
I connect my xperia phone to the wifi and it keeps disconnected around every 10 minutes. And the log keeps showing:

> ```
> WPA: group key handshake completed (RSN)

> I google for a solution and I added the followings to /etc/config/wireless

> ```
> option wmm '0'
> option wpa_group_rekey '0'
> option wpa_pair_rekey '0'
> option wpa_master_rekey '0'

> Things become better. But after 18-20 hours, clients disconnect again and the AP disappears from the air. I reference to the log and it shows:

> ```
> hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disconnected due to excessive missing ACKs

> I asked google for help again and added the following line to /etc/config/wireless:

> ```
> option disassoc_low_ack '0'

> After a few days testing, clients also keep disconnected, the log did not show something related to ACKs but:

> ```
> hostapd: wlan0: STA 00:00:00:00:00:02 IEEE 802.11: disassociated due to inactivity
> daemon.info hostapd: wlan0: STA 00:00:00:00:00:02 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
> daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
> ```

> BTW, when clients disconnected, only rebooting recover the AP.

> The full /etc/config/wireless:

> ```
> config wifi-device 'radio0'
>         option type 'mac80211'
>         option channel '11'
>         option hwmode '11g'
>         option path 'platform/10300000.wmac'
>         option htmode 'HT20'
>         option country '00'
>         option legacy_rates '1'
> 
> config wifi-iface 'default_radio0'
>         option device 'radio0'
>         option network 'lan'
>         option mode 'ap'
>         option ssid 'LEDE'
>         option wmm '0'
>         option encryption 'psk-mixed'
>         option key 'keep it secret'
>         option wpa_group_rekey '0'
>         option wpa_pair_rekey '0'
>         option wpa_master_rekey '0'
>         option disassoc_low_ack '0'
> ```

Openwrtfunboy, 2020-03-30:
Same in MT7603|MT7621

brentonv, 2020-03-31:
Same in MT7628AN

NatanNMB15, 2020-05-10:
Same in Wifi SoC Atheros AR9132 (Router TL-WR1043ND V1.8).

db260179, 2020-11-22:
https://bugs.openwrt.org/index.php?do=details&task_id=3275

Remove the option country and let the driver decide, some unknown bug

brentonv, 2020-11-25:
@db260179 sorry mate deleting config `option country` doesn't fix anything. This crontab is the only thing I've found to keep the device reliable  `0 */1 * * * ifdown wwan && ifup wwan`

Areopagitics, 2022-01-08:
I'm using [https://openwrt.org/toh/hwdata/tp-link/tp-link_tl-wr841n_v13](url) and the wifi is still unstable. I found that by disabling WMM it hasn't disconnected for a few days, but now my mbits/s have dropped to about half.

It would be nice to get this fixed.

thmang82, 2022-08-16:
Has this ever been fixed? 

Another user and I have a very similar issue: https://github.com/openwrt/openwrt/issues/9555
Our MT76 based devices randomly "loose" some clients after a few hours. The clients think they are connected, but you cannot ping them. They just get stuck and remain unreachable :-(

This is with 21.02.3, TP-Link TL-WR802N v4 and Cudy WR2100