openwrt/mt76 #380: mt7615 really brocken in sta mode

bricco1981, 2020-03-29:
hallo my upload is 30 mbps but with mt7615 is only 4 mbps.
this is really bad openwrt is the best, but  this radio is really broke

dword1511, 2020-04-10:
Same here with MT7612U in STA mode (but not sure if it is caused by the same thing). Using the version in `Linux 5.4.0-4-amd64 #1 SMP Debian 5.4.19-1` on a Bay Trail machine (but the same has been observed on multiple different x64 machines).
Download with `iperf3` TCP:
```
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   323 MBytes   271 Mbits/sec  390             sender
[  5]   0.00-10.01  sec   320 MBytes   268 Mbits/sec                  receiver
```
Download with `iperf3`UDP:
```
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-10.00  sec   596 MBytes   500 Mbits/sec  0.000 ms  0/431596 (0%)  sender
[  5]   0.00-10.05  sec   423 MBytes   353 Mbits/sec  0.065 ms  117173/423441 (28%)  receiver
```
Upload with `iperf3` TCP:
```
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  2.87 MBytes  2.41 Mbits/sec  954             sender
[  5]   0.00-10.00  sec  2.80 MBytes  2.35 Mbits/sec                  receiver
```
Upload with `iperf3`UDP:
```
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-10.01  sec   107 MBytes  90.0 Mbits/sec  0.000 ms  0/77815 (0%)  sender
[  5]   0.00-10.00  sec  56.9 MBytes  47.7 Mbits/sec  0.105 ms  35418/76595 (46%)  receiver
```
From the last test, it seems to me the driver is not dispatching packets to be TX'ed fast enough (set `-b 500M`, got 90 Mbps w/o any packet loss on sender side)

dword1511, 2020-04-10:
Actually, the version in `backports-5.5.11-1` seems to have some improvements, although the speed is still not very stable.
Upload with `iperf3` UDP:
```
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-10.00  sec   235 MBytes   197 Mbits/sec  0.000 ms  0/170231 (0%)  sender
[  5]   0.00-10.00  sec   215 MBytes   180 Mbits/sec  0.069 ms  14498/170154 (8.5%)  receiver
```
TCP upload performance is largely the same:
```
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  2.64 MBytes  2.21 Mbits/sec  812             sender
[  5]   0.00-10.00  sec  2.58 MBytes  2.16 Mbits/sec                  receiver
```

bricco1981, 2020-04-22:
looks much better on kernel 5.4 in general, but still same problem with upload, i should get 25/30 mbps, and i only get 5 mbps.

edmundwatson, 2020-06-03:
I have similar finding on a Wavlink WL-WN575A3  OpenWrt 19.07.2 r10947-65030d81f3. I get more results if you need