openwrt/mt76 #383: Support of MT7615D available?

jneuhauser, 2020-03-30:
Hello @nbd168,

firstly thanks for your efforts to this great open source driver.
I want to switch my apu2 board from qualcomm wireless to mt76 based minipcie cards.

Is the mt7615d hw configuration (2x2 5 ghz ac and 2x2 2.4 ghz n) supported by this driver?
Here is one miniPCIe board from UniElec:
http://unielecinc.com/q/news/cn/m/product/detail.html?qd_guid=nhefShstqf

Many thanks,
Johann

mrkiko, 2020-03-30:
DBDC is supported but is experimental.

Personal curiosity: why the switch?

thanks!

Enrico

jneuhauser, 2020-03-31:
Thank you for the info...

Because is dislike qualcomm ;-)
And because a friend with a apu2c4 would by my used ones, but i need a replacement.

qiuzi, 2020-05-02:
@mrkiko How to open the experimental DBDC?

closed 2020-05-02 by jneuhauser