openwrt/mt76 #384: MT7621 | MT7603EN: Router rebooting and clients disconnecting. iPhone is evil.

Openwrtfunboy, 2020-03-30:
When about 8 devices at different distances are connected to the router, but several devices are close to it(2-4), it reboots and / or disconnects clients. I noticed that the router reboots and disconnects clients only when there is an iPhone device next to the router. For example, this is an iPhone 6 plus.

I had to block the user iPhone 6 and only after that the router stopped rebooting.

Openwrtfunboy, 2020-03-30:
config wifi-iface 'wifinet3'
        option encryption 'psk2+ccmp'
        option device 'radio0'
        option disassoc_low_ack '0'
        option key 'password'
        option mode 'ap'
        option network 'lan'
        option wpa_disable_eapol_key_retries '1'
        option ssid 'NameWifi'

My config

Openwrtfunboy, 2020-03-30:
To block i add this lines to /etc/config/wireless
        option macfilter 'deny'
        list maclist 'XX:XX:XX:XX:XX:XX'

lukasz1992, 2020-03-31:
If you reproduce the problem and after reboot do:

> cat /sys/kernel/debug/crashlog

does it show anything?

Openwrtfunboy, 2020-03-31:
> If you reproduce the problem and after reboot do:
> 
> > cat /sys/kernel/debug/crashlog
> 
> does it show anything?

Yes

root@OpenWrt:~# cat /sys/kernel/debug/crashlog
Time: 1585680762.850259
Modules:	pppoe@8ec24000+23e0	ppp_async@8f7d0000+1b90	pppox@8ec4c000+4b0	ppp_generic@8f140000+5970	nf_conntrack_ipv6@8ec7a000+1590	mt76x2e@8ec84000+26c0	mt76x2_common@8ec7c000+2a40	mt76x02_lib@8f040000+9640	mt7603e@8ecb0000+90a0	mt76@8eca0000+7920	mac80211@8ed00000+6c840	iptable_nat@8efa9000+2c0	ipt_REJECT@8ef12000+3c0	ipt_MASQUERADE@8ee87000+2a0	cfg80211@8ecc0000+36b80	xt_time@8ec91000+6e0	xt_tcpudp@8ec8f000+700	xt_tcpmss@8ec8c000+420	xt_string@8ec89000+2c0	xt_statistic@8ec79000+340	xt_state@8ec77000+2c0	xt_recent@8ec82000+1bf0	xt_quota@8ec72000+300	xt_pkttype@8ec6f000+240	xt_owner@8ec6d000+360	xt_nat@8ec5e000+600	xt_multiport@8ec5b000+500	xt_mark@8ec59000+2a0	xt_mac@8ec57000+260	xt_limit@8ec53000+4a0	xt_length@8ec51000+2c0	xt_hl@8ec3f000+320	xt_helper@8ec3c000+360	xt_ecn@8ec3a000+540	xt_dscp@8ec37000+3e0	xt_conntrack@8ec34000+900	xt_connmark@8ec23000+4e0	xt_connlimit@8ec2e000+1510	xt_connbytes@8ec1e000+640	xt_comment@8ec1c000+1e0	xt_bpf@8ec19000+520	xt_addrtype@8f7c7000+940xt_TCPMSS@8f78d000+ae0	xt_REDIRECT@8ec11000+2c0	xt_NFQUEUE@8ec0f000+d00xt_LOG@8ec0c000+300	xt_HL@8ec0a000+540	xt_FLOWOFFLOAD@8ec07000+c10	xt_DSCP@8f782000+5e0	xt_CT@8f780000+a60	xt_CLASSIFY@8ee7e000+240	ts_kmp@8ee78000+520	ts_fsm@8f714000+ac0	ts_bm@8efe8000+580	slhc@8f7f8000+13c0	nfnetlink_queue@8f7f4000+2840	nf_reject_ipv4@8ee9f000+840	nf_nat_redirect@8f6b6000+580	nf_nat_masquerade_ipv4@8ee3f000+650	nf_conntrack_ipv4@8f7c4000+1420	nf_nat_ipv4@8f7bf000+f30	nf_nat@8f7e0000+37d0	nf_log_ipv4@8ef9f000+d60	nf_flow_table_hw@8ef19000+8d0	nf_flow_table@8f784000+3350	nf_defrag_ipv6@8f79c000+12e0	nf_defrag_ipv4@8eeb9000+440	nf_conntrack_rtcache@8f77f000+aa0	iptable_raw@8ef49000+260	iptable_mangle@8f7b3000+380	iptable_filter@8f7b5000+280	ipt_ECN@8f7b8000+580	ip_tables@8eff4000+2b30	crc_ccitt@8ee7a000+3e0	compat@8efce000+1740	sch_cake@8f7d8000+74d0	nf_conntrack@8f7a0000+10550	sch_tbf@8ee14000+1700	sch_ingress@8ee7c000+4c0	sch_htb@8effc000+36d0	sch_hfsc@8efd8000+3a00	em_u32@8ee67000+220	cls_u32@8f798000+22d0	cls_tcindex@8efca000+1500	cls_route@8efd2000+1470	cls_matchall@8efea000+9e0	cls_fw@8ee1d000+f20	cls_flow@8f778000+1680	cls_basic@8efe0000+c40	act_skbedit@8efe4000+9f0	act_mirred@8ee07000+f30	ledtrig_usbport@8ee1f000+ab0	xt_set@8eff8000+1b20	ip_set_list_set@8ee12000+1c80	ip_set_hash_netportnet@8ee68000+6c60	ip_set_hash_netport@8ee60000+61e0	ip_set_hash_netnet@8ee58000+6780	ip_set_hash_netiface@8ee50000+62c0	ip_set_hash_net@8ee48000+5c00	ip_set_hash_mac@8f738000+2920	ip_set_hash_ipportnet@8ee40000+6740	ip_set_hash_ipportip@8ee38000+54a0	ip_set_hash_ipport@8ee30000+5140	ip_set_hash_ipmark@8ee20000+4e60	ip_set_hash_ip@8ee18000+4d80	ip_set_bitmap_port@8ee10000+1800	ip_set_bitmap_ipmac@8ee0c000+1ba0	ip_set_bitmap_ip@8ee08000+1b00	ip_set@8ee00000+5690	nfnetlink@8f6fe000+1110	nf_log_ipv6@8eff2000+1020	nf_log_common@8efe2000+aa0	ip6table_mangle@8f731000+4a0	ip6table_filter@8f72e000+280	ip6_tables@8efd4000+2a20	ip6t_REJECT@8f707000+400	x_tables@8efc4000+3290	nf_reject_ipv6@8f705000+9e0	ifb@8f708000+ce0	leds_gpio@8f600000+b60	xhci_plat_hcd@8f66c000+15c0	xhci_pci@8f664000+bc0	xhci_mtk@8f616000+1460	xhci_hcd@8f640000+160f0	gpio_button_hotplug@8f60a000+1ac0	usbcore@8f620000+1fd50	nls_base@8f5fc000+1430	usb_common@8f5f1000+8c0
<6>[    2.500887] spi-mt7621 1e000b00.spi: sys_freq: 220000000
<6>[    2.516090] m25p80 spi0.0: w25q256 (32768 Kbytes)
<5>[    2.520838] 4 fixed-partitions partitions found on MTD device spi0.0
<5>[    2.527195] Creating 4 MTD partitions on "spi0.0":
<5>[    2.531974] 0x000000000000-0x000000030000 : "u-boot"
<5>[    2.538155] 0x000000030000-0x000000040000 : "u-boot-env"
<5>[    2.544640] 0x000000040000-0x000000050000 : "factory"
<5>[    2.550815] 0x000000050000-0x000002000000 : "firmware"
<5>[    2.557327] 2 uimage-fw partitions found on MTD device firmware
<5>[    2.563234] Creating 2 MTD partitions on "firmware":
<5>[    2.568223] 0x000000000000-0x0000001c9664 : "kernel"
<5>[    2.574277] 0x0000001c9664-0x000001fb0000 : "rootfs"
<5>[    2.580340] mtd: device 5 (rootfs) set to be root filesystem
<5>[    2.586181] 1 squashfs-split partitions found on MTD device rootfs
<5>[    2.592347] 0x000000450000-0x000001fb0000 : "rootfs_data"
<6>[    2.599670] libphy: Fixed MDIO Bus: probed
<6>[    2.666580] libphy: mdio: probed
<6>[    4.069401] mtk_soc_eth 1e100000.ethernet: loaded mt7530 driver
<6>[    4.076038] mtk_soc_eth 1e100000.ethernet eth0: mediatek frame engine at 0xbe100000, irq 21
<6>[    4.086921] NET: Registered protocol family 10
<6>[    4.092986] Segment Routing with IPv6
<6>[    4.096789] NET: Registered protocol family 17
<6>[    4.101293] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
<6>[    4.114201] 8021q: 802.1Q VLAN Support v1.8
<6>[    4.121056] hctosys: unable to open rtc device (rtc0)
<6>[    4.135333] VFS: Mounted root (squashfs filesystem) readonly on device 31:5.
<6>[    4.147233] Freeing unused kernel memory: 1260K
<4>[    4.151765] This architecture does not have kernel memory protection.
<14>[    4.970724] init: Console is alive
<14>[    4.974383] init: - watchdog -
<14>[    5.999630] kmodloader: loading kernel modules from /etc/modules-boot.d/*
<6>[    6.175870] mtk_soc_eth 1e100000.ethernet eth0: port 4 link up
[... cut ...]
<4>[ 1902.326311] [<8eca1060>] 0x8eca1060 [mt76@8eca0000+0x7920]
<4>[ 1902.331776] [<80360ec0>] 0x80360ec0
<4>[ 1902.335268] [<803637c0>] 0x803637c0
<4>[ 1902.338747] [<804a2878>] 0x804a2878
<4>[ 1902.342218] [<80032f84>] 0x80032f84
<4>[ 1902.345692] [<80259ed0>] 0x80259ed0
<4>[ 1902.349165] [<80007488>] 0x80007488
<4>[ 1902.352632] 
<4>[ 1902.354115] Code: 00000000  8e020014  00e23821 <8ce20000> 10000009  cc400000  1040ffbd  00000000  8e060010 
<4>[ 1902.363845] 
<4>[ 1902.365616] ---[ end trace 15a5dae4a888264c ]---

Openwrtfunboy, 2020-03-31:
Another one
root@OpenWrt:~# cat /sys/kernel/debug/crashlog
Time: 1585692875.534090
Modules:	pppoe@8f7ec000+23e0	ppp_async@8ec14000+1b90	pppox@8ec44000+4b0	ppp_generic@8f0b0000+5970	nf_conntrack_ipv6@8ec5a000+1590	mt76x2e@8ec50000+26c0	mt76x2_common@8ee68000+2a40	mt76x02_lib@8edb0000+9640	mt7603e@8ecc0000+90a0	mt76@8ec70000+7920	mac80211@8ed00000+6c840	iptable_nat@8f690000+2c0	ipt_REJECT@8f6ae000+3c0	ipt_MASQUERADE@8f685000+2a0	cfg80211@8ec80000+36b80	xt_time@8ec58000+6e0	xt_tcpudp@8ec46000+700	xt_tcpmss@8ec43000+420	xt_string@8ec40000+2c0	xt_statistic@8ec3e000+340	xt_state@8ec3c000+2c0	xt_recent@8ec4a000+1bf0	xt_quota@8ee6f000+300	xt_pkttype@8ee6c000+240	xt_owner@8ec02000+360	xt_nat@8ec0b000+600	xt_multiport@8ec08000+500	xt_mark@8f7fe000+2a0	xt_mac@8f679000+260	xt_limit@8f523000+4a0	xt_length@8ef27000+2c0	xt_hl@8ef1a000+320	xt_helper@8ef88000+360	xt_ecn@8eea2000+540	xt_dscp@8ef99000+3e0	xt_conntrack@8f74d000+900	xt_connmark@8f76f000+4e0	xt_connlimit@8f78e000+1510	xt_connbytes@8ee07000+640	xt_comment@8f76e000+1e0	xt_bpf@8eea6000+520	xt_addrtype@8eea4000+940xt_TCPMSS@8ef7f000+ae0	xt_REDIRECT@8ee3b000+2c0	xt_NFQUEUE@8f6ad000+d00xt_LOG@8f6b5000+300	xt_HL@8efdc000+540	xt_FLOWOFFLOAD@8efa7000+c10	xt_DSCP@8ee3d000+5e0	xt_CT@8f77e000+a60	xt_CLASSIFY@8f70c000+240	ts_kmp@8ef8b000+520	ts_fsm@8eff5000+ac0	ts_bm@8ee0e000+580	slhc@8f7a2000+13c0	nfnetlink_queue@8f7a4000+2840	nf_reject_ipv4@8ee50000+840	nf_nat_redirect@8f715000+580	nf_nat_masquerade_ipv4@8ee7e000+650	nf_conntrack_ipv4@8f7e8000+1420	nf_nat_ipv4@8ef2c000+f30	nf_nat@8ec10000+37d0	nf_log_ipv4@8f73a000+d60	nf_flow_table_hw@8f738000+8d0	nf_flow_table@8ec0c000+3350	nf_defrag_ipv6@8f796000+12e0	nf_defrag_ipv4@8eff6000+440	nf_conntrack_rtcache@8ef38000+aa0	iptable_raw@8f61f000+260	iptable_mangle@8efa8000+380	iptable_filter@8f7d4000+280	ipt_ECN@8f7d6000+580	ip_tables@8f7f0000+2b30	crc_ccitt@8f7d8000+3e0	compat@8f6d4000+1740	sch_cake@8f7e0000+74d0	nf_conntrack@8f7c0000+10550	sch_tbf@8f762000+1700	sch_ingress@8efc1000+4c0	sch_htb@8f79c000+36d0	sch_hfsc@8f740000+3a00	em_u32@8ee2f000+220	cls_u32@8f790000+22d0	cls_tcindex@8f69c000+1500	cls_route@8f766000+1470	cls_matchall@8f77a000+9e0	cls_fw@8eff4000+f20	cls_flow@8f696000+1680	cls_basic@8ef4e000+c40	act_skbedit@8f75e000+9f0	act_mirred@8eeb4000+f30	ledtrig_usbport@8f770000+ab0	xt_set@8ee64000+1b20	ip_set_list_set@8ee60000+1c80	ip_set_hash_netportnet@8ee58000+6c60	ip_set_hash_netport@8ee48000+61e0	ip_set_hash_netnet@8ee28000+6780	ip_set_hash_netiface@8ee40000+62c0	ip_set_hash_net@8ee08000+5c00	ip_set_hash_mac@8ee38000+2920	ip_set_hash_ipportnet@8ee30000+6740	ip_set_hash_ipportip@8ee20000+54a0	ip_set_hash_ipport@8ee18000+5140	ip_set_hash_ipmark@8ee10000+4e60	ip_set_hash_ip@8ee00000+4d80	ip_set_bitmap_port@8f776000+1800	ip_set_bitmap_ipmac@8f772000+1ba0	ip_set_bitmap_ip@8f764000+1b00	ip_set@8f768000+5690	nfnetlink@8f75c000+1110	nf_log_ipv6@8f700000+1020	nf_log_common@8eea7000+aa0	ip6table_mangle@8eea5000+4a0	ip6table_filter@8f68c000+280	ip6_tables@8f744000+2a20	ip6t_REJECT@8eea3000+400	x_tables@8f71c000+3290	nf_reject_ipv6@8f6cc000+9e0	ifb@8efbc000+ce0	leds_gpio@8f61a000+b60	xhci_plat_hcd@8f686000+15c0	xhci_pci@8f616000+bc0	xhci_mtk@8f64c000+1460	xhci_hcd@8f660000+160f0	gpio_button_hotplug@8f640000+1ac0	usbcore@8f620000+1fd50	nls_base@8f612000+1430	usb_common@8f5fe000+8c0
<6>[    2.472466] console [ttyS0] enabled
<6>[    2.479339] bootconsole [early0] disabled
<4>[    2.490052] MediaTek Nand driver init, version v2.1 Fix AHB virt2phys error
<6>[    2.497488] spi-mt7621 1e000b00.spi: sys_freq: 220000000
<6>[    2.512582] m25p80 spi0.0: w25q256 (32768 Kbytes)
<5>[    2.517359] 4 fixed-partitions partitions found on MTD device spi0.0
<5>[    2.523680] Creating 4 MTD partitions on "spi0.0":
<5>[    2.528477] 0x000000000000-0x000000030000 : "u-boot"
<5>[    2.534700] 0x000000030000-0x000000040000 : "u-boot-env"
<5>[    2.541103] 0x000000040000-0x000000050000 : "factory"
<5>[    2.547282] 0x000000050000-0x000002000000 : "firmware"
<5>[    2.553739] 2 uimage-fw partitions found on MTD device firmware
<5>[    2.559691] Creating 2 MTD partitions on "firmware":
<5>[    2.564670] 0x000000000000-0x0000001c9664 : "kernel"
<5>[    2.570780] 0x0000001c9664-0x000001fb0000 : "rootfs"
<5>[    2.576864] mtd: device 5 (rootfs) set to be root filesystem
<5>[    2.582666] 1 squashfs-split partitions found on MTD device rootfs
<5>[    2.588901] 0x000000450000-0x000001fb0000 : "rootfs_data"
<6>[    2.596153] libphy: Fixed MDIO Bus: probed
<6>[    2.665931] libphy: mdio: probed
<6>[    4.068790] mtk_soc_eth 1e100000.ethernet: loaded mt7530 driver
<6>[    4.075505] mtk_soc_eth 1e100000.ethernet eth0: mediatek frame engine at 0xbe100000, irq 21
<6>[    4.086674] NET: Registered protocol family 10
<6>[    4.092755] Segment Routing with IPv6
<6>[    4.096542] NET: Registered protocol family 17
<6>[    4.101045] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
<6>[    4.113958] 8021q: 802.1Q VLAN Support v1.8
<6>[    4.120802] hctosys: unable to open rtc device (rtc0)
<6>[    4.135075] VFS: Mounted root (squashfs filesystem) readonly on device 31:5.
<6>[    4.146975] Freeing unused kernel memory: 1260K
<4>[    4.151507] This architecture does not have kernel memory protection.
<14>[    4.972014] init: Console is alive
<14>[    4.975745] init: - watchdog -
<14>[    6.000983] kmodloader: loading kernel modules from /etc/modules-boot.d/*
[... cut ...]
<4>[12097.524904] [<803637c0>] 0x803637c0
<4>[12097.528375] [<80078240>] 0x80078240
<4>[12097.531868] [<804a2878>] 0x804a2878
<4>[12097.535351] [<80032f84>] 0x80032f84
<4>[12097.538824] [<80259ed0>] 0x80259ed0
<4>[12097.542305] [<80007488>] 0x80007488
<4>[12097.545781] 
<4>[12097.547261] Code: 00000000  8e020014  02421021 <8c470000> 41666000  30c60001  000000c0  8f85000c  8e020000 
<4>[12097.556987] 
<4>[12097.558666] ---[ end trace e849372e155057aa ]---

===================================
Time: 1585692875.542426

Openwrtfunboy, 2020-04-02:
Another
root@OpenWrt:~# cat /sys/kernel/debug/crashlog
Time: 1585770206.42749
Modules:	pppoe@8ee2c000+23e0	ppp_async@8ee04000+1b90	pppox@8f01e000+4b0	ppp_generic@8f018000+5970	nf_conntrack_ipv6@8f7ae000+1590	mt76x2e@8ee14000+26c0	mt76x2_common@8f798000+2a40	mt76x02_lib@8ed80000+9640	mt7603e@8f7e0000+90a0	mt76@8f7b0000+7920	mac80211@8ec80000+6c840	iptable_nat@8ee97000+2c0	ipt_REJECT@8ef35000+3c0	ipt_MASQUERADE@8ef8e000+2a0	cfg80211@8ec00000+36b80	xt_time@8f7b8000+6e0	xt_tcpudp@8ef96000+700	xt_tcpmss@8ee7f000+420	xt_statistic@8ee7d000+340	xt_state@8ee13000+2c0	xt_recent@8f750000+1bf0	xt_nat@8f709000+600	xt_multiport@8eeab000+500	xt_mark@8ee70000+2a0	xt_mac@8f7ad000+260	xt_limit@8f7ab000+4a0	xt_length@8f7a9000+2c0	xt_hl@8f7a7000+320	xt_helper@8f7a4000+360	xt_ecn@8f7a1000+540	xt_dscp@8f79f000+3e0	xt_conntrack@8f78d000+900	xt_connmark@8f789000+4e0xt_connlimit@8f796000+1510	xt_connbytes@8f785000+640	xt_comment@8f783000+1e0	xt_TCPMSS@8f781000+ae0	xt_REDIRECT@8ee36000+2c0	xt_LOG@8ee89000+300	xt_HL@8eea6000+540	xt_FLOWOFFLOAD@8ee34000+c10	xt_DSCP@8ee3e000+5e0	xt_CT@8efb0000+a60	xt_CLASSIFY@8efb5000+240	slhc@8f70e000+13c0	nf_reject_ipv4@8efc5000+840	nf_nat_redirect@8efb1000+580	nf_nat_masquerade_ipv4@8f6f4000+650	nf_conntrack_ipv4@8ee30000+1420	nf_nat_ipv4@8ee62000+f30	nf_nat@8f740000+37d0	nf_log_ipv4@8ee06000+d60	nf_flow_table_hw@8ee66000+8d0	nf_flow_table@8f6d4000+3350	nf_defrag_ipv6@8ee0e000+12e0	nf_defrag_ipv4@8f733000+440	nf_conntrack_rtcache@8f735000+aa0	iptable_raw@8f738000+260	iptable_mangle@8effa000+380	iptable_filter@8ee1c000+280	ipt_ECN@8ee23000+580	ip_tables@8eff0000+2b30	crc_ccitt@8ee5f000+3e0	compat@8ee0c000+1740	sch_cake@8f6f8000+74d0	nf_conntrack@8f720000+10550	sch_tbf@8eeac000+1700	sch_ingress@8ee60000+4c0	sch_htb@8f704000+36d0	sch_hfsc@8f700000+3a00	em_u32@8ee5e000+220	cls_u32@8ee74000+22d0	cls_tcindex@8ee6e000+1500	cls_route@8ee6a000+1470	cls_matchall@8ee58000+9e0	cls_fw@8ee25000+f20	cls_flow@8ee4a000+1680	cls_basic@8ee21000+c40	act_skbedit@8ee1f000+9f0	act_mirred@8ee1b000+f30	ledtrig_usbport@8ee03000+ab0	nf_log_ipv6@8ee18000+1020	nf_log_common@8eff9000+aa0	ip6table_mangle@8efea000+4a0	ip6table_filter@8efa9000+280	ip6_tables@8ee08000+2a20	ip6t_REJECT@8efac000+400	x_tables@8ef90000+3290	nf_reject_ipv6@8efe9000+9e0	ifb@8eff7000+ce0	leds_gpio@8f5f1000+b60	xhci_plat_hcd@8f5fe000+15c0	xhci_pci@8fe83000+bc0	xhci_mtk@8f624000+1460	xhci_hcd@8f640000+160f0	gpio_button_hotplug@8f5f8000+1ac0	usbcore@8f600000+1fd50	nls_base@8f5ea000+1430	usb_common@8f513000+8c0
<6>[    2.088765] pci 0000:00:01.0: PCI bridge to [bus 02]
<6>[    2.093699] pci 0000:00:01.0:   bridge window [mem 0x60200000-0x602fffff]
<6>[    2.101918] clocksource: Switched to clocksource GIC
<6>[    2.108493] NET: Registered protocol family 2
<6>[    2.113770] TCP established hash table entries: 2048 (order: 1, 8192 bytes)
<6>[    2.120661] TCP bind hash table entries: 2048 (order: 2, 16384 bytes)
<6>[    2.127105] TCP: Hash tables configured (established 2048 bind 2048)
<6>[    2.133553] UDP hash table entries: 256 (order: 1, 8192 bytes)
<6>[    2.139318] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes)
<6>[    2.145811] NET: Registered protocol family 1
<7>[    2.150135] PCI: CLS 0 bytes, default 32
<6>[    2.381855] 4 CPUs re-calibrate udelay(lpj = 2924544)
<4>[    2.388386] Crashlog allocated RAM at address 0x3f00000
<6>[    2.394033] workingset: timestamp_bits=30 max_order=17 bucket_order=0
<6>[    2.408122] squashfs: version 4.0 (2009/01/31) Phillip Lougher
<6>[    2.413923] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
<5>[    2.425015] random: fast init done
<6>[    2.430683] bounce: pool size: 64 pages
<6>[    2.434494] io scheduler noop registered
<6>[    2.438319] io scheduler deadline registered (default)
<6>[    2.444121] gpio-export gpio_export: 1 gpio(s) exported
<6>[    2.450074] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
<6>[    2.459591] console [ttyS0] disabled
<6>[    2.463260] 1e000c00.uartlite: ttyS0 at MMIO 0x1e000c00 (irq = 19, base_baud = 3125000) is a 16550A
<6>[    2.472244] console [ttyS0] enabled
<6>[    2.479098] bootconsole [early0] disabled
<4>[    2.489524] MediaTek Nand driver init, version v2.1 Fix AHB virt2phys error
<6>[    2.496938] spi-mt7621 1e000b00.spi: sys_freq: 220000000
<6>[    2.512131] m25p80 spi0.0: w25q256 (32768 Kbytes)
<5>[    2.516878] 4 fixed-partitions partitions found on MTD device spi0.0
<5>[    2.523235] Creating 4 MTD partitions on "spi0.0":
<5>[    2.528014] 0x000000000000-0x000000030000 : "u-boot"
<5>[    2.534005] 0x000000030000-0x000000040000 : "u-boot-env"
<5>[    2.540213] 0x000000040000-0x000000050000 : "factory"
<5>[    2.546225] 0x000000050000-0x000002000000 : "firmware"
<5>[    2.552514] 2 uimage-fw partitions found on MTD device firmware
<5>[    2.558419] Creating 2 MTD partitions on "firmware":
<5>[    2.563397] 0x000000000000-0x0000001c9664 : "kernel"
<5>[    2.569278] 0x0000001c9664-0x000001fb0000 : "rootfs"
<5>[    2.575153] mtd: device 5 (rootfs) set to be root filesystem
<5>[    2.580957] 1 squashfs-split partitions found on MTD device rootfs
<5>[    2.587147] 0x000000450000-0x000001fb0000 : "rootfs_data"
<6>[    2.594228] libphy: Fixed MDIO Bus: probed
<6>[    2.663938] libphy: mdio: probed
<6>[    4.066829] mtk_soc_eth 1e100000.ethernet: loaded mt7530 driver
<6>[    4.073428] mtk_soc_eth 1e100000.ethernet eth0: mediatek frame engine at 0xbe100000, irq 21
[... cut ...]
<4>[ 5057.758604] [<80045f60>] 0x80045f60
<4>[ 5057.762089] [<8004bea8>] 0x8004bea8
<4>[ 5057.765564] [<8004bd78>] 0x8004bd78
<4>[ 5057.769054] [<8004bd78>] 0x8004bd78
<4>[ 5057.772544] [<8004bd78>] 0x8004bd78
<4>[ 5057.776017] [<80006f78>] 0x80006f78
<4>[ 5057.779505] 
<4>[ 5057.780986] Code: 8e10060c  8e020004  8e030000 <ac620004> ac430000  24020100  ae020000  24020200  10000002 
<4>[ 5057.790733] 
<4>[ 5057.792487] ---[ end trace f9ab7dd3440b1d4e ]---

===================================
Time: 1585770206.52196

Openwrtfunboy, 2020-04-02:
I noticed that this happens even when the iPhone does not connect.

Openwrtfunboy, 2020-04-02:
Also i have another crashlog reports. Did me need publish it?

lukasz1992, 2020-04-03:
@Openwrtfunboy I am not a developer (I wish I was), but providing: crashlogs, dumps of config and version of OpenWRT/mt76 are the most helpful things you can do.

I think the first crashlog you provided is very similar to https://github.com/openwrt/mt76/issues/366

Openwrtfunboy, 2020-04-03:
8 people are connected to the network, only one of them is very close to the router and because of it the router rebooted and disconnected clients. I attach the log below.

root@OpenWrt:~# cat /sys/kernel/debug/crashlog
Time: 1585946894.858343
Modules:	pppoe@8eff0000+23e0	ppp_async@8f764000+1b90	pppox@8ee06000+4b0	ppp_generic@8f010000+5970	nf_conntrack_ipv6@8ee3c000+1590	mt76x2e@8f78c000+26c0	mt76x2_common@8ee00000+2a40	mt76x02_lib@8ed10000+9640	mt7603e@8f7e0000+90a0	mt76@8f768000+7920	mac80211@8ec80000+6c840	iptable_nat@8eeb4000+2c0	ipt_REJECT@8f65c000+3c0	ipt_MASQUERADE@8f753000+2a0	cfg80211@8ec00000+36b80	xt_time@8f6b7000+6e0	xt_tcpudp@8f642000+700	xt_tcpmss@8ee9b000+420	xt_statistic@8eeb2000+340	xt_state@8efc3000+2c0	xt_recent@8f7b6000+1bf0	xt_nat@8efc5000+600	xt_multiport@8ef31000+500	xt_mark@8f7a8000+2a0	xt_mac@8f7a6000+260	xt_limit@8f7a0000+4a0	xt_length@8f79d000+2c0	xt_hl@8f789000+320	xt_helper@8f786000+360	xt_ecn@8f784000+540	xt_dscp@8f781000+3e0	xt_conntrack@8effa000+900	xt_connmark@8eff8000+4e0xt_connlimit@8ee6c000+1510	xt_connbytes@8f75c000+640	xt_comment@8f759000+1e0	xt_TCPMSS@8f756000+ae0	xt_REDIRECT@8f754000+2c0	xt_LOG@8f752000+300	xt_HL@8f749000+540	xt_FLOWOFFLOAD@8f751000+c10	xt_DSCP@8ee1d000+5e0	xt_CT@8ee5b000+a60	xt_CLASSIFY@8efec000+240	slhc@8f766000+13c0	nf_reject_ipv4@8f71e000+840	nf_nat_redirect@8f71a000+580	nf_nat_masquerade_ipv4@8f718000+650	nf_conntrack_ipv4@8ee66000+1420	nf_nat_ipv4@8ee56000+f30	nf_nat@8f730000+37d0	nf_log_ipv4@8ee53000+d60	nf_flow_table_hw@8f713000+8d0	nf_flow_table@8efb0000+3350	nf_defrag_ipv6@8ee08000+12e0	nf_defrag_ipv4@8f719000+440	nf_conntrack_rtcache@8f71b000+aa0	iptable_raw@8efe7000+260	iptable_mangle@8efe9000+380	iptable_filter@8ee43000+280	ipt_ECN@8ee4c000+580	ip_tables@8ee58000+2b30	crc_ccitt@8ee50000+3e0	compat@8f6e2000+1740	sch_cake@8f720000+74d0	nf_conntrack@8f700000+10550	sch_tbf@8ee68000+1700	sch_ingress@8ee4e000+4c0	sch_htb@8ee60000+36d0	sch_hfsc@8efa4000+3a00	em_u32@8ee48000+220	cls_u32@8ee40000+22d0	cls_tcindex@8ee32000+1500	cls_route@8ee30000+1470	cls_matchall@8ee1e000+9e0	cls_fw@8ee1c000+f20	cls_flow@8ee28000+1680	cls_basic@8ee0f000+c40	act_skbedit@8efed000+9f0	act_mirred@8efeb000+f30	ledtrig_usbport@8efe8000+ab0	nf_log_ipv6@8ee04000+1020	nf_log_common@8f67c000+aa0	ip6table_mangle@8efe6000+4a0	ip6table_filter@8efe1000+280	ip6_tables@8eff4000+2a20	ip6t_REJECT@8f600000+400	x_tables@8f6e4000+3290	nf_reject_ipv6@8f65f000+9e0	ifb@8f663000+ce0	leds_gpio@8f5df000+b60	xhci_plat_hcd@8f618000+15c0	xhci_pci@8f5da000+bc0	xhci_mtk@8f608000+1460	xhci_hcd@8f620000+160f0	gpio_button_hotplug@8f5ce000+1ac0	usbcore@8f5e0000+1fd50	nls_base@8f5c0000+1430	usb_common@8f5b4000+8c0
<6>[    2.095905] pci 0000:00:01.0:   bridge window [mem 0x60200000-0x602fffff]
<6>[    2.104186] clocksource: Switched to clocksource GIC
<6>[    2.110867] NET: Registered protocol family 2
<6>[    2.115971] TCP established hash table entries: 2048 (order: 1, 8192 bytes)
<6>[    2.122861] TCP bind hash table entries: 2048 (order: 2, 16384 bytes)
<6>[    2.129319] TCP: Hash tables configured (established 2048 bind 2048)
<6>[    2.135732] UDP hash table entries: 256 (order: 1, 8192 bytes)
<6>[    2.141499] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes)
<6>[    2.147977] NET: Registered protocol family 1
<7>[    2.152303] PCI: CLS 0 bytes, default 32
<6>[    2.384126] 4 CPUs re-calibrate udelay(lpj = 2924544)
<4>[    2.390434] Crashlog allocated RAM at address 0x3f00000
<6>[    2.396234] workingset: timestamp_bits=30 max_order=17 bucket_order=0
<6>[    2.409413] squashfs: version 4.0 (2009/01/31) Phillip Lougher
<6>[    2.415209] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
<5>[    2.425426] random: fast init done
<6>[    2.431041] bounce: pool size: 64 pages
<6>[    2.434884] io scheduler noop registered
<6>[    2.438713] io scheduler deadline registered (default)
<6>[    2.444785] gpio-export gpio_export: 1 gpio(s) exported
<6>[    2.450824] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
<6>[    2.460971] console [ttyS0] disabled
<6>[    2.464580] 1e000c00.uartlite: ttyS0 at MMIO 0x1e000c00 (irq = 19, base_baud = 3125000) is a 16550A
<6>[    2.473539] console [ttyS0] enabled
<6>[    2.480436] bootconsole [early0] disabled
<4>[    2.490893] MediaTek Nand driver init, version v2.1 Fix AHB virt2phys error
<6>[    2.498333] spi-mt7621 1e000b00.spi: sys_freq: 220000000
<6>[    2.513644] m25p80 spi0.0: w25q256 (32768 Kbytes)
<5>[    2.518453] 4 fixed-partitions partitions found on MTD device spi0.0
<5>[    2.524796] Creating 4 MTD partitions on "spi0.0":
<5>[    2.529574] 0x000000000000-0x000000030000 : "u-boot"
<5>[    2.535733] 0x000000030000-0x000000040000 : "u-boot-env"
<5>[    2.542105] 0x000000040000-0x000000050000 : "factory"
<5>[    2.548315] 0x000000050000-0x000002000000 : "firmware"
<5>[    2.554860] 2 uimage-fw partitions found on MTD device firmware
<5>[    2.560769] Creating 2 MTD partitions on "firmware":
<5>[    2.565765] 0x000000000000-0x0000001c9664 : "kernel"
<5>[    2.571842] 0x0000001c9664-0x000001fb0000 : "rootfs"
<5>[    2.578012] mtd: device 5 (rootfs) set to be root filesystem
<5>[    2.583841] 1 squashfs-split partitions found on MTD device rootfs
<5>[    2.590055] 0x000000450000-0x000001fb0000 : "rootfs_data"
<6>[    2.597332] libphy: Fixed MDIO Bus: probed
<6>[    2.666195] libphy: mdio: probed
<6>[    4.069056] mtk_soc_eth 1e100000.ethernet: loaded mt7530 driver
[... cut ...]
<4>[ 4114.249491] [<8f769060>] 0x8f769060 [mt76@8f768000+0x7920]
<4>[ 4114.254963] [<80360ec0>] 0x80360ec0
<4>[ 4114.258457] [<803637c0>] 0x803637c0
<4>[ 4114.261953] [<804a2878>] 0x804a2878
<4>[ 4114.265424] [<80032f84>] 0x80032f84
<4>[ 4114.268895] [<80259ed0>] 0x80259ed0
<4>[ 4114.272373] [<80007488>] 0x80007488
<4>[ 4114.275844] 
<4>[ 4114.277323] Code: 00000000  8e020014  00e23821 <8ce20000> 10000009  cc400000  1040ffbd  00000000  8e060010 
<4>[ 4114.287050] 
<4>[ 4114.288777] ---[ end trace 692700edfe58e435 ]---

Openwrtfunboy, 2020-04-03:
@nbd168 @LorenzoBianconi 

Can we get fix, please?

r0ck3r, 2020-04-20:
Same for me on Mi Router 3G, but not related to distance from router to device, as far I understand.

My crashlogs, first:
`Time: 1587317574.935027
Modules:	pppoe@8f2b4000+2430	ppp_async@8f2b8000+1be0	ebtable_nat@8f2fb000+3b0	ebtable_filter@8f10c000+3b0	ebtable_broute@8e7a4000+350	pppox@8e6eb000+58a	ppp_generic@8e970000+5b42	mt76x2e@8f234000+2850	mt76x2_common@8f2ac000+2d48	mt76x02_lib@8f3d0000+a401	mt7603e@8e800000+976c	mt76@8f370000+8224	mac80211@8e880000+72008	iptable_nat@8f2a6000+310	iptable_mangle@8f306000+3d0	iptable_filter@8f34b000+2d0	ipt_REJECT@8f348000+410	ipt_MASQUERADE@8f341000+2f0	ip_tables@8f358000+294d	ebtables@8f350000+33d5	ebt_vlan@8f330000+3f0	ebt_stp@8f32e000+790	ebt_snat@8f32a000+390	ebt_redirect@8f31d000+3b0	ebt_pkttype@8f319000+270	ebt_mark_m@8f316000+290	ebt_mark@8f313000+2f0	ebt_limit@8f2b0000+420	ebt_ip@8f30d000+570	ebt_dnat@8f309000+450	ebt_arpreply@8f2ab000+4f0	ebt_arp@8f232000+6b0	ebt_among@8f233000+a50	ebt_802_3@8e69f000+310	cfg80211@8f2c0000+39624	xt_time@8e418000+730	xt_tcpudp@8e41b000+750	xt_state@8e41f000+310	xt_nat@8e6b4000+650	xt_multiport@8e428000+550	xt_mark@8e6c0000+2f0	xt_mac@8e6b9000+2b0	xt_limit@8e6b3000+4f0	xt_conntrack@8f225000+950	xt_comment@8e7a1000+230	xt_TCPMSS@8e7a0000+870	xt_REDIRECT@8e7a5000+310	xt_LOG@8f20a000+2f0	xt_FLOWOFFLOAD@8e44a000+c40	x_tables@8f278000+37cf	wireguard@8e400000+1660e	ums_usbat@8e7f4000+20d0	ums_sddr55@8e7a8000+1790	ums_sddr09@8e6d0000+2410	ums_karma@8e7bc000+6f0	ums_jumpshot@8f226000+1010	ums_isd200@8e7b8000+1760	ums_freecom@8f21c000+8d0	ums_datafab@8f230000+1370	ums_cypress@8e6f3000+a70	ums_alauda@8f220000+22f0	slhc@8e68c000+149b	nf_reject_ipv4@8f214000+883	nf_nat_redirect@8f210000+5fb	nf_nat_masquerade_ipv4@8f209000+71c	nf_conntrack_ipv4@8f25a000+1150	nf_nat_ipv4@8e44d000+d91	nf_nat@8e784000+354b	nf_log_ipv4@8e7e9000+db0	nf_log_common@8f25e000+acf	nf_flow_table_hw@8e444000+920	nf_flow_table@8e420000+363f	nf_defrag_ipv4@8e468000+436	nf_conntrack_rtcache@8e686000+990	nf_conntrack@8f260000+f885	crc_ccitt@8f21b000+42b	compat@8e6d4000+19b7	fuse@8f240000+13220	ledtrig_usbport@8e7ff000+b00	ip_gre@8e6f4000+3435	gre@8e7dd000+8b3	udp_tunnel@8e7d6000+aef	ip_tunnel@8e6f0000+2d30	tun@8e6e0000+5e31	loop@8e460000+4130	vfat@8e43c000+2210	fat@8e4f0000+d24e	cifs@8e480000+5b5e2	nls_utf8@8e6c1000+390	nls_iso8859_1@8e6bf000+b70	nls_cp437@8e6d6000+1170	sha256_generic@8e6cc000+29f9	md5@8e6bc000+11b0	md4@8e6b8000+cb0	hmac@8e6b6000+a10	ecb@8e6b2000+5b0	des_generic@8e6a8000+48eb	usb_storage@8f0f0000+9cab	leds_gpio@8f101000+bb0	xhci_plat_hcd@8f79c000+1590	xhci_pci@8f052000+c10	xhci_mtk@8f780000+12f0	xhci_hcd@8f060000+16ea7	ehci_platform@8f0ee000+1270	sd_mod@8f7b0000+7460	scsi_mod@8f020000+16e32	ehci_hcd@8f010000+89d7	gpio_button_hotplug@8f048000+1b10	ext4@8f080000+6018e	mbcache@8f798000+fae	jbd2@8f7f0000+e4a1	usbcore@8f7c0000+2149d	nls_base@8f782000+1570	usb_common@8f74c000+a17	crc32c_generic@8f749000+5b0
<6>[  183.239958] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
<6>[  185.807872] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
<5>[  320.517850] EXT4-fs (sda2): error count since last fsck: 13
<5>[  320.523439] EXT4-fs (sda2): initial error at time 1571823187: __ext4_new_inode:1120
<5>[  320.531177] EXT4-fs (sda2): last error at time 1582553602: ext4_mb_generate_buddy:757
<6>[ 9440.289328] do_page_fault(): sending SIGSEGV to minidlna for invalid read access from 00000010
<6>[ 9440.298380] epc = 0041a3b5 in minidlna[400000+2c000]
<6>[ 9440.303941] ra  = 00405f23 in minidlna[400000+2c000]
<6>[15448.612360] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
<6>[15461.441346] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
<6>[22037.043209] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
<6>[22039.604996] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
<6>[48740.578519] usb 2-1: reset SuperSpeed USB device number 2 using xhci-mtk
<6>[62191.174820] mtk_soc_eth 1e100000.ethernet eth0: port 2 link down
<6>[62204.542088] mtk_soc_eth 1e100000.ethernet eth0: port 2 link up
<6>[62728.126635] rs IN=br-lan OUT= MAC=64:09:80:7c:ca:1b:a8:9c:ed:ae:cb:a6:08:00 SRC=192.168.10.21 DST=91.132.60.13 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=12442 DF PROTO=TCP SPT=48564 DPT=80 WINDOW=65535 RES=0x00 SYN URGP=0 
<6>[62728.146974] rs IN=br-lan OUT= MAC=64:09:80:7c:ca:1b:a8:9c:ed:ae:cb:a6:08:00 SRC=192.168.10.21 DST=91.132.60.13 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=45925 DF PROTO=TCP SPT=48568 DPT=80 WINDOW=65535 RES=0x00 SYN URGP=0 
<6>[62854.975871] rs IN=br-lan OUT= MAC=64:09:80:7c:ca:1b:a8:9c:ed:ae:cb:a6:08:00 SRC=192.168.10.21 DST=91.132.60.13 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=908 DF PROTO=TCP SPT=48774 DPT=80 WINDOW=65535 RES=0x00 SYN URGP=0 
<6>[62854.995745] rs IN=br-lan OUT= MAC=64:09:80:7c:ca:1b:a8:9c:ed:ae:cb:a6:08:00 SRC=192.168.10.21 DST=91.132.60.13 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=28683 DF PROTO=TCP SPT=48776 DPT=80 WINDOW=65535 RES=0x00 SYN URGP=0 
<6>[62878.471202] rs IN=br-lan OUT= MAC=64:09:80:7c:ca:1b:a8:9c:ed:ae:cb:a6:08:00 SRC=192.168.10.21 DST=91.132.60.13 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=62469 DF PROTO=TCP SPT=48794 DPT=80 WINDOW=65535 RES=0x00 SYN URGP=0 
<6>[62878.490724] rs IN=br-lan OUT= MAC=64:09:80:7c:ca:1b:a8:9c:ed:ae:cb:a6:08:00 SRC=192.168.10.21 DST=91.132.60.13 LEN=60 TOS=0x00 PREC=0x00 TTL=64 ID=51222 DF PROTO=TCP SPT=48796 DPT=80 WINDOW=65535 RES=0x00 SYN URGP=0 
[... cut ...]
<4>[40416.747365] [<8042e310>] __do_softirq+0x128/0x2ec
<4>[40416.752071] [<80033f84>] irq_exit+0xac/0xc8
<4>[40416.756237] [<8024f4f0>] plat_irq_dispatch+0xfc/0x138
<4>[40416.761284] [<80007588>] except_vec_vi_end+0xb8/0xc4
<4>[40416.766233] [<80008f50>] r4k_wait_irqoff+0x1c/0x24
<4>[40416.771006] Code: 00000000  8e020014  00e23821 <8ce20000> 10000009  cc400000  1040ffbd  00000000  8e060010 
<4>[40416.780736] 
<4>[40416.782385] ---[ end trace f2825c0c61368991 ]---

===================================
Time: 158736
`

r0ck3r, 2020-04-20:
Another one crashlog. Happened when I changed tx power from Driver defaults to 20 dBm and channel from 2 to 10. But I was not able to repeat it

`
Time: 1587383718.84874
Modules:	pppoe@8f374000+2430	ppp_async@8f37a000+1be0	ebtable_nat@8e598000+3b0	ebtable_filter@8e805000+3b0	ebtable_broute@8f3ab000+350	pppox@8e75c000+58a	ppp_generic@8ea00000+5b42	mt76x2e@8e4a8000+2850	mt76x2_common@8f358000+2d48	mt76x02_lib@8e900000+a401	mt7603e@8f380000+976c	mt76@8f3a0000+8224	mac80211@8e880000+72008	iptable_nat@8e4d2000+310	iptable_mangle@8e7f3000+3d0	iptable_filter@8e47a000+2d0	ipt_REJECT@8e7da000+410	ipt_MASQUERADE@8f27b000+2f0	ip_tables@8f348000+294d	ebtables@8f340000+33d5	ebt_vlan@8f29e000+3f0	ebt_stp@8f2f3000+790	ebt_snat@8f2f1000+390	ebt_redirect@8f2ed000+3b0	ebt_pkttype@8f2ea000+270	ebt_mark_m@8f2e6000+290	ebt_mark@8f2ca000+2f0	ebt_limit@8f2c6000+420	ebt_ip@8f2c3000+570	ebt_dnat@8f2c0000+450	ebt_arpreply@8f2d5000+4f0	ebt_arp@8e7f5000+6b0	ebt_among@8f1c9000+a50	ebt_802_3@8e487000+310	cfg80211@8f300000+39624	xt_time@8f29a000+730	xt_tcpudp@8f29d000+750	xt_state@8f22e000+310	xt_nat@8f1e5000+650	xt_multiport@8e45b000+550	xt_mark@8e455000+2f0	xt_mac@8e452000+2b0	xt_limit@8e44e000+4f0	xt_conntrack@8e44c000+950	xt_comment@8f259000+230	xt_TCPMSS@8f258000+870	xt_REDIRECT@8f269000+310	xt_LOG@8f263000+2f0	xt_FLOWOFFLOAD@8e462000+c40	x_tables@8e42c000+37cf	wireguard@8f280000+1660e	ums_usbat@8f278000+20d0	ums_sddr55@8f270000+1790	ums_sddr09@8f260000+2410	ums_karma@8e45f000+6f0	ums_jumpshot@8f25a000+1010	ums_isd200@8e7f8000+1760	ums_freecom@8e467000+8d0	ums_datafab@8f218000+1370	ums_cypress@8e46f000+a70	ums_alauda@8e444000+22f0	slhc@8f256000+149b	nf_reject_ipv4@8e476000+883	nf_nat_redirect@8f1f7000+5fb	nf_nat_masquerade_ipv4@8e40a000+71c	nf_conntrack_ipv4@8f216000+1150	nf_nat_ipv4@8f1b1000+d91	nf_nat@8e438000+354b	nf_log_ipv4@8f20e000+db0	nf_log_common@8e486000+acf	nf_flow_table_hw@8f1f9000+920	nf_flow_table@8e790000+363f	nf_defrag_ipv4@8e75d000+436	nf_conntrack_rtcache@8e485000+990	nf_conntrack@8f230000+f885	crc_ccitt@8e42a000+42b	compat@8e646000+19b7	fuse@8f240000+13220	ledtrig_usbport@8e5bd000+b00	ip_gre@8e440000+3435	gre@8e7dc000+8b3	udp_tunnel@8e7df000+aef	ip_tunnel@8e7f0000+2d30	tun@8e7e8000+5e31	loop@8e760000+4130	vfat@8f1c0000+2210	fat@8e490000+d24e	cifs@8e700000+5b5e2	nls_utf8@8e40b000+390	nls_iso8859_1@8f1e7000+b70	nls_cp437@8f1fe000+1170	sha256_generic@8f1f4000+29f9	md5@8f1ee000+11b0	md4@8f1de000+cb0	hmac@8f1db000+a10	ecb@8f1cc000+5b0	des_generic@8f1d0000+48eb	usb_storage@8f170000+9cab	leds_gpio@8f7ff000+bb0	xhci_plat_hcd@8f792000+1590	xhci_pci@8f066000+c10	xhci_mtk@8f79e000+12f0	xhci_hcd@8f0a0000+16ea7	ehci_platform@8f7b6000+1270	sd_mod@8f058000+7460	scsi_mod@8f040000+16e32	ehci_hcd@8f7f0000+89d7	gpio_button_hotplug@8f7d6000+1b10	ext4@8f100000+6018e	mbcache@8f7ae000+fae	jbd2@8f7e0000+e4a1	usbcore@8f000000+2149d	nls_base@8f794000+1570	usb_common@8f781000+a17	crc32c_generic@8f77e000+5b0
<14>[    6.042899] init: Console is alive
<14>[    6.046544] init: - watchdog -
<14>[    7.310477] kmodloader: loading kernel modules from /etc/modules-boot.d/*
<6>[    7.439029] usbcore: registered new interface driver usbfs
<6>[    7.444580] usbcore: registered new interface driver hub
<6>[    7.450076] usbcore: registered new device driver usb
<6>[    7.480704] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
<5>[    7.493555] SCSI subsystem initialized
<6>[    7.500874] ehci-platform: EHCI generic platform driver
<4>[    7.512300] xhci-mtk 1e1c0000.xhci: 1e1c0000.xhci supply vusb33 not found, using dummy regulator
<6>[    7.521359] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
<6>[    7.526682] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 1
<6>[    7.537033] xhci-mtk 1e1c0000.xhci: hcc params 0x01401198 hci version 0x96 quirks 0x0000000000210010
<6>[    7.546214] xhci-mtk 1e1c0000.xhci: irq 20, io mem 0x1e1c0000
<6>[    7.553001] hub 1-0:1.0: USB hub found
<6>[    7.556892] hub 1-0:1.0: 2 ports detected
<6>[    7.561457] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
<6>[    7.566875] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 2
<6>[    7.574342] xhci-mtk 1e1c0000.xhci: Host supports USB 3.0  SuperSpeed
<6>[    7.580981] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
<6>[    7.589916] hub 2-0:1.0: USB hub found
<6>[    7.593739] hub 2-0:1.0: 1 port detected
<6>[    7.605814] usbcore: registered new interface driver usb-storage
<14>[    7.612234] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
<14>[    7.627163] init: - preinit -
<6>[    8.296979] usb 2-1: new SuperSpeed USB device number 2 using xhci-mtk
<6>[    8.329004] usb-storage 2-1:1.0: USB Mass Storage device detected
<6>[    8.335717] scsi host0: usb-storage 2-1:1.0
<5>[    8.415545] random: procd: uninitialized urandom read (4 bytes read)
<6>[    8.557595] mtk_soc_eth 1e100000.ethernet: PPE started
<5>[    9.397517] scsi 0:0:0:0: Direct-Access     ADATA    HV300            0    PQ: 0 ANSI: 6
<5>[    9.408235] sd 0:0:0:0: [sda] Spinning up disk...
<4>[   10.436813] .
<4>[   11.476812] .
<14>[   11.773132] mount_root: loading kmods from internal overlay
<14>[   11.808325] kmodloader: loading kernel modules from //etc/modules-boot.d/*
<14>[   11.818946] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
<5>[   12.324146] UBIFS (ubi0:1): background thread "ubifs_bgt0_1" started, PID 513
[... cut ...]
<4>[ 8977.921085]         ...
<4>[ 8977.923531] Call Trace:
<4>[ 8977.925976] [<80336e1c>] skb_release_data+0x7c/0x1c8
<4>[ 8977.930948] [<80336a70>] consume_skb+0x60/0x88
<4>[ 8977.935406] [<8f382a44>] mt7603_unregister_device+0x260/0x2cc [mt7603e]
<4>[ 8977.942002] Code: 10000017  00809825  8e240000 <8c820014> 30430001  10600002  00000000  2444ffff  0000000f 
<4>[ 8977.951731] 
<4>[ 8977.954160] ---[ end trace d116f66a6780da93 ]---

===================================
Time: 1587383718.92231

`

r0ck3r, 2020-04-21:
Another one:
`
Time: 1587458726.406377
Modules:	pppoe@8ea10000+2430	ppp_async@8ea0c000+1be0	ebtable_nat@8f1d8000+3b0	ebtable_filter@8f2e5000+3b0	ebtable_broute@8e827000+350	pppox@8f235000+58a	ppp_generic@8ea00000+5b42	mt76x2e@8e4c0000+2850	mt76x2_common@8e830000+2d48	mt76x02_lib@8e860000+a401	mt7603e@8e810000+976c	mt76@8f390000+8224	mac80211@8e880000+72008	iptable_nat@8e6f4000+310	iptable_mangle@8e763000+3d0	iptable_filter@8e4d4000+2d0	ipt_REJECT@8f31a000+410	ipt_MASQUERADE@8f313000+2f0	ip_tables@8f330000+294d	ebtables@8f324000+33d5	ebt_vlan@8f37d000+3f0	ebt_stp@8f307000+790	ebt_snat@8f303000+390	ebt_redirect@8f300000+3b0	ebt_pkttype@8f2f5000+270	ebt_mark_m@8e724000+290	ebt_mark@8f227000+2f0	ebt_limit@8f25f000+420	ebt_ip@8e768000+570	ebt_dnat@8e620000+450	ebt_arpreply@8e779000+4f0	ebt_arp@8e78f000+6b0	ebt_among@8f22f000+a50	ebt_802_3@8e4e3000+310	cfg80211@8f340000+39624	xt_time@8f237000+730	xt_tcpudp@8f234000+750	xt_state@8f231000+310	xt_nat@8f29a000+650	xt_multiport@8f29d000+550	xt_mark@8f1fe000+2f0	xt_mac@8e70f000+2b0	xt_limit@8e714000+4f0	xt_conntrack@8f259000+950	xt_comment@8f25d000+230	xt_TCPMSS@8e70b000+870	xt_REDIRECT@8e707000+310	xt_LOG@8f266000+2f0	xt_FLOWOFFLOAD@8f267000+c40	x_tables@8f218000+37cf	wireguard@8f280000+1660e	ums_usbat@8e4b8000+20d0	ums_sddr55@8e766000+1790	ums_sddr09@8f224000+2410	ums_karma@8f261000+6f0	ums_jumpshot@8e772000+1010	ums_isd200@8e45c000+1760	ums_freecom@8e727000+8d0	ums_datafab@8e4c6000+1370	ums_cypress@8e710000+a70	ums_alauda@8f214000+22f0	slhc@8e6d4000+149b	nf_reject_ipv4@8e718000+883	nf_nat_redirect@8f256000+5fb	nf_nat_masquerade_ipv4@8f25a000+71c	nf_conntrack_ipv4@8f19e000+1150	nf_nat_ipv4@8e720000+d91	nf_nat@8f1c8000+354b	nf_log_ipv4@8e739000+db0	nf_log_common@8e69c000+acf	nf_flow_table_hw@8e726000+920	nf_flow_table@8e77c000+363f	nf_defrag_ipv4@8e728000+436	nf_conntrack_rtcache@8e72b000+990	nf_conntrack@8f270000+f885	crc_ccitt@8e58d000+42b	compat@8f264000+19b7	fuse@8f240000+13220	ledtrig_usbport@8e736000+b00	ip_gre@8f20c000+3435	gre@8e4e2000+8b3	udp_tunnel@8e4db000+aef	ip_tunnel@8f200000+2d30	tun@8e4f8000+5e31	loop@8f1d0000+4130	vfat@8e78c000+2210	fat@8e470000+d24e	cifs@8e400000+5b5e2	nls_utf8@8e5cf000+390	nls_iso8859_1@8e58c000+b70	nls_cp437@8e6ca000+1170	sha256_generic@8e774000+29f9	md5@8f18a000+11b0	md4@8e62d000+cb0	hmac@8f1fb000+a10	ecb@8f1f7000+5b0	des_generic@8e6c0000+48eb	usb_storage@8f0f0000+9cab	leds_gpio@8f101000+bb0	xhci_plat_hcd@8f7ce000+1590	xhci_pci@8f032000+c10	xhci_mtk@8f778000+12f0	xhci_hcd@8f060000+16ea7	ehci_platform@8f02c000+1270	sd_mod@8f7f8000+7460	scsi_mod@8f7e0000+16e32	ehci_hcd@8f780000+89d7	gpio_button_hotplug@8f03e000+1b10	ext4@8f080000+6018e	mbcache@8f79c000+fae	jbd2@8f7d0000+e4a1	usbcore@8f000000+2149d	nls_base@8f77a000+1570	usb_common@8f75c000+a17	crc32c_generic@8f758000+5b0
<6>[   36.798345] br-lan: port 4(tap0) entered disabled state
<6>[   36.804072] device tap0 entered promiscuous mode
<6>[   36.809036] br-lan: port 4(tap0) entered blocking state
<6>[   36.814276] br-lan: port 4(tap0) entered forwarding state
<6>[   37.848926] br-lan: port 5(grelan0) entered blocking state
<6>[   37.854466] br-lan: port 5(grelan0) entered disabled state
<6>[   37.860972] device grelan0 entered promiscuous mode
<6>[   37.866401] br-lan: port 5(grelan0) entered blocking state
<6>[   37.871987] br-lan: port 5(grelan0) entered forwarding state
<6>[   37.921617] br-lan: port 6(grelan1) entered blocking state
<6>[   37.927230] br-lan: port 6(grelan1) entered disabled state
<6>[   37.933403] device grelan1 entered promiscuous mode
<6>[   37.938714] br-lan: port 6(grelan1) entered blocking state
<6>[   37.944289] br-lan: port 6(grelan1) entered forwarding state
<6>[   96.370683] br-lan: port 3(wlan1) entered blocking state
<6>[   96.376108] br-lan: port 3(wlan1) entered forwarding state
<6>[  104.706339] device wlan0 left promiscuous mode
<6>[  104.711019] br-lan: port 2(wlan0) entered disabled state
<6>[  182.702899] br-lan: port 2(wlan0) entered blocking state
<6>[  182.708245] br-lan: port 2(wlan0) entered disabled state
<6>[  182.714246] device wlan0 entered promiscuous mode
<6>[  184.678074] br-lan: port 2(wlan0) entered blocking state
<6>[  184.683414] br-lan: port 2(wlan0) entered forwarding state
<6>[  265.405837] device wlan0 left promiscuous mode
<6>[  265.410438] br-lan: port 2(wlan0) entered disabled state
<6>[  266.287075] br-lan: port 2(wlan0) entered blocking state
<6>[  266.292418] br-lan: port 2(wlan0) entered disabled state
<6>[  266.298382] device wlan0 entered promiscuous mode
<6>[  266.303433] br-lan: port 2(wlan0) entered blocking state
<6>[  266.308891] br-lan: port 2(wlan0) entered forwarding state
<6>[  266.645748] br-lan: port 2(wlan0) entered disabled state
<6>[  268.234072] br-lan: port 2(wlan0) entered blocking state
<6>[  268.239410] br-lan: port 2(wlan0) entered forwarding state
<6>[  284.645913] device wlan0 left promiscuous mode
<6>[  284.650530] br-lan: port 2(wlan0) entered disabled state
<6>[  285.660271] br-lan: port 2(wlan0) entered blocking state
<6>[  285.665694] br-lan: port 2(wlan0) entered disabled state
<6>[  285.671692] device wlan0 entered promiscuous mode
<6>[  285.676804] br-lan: port 2(wlan0) entered blocking state
<6>[  285.682160] br-lan: port 2(wlan0) entered forwarding state
<6>[  285.895860] br-lan: port 2(wlan0) entered disabled state
<6>[  287.621013] br-lan: port 2(wlan0) entered blocking state
<6>[  287.626363] br-lan: port 2(wlan0) entered forwarding state
[... cut ...]
<4>[74993.602000] Call Trace:
<4>[74993.604578] [<8e8dc448>] minstrel_calc_rate_stats+0xad4/0x3164 [mac80211]
<4>[74993.611495] Code: 14e0000b  24040003  8ca50000 <8ca5007c> 14a4000d  2404001a  70442802  00ac1021  90420002 
<4>[74993.621269] 
<4>[74993.622897] ---[ end trace 733750a9f47f055b ]---

===================================
Time: 1587458726.4`

The most strange think is that it was working pretty stable on this rom, but about 20 days ago it started to reboot. Maybe it is some hardware failure?

r0ck3r, 2020-05-04:
Tried to change power supply with no luck, but then I have compiled image from branch 19.07 and that commit: https://github.com/openwrt/openwrt/tree/5feb0df9bb3bd83ec49af17d2abc261fc58e2551

Works fine for 4 days without any reboot.

koala0529, 2020-12-01:
Hostname	OpenWrt
Model	Newifi-D2
Architecture	MediaTek MT7621 ver:1 eco:3
Firmware Version	OpenWrt 19.07.4 r11208-ce6496d796 / LuCI openwrt-19.07 branch git-20.325.64304-92c01e2
Kernel Version	4.14.195
Local Time	2020-12-01 01:23:28
Uptime	2d 10h 32m 13s
Load Average	0.36, 0.22, 0.14

我知道这个路由器USB3.0 host不支持uas，所以用的quriks的方式匹配的usb-storage驱动，移动硬盘是希捷2T；通过SAMBA读写数据就会产生reset SuperSpeed USB device number 2 using xhci-mtk这种复位，也不知道为什么，好像内核有bug,那位大神知道怎么修改吗？