openwrt/mt76 #385: mt7615: frequent STA disconnections and connection interruptions

lantsem, 2020-04-02:
AP: ASUS RT-AC65P, OpenWrt master r12787+2-af5ccfbac7, mt76 from master 2020-03-17-2097f74f.
STA: phones, tablet and laptops.

AP has simple configuration:

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'VHT80'

In log I see high disconnection rate for two my phones with WCN3680. They can lie on the table 1-2m from AP, have a great signal level and randomly disconnects from AP. No errors in log, no missing ACKs, simply disconnects. Example of logread | grep hostapd:

Thu Apr  2 10:17:22 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 88:79:7e:d6:42:40
Thu Apr  2 10:17:22 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 WPA: pairwise key handshake completed (RSN)
Thu Apr  2 10:18:30 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 88:79:7e:d6:42:40
Thu Apr  2 10:18:30 2020 daemon.notice hostapd: wlan1: STA 88:79:7e:d6:42:40 IEEE 802.11: did not acknowledge authentication response
Thu Apr  2 10:18:30 2020 daemon.notice hostapd: wlan1: STA 88:79:7e:d6:42:40 IEEE 802.11: did not acknowledge authentication response
Thu Apr  2 10:18:30 2020 daemon.notice hostapd: wlan1: STA 88:79:7e:d6:42:40 IEEE 802.11: did not acknowledge authentication response
Thu Apr  2 10:18:31 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 IEEE 802.11: authenticated
Thu Apr  2 10:18:55 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 IEEE 802.11: authenticated
Thu Apr  2 10:19:39 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 IEEE 802.11: authenticated
Thu Apr  2 10:19:39 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 IEEE 802.11: associated (aid 1)
Thu Apr  2 10:19:39 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 88:79:7e:d6:42:40
Thu Apr  2 10:19:39 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 WPA: pairwise key handshake completed (RSN)
Thu Apr  2 10:19:57 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 88:79:7e:d6:42:40
Thu Apr  2 10:19:57 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 IEEE 802.11: authenticated
Thu Apr  2 10:19:57 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 IEEE 802.11: associated (aid 1)
Thu Apr  2 10:19:57 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 88:79:7e:d6:42:40
Thu Apr  2 10:19:57 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 WPA: pairwise key handshake completed (RSN)
Thu Apr  2 10:29:55 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 88:79:7e:d6:42:40
Thu Apr  2 10:29:57 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 IEEE 802.11: authenticated
Thu Apr  2 10:29:57 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 IEEE 802.11: associated (aid 1)
Thu Apr  2 10:29:58 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 88:79:7e:d6:42:40
Thu Apr  2 10:29:58 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 WPA: pairwise key handshake completed (RSN)
Thu Apr  2 10:36:45 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 88:79:7e:d6:42:40
Thu Apr  2 10:36:47 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 IEEE 802.11: authenticated
Thu Apr  2 10:36:47 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 IEEE 802.11: associated (aid 1)
Thu Apr  2 10:36:47 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 88:79:7e:d6:42:40
Thu Apr  2 10:36:47 2020 daemon.info hostapd: wlan1: STA 88:79:7e:d6:42:40 WPA: pairwise key handshake completed (RSN)

But I suppose that others STA, although they do not perform reconnections, have connection interruptions. Tablet with online game loses connection to game server from time to time. Sometimes exactly in the same moments as phones was disconnected.

AR9220+ath9k based AP is stable with this STAs.

lantsem, 2020-04-02:
When I came back to the AP after some time and phone got connected to Wi-Fi, I got this trace in kernel log:

[82419.785218] ------------[ cut here ]------------
[82419.794597] WARNING: CPU: 3 PID: 0 at backports-5.4.27-1/net/mac80211/sta_info.c:1937 ieee80211_sta_update_pending_airtime+0x110/0x214 [mac80211]
[82419.820568] STA 88:79:7e:d6:42:45 AC 2 txq pending airtime underflow: 4294967252, 44
[82419.820575] Modules linked in: nf_conntrack_ipv6 mt7615e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_ftp iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables compat act_connmark nf_conntrack sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow
[82419.976938]  cls_basic act_skbedit act_mirred nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[82420.021644] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.14.172 #0
[82420.033757] Stack : 00000000 00000000 00000001 800775d4 80600000 80597e58 00000000 00000000
[82420.050389]         80561b00 8fc0fcac 8fc3d30c 805d29e7 8055c840 00000001 8fc0fc50 5326164d
[82420.067018]         00000000 00000000 80740000 00000000 807384a0 0000012f 00000007 00000000
[82420.083646]         00000000 00000000 0000548c ffffffff 00000000 80600000 00000000 8f307fc8
[82420.100273]         8f36da6c 00000791 00000000 00000000 00000000 802d3e48 0000000c 8073000c
[82420.116902]         ...
[82420.121761] Call Trace:
[82420.126644] [<8000c4d4>] show_stack+0x58/0x100
[82420.135486] [<804976b4>] dump_stack+0xa4/0xe0
[82420.144160] [<8002fdf0>] __warn+0xe0/0x140
[82420.152302] [<8002fa34>] warn_slowpath_fmt+0x30/0x3c
[82420.162289] [<8f307fc8>] ieee80211_sta_update_pending_airtime+0x110/0x214 [mac80211]
[82420.177740] [<8f301e4c>] ieee80211_report_low_ack+0x160/0x4f4 [mac80211]
[82420.191277] ---[ end trace c6bf0aed89221536 ]---

Not sure if this relates to disconnects.

andrey-zibrov, 2020-04-22:
Please [compare](https://github.com/openwrt/mt76/issues/386)

nbd168, 2020-04-29:
Please try the latest version from master

lantsem, 2020-05-02:
Tested 2020-04-27-5f3ccc72. Phones are still disconnecting, laptops and tablet not.

lantsem, 2020-05-03:
Connection with phone is stable after boot, problem manifests itself after ~24h and most of disconnects occur when using the phone or when the phone is charging. After wifi reload connection is again stable for a day.

Sixtiz, 2020-05-05:
I get the same kind of behavior here with a D-Link DIR-878, same MediaTek SoC. Same exception in the logs and Android phone loses connection after a while, but laptops are fine...

lantsem, 2020-05-06:
BTW, when my phones enter the reconnection cycle I see the same symptoms as in #350.

shir474, 2020-05-07:
I'm getting the same warning.
OpenWrt - 5019a06fc1027fad35fc9847735218c1057f8483
mt76 - 3e6968593b613006a906822e08a7725e07cfebde

[60985.426443] WARNING: CPU: 1 PID: 1255 at backports-5.4.27-1/net/mac80211/sta_info.c:1937 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[60985.439761] STA b0:48:1a:f0:b2:24 AC 3 txq pending airtime underflow: 4294967228, 68
[60985.439774] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7615e mt7615_common mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat i2c_tiny_usb i2c_gpio i2c_smbus i2c_algo_bit i2c_mux_gpio i2c_mux i2c_dev ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common
[60985.534717] CPU: 1 PID: 1255 Comm: hostapd Not tainted 5.4.38 #0
[60985.540699] Stack : 8f3f9e60 8006ea7c 80610000 80613578 80670000 80613540 806126a4 8fc0fc44
[60985.549026]         807b0000 8feef9d4 80658be3 805ad3b4 00000001 00000001 8fc0fbe8 560b6876
[60985.557350]         00000000 00000000 807f0000 00000000 00000030 00000193 2e352064 38332e34
[60985.565675]         00000000 00000bda 00000000 000828bd 80000000 80670000 00000000 8ed08790
[60985.574001]         00000009 00000000 00000000 00000000 00000003 802e1ae8 00000004 807b0004
[60985.582331]         ...
[60985.584769] Call Trace:
[60985.587231] [<8000b5fc>] show_stack+0x30/0x100
[60985.591692] [<804fbdb4>] dump_stack+0xa4/0xdc
[60985.596046] [<80027c14>] __warn+0xc0/0x10c
[60985.600131] [<80027cec>] warn_slowpath_fmt+0x8c/0xac
[60985.605180] [<8ed08790>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[60985.612958] [<8ed01ebc>] ieee80211_report_low_ack+0xf0/0x43c [mac80211]
[60985.619612] [<8ed02220>] ieee80211_free_txskb+0x18/0x30 [mac80211]
[60985.625825] [<8ecd539c>] mt76_tx_complete_skb+0xa8/0xbc [mt76]
[60985.631665] [<8f3da264>] mt7615_txp_skb_unmap+0x1f0/0x22c [mt7615_common]

lantsem, 2020-06-02:
Tested with OpenWrt master r13419-8d2c031f21, mt76 2020-05-30-85c51608-2. Problem is still there.

ryderlee1110, 2020-06-18:
can you try the latest mt76 version?

duga3, 2020-06-19:
Works fine on DIR-882 without any kernel errors or disconnects since 5 days:
* mt76 2020-05-30 85c51608
* OpenWrt master, r13520+1-68b94f0

shir474, 2020-06-21:
Openwrt - d93da0d0169b2ca90b888e4bb782c259e5ddf2fe
mt76 - d6d9a7ea428dca5a514c15250b4e269bc0d0b0dd

[10871.092325] ------------[ cut here ]------------
[10871.097120] WARNING: CPU: 1 PID: 1257 at backports-5.7-rc3-1/net/mac80211/sta_info.c:1929 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[10871.110495] STA 84:ab:1a:3e:a8:b7 AC 3 txq pending airtime underflow: 4294967192, 104
[10871.110500] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7615e mt7615_common mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat i2c_tiny_usb i2c_gpio i2c_smbus i2c_algo_bit i2c_mux_gpio i2c_mux i2c_dev ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common
[10871.205498] CPU: 1 PID: 1257 Comm: hostapd Not tainted 5.4.46 #0
[10871.211479] Stack : 8f3bde80 8006eac0 80610000 80616740 80670000 80616708 8061586c 8fc0fc44
[10871.219805]         807b0000 8fdfbeb4 8065cbe3 805af870 00000001 00000001 8fc0fbe8 b7dd635c
[10871.228130]         00000000 00000000 807f0000 00000000 00000030 000001ae 2e352064 36342e34
[10871.236456]         00000000 0000021c 00000000 000322ba 80000000 80670000 00000000 8ed08a84
[10871.244782]         00000009 00000000 00000000 8f3be000 00000003 802e1c18 00000004 807b0004
[10871.253107]         ...
[10871.255546] Call Trace:
[10871.258017] [<8000b5fc>] show_stack+0x30/0x100
[10871.262454] [<804fd9e0>] dump_stack+0xa4/0xdc
[10871.266804] [<80027c14>] __warn+0xc0/0x10c
[10871.270890] [<80027cec>] warn_slowpath_fmt+0x8c/0xac
[10871.275948] [<8ed08a84>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[10871.283718] [<8ed0205c>] ieee80211_report_low_ack+0xf0/0x43c [mac80211]
[10871.290364] [<8ed023c0>] ieee80211_free_txskb+0x18/0x30 [mac80211]
[10871.296569] [<8ecd5b6c>] mt76_tx_complete_skb+0xa8/0xbc [mt76]
[10871.302403] [<8edaaf7c>] mt7615_txp_skb_unmap+0x1f4/0x230 [mt7615_common]
[10871.309297] ---[ end trace 4a9c58d0e09c5f78 ]---

lantsem, 2020-06-22:
@shir474 this warning is discussed in #406.

lantsem, 2020-06-22:
@ryderlee1110 tested with OpenWrt r13597-be56b29707 and mt76 2020-06-16-d6d9a7ea. Phones are still often disconnecting after first 24h (GTK rekey). Will try with reduced rekey interval as a workaround for #350.

lantsem, 2020-06-26:
4 days uptime with 6h GTK rekey interval and wifi is stable - no more frequent phone disconnects and no symptoms of #350. I keep thinking that those two issues (this and #350) are caused by the same bug.

@ryderlee1110, @nbd168 guys, what do you think?

lantsem, 2020-06-30:
Phones are disconnecting from time to time when in use even with reduced GTK rekey interval or wifi reload. Still not fully stable.

lantsem, 2020-07-20:
Close it, since I no longer have a device.

closed 2020-07-20 by lantsem