openwrt/mt76 #386: 2.4G and 5G WIFI cannot start

hcwang, 2020-04-05:
With the new version of mac80211 driver, the WIFI of xiaomi r3p cannot start. The below is log:

```
[10031.886744] ------------[ cut here ]------------
[10031.891549] WARNING: CPU: 0 PID: 0 at target-mipsel_24kc_musl/linux-ramips_mt7621/backports-5.4.27-1/net/mac80211/sta_info.c:1937 ieee80211_sta_update_pending_airtime+0x104/0x204 [mac80211]
[10031.908437] STA 80:5e:4f:de:36:b3 AC 0 txq pending airtime underflow: 4294967032, 264
[10031.908447] Modules linked in: xt_FULLCONENAT pppoe ppp_async pppox ppp_mppe ppp_generic mt7615e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_bpf xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm slhc nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conntrack_rtcache
[10031.986852]  nf_conntrack_netlink macvlan iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat fuse tcp_bbr ledtrig_usbport cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 nf_nat nf_conntrack nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sit tunnel4 ip_tunnel tun vfat fat ntfs hfsplus hfs nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb authenc usb_storage
[10032.058287]  leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd sd_mod scsi_mod gpio_button_hotplug f2fs ext4 mbcache jbd2 usbcore nls_base usb_common crc32c_generic crc32_generic
[10032.074291] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.172 #0
[10032.080355] Stack : 00000000 00000200 60000211 8006c638 805b0000 80557c20 00000000 00000000
[10032.088688]         80522578 8fc09c5c 80590d3c 805908c7 8051d3b0 00000001 8fc09c00 1cc2824c
[10032.097020]         00000000 00000000 806f0000 00007c58 00000000 0000018b 00000008 00000000
[10032.105350]         00000000 80590000 00012233 00000000 00000000 805b0000 00000000 8f3077d8
[10032.113680]         00000009 00000791 8de54480 00000200 00000000 802a4898 00000000 806f0000
[10032.122010]         ...
[10032.124445] Call Trace:
[10032.126907] [<8000c6f0>] show_stack+0x58/0x100
[10032.131363] [<8045b484>] dump_stack+0xa4/0xe0
[10032.135705] [<8002aec8>] __warn+0xe0/0x114
[10032.139781] [<8002af2c>] warn_slowpath_fmt+0x30/0x3c
[10032.144848] [<8f3077d8>] ieee80211_sta_update_pending_airtime+0x104/0x204 [mac80211]
[10032.152609] [<8f303458>] ieee80211_tx_monitor+0xf50/0x11b0 [mac80211]
[10032.159152] ---[ end trace debe7b9a376c04ce ]---
```

andrey-zibrov, 2020-04-22:
https://openwrt.org/toh/asus/asus_rt-ac85p#notes

Model: ASUS RT-AC65P
Architecture: MediaTek MT7621 ver:1 eco:3
Firmware: OpenWrt SNAPSHOT r12985-508462a399 / LuCI Master git-20.108.47759-d516e22
Kernel Version: 5.4.32

```
[ 3449.732818] ------------[ cut here ]------------
[ 3449.742233] WARNING: CPU: 0 PID: 0 at backports-5.4.27-1/net/mac80211/sta_info.c:1937 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[ 3449.768257] STA 6c:96:cf:d7:69:1f AC 2 txq pending airtime underflow: 4294967140, 156
[ 3449.768269] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt7615e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_recent xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun usb_storage uhci_hcd ohci_platform ohci_hcd sd_mod scsi_mod ext4 mbcache jbd2 crc32c_generic leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[ 3449.933358] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.32 #0
[ 3449.945126] Stack : 8eaf1e60 800774cc 806b0000 806b6474 80720000 806b643c 806b5590 8fc0dcb4
[ 3449.961748]         80860000 80701248 80700d83 8064c7e4 00000000 00000001 8fc0dc58 3c00a33e
[ 3449.978360]         00000000 00000000 808a0000 00000000 00000030 00000250 342e3520 2032332e
[ 3449.994972]         00000000 000000ab 00000000 000e3dee 00000000 80720000 00000000 8e008790
[ 3450.011584]         00000009 00000000 00000000 00000000 00000002 8033cfd8 01800117 01800157
[ 3450.028195]         ...
[ 3450.033046] Call Trace:
[ 3450.037922] [<8000b72c>] show_stack+0x30/0x100
[ 3450.046770] [<8058fac4>] dump_stack+0xa4/0xdc
[ 3450.055436] [<80028038>] __warn+0xc0/0x10c
[ 3450.063573] [<80028110>] warn_slowpath_fmt+0x8c/0xac
[ 3450.073519] [<8e008790>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[ 3450.088970] [<8e001ebc>] ieee80211_report_low_ack+0xf0/0x43c [mac80211]
[ 3450.102171] [<8e002220>] ieee80211_free_txskb+0x18/0x30 [mac80211]
[ 3450.114500] [<8eb652fc>] mt76_tx_complete_skb+0xa8/0xbc [mt76]
[ 3450.126114] [<8eac6d34>] mt7615_txp_skb_unmap+0x1e0/0x21c [mt7615e]
[ 3450.138586] [<80078dbc>] __handle_irq_event_percpu+0x78/0x1c4
[ 3450.150286] ---[ end trace dd01a3d0f58b9209 ]---
```

closed 2020-04-26 by hcwang

andrey-zibrov, 2020-05-31:
Model	ASUS RT-AC65P
Architecture	MediaTek MT7621 ver:1 eco:3
Firmware Version	OpenWrt SNAPSHOT r13406-f166cf9ca0 / LuCI Master git-20.150.71479-704241d
Kernel Version	5.4.42

```
[23191.741595] ------------[ cut here ]------------
[23191.751047] WARNING: CPU: 0 PID: 0 at backports-5.7-rc3-1/net/mac80211/sta_info.c:1929 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[23191.777254] STA 6c:96:cf:d7:69:1f AC 2 txq pending airtime underflow: 4294967220, 76
[23191.777262] Modules linked in: ledtrig_heartbeat pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt7615e mt7615_common mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[23191.915564] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.42 #0
[23191.927333] Stack : 8ea85e80 80077280 806c0000 806bc31c 80720000 806bc2e4 806bb438 8fc0dc44
[23191.943953]         80860000 80707248 80706d83 806512c8 00000000 00000001 8fc0dbe8 9341354b
[23191.960566]         00000000 00000000 808a0000 00000000 00000030 00000675 342e3520 2032342e
[23191.977178]         00000000 00000481 00000000 000df86c 00000000 80720000 00000000 8ea08a84
[23191.993790]         00000009 00000000 00000000 00000000 00000002 8033d688 00000000 80860000
[23192.010402]         ...
[23192.015254] Call Trace:
[23192.020129] [<8000b72c>] show_stack+0x30/0x100
[23192.028990] [<805943e4>] dump_stack+0xa4/0xdc
[23192.037657] [<80028058>] __warn+0xc0/0x10c
[23192.045795] [<80028130>] warn_slowpath_fmt+0x8c/0xac
[23192.055774] [<8ea08a84>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[23192.071224] [<8ea0205c>] ieee80211_report_low_ack+0xf0/0x43c [mac80211]
[23192.084425] [<8ea023c0>] ieee80211_free_txskb+0x18/0x30 [mac80211]
[23192.096753] [<8e9e53cc>] mt76_tx_complete_skb+0xa8/0xbc [mt76]
[23192.108366] [<8e9d9eb8>] mt7615_txp_skb_unmap+0x1f0/0x22c [mt7615_common]
[23192.122126] ---[ end trace eb35a3a58c7e1e54 ]---
```

dwmw2, 2020-06-01:
Still seeing this on RT-AC85P.

```
May 31 20:16:49 hostapd: nl80211: kernel reports: key addition failed
May 31 20:16:49 hostapd: wlan1-1: AP-STA-POLL-OK 7c:d9:5c:b5:83:5a
May 31 20:16:49 kernel: [18896.317402] ------------[ cut here ]------------
May 31 20:16:49 kernel: [18896.326817] WARNING: CPU: 0 PID: 0 at backports-5.7-rc3-1/net/mac80211/sta_info.c:1929 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
May 31 20:16:49 kernel: [18896.352992] STA 7c:d9:5c:b5:83:5a AC 2 txq pending airtime underflow: 4294967248, 48
May 31 20:16:49 kernel: [18896.352996] Modules linked in: ath9k ath9k_common pppoe ppp_async iptable_nat ath9k_hw ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT ti_usb_3410_5052 pppox ppp_generic pl2303 nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt7615e mt7615_common mt76 mac80211 ipt_REJECT ftdi_sio cp210x cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG usbserial slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat cdc_acm ledtrig_usbport ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 autofs4 usb_storage ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug f2fs usbcore nls_base usb_common crc32_generic
May 31 20:16:49 kernel: [18896.522216] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.42 #0
May 31 20:16:49 kernel: [18896.533982] Stack : 8e16de80 80070f54 80680000 8067fa50 806e0000 8067fa18 8067eb7c 8fc0dc44
May 31 20:16:49 kernel: [18896.550600]         80820000 806c9178 806c8cc3 80615cb0 00000000 00000001 8fc0dbe8 c1c887ea
May 31 20:16:49 kernel: [18896.567212]         00000000 00000000 80860000 00000000 00000030 00000198 342e3520 2032342e
May 31 20:16:49 kernel: [18896.583824]         00000000 000003ac 00000000 0007f7e8 00000000 806e0000 00000000 8eb08a84
May 31 20:16:49 kernel: [18896.600436]         00000009 00000000 00000000 00000000 00000002 8030e7dc 01800117 01800157
May 31 20:16:49 kernel: [18896.617047]         ...
May 31 20:16:49 kernel: [18896.621898] Call Trace:
May 31 20:16:49 kernel: [18896.626771] [<8000b5fc>] show_stack+0x30/0x100
May 31 20:16:49 kernel: [18896.635628] [<8055b550>] dump_stack+0xa4/0xdc
May 31 20:16:49 kernel: [18896.644296] [<80029d4c>] __warn+0xc0/0x10c
May 31 20:16:49 kernel: [18896.652434] [<80029e24>] warn_slowpath_fmt+0x8c/0xac
May 31 20:16:49 kernel: [18896.662359] [<8eb08a84>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
May 31 20:16:49 kernel: [18896.677861] [<8eb0205c>] ieee80211_report_low_ack+0xf0/0x43c [mac80211]
May 31 20:16:49 kernel: [18896.691061] [<8eb023c0>] ieee80211_free_txskb+0x18/0x30 [mac80211]
May 31 20:16:49 kernel: [18896.703386] [<8e95539c>] mt76_tx_complete_skb+0xa8/0xbc [mt76]
May 31 20:16:49 kernel: [18896.714998] [<8ea59eb8>] mt7615_txp_skb_unmap+0x1f0/0x22c [mt7615_common]
May 31 20:16:49 kernel: [18896.728738] ---[ end trace b9040f73575d422c ]---
May 31 20:16:49 hostapd: wlan1-1: STA 7c:d9:5c:b5:83:5a IEEE 802.11: associated (aid 2)
May 31 20:16:49 hostapd: wlan1-1: AP-STA-CONNECTED 7c:d9:5c:b5:83:5a
```

lynxis, 2021-12-03:
With an asus rt-ac65p I haven't seen this issue. The AP runs in a small home use case stable for > 10 days.
My device runs with OpenWrt 21.02.1 / mt76 22b690334c0f49b11534cc2e331c9d5e17c4a0bc 

@dwmw2 is it reproducible with your device (within a reasonable time frame)? Is it still happening with openwrt 21.02.1?

dwmw2, 2021-12-03:
Er... do I own an RT-AC85P? What does it look like?