openwrt/mt76 #387: Really bad 802.11s throughput between a QCA9880 and MT7612 device (5Ghz) in one direction

Rising-Sun, 2020-04-08:
Hi,

I am trying to set up a WPA3 mesh link between these devices. but the throughput between my Devolo 1750e and Xiaomi mir3g is really bad in one direction using 5Ghz wifi.
Unfortunately I don't have another wifi 5Ghz router to test if the devolo or the Xiaomi is the cause but I have read a QCA9880 NIC works well with a Intel NIC in mesh mode, so I suppose the cause lays in the mt76 driver.
The throughout results are using the mt76 driver @bf60f43b12fb14d2acff2c6608e37645c3ee4242 but the driver in the current trunk behaved the same.
Can I collect some useful data for you?

```
root@OpenWrt-Xiaomi:~# batctl tp $(mac of devolo)
Test duration 10236ms.
Sent 28640304 Bytes.
Throughput: 2.67 MB/s (22.38 Mbps)
root@OpenWrt-Xiaomi:~# batctl tp $(mac of devolo)
Test duration 10280ms.
Sent 39606984 Bytes.
Throughput: 3.67 MB/s (30.82 Mbps)
```

```
root@OpenWrt-devolo:~# batctl tp $(mac of Xiaomi)
Test duration 10164ms.
Sent 19188 Bytes.
Throughput: 1.84 KB/s (15.10 Kbps)
root@OpenWrt-devolo:~# batctl tp $(mac of Xiaomi)
Test duration 10268ms.
Sent 13284 Bytes.
Throughput: 1.26 KB/s (10.34 Kbps)
```

Thank you and the team for the great work for the mt76 driver!

Openwrtfunboy, 2020-04-08:
**Thank you and the team for the great work for the mt76 driver!**

Sounds like a sarcasm xD

Rising-Sun, 2020-05-22:
I got a new device to play around and seeing the "hostapd: bring back mesh patches" and mac80211 updates, I run few more tests. I did run the iperf3 server on the router itself. Routers hat batman_adv as mesh protocol with a mtu of 1532 set. Mesh forwrding was disabled. on channel 36 

Now consisting of these devices (with kernel 5.4 on every devices):
Asus Lyra MAP 2200 (192.168.178.18) (ath10k mainline)
Devolo 1750e (192.168.178.5) (ath10k mainline)
Xiaomi mir3g (192.1687.178.18) (with DSA patch pr-2798)

iperf3 over the mesh connection:

devolo as an iperf3 client:

```

[  5] local 192.168.178.5 port 50348 connected to 192.168.178.10 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  1.25 MBytes  10.4 Mbits/sec  132   4.24 KBytes
[  5]   1.00-2.00   sec  1.12 MBytes  9.41 Mbits/sec  110   4.24 KBytes
[  5]   2.00-3.00   sec  1.11 MBytes  9.31 Mbits/sec  106   5.66 KBytes
[  5]   3.00-4.00   sec  1004 KBytes  8.23 Mbits/sec  108   2.83 KBytes
[  5]   4.00-5.00   sec  1015 KBytes  8.32 Mbits/sec  100   2.83 KBytes
[  5]   5.00-6.00   sec  1.19 MBytes  10.0 Mbits/sec  108   4.24 KBytes
[  5]   6.00-7.00   sec  1.19 MBytes  10.0 Mbits/sec  118   4.24 KBytes
[  5]   7.00-8.00   sec  1.18 MBytes  9.90 Mbits/sec  105   4.24 KBytes
[  5]   8.00-9.00   sec  1.25 MBytes  10.5 Mbits/sec  117   4.24 KBytes
[  5]   9.00-10.00  sec  1.08 MBytes  9.10 Mbits/sec  108   2.83 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  11.4 MBytes  9.53 Mbits/sec  1112             sender
[  5]   0.00-10.00  sec  11.3 MBytes  9.50 Mbits/sec                  receiver

iperf Done.
root@OpenWrt-devolo:~# iperf3  -c 192.168.178.18
Connecting to host 192.168.178.18, port 5201
[  5] local 192.168.178.5 port 55840 connected to 192.168.178.18 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  10.4 MBytes  87.4 Mbits/sec    0    445 KBytes
[  5]   1.00-2.05   sec  14.4 MBytes   115 Mbits/sec    0    823 KBytes
[  5]   2.05-3.04   sec  16.2 MBytes   138 Mbits/sec    0    823 KBytes
[  5]   3.04-4.07   sec  16.2 MBytes   133 Mbits/sec    0    823 KBytes
[  5]   4.07-5.01   sec  15.0 MBytes   133 Mbits/sec    0    823 KBytes
[  5]   5.01-6.01   sec  16.2 MBytes   137 Mbits/sec    0    823 KBytes
[  5]   6.01-7.01   sec  16.0 MBytes   134 Mbits/sec    0    871 KBytes
[  5]   7.01-8.06   sec  17.5 MBytes   140 Mbits/sec    0    871 KBytes
[  5]   8.06-9.03   sec  16.2 MBytes   141 Mbits/sec    0    871 KBytes
[  5]   9.03-10.05  sec  16.2 MBytes   134 Mbits/sec    0    871 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.05  sec   155 MBytes   129 Mbits/sec    0             sender
[  5]   0.00-10.05  sec   155 MBytes   129 Mbits/sec                  receiver

iperf Done.
root@OpenWrt-devolo:~# iperf3  -c 192.168.178.10
Connecting to host 192.168.178.10, port 5201
[  5] local 192.168.178.5 port 50356 connected to 192.168.178.10 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   103 KBytes   845 Kbits/sec   29   2.83 KBytes
[  5]   1.00-2.00   sec  52.3 KBytes   428 Kbits/sec   14   2.83 KBytes
[  5]   2.00-3.00   sec  52.3 KBytes   429 Kbits/sec   16   2.83 KBytes
[  5]   3.00-4.00   sec  52.3 KBytes   429 Kbits/sec   16   2.83 KBytes
[  5]   4.00-5.00   sec  52.3 KBytes   429 Kbits/sec   12   2.83 KBytes
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec   16   2.83 KBytes
[  5]   6.00-7.00   sec   105 KBytes   858 Kbits/sec   20   2.83 KBytes
[  5]   7.00-8.00   sec   107 KBytes   880 Kbits/sec   20   2.83 KBytes
[  5]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec   10   2.83 KBytes
[  5]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec    7   1.41 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   525 KBytes   430 Kbits/sec  160             sender
[  5]   0.00-23.89  sec   525 KBytes   180 Kbits/sec                  receiver

[... 287 lines cut ...]

mir3g as a client:
```
root@OpenWrt-Xiaomi:~# iperf3  -c 192.168.178.5
Connecting to host 192.168.178.5, port 5201
[  5] local 192.168.178.10 port 50046 connected to 192.168.178.5 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  43.1 MBytes   361 Mbits/sec   24    383 KBytes
[  5]   1.00-2.00   sec  42.9 MBytes   360 Mbits/sec   11    329 KBytes
[  5]   2.00-3.00   sec  42.2 MBytes   354 Mbits/sec    0    402 KBytes
[  5]   3.00-4.01   sec  43.3 MBytes   361 Mbits/sec   16    365 KBytes
[  5]   4.01-5.00   sec  30.3 MBytes   255 Mbits/sec   16    307 KBytes
[  5]   5.00-6.00   sec  39.6 MBytes   332 Mbits/sec    0    378 KBytes
[  5]   6.00-7.00   sec  43.6 MBytes   365 Mbits/sec   23    324 KBytes
[  5]   7.00-8.00   sec  42.4 MBytes   355 Mbits/sec    0    397 KBytes
[  5]   8.00-9.00   sec  43.0 MBytes   362 Mbits/sec    3    342 KBytes
[  5]   9.00-10.00  sec  42.9 MBytes   359 Mbits/sec    9    288 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   413 MBytes   347 Mbits/sec  102             sender
[  5]   0.00-10.00  sec   412 MBytes   345 Mbits/sec                  receiver

iperf Done.

```

bg24623, 2020-07-22:
I seem to be having a related issue. I have a Dlink DIR-860L that I occasionally run in client mode (to get convenient wired internet ports for some device testing). My main router is an ath10k based Linksys EA8500. My internet connection normally tests out at around 70mbps down and 6mbps up. Running an version of OpenWRT 18.06.x, the 5ghz radio (7612e) in client mode will max out these speeds (as it should). If I use the 2.4Ghz radio (7602e) download rates are a bit slower, but the upload will still max out at roughly 6mbps.

Using any version of OpenWRT 19.07.x or later up through master (as of 21 July 2020, eventually presumed to be 20.x ) in the same fashion (DIR-860L as a client to an EA8500). Operating in client mode at 5GHz, I still get 70mbps down, but I only get a 100-200 kbps of upload speed. Using the 2.4GHz radio in the same fashion, I get the same results I did previously with 18.06.x. Download speeds of 40-50 mpbs and upload speed at around my 6mpbs max. 

I have no explanation at the moment for why the 5GHz client upload speed is so slow on the later builds. Given that performance with the 2.4GHz radio seems to be unaffected, it seems unlikely to be an issue with some other portion of OpenWRT, but I have no way to say for certain at the moment. I may do some build and bisecting to see if I can figure out exactly when this got broken. I suspect it is the same issue affecting your 802.11s speeds here.

bg24623, 2020-07-23:
I did some checking on the very slow 5ghz client upload issue that I believe may be related to this, and it stared with this OpenWRT commit, which updated the mt76 driver: [https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=8852a09d85e5b0c2ffab55578a9624edb2fb6dd1](url)

JosiahWhite, 2020-08-12:
Just wanted to chime in here and say I am having a similar issue with a similar setup:
B.A.T.M.A.N mesh with 802.11s between a QCA9880(TP-Link Archer C7v2) and 3xMT7612(Xiaomi Router 4A gigabit)
download from QCA9880 -> MT7612 gives 200-800KB/s
download from MT7612 -> QCA9880 gives 35-40MB/s

I have tried every network mode from N at 20mhz width to AC at 80mhz, both have similar slow downlink
I have also tried switching between an unencrypted mesh and an encrypted mesh with no change. The odd part is using the QCA9880 device in AP mode with the MT7612 device in client mode results in good speeds in both directions.
Willing to provide any other useful information.

drandyhaas, 2020-09-15:
Hi, I'm just discovering what I think is the same issue. I *believe* it is due to the extra info (28 bytes?) put on the end of each packet by batman. The switch in the mt7621 seems to like to silently drop packets that are too large, above its MTU. 
Try doing ping -s 1900 host, which sens a 1900 byte packet, and you'll see like half the large packets are dropped by the mt7621. 

`
drandyhaas@penguin:~/Downloads$ ping -s 1472 192.168.2.210
PING 192.168.2.210 (192.168.2.210) 1472(1500) bytes of data.
1480 bytes from 192.168.2.210: icmp_seq=1 ttl=62 time=5.75 ms
1480 bytes from 192.168.2.210: icmp_seq=2 ttl=62 time=6.28 ms
1480 bytes from 192.168.2.210: icmp_seq=3 ttl=62 time=10.1 ms
1480 bytes from 192.168.2.210: icmp_seq=4 ttl=62 time=10.6 ms
1480 bytes from 192.168.2.210: icmp_seq=5 ttl=62 time=6.49 ms
1480 bytes from 192.168.2.210: icmp_seq=6 ttl=62 time=9.27 ms
1480 bytes from 192.168.2.210: icmp_seq=7 ttl=62 time=9.67 ms
1480 bytes from 192.168.2.210: icmp_seq=8 ttl=62 time=11.8 ms
1480 bytes from 192.168.2.210: icmp_seq=9 ttl=62 time=7.20 ms
1480 bytes from 192.168.2.210: icmp_seq=10 ttl=62 time=9.15 ms
^C
--- 192.168.2.210 ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 25ms
rtt min/avg/max/mdev = 5.751/8.619/11.750/1.946 ms
drandyhaas@penguin:~/Downloads$ ping -s 1473 192.168.2.210
PING 192.168.2.210 (192.168.2.210) 1473(1501) bytes of data.
1481 bytes from 192.168.2.210: icmp_seq=3 ttl=62 time=15.7 ms
1481 bytes from 192.168.2.210: icmp_seq=5 ttl=62 time=9.51 ms
1481 bytes from 192.168.2.210: icmp_seq=6 ttl=62 time=11.10 ms
1481 bytes from 192.168.2.210: icmp_seq=9 ttl=62 time=11.7 ms
1481 bytes from 192.168.2.210: icmp_seq=12 ttl=62 time=11.0 ms
^C
--- 192.168.2.210 ping statistics ---
13 packets transmitted, 5 received, 61.5385% packet loss, time 204ms
rtt min/avg/max/mdev = 9.505/11.987/15.726/2.056 ms
`

AzimsTech, 2023-02-05:
I am facing the same problem. 

1. **EA7500v2** :  MT7615N (4x4:4)
2. **DIR-842 C2** :  QCA9886 (2x2:2)

Speed test result:
[image]

eastlake5, 2026-10-28:
I ran into what looks like the same problem with an MT7915 (Linksys E8450, OpenWrt 25.12.5) meshed with ath10k (QCA9984) and ath11k (IPQ8074) nodes: traffic from the Qualcomm node to the MT7915 node was 40-240 Mbit/s instead of ~800 Mbit/s, while the other direction was fine. With Claude Code’s help, I was able to fix the issue in my environment. Here is what I found.

Root cause: with hw rx A-MSDU de-aggregation, mt76_rx_complete() hands each subframe to mac80211 separately, but only the first subframe gets
RX_FLAG_DUP_VALIDATED and none gets RX_FLAG_AMSDU_MORE.  mac80211's duplicate check then drops the 3rd and later subframes of every *retransmitted* A-MSDU as duplicates.  Mesh frames keep the 802.11 header, so they go through that check; frames with rx header translation (AP/STA) do not, which is why this shows up mostly with 802.11s.  The code dates back to 4c0811c647b9 ("mt76: mt7915: enable hw rx-amsdu de-aggregation") and is still the same in current HEAD.

How to check: the loss is visible as "rx drop misc" in `iw dev <mesh-if> station dump` and as rx_duplicates of the peer station in debugfs (/sys/kernel/debug/ieee80211/phyX/netdev:<mesh-if>/stations/<peer>/rx_duplicates), growing together with the packet loss.

The patch below (against mt76 HEAD 467ae2e3; the same change applies to the mt76 in OpenWrt 25.12.5, 39c960c3) copies RX_FLAG_DUP_VALIDATED to all subframes and sets RX_FLAG_AMSDU_MORE on all but the last one, as the mac80211 documentation describes for drivers that report A-MSDU subframes separately.  With it, rx_duplicates stays at 0, UDP at 900 Mbit/s goes through without loss, and TCP reaches 830-870 Mbit/s in that direction (ch 36 HE80 with ath11k, ch 120 VHT80 with ath10k).  The workarounds mentioned here (disabling A-MSDU, smaller MTU) only avoid the trigger.  To try it on OpenWrt, put it in package/kernel/mt76/patches/ and rebuild kmod-mt76-core.

```diff
--- a/mac80211.c
+++ b/mac80211.c
@@ -1553,9 +1553,19 @@ void mt76_rx_complete(struct mt76_dev *dev, struct sk_buff_head *frames,
 	spin_lock(&dev->rx_lock);
 	while ((skb = __skb_dequeue(frames)) != NULL) {
 		struct sk_buff *nskb = skb_shinfo(skb)->frag_list;
+		struct mt76_rx_status *mstat = (struct mt76_rx_status *)skb->cb;
+		u32 dup_flag = mstat->flag & RX_FLAG_DUP_VALIDATED;
 
 		mt76_check_ccmp_pn(skb);
 		skb_shinfo(skb)->frag_list = NULL;
+		/*
+		 * All A-MSDU subframes share the sequence number of the MPDU.
+		 * Mark all but the last one with RX_FLAG_AMSDU_MORE and pass the
+		 * reorder result to every subframe, so that mac80211 does not
+		 * drop the subframes of a retransmitted A-MSDU as duplicates.
+		 */
+		if (nskb)
+			mstat->flag |= RX_FLAG_AMSDU_MORE;
 		mt76_rx_convert(dev, skb, &hw, &sta);
 		ieee80211_rx_list(hw, sta, skb, &list);
 
@@ -1565,6 +1575,10 @@ void mt76_rx_complete(struct mt76_dev *dev, struct sk_buff_head *frames,
 			nskb = nskb->next;
 			skb->next = NULL;
 
+			mstat = (struct mt76_rx_status *)skb->cb;
+			mstat->flag |= dup_flag;
+			if (nskb)
+				mstat->flag |= RX_FLAG_AMSDU_MORE;
 			mt76_rx_convert(dev, skb, &hw, &sta);
 			ieee80211_rx_list(hw, sta, skb, &list);
 		}
```

It's been a while since the last update and I am not sure if anyone is still checking this thread, but I hope this helps.

castiel652, 2026-10-28:
> hw rx A-MSDU de-aggregation

mt7915 doesn't support hw tx/rx a-msdu for mesh.

eastlake5, 2026-10-28:
Agreed on tx: in mesh mode the MT7915 sends one MSDU per frame, with no hw A-MSDU.

Rx is different, though. Hardware de-aggregation is enabled for the whole device, regardless of interface type (mt7915/init.c: mt76_set(dev, MT_MDP_DCR0, MT_MDP_DCR0_DAMSDU_EN)). The `!hdr_trans && status->amsdu` branch in mt7915_mac_fill_rx() handles A-MSDU subframes that still carry an 802.11 header, which is the mesh case.

Measured in my environment: in one UDP run, 2520 packets were lost and rx_duplicates grew by exactly 2520. With the patch, both stay at 0.

castiel652, 2026-10-28:
Well in that case I think we should figure out why rx_reorder doesn't work.

`RX_FLAG_DUP_VALIDATED` is only set in `mt76_rx_aggr_reorder`.
Start by adding debug print and see when does it bail out