openwrt/mt76 #389: mt7615e 0000:01:00.0: Failed to get patch semaphore

ShangkunLuo, 2020-04-10:
`[   11.170452] mt7615e 0000:01:00.0: card - bus=0x1, slot = 0x0 irq=4
[   11.233668] urngd: jent-rng init failed, err: 2
[   11.245011] mt7615e 0000:01:00.0: Invalid MAC address, using random address ea:47:3f:16:9e:fc
[   11.267880] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   11.369304] PPP generic driver version 2.4.2
[   11.381196] NET: Registered protocol family 24
[   11.428316] rt2800_wmac 10180000.wmac: loaded eeprom from mtd device "factory"
[   11.442790] ieee80211 phy1: rt2x00_set_rt: Info - RT chipset 6352, rev 0500 detected
[   11.458240] ieee80211 phy1: rt2x00_set_rf: Info - RF chipset 7620 detected
[   11.472699] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   11.496105] kmodloader: done loading kernel modules from /etc/modules.d/*
[   31.555841] mt7615e 0000:01:00.0: Message 16 (seq 1) timeout
[   31.567145] mt7615e 0000:01:00.0: Failed to get patch semaphore
`
**Why is the file mt7663pr2h.bin missing？
Who has the file mt7663pr2h.bin？
I use the latest commit ID 03daa60ca69c40bf3e87c70221da57465634440f
Thank you very much.**

nbd168, 2020-04-10:
What device are you using?

ShangkunLuo, 2020-04-13:
> What device are you using?

mt7620 + mt7613

mrkiko, 2020-04-14:
Hi!!

I guess the name of the device on which these chipsets are present, the router name or something like that.