openwrt/mt76 #390: wifi network freeze Intermittently

ptpt52, 2020-04-10:
The ping lost or time increase Intermittently
This issue cause network freeze
This happens in mt7603(2.4G) or mt7612e(5G)  on mt7621

ping log:

PING 192.168.16.1 (192.168.16.1) 56(84) bytes of data.
64 bytes from 192.168.16.1: icmp_seq=1 ttl=64 time=4.37 ms
64 bytes from 192.168.16.1: icmp_seq=2 ttl=64 time=1.18 ms
64 bytes from 192.168.16.1: icmp_seq=3 ttl=64 time=1.24 ms
64 bytes from 192.168.16.1: icmp_seq=4 ttl=64 time=2.70 ms
64 bytes from 192.168.16.1: icmp_seq=5 ttl=64 time=1.17 ms
64 bytes from 192.168.16.1: icmp_seq=6 ttl=64 time=1.20 ms
64 bytes from 192.168.16.1: icmp_seq=7 ttl=64 time=2.11 ms
64 bytes from 192.168.16.1: icmp_seq=8 ttl=64 time=1.15 ms
64 bytes from 192.168.16.1: icmp_seq=9 ttl=64 time=18.0 ms
64 bytes from 192.168.16.1: icmp_seq=10 ttl=64 time=1.25 ms
64 bytes from 192.168.16.1: icmp_seq=11 ttl=64 time=2.78 ms
64 bytes from 192.168.16.1: icmp_seq=12 ttl=64 time=2.30 ms
64 bytes from 192.168.16.1: icmp_seq=13 ttl=64 time=2.24 ms
64 bytes from 192.168.16.1: icmp_seq=14 ttl=64 time=2.61 ms
`64 bytes from 192.168.16.1: icmp_seq=15 ttl=64 time=224 ms`
`64 bytes from 192.168.16.1: icmp_seq=16 ttl=64 time=208 ms`
64 bytes from 192.168.16.1: icmp_seq=17 ttl=64 time=2.31 ms
64 bytes from 192.168.16.1: icmp_seq=18 ttl=64 time=2.45 ms
64 bytes from 192.168.16.1: icmp_seq=19 ttl=64 time=1.15 ms
64 bytes from 192.168.16.1: icmp_seq=20 ttl=64 time=2.67 ms
64 bytes from 192.168.16.1: icmp_seq=21 ttl=64 time=2.71 ms
64 bytes from 192.168.16.1: icmp_seq=22 ttl=64 time=0.973 ms
64 bytes from 192.168.16.1: icmp_seq=23 ttl=64 time=1.07 ms
64 bytes from 192.168.16.1: icmp_seq=24 ttl=64 time=2.75 ms
64 bytes from 192.168.16.1: icmp_seq=25 ttl=64 time=4.18 ms
64 bytes from 192.168.16.1: icmp_seq=26 ttl=64 time=3.37 ms
64 bytes from 192.168.16.1: icmp_seq=27 ttl=64 time=2.64 ms
64 bytes from 192.168.16.1: icmp_seq=28 ttl=64 time=2.51 ms
64 bytes from 192.168.16.1: icmp_seq=29 ttl=64 time=2.61 ms
64 bytes from 192.168.16.1: icmp_seq=30 ttl=64 time=2.70 ms
64 bytes from 192.168.16.1: icmp_seq=31 ttl=64 time=1.17 ms
`64 bytes from 192.168.16.1: icmp_seq=32 ttl=64 time=218 ms`
`64 bytes from 192.168.16.1: icmp_seq=33 ttl=64 time=96.1 ms`
`64 bytes from 192.168.16.1: icmp_seq=34 ttl=64 time=79.5 ms`
64 bytes from 192.168.16.1: icmp_seq=35 ttl=64 time=1.22 ms
64 bytes from 192.168.16.1: icmp_seq=36 ttl=64 time=2.70 ms
64 bytes from 192.168.16.1: icmp_seq=37 ttl=64 time=1.15 ms
64 bytes from 192.168.16.1: icmp_seq=38 ttl=64 time=2.45 ms

ptpt52, 2020-04-10:
@nbd168  any idea?

Openwrtfunboy, 2020-04-10:
Same on SNAPSHOT from 8 APR and openwrt 19.07.02

gwlim, 2020-04-29:
This is could be due to 
rcu_sched self-detected stall on MT7621 CPU.
I am getting this a lot when logging into LuCI when rpcd timeout

Hao123br, 2020-11-28:
Had a similar problem with MT7628AN, ping would get lost or really high (2000+ ms). Openwrt commit 0bb5d39b7adab53ca35d9c44fdf0200e3377f72d fixed it.