openwrt/mt76 #393: Comfast CF-926AC does not change its USB ID

BKPepe, 2020-04-11:
In OpenWrt 19.07, I installed `kmod-mt76x2u` (version: 4.14.172+2020-03-10-08054d5a-1-d3f64c5022212753017f85fd9365b83f.0) for Comfast CF-926AC. It is USB 3.0 dual-band 2T2R Wi-Fi dongle with chipset Mediatek MT7612U is detected as mass storage device.

On router:
dmesg:
```
[  706.765409] usb 2-1: new high-speed USB device number 3 using xhci-hcd
[  706.956529] usb-storage 2-1:1.0: USB Mass Storage device detected
[  706.973423] scsi host3: usb-storage 2-1:1.0
[  707.985568] scsi 3:0:0:0: CD-ROM            MediaTek Flash autorun    0.01 PQ: 0 ANSI: 0 CC
```

lsusb -v
```
Bus 002 Device 006: ID 0e8d:2870 MediaTek Inc. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0e8d MediaTek Inc.
  idProduct          0x2870 
  bcdDevice            0.01
  iManufacturer           1 Љ
  iProduct                2 Љ
  iSerial                 3 Љ
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              160mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
[... 151 lines cut ...]
        bDescriptorType         5
        bEndpointAddress     0x07  EP 7 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x09  EP 9 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               1
```

Based on the USB ID, I see that [this](https://github.com/openwrt/mt76/blob/master/mt76x2/usb.c#L16) should be supported with this driver.

If there's anything what you need, let me know.

LorenzoBianconi, 2020-04-11:
> In OpenWrt 19.07, I installed `kmod-mt76x2u` (version: 4.14.172+2020-03-10-08054d5a-1-d3f64c5022212753017f85fd9365b83f.0) for Comfast CF-926AC. It is USB 3.0 dual-band 2T2R Wi-Fi dongle with chipset [...]

the idProduct here is wrong. Could you please try latest version please?

> bcdDevice            0.01
>   iManufacturer           1 Љ
>   iProduct                2 Љ
>   iSerial                 3 Љ
>   bNumConfigurations      1
>   Configuration Descriptor: [...]

BKPepe, 2020-04-11:
I am using the latest changes in `openwrt-19.07` branch from this repository and from openwrt and its feed as well. I tried to compile `master` branch of mt76 for OpenWrt 19.07 and it fails with this ouput where all warning being treated as errors:

```
/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/mt76-2020-04-09-03daa60c/mac80211.c:286:31: error: 'NL80211_EXT_FEATURE_AQL' undeclared (first use in this function); did you mean 'NL80211_EXT_FEATURE_TXQS'?
  wiphy_ext_feature_set(wiphy, NL80211_EXT_FEATURE_AQL);
                               ^~~~~~~~~~~~~~~~~~~~~~~
                               NL80211_EXT_FEATURE_TXQS
/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/mt76-2020-04-09-03daa60c/mac80211.c:286:31: note: each undeclared identifier is reported only once for each function it appears in
  CC [M]  /build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/mt76-2020-04-09-03daa60c/mt76x02_util.o
/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/mt76-2020-04-09-03daa60c/mac80211.c: In function 'mt76_airtime_report':
/build/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/mt76-2020-04-09-03daa60c/mac80211.c:764:12: error: implicit declaration of function 'ieee80211_calc_rx_airtime'; did you mean 'ieee80211_check_tim'? [-Werror=implicit-function-declaration]
  airtime = ieee80211_calc_rx_airtime(dev->hw, &info, len);
            ^~~~~~~~~~~~~~~~~~~~~~~~~
            ieee80211_check_tim
cc1: all warnings being treated as errors

```

AQL for ath10k and ath10k-ct landed recently in the OpenWrt master branch. But I got your point, I will try the latest development OpenWrt and compile it with the latest mt76 changes to see if works for me and get back to you.

LorenzoBianconi, 2020-04-11:
> I am using the latest changes in `openwrt-19.07` branch from this repository and from openwrt and its feed as well. I tried to compile `master` branch of mt76 for OpenWrt 19.07 and it fails with [...]

You need to use OpenWrt master to fix them

Openwrtfunboy, 2020-04-11:
> You need to use OpenWrt master to fix them

Are you planning to fix many of the wifi issues that are now? Or will you continue to ignore them as well as ignore the community?
@LorenzoBianconi

LorenzoBianconi, 2020-04-12:
> Are you planning to fix many of the wifi issues that are now? Or will you continue to ignore them as well as ignore the community?
> @LorenzoBianconi

Just use OpenWrt master with mt76 master. This is not a mt76 issue, mac80211 package is just too old for new mt76

Openwrtfunboy, 2020-04-12:
> Just use OpenWrt master with mt76 master. This is not a mt76 issue, mac80211 package is just too old for new mt76

i use latest snapshot like others peoples and still have problems

LorenzoBianconi, 2020-04-12:
> i use latest snapshot like others peoples and still have problems

@Openwrtfunboy are you referring to this particular issue? (Comfast CF-926AC does not change its USB ID)
If not, please do not ask about other problems, otherwise it will impossible to track it down

lukasz1992, 2020-04-13:
Isn't there a kind od switch like for 3g/4g USB modems (usbmodeswitch?)?

BKPepe, 2020-05-07:
I'd like to apologize, @LorenzoBianconi, that it took me so long to try it on OpenWrt master. I tried it, but I still have the same issue. May I ask you if you are willing to tell me what I need to get it working or which output should it to you?

```
root@omnia:~# uname -a
Linux omnia 5.4.38 #0 SMP Thu May 7 00:06:36 2020 armv7l GNU/Linux
root@omnia:~# opkg list-installed | grep mt76
kmod-mt76 - 5.4.38+2020-04-27-5f3ccc72-2-42fd8fce3346202d14f67f3889b3b9b4.0
kmod-mt76-core - 5.4.38+2020-04-27-5f3ccc72-2-42fd8fce3346202d14f67f3889b3b9b4.0
kmod-mt76-usb - 5.4.38+2020-04-27-5f3ccc72-2-42fd8fce3346202d14f67f3889b3b9b4.0
kmod-mt7603 - 5.4.38+2020-04-27-5f3ccc72-2-42fd8fce3346202d14f67f3889b3b9b4.0
kmod-mt76x0-common - 5.4.38+2020-04-27-5f3ccc72-2-42fd8fce3346202d14f67f3889b3b9b4.0
kmod-mt76x02-common - 5.4.38+2020-04-27-5f3ccc72-2-42fd8fce3346202d14f67f3889b3b9b4.0
kmod-mt76x02-usb - 5.4.38+2020-04-27-5f3ccc72-2-42fd8fce3346202d14f67f3889b3b9b4.0
kmod-mt76x0u - 5.4.38+2020-04-27-5f3ccc72-2-42fd8fce3346202d14f67f3889b3b9b4.0
kmod-mt76x2 - 5.4.38+2020-04-27-5f3ccc72-2-42fd8fce3346202d14f67f3889b3b9b4.0
kmod-mt76x2-common - 5.4.38+2020-04-27-5f3ccc72-2-42fd8fce3346202d14f67f3889b3b9b4.0
kmod-mt76x2u - 5.4.38+2020-04-27-5f3ccc72-2-42fd8fce3346202d14f67f3889b3b9b4.0
root@omnia:~# lsusb
lBus 003 Device 002: ID 8564:7000 Transcend Information, Inc.
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 0e8d:2870 MediaTek Inc.
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
root@omnia:~# lsusb -t
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 480M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=orion-ehci/1p, 480M
```

dmesg:

```
[ 2114.441370] usb 4-1: new high-speed USB device number 3 using xhci-hcd
[ 2114.622603] usb-storage 4-1:1.0: USB Mass Storage device detected
[ 2114.628853] scsi host3: usb-storage 4-1:1.0
[ 2115.701698] scsi 3:0:0:0: CD-ROM            MediaTek Flash autorun    0.01 PQ: 0 ANSI: 0 CCS

```

BKPepe, 2020-06-22:
Still present with the kernel 5.4.46 and the latest mt76 driver from OpenWrt.

```
root@turris:~# opkg list-installed | grep mt76
kmod-mt76-core - 5.4.46+2020-05-30-85c51608-2-f07be2a948e7e5be5378934c4f9a8033.0
kmod-mt76-usb - 5.4.46+2020-05-30-85c51608-2-f07be2a948e7e5be5378934c4f9a8033.0
kmod-mt76x02-common - 5.4.46+2020-05-30-85c51608-2-f07be2a948e7e5be5378934c4f9a8033.0
kmod-mt76x02-usb - 5.4.46+2020-05-30-85c51608-2-f07be2a948e7e5be5378934c4f9a8033.0
kmod-mt76x2-common - 5.4.46+2020-05-30-85c51608-2-f07be2a948e7e5be5378934c4f9a8033.0
kmod-mt76x2u - 5.4.46+2020-05-30-85c51608-2-f07be2a948e7e5be5378934c4f9a8033.0
root@turris:~# uname -a
Linux turris 5.4.46 #0 SMP Sun Jun 21 23:33:44 2020 armv7l GNU/Linux
root@turris:~# lsusb -t
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 4, If 0, Class=Mass Storage, Driver=usb-storage, 480M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=orion-ehci/1p, 480M
```

dmesg:
```
[  693.373421] usb 4-1: new high-speed USB device number 4 using xhci-hcd
[  693.554657] usb-storage 4-1:1.0: USB Mass Storage device detected
[  693.560917] scsi host2: usb-storage 4-1:1.0
[  694.613760] scsi 2:0:0:0: CD-ROM            MediaTek Flash autorun    0.01 PQ: 0 ANSI: 0 CCS
```

@LorenzoBianconi Is there anything that I can do?

pgqz53, 2020-10-08:
@BKPepe This is NOT a mt76 issue as mt76 is not responsible for switching the device to the WiFi adapter mode. Please install `usb-modeswitch` and use this to perform the standard eject on this mass storage medium. This ID has already been shipped with `usb-modeswitch`, so all you need is to install `usb-modeswitch` and reboot your device. Here's the rule file just in case you need it (you shouldn't):
```
$ cat /usr/share/usb_modeswitch/0e8d\:2870 
# COMFAST CF-WU782AC WiFi Dongle
TargetVendor=0x0e8d
TargetProduct=0x7612
StandardEject=1
```

eacudilla, 2020-10-23:
Got the same problem on raspberry pi 4 running openwrt. The MT7612u only switch when you replug after boot. It will not automatically switch after reboot. Looking in usb-mode.json of openwrt the product key and cendor exist. usb_modeswitch doesnt work in openwrt anymore.

amk316, 2020-10-27:
usb-mode.json
```
{
	"messages" : [
		"5553424312345678000000000000061b004600000000000000000000000000",
],
"devices" : {
		"0e8d:2870": {
			"*": {
				"t_vendor": 3725,
				"t_product": [ 10352 ],
				"mode": "StandardEject",
				"msg": [ 0 ]
			}
		},
	}
}
```

hagaygo, 2021-10-05:
Tried the same dongle on a RPI 4 using latest OpenWRT 21.02 and i experience same problems.

After boot the dongle is registered as Mass storage device :

```
[    1.364738] usb 1-1.3: New USB device found, idVendor=0e8d, idProduct=2870, bcdDevice= 0.01
[    1.373093] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.373096] usb 1-1.3: Product: Љ
[    1.373098] usb 1-1.3: Manufacturer: Љ
[    1.373101] usb 1-1.3: SerialNumber: Љ
[    1.402720] usb-storage 1-1.3:1.0: USB Mass Storage device detected
```
only after physically ejecting the device and plugin back in the device is switched to wifi adapter :

```
[   74.348576] usb 1-1.3: USB disconnect, device number 3
[   77.019968] usb 1-1.3: new high-speed USB device number 4 using xhci_hcd
[   77.124896] usb 1-1.3: New USB device found, idVendor=0e8d, idProduct=2870, bcdDevice= 0.01
[   77.133274] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   77.140606] usb 1-1.3: Product: Љ
[   77.144020] usb 1-1.3: Manufacturer: Љ
[   77.147856] usb 1-1.3: SerialNumber: Љ
[   77.153698] usb-storage 1-1.3:1.0: USB Mass Storage device detected
[   77.160178] scsi host0: usb-storage 1-1.3:1.0
[   77.454089] usb 1-1.3: USB disconnect, device number 4
[   78.155991] usb 1-1.3: new high-speed USB device number 5 using xhci_hcd
[   78.261454] usb 1-1.3: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
[   78.269830] usb 1-1.3: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[   78.277165] usb 1-1.3: Product: Wireless
[   78.281190] usb 1-1.3: Manufacturer: MediaTek Inc.
[   78.285993] usb 1-1.3: SerialNumber: 000000000
[   78.376405] usb 1-1.3: reset high-speed USB device number 5 using xhci_hcd
[   78.486281] mt76x2u 1-1.3:1.0: ASIC revision: 76120044
[   78.524852] mt76x2u 1-1.3:1.0: ROM patch build: 20141115060606a
[   78.704723] mt76x2u 1-1.3:1.0: Firmware Version: 0.0.00
[   78.709963] mt76x2u 1-1.3:1.0: Build: 1
[   78.713797] mt76x2u 1-1.3:1.0: Build Time: 201507311614____
```

Tried running usbmode -s and with the supplied file , nothing happens and not output on dmesg.

Any way to make the device switched on boot ?

1yura, 2021-11-09:
If someone posts a photo of the cf-926ac board, I'll show you what parts need to be removed to disable the functionality "CD-ROM".

hagaygo, 2021-11-10:
@1yura thanks for the trying to help , but for me at least , "hacking" the hardware is not an option to overcome a software problem.

Boos4721, 2022-03-31:
I'm facing the same Problem, how can i slove it ?

PotatoLord3, 2022-04-08:
> If someone posts a photo of the cf-926ac board, I'll show you what parts need to be removed to disable the functionality "CD-ROM".

[https://photos.app.goo.gl/K12e13veCoVzVAUs5](url)

1yura, 2022-04-08:
<details>
  <summary>You need to remove these three resistors (highlighted in red):</summary>
  
[image]

  
[image]

</details>

PotatoLord3, 2022-04-08:
> You need to remove these three resistors (highlighted in red):

Thank you! it worked.

egidio85, 2022-05-06:
> 

Hi I have the same proble with dongle cf-926acV2 can you please send me an image with better quality with the detail of resistors to be removed please?
Many thanks

1yura, 2022-05-06:
I do not have an image of 926acV2. There are only 3 resistors in the red. Other 3 are not installed. If your resistors in a different place, just remove all resistors from pins 45, 46, 47 of mt7612un.

egidio85, 2022-05-08:
thank you so much.
It worked for me.

baronor, 2022-11-01:
Anyone has a solution how can do it softwarely?

dieskim, 2023-02-09:
Any update on how to make this decide mode switch on boot other than a hardware hack?

thatguy-za, 2023-06-06:
Hey @1yura, I've just picked up a Comfast CF-926AC V2 and would like to do the resistor mod. Mine has four resisters in the red area from the v1. Should all of them be removed? Thanks

https://imgur.com/gallery/g8uq7Xy

fanoush, 2023-11-26:
I just found this issue and I have same cf-926ac V2 usb dongle and some time ago I wanted to use it with openwrt but had same issue - it did not work after reboot. I had different dongle so used that one with openwrt and now I tried this cf-926ac V2  in ubuntu 22.04 and there it works just fine! After every reboot or shutdown and power on I see in dmesg output
```
[   18.525857] usb 4-2: new SuperSpeed USB device number 2 using xhci_hcd
[   18.549593] usb 4-2: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
[   18.549611] usb 4-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   18.549618] usb 4-2: Product: 802.11ac WLAN
[   18.549623] usb 4-2: Manufacturer: MediaTek Inc.
[   18.549628] usb 4-2: SerialNumber: 000000000
[   18.780008] usb 4-2: reset SuperSpeed USB device number 2 using xhci_hcd
[   18.810084] mt76x2u 4-2:1.0: ASIC revision: 76120044
[   18.886214] mt76x2u 4-2:1.0: ROM patch build: 20141115060606a
[   19.038189] mt76x2u 4-2:1.0: Firmware Version: 0.0.00
[   19.038207] mt76x2u 4-2:1.0: Build: 1
[   19.038212] mt76x2u 4-2:1.0: Build Time: 201507311614____
[   20.057576] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   20.061091] usbcore: registered new interface driver mt76x2u
[   20.102556] mt76x2u 4-2:1.0 wlx200db0c249b3: renamed from wlan0
```
however it is probably more complicated because for some reason don't see the usb storage mode in kernel output above these lines. So I disabled automatic mode switching in /etc/usb_modeswitch.conf and I still get the wifi after next reboot. So somehow in this pc the dongle does not get reset when rebooting the device so usb modeswitch is probably not even called at reboot time.

But anyway since I disabled mode switch via usb_modeswitch then when I removed the dongle and plugged  it back I get the usb storage device detected as cdrom /dev/sr0. And the interesting part is that what helps me in this state is simply ejecting the cdrom via eject command `eject /dev/sr0` - this triggers the switch and device gets detected as wifi and wireless interface comes up :-)

So you may try cdrom eject command in openwrt too and see what happens.

dyba404, 2023-12-14:
"
> I just found this issue and I have same cf-926ac V2 usb dongle and some time ago I wanted to use it with openwrt but had same issue - it did not work after reboot. I had different dongle so used [...]

> however it is probably more complicated because for some reason don't see the usb storage mode in kernel output above these lines. So I disabled automatic mode switching in /etc/usb_modeswitch.conf [...]

"

-could you elaborate more, you say that the adapter works on ubuntu 22.04, I am trying to make it work on kali-linux, I saw your comment and installed ubuntu 22.04, but I could not make it work. 
-I have the same wifi adapter (comfast cf-926ac v2), I removed the 3 resistors because it works in openwrt [wifi pineapple (gl.inet gl-ar300m16-ext)], it works correctly to un-authenticate clients from a network (monitor mode), the recon module and act as a client, so I said it should also work well kali-linux, but there are problems with drivers I think

fanoush, 2023-12-15:
> could you elaborate more, you say that the adapter works on ubuntu 22.04, I am trying to make it work on kali-linux, I saw your comment and installed ubuntu 22.04, but I could not make it work.

well, I don't have much to say as it simply works with no issues out of box (in Ubuntu Mate 22.04), just plugged it in an that's all. the only thing that was surprising for me that even after reboot it worked out of box in ubuntu unlike with openwrt. I did not remove any resistors. 

if you have same vendor and product ids like listed in previous posts it should work (mine is `idVendor=0e8d, idProduct=7612`)

dyba404, 2023-12-16:
> well, I don't have much to say as it simply works with no issues out of box (in Ubuntu Mate 22.04), just plugged it in an that's all. the only thing that was surprising for me that even after reboot [...]

-I understand, I tested it on virtual machines, maybe you can "package the driver" and upload it to your repository to rule out the problem, I have the same `idVendor=0e8d, idProduct=7612`.
you would help me a lot to me and other people who have this device, and also you could check if it works as monitor mode and test if it is the case, in openwrt (wifi pineapple), if it works the monitor mode and packet injection, but I would like to make it work in kali-linux.

markndelaney, 2024-05-10:
Solution:

opkg update usbutils
opkg install usb-modeswitch

Put the following in /etc/rc.local

/bin/sleep 5
/usr/bin/usbreset 0e8d:2870
/bin/sleep 5               
/sbin/usbmode -s -c /etc/usb-mode.json

Note:  The sleeps are important - so this is likely a bootup timing issue.    My guess is that the usb-storage driver is somehow keeping the usb resource and stopping usbmode from doing its thing.   The usbreset appears to release the usb device and allows usbmode to work after a delay.

Tested against Wifi 'AC1300 Dual Band USB Adapter'.  
Contains:  MT7612U chipset.

ID 0e8d:7612 MediaTek Inc. 802.11ac WLAN

fakemanhk, 2024-06-05:
> I do not have an image of 926acV2. There are only 3 resistors in the red. Other 3 are not installed. If your resistors in a different place, just remove all resistors from pins 45, 46, 47 of mt7612un.

Sorry for digging up this old reply, I have the MT7612 but different model (COMFAST CF-WU782AC/CF-WU785AC), do you know if we can do the same hardware mod to put them back to single state?

1yura, 2024-06-07:
> do you know if we can do the same hardware mod to put them back to single state?
This hardware mod applicable to any adapter based on the mt7612un chip. 
After this, the SPI flash memory 25q... is not used and can also be removed and used for hobby or router upgrade.

fakemanhk, 2024-06-07:
> Solution:
> 
> opkg update usbutils opkg install usb-modeswitch
> 
> Put the following in /etc/rc.local
> 
> /bin/sleep 20 /usr/bin/usbreset 0e8d:2870 /bin/sleep 5 /sbin/usbmode -s -c /etc/usb-mode.json
> 
> Note: The sleeps are important - so this is likely a bootup timing issue. My guess is that the usb-storage driver is somehow keeping the usb resource and stopping usbmode from doing its thing. The usbreset appears to release the usb device and allows usbmode to work after a delay.
> 
> Tested against: Raspberry Pi 4B running: OpenWrt 23.05.2 r23630-842932a63d USB dongle: 'Wifi 5 AC1300 Dual Band USB Adapter'. Containing: MT7612U chipset.
> 
> lsusb shortly after boot: ID 0e8d:7612 MediaTek Inc. 802.11ac WLAN

Question: When I use OpenWrt on RPi with manual unplug/plug method, though usb-modeswitch works and I can see the adaptor, but the AP I created before no longer works and I have to create a new one, which means after reboot I need to setup again, is your method capable to preserve all settings?

fakemanhk, 2024-06-08:
> > do you know if we can do the same hardware mod to put them back to single state?
> 
> This hardware mod applicable to any adapter based on the mt7612un chip. After this, the SPI flash memory 25q... is not used and can also be removed and used for hobby or router upgrade.

OK confirmed!! I opened up my 2 COMFAST MT7612U based adaptors (CF-WU782ACv2, CF-WU785AC), removing the 3 resistors according to the layout shown above works! (I disabled USB modeswitch on my Linux laptop and it can be recognized)

UIU2025, 2025-03-06:
hey everyone, i ordered the comfast mt7612u from aliexpress and i had this issue with the storage mode- removing the resistors did help ! i want to say my model was cf-922ac and the process was the same as in the pictures on this page,

my question is- i want to order the BE6500 by comfast (CF-983BE) and i'm worried i will have the same issue with no way of solving it like removing transistors - did anyone had experience with this model and if they had to remove the resistors or is it a completely new beast ? 

thank you everyone for being so helpful i would never thought of removing resistors to fix this issue ! i do now :)

JackRob, 2025-12-17:
Hi All,

Newer board revision here. Snipped off these 3 resistors and bypassed CDROM mode.
Also snipped the EPROM/NAND chip (bottom left) for good measure (not needed)💯 

[image]

stefannagy, 2026-05-30:
I think the problem is that the wifi adapter acts as a CD-ROM before usbmode does its thing and that the necessary kernel modules are not loaded. It seems to me that this is not relevant if usbmode is fast enough after connecting the device, but at startup it takes some time for usbmode to get loaded.

Please try to install `kmod-scsi-cdrom` and after that make sure sr_mod (the kernel module for SCSI CD-ROM support) is loaded early in the boot process: `~# ln -s /etc/modules.d/45-scsi-cdrom /etc/modules-boot.d/45-scsi-cdrom`

I have a different wifi adapter but mine also acts like a CDROM and I had the same issue. With this I solved the problem.

jefferzombie, 2026-11-11:
> OK confirmed!! I opened up my 2 COMFAST MT7612U based adaptors (CF-WU782ACv2, CF-WU785AC), removing the 3 resistors according to the layout shown above works! (I disabled USB modeswitch on my Linux laptop and it can be recognized)

Hi there! I have CF-WU785AC, so can you indicate the right resistors to remove? There are an total of 4 resistors, 2, 3, 6 and 7

[image]

fakemanhk, 2026-11-11:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

You need to find the one corresponding to the MT7612U chip's pin 45-47

jefferzombie, 2026-11-11:
> You need to find the one corresponding to the MT7612U chip's pin 45-47
> […](#)

Ok, but there no identification of the correspondent pins on the board. Do you not remember the pins of CF-WU785AC?