openwrt/mt76 #394: Kernel warning right after a false radar detect

dchard, 2020-04-22:
In the last two weeks this is the third time my device does a false radar detect on channel 100, but this is the first time it generated a kernel warning as well:

```
Wed Apr 22 15:22:29 2020 daemon.notice hostapd: wlan0: DFS-RADAR-DETECTED freq=5500 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5530 cf2=0
Wed Apr 22 15:22:29 2020 daemon.notice hostapd: wlan0: DFS-NEW-CHANNEL freq=5745 chan=149 sec_chan=1
Wed Apr 22 15:22:29 2020 daemon.info hostapd: wlan0: IEEE 802.11 driver starting channel switch: freq=5745, ht=1, vht_ch=0x0, offset=1, width=3 (80 MHz), cf1=5775, cf2=0
Wed Apr 22 15:22:29 2020 daemon.notice hostapd: wlan0: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5745 ht_enabled=1 ch_offset=1 ch_width=80 MHz cf1=5775 cf2=0 dfs=0
Wed Apr 22 15:22:30 2020 daemon.info hostapd: wlan0: IEEE 802.11 driver had channel switch: freq=5745, ht=1, vht_ch=0x0, offset=1, width=3 (80 MHz), cf1=5775, cf2=0
Wed Apr 22 15:22:30 2020 daemon.notice hostapd: wlan0: CTRL-EVENT-CHANNEL-SWITCH freq=5745 ht_enabled=1 ch_offset=1 ch_width=80 MHz cf1=5775 cf2=0 dfs=0
Wed Apr 22 15:22:30 2020 daemon.notice hostapd: wlan0: AP-CSA-FINISHED freq=5745 dfs=0
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.366899] ------------[ cut here ]------------
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.376455] WARNING: CPU: 2 PID: 20 at backports-5.4.27-1/net/mac80211/tx.c:4389 ieee80211_next_txq+0x2ac/0x2c0 [mac80211]
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.398699] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 iptable_mangle iptable_filter ipt_REJECT ip_tables cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG x_tables slhc nf_reject_ipv4 nf_log_ipv4 nf_log_common nf_defrag_ipv4 crc_ccitt compat ledtrig_usbport ledtrig_heartbeat leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.508800] CPU: 2 PID: 20 Comm: ksoftirqd/2 Not tainted 5.4.31 #0
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.521264] Stack : 86c48c00 8007729c 80610000 8060da7c 80670000 8060da44 8060cb98 87c91bcc
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.538059]         807b0000 87c883a4 806546c3 805aae08 00000002 00000001 87c91b70 5c149b15
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.554854]         00000000 00000000 807f0000 00000000 30230000 0000016a 00000010 342e3520
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.571649]         00000000 0000b543 00000000 6f736b20 00000000 80670000 00000000 86d30920
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.588441]         00000009 00000000 87772480 8739cb30 00000010 80324580 00000008 807b0008
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.605237]         ...
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.610273] Call Trace:
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.615325] [<8000b72c>] show_stack+0x30/0x100
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.624354] [<804f927c>] dump_stack+0xa4/0xdc
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.633197] [<80028038>] __warn+0xc0/0x10c
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.641511] [<800280e0>] warn_slowpath_fmt+0x5c/0xac
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.651679] [<86d30920>] ieee80211_next_txq+0x2ac/0x2c0 [mac80211]
Wed Apr 22 15:22:30 2020 kern.warn kernel: [932654.664461] ---[ end trace a1e01e64b1e363ea ]---
```

Device is a Dlink 860L (mt7621), I am on kernel 5.4.31 (11 days old master). I am stating it is a false detect, as first of all there is no radar in my area on channel 100 (5530MHz +-40MHz, not even by far), and second I checked with a realtime spectrum analyzer. Even the radar relatively close by I cannot pick up, as there is a mountain in between me and the radar itself.

If more info is needed, please let me know.

ryderlee1110, 2020-05-25:
DFS is really tricky to get it working right perfectly. Maybe You can drop this in your code to get rid of those disturbing stuff - https://github.com/openwrt/mt76/blob/master/mt7615/init.c#L227

dchard, 2020-05-25:
@ryderlee1110 thanks for the answer! However, I wonder how can this help as I have false detect at 80MHz mode, and the line you referring to adds 20MHz NOHT mode. Or maybe if this line is missing, it affects the whole detection process? Would be nice if you can give some more information. :-)

ryderlee1110, 2020-05-25:
I meant drop .radar_detect_widths.

thegroove, 2020-07-23:
Not sure if this is the right place to drop this, but it seemed similar enough, so here it is:

`Thu Jul 23 15:36:06 2020 daemon.notice hostapd: wlan0: DFS-RADAR-DETECTED freq=5260 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5290 cf2=0
Thu Jul 23 15:36:06 2020 daemon.notice hostapd: wlan0: DFS-NEW-CHANNEL freq=5745 chan=149 sec_chan=1
Thu Jul 23 15:36:06 2020 daemon.notice hostapd: wlan1: DFS-RADAR-DETECTED freq=5260 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5290 cf2=0
Thu Jul 23 15:36:06 2020 daemon.info hostapd: wlan0: IEEE 802.11 driver starting channel switch: freq=5745, ht=1, vht_ch=0x0, offset=1, width=3 (80 MHz), cf1=5775, cf2=0
Thu Jul 23 15:36:06 2020 daemon.notice hostapd: wlan0: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5745 ht_enabled=1 ch_offset=1 ch_width=80 MHz cf1=5775 cf2=0 dfs=0
Thu Jul 23 15:36:07 2020 daemon.info hostapd: wlan0: IEEE 802.11 driver had channel switch: freq=5745, ht=1, vht_ch=0x0, offset=1, width=3 (80 MHz), cf1=5775, cf2=0
Thu Jul 23 15:36:07 2020 daemon.notice hostapd: wlan0: CTRL-EVENT-CHANNEL-SWITCH freq=5745 ht_enabled=1 ch_offset=1 ch_width=80 MHz cf1=5775 cf2=0 dfs=0
Thu Jul 23 15:36:07 2020 daemon.notice hostapd: wlan0: AP-CSA-FINISHED freq=5745 dfs=0
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401053.863165] ------------[ cut here ]------------
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401053.867971] WARNING: CPU: 0 PID: 9 at ieee80211_next_txq+0x1c4/0x1d0 [mac80211]
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401053.875463] Modules linked in: ath9k ath9k_common pppoe ppp_async mt76x0e mt76x0_common iptable_nat ath9k_hw ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc r8169 nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables forcedeth e1000e crc_ccitt compat bnx2 i2c_dev ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ixgbe igb e1000 mdio button_hotplug tg3 ptp realtek broadcom bcm_phy_lib pps_core mii libphy
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401053.945692] CPU: 0 PID: 9 Comm: ksoftirqd/0 Not tainted 5.4.52 #0
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401053.951928] Hardware name: Hewlett-Packard HP t610 PLUS WW Thin Client/17E2, BIOS K30 v01.10 10/26/2012
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401053.961508] RIP: 0010:ieee80211_next_txq+0x1c4/0x1d0 [mac80211]
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401053.967572] Code: 8b 7d d0 e8 fe 12 62 e1 48 89 d8 48 83 c4 10 5b 41 5c 41 5d 41 5e 41 5f 5d c3 66 66 2e 0f 1f 84 00 00 00 00 00 90 55 48 89 e5 <0f> 0b 5d c3 0f 1f 84 00 00 00 00 00 55 48 89 e5 53 8b 07 83 f8 03
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401053.986518] RSP: 0018:ffffc90000067c50 EFLAGS: 00010246
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401053.991888] RAX: 0000000000000003 RBX: ffff8881187f0fc0 RCX: 0000000000000000
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401053.999205] RDX: ffffc90000067d00 RSI: ffff888117ee5438 RDI: ffff8881187f0fc0
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.006520] RBP: ffffc90000067c50 R08: 000000000000ff7f R09: 0000000000000100
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.013839] R10: 000000000000001e R11: 000000000000001f R12: ffff888117ee5438
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.021157] R13: ffff888117f8c600 R14: ffff8881187f0fc0 R15: ffff888117d04658
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.028468] FS:  0000000000000000(0000) GS:ffff88811ac00000(0000) knlGS:0000000000000000
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.036735] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.042623] CR2: 00007f60b6aa8000 CR3: 0000000117d12000 CR4: 00000000000006f0
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.049939] Call Trace:
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.052551]  ieee80211_nullfunc_get+0xc10/0xc20 [mac80211]
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.058182]  ? mt76_rx_complete+0x31/0x260 [mt76]
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.063038]  ieee80211_beacon_get_tim+0x30/0x100 [mac80211]
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.068755]  ? mt76x02_resync_beacon_timer+0xe0/0xe0 [mt76x02_lib]
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.075076]  mt76x02_update_beacon_iter+0x2c/0x40 [mt76x02_lib]
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.081144]  ieee80211_queue_stopped+0xc3/0x150 [mac80211]
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.086781]  ieee80211_iterate_active_interfaces_atomic+0xc/0x10 [mac80211]
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.093885]  mt76x02_rx_poll_complete+0x110/0x350 [mt76x02_lib]
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.099943]  tasklet_action_common.isra.19+0x51/0xc0
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.105051]  tasklet_action+0x1d/0x20
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.108863]  __do_softirq+0xc9/0x20e
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.112588]  run_ksoftirqd+0x26/0x40
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.116310]  smpboot_thread_fn+0xcb/0x170
Thu Jul 23 15:36:07 2020 kern.warn kernel: [401054.120468]  kthread+0x10b/0x130
[... cut ...]
Thu Jul 23 16:07:56 2020 daemon.notice hostapd: wlan0: DFS-NOP-FINISHED freq=5320 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5320 cf2=0`

19.x causes CPU stalls on any kind of load other than very light, so this is already a lot better, but maybe this info can offer some insights to ongoing development. This is the first time this has happened in a week or so. Not a very high-traffic network, just an average household setup.

It doesn't seem to have caused anything fatal, just the hiccup from switching the channel. I have no idea if this is a false radar detect or a legit one.

ryderlee1110, 2020-07-23:
Did you use the latest mt76 master?

thegroove, 2020-07-23:
@ryderlee1110 yes it is, lspci -vv:

```
04:00.0 Network controller: MEDIATEK Corp. Device 7612
        Subsystem: MEDIATEK Corp. Device 7612
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 17
        Region 0: Memory at fe600000 (64-bit, non-prefetchable) [size=1M]
        Expansion ROM at fe700000 [disabled] [size=64K]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Capabilities: [70] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 0.000W
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop-
                        MaxPayload 128 bytes, MaxReadReq 128 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 unlimited
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 64 bytes, Disabled- CommClk+
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s (ok), Width x1 (ok)
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR-
                         10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS- TPHComp- ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- OBFF Disabled,
                         AtomicOpsCtl: ReqEn-
                LnkCap2: Supported Link Speeds: 2.5GT/s, Crosslink- Retimer- 2Retimers- DRS-
                LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete- EqualizationPhase1-
                         EqualizationPhase2- EqualizationPhase3- LinkEqualizationRequest-
                         Retimer- 2Retimers- CrosslinkRes: unsupported
        Capabilities: [100 v2] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap+ ECRCGenEn- ECRCChkCap+ ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Capabilities: [148 v1] Device Serial Number 00-00-00-00-00-00-00-00
        Capabilities: [158 v1] Latency Tolerance Reporting
                Max snoop latency: 0ns
                Max no snoop latency: 0ns
        Capabilities: [160 v1] L1 PM Substates
                L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
                          PortCommonModeRestoreTime=50us PortTPowerOnTime=10us
                L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
                           T_CommonMode=0us LTR1.2_Threshold=0ns
                L1SubCtl2: T_PwrOn=10us
        Kernel driver in use: mt76x2e
```

dchard, 2020-07-24:
> @dchard Regarding mt7615, I think the false radar detect should be fixed by the latest commit

Mine is 7602 (2.4G), and 7612 (5G). Have not seen the issue in the last month or so.  Will keep an eye on it.