openwrt/mt76 #395: MCS 8-9 for mt76x0u

lucabianchi91, 2020-04-24:
Hi,
we have a mt7610un wifi dongle which is running properly in IBSS mode. The datasheet of the module says that MSC 8-9 are supported but `iw` reports up to MCS 7. Is there something we can do to activate them?
Many thanks

LorenzoBianconi, 2020-04-24:
@lucabianchi91 could you please try this [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/67151a3a910f52b689d316589377333abc2be902)? Do you notice any throughput increase or decrease?

lucabianchi91, 2020-04-28:
Sure, we will try as soon as we can access the office, hopefully Wednesday

lucabianchi91, 2020-04-29:
https://github.com/openwrt/mt76/commit/c2a3cced36de8bd318ae662eb3f0333329d12991
solved the issue, many thanks

closed 2020-04-29 by lucabianchi91