openwrt/mt76 #398: ASUS USB-AC51 / low 5ghz TX rate

hoqogi, 2020-04-29:
How to make it work full-speed for upload on 5ghz wifi? 

I get 100mbit down / ~3mbit up with AC51 usb dongle on my ubuntu.
Kernel is 5.4.0, but it was always like this, I guess.

I tried compiling the driver from this repo to see if the master is better than what's in my kernel, but it won't even build.

Maybe I need some specific version of this driver? I remember having better speeds with this device.

LorenzoBianconi, 2020-04-29:
How to make it work full-speed for upload on 5ghz wifi?

> I get 100mbit down / ~3mbit up with AC51 usb dongle on my ubuntu.
> Kernel is 5.4.0, but it was always like this, I guess.
>
> I tried compiling the driver from this repo to see if the master is better [...]
Could you please try the latest version available on openwrt mt76 tree?

hoqogi, 2020-04-29:
Do you mean this repo or somewhere else?

I'm struggling to build it. Here is my log: 
```
user@workstation:/tmp$ cd mt76/
user@workstation:/tmp/mt76$ make -C /lib/modules/`uname -r`/build M=$PWD
make: Entering directory '/usr/src/linux-headers-5.4.0-28-generic'
  CC [M]  /tmp/mt76/mt7603/pci.o
  CC [M]  /tmp/mt76/mt7603/soc.o
  CC [M]  /tmp/mt76/mt7603/main.o
  CC [M]  /tmp/mt76/mt7603/init.o
  CC [M]  /tmp/mt76/mt7603/mcu.o
  CC [M]  /tmp/mt76/mt7603/core.o
  CC [M]  /tmp/mt76/mt7603/dma.o
  CC [M]  /tmp/mt76/mt7603/mac.o
  CC [M]  /tmp/mt76/mt7603/eeprom.o
  CC [M]  /tmp/mt76/mt7603/beacon.o
  CC [M]  /tmp/mt76/mt7603/debugfs.o
  LD [M]  /tmp/mt76/mt7603/mt7603e.o
  CC [M]  /tmp/mt76/mt76x0/usb.o
  CC [M]  /tmp/mt76/mt76x0/usb_mcu.o
  LD [M]  /tmp/mt76/mt76x0/mt76x0u.o
  CC [M]  /tmp/mt76/mt76x0/pci.o
  CC [M]  /tmp/mt76/mt76x0/pci_mcu.o
  LD [M]  /tmp/mt76/mt76x0/mt76x0e.o
  CC [M]  /tmp/mt76/mt76x0/init.o
  CC [M]  /tmp/mt76/mt76x0/main.o
  CC [M]  /tmp/mt76/mt76x0/eeprom.o
  CC [M]  /tmp/mt76/mt76x0/phy.o
  LD [M]  /tmp/mt76/mt76x0/mt76x0-common.o
  CC [M]  /tmp/mt76/mt76x2/eeprom.o
  CC [M]  /tmp/mt76/mt76x2/mac.o
  CC [M]  /tmp/mt76/mt76x2/init.o
  CC [M]  /tmp/mt76/mt76x2/phy.o
  CC [M]  /tmp/mt76/mt76x2/mcu.o
  LD [M]  /tmp/mt76/mt76x2/mt76x2-common.o
  CC [M]  /tmp/mt76/mt76x2/pci.o
  CC [M]  /tmp/mt76/mt76x2/pci_main.o
  CC [M]  /tmp/mt76/mt76x2/pci_init.o
  CC [M]  /tmp/mt76/mt76x2/pci_mcu.o
  CC [M]  /tmp/mt76/mt76x2/pci_phy.o
  LD [M]  /tmp/mt76/mt76x2/mt76x2e.o
  CC [M]  /tmp/mt76/mt76x2/usb.o
  CC [M]  /tmp/mt76/mt76x2/usb_init.o
  CC [M]  /tmp/mt76/mt76x2/usb_main.o
  CC [M]  /tmp/mt76/mt76x2/usb_mac.o
  CC [M]  /tmp/mt76/mt76x2/usb_mcu.o
  CC [M]  /tmp/mt76/mt76x2/usb_phy.o
  LD [M]  /tmp/mt76/mt76x2/mt76x2u.o
  AR      /tmp/mt76/built-in.a
  CC [M]  /tmp/mt76/mmio.o
  CC [M]  /tmp/mt76/util.o
  CC [M]  /tmp/mt76/trace.o
  CC [M]  /tmp/mt76/dma.o
  CC [M]  /tmp/mt76/mac80211.o
/tmp/mt76/mac80211.c: In function ‘mt76_phy_init’:
/tmp/mt76/mac80211.c:286:31: error: ‘NL80211_EXT_FEATURE_AQL’ undeclared (first use in this function); did you mean ‘NL80211_EXT_FEATURE_TXQS’?
  286 |  wiphy_ext_feature_set(wiphy, NL80211_EXT_FEATURE_AQL);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~
      |                               NL80211_EXT_FEATURE_TXQS
/tmp/mt76/mac80211.c:286:31: note: each undeclared identifier is reported only once for each function it appears in
/tmp/mt76/mac80211.c: In function ‘mt76_airtime_report’:
/tmp/mt76/mac80211.c:763:12: error: implicit declaration of function ‘ieee80211_calc_rx_airtime’; did you mean ‘ieee80211_check_tim’? [-Werror=implicit-function-declaration]
  763 |  airtime = ieee80211_calc_rx_airtime(dev->hw, &info, len);
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~
      |            ieee80211_check_tim
cc1: all warnings being treated as errors
make[1]: *** [scripts/Makefile.build:275: /tmp/mt76/mac80211.o] Error 1
make: *** [Makefile:1719: /tmp/mt76] Error 2
make: Leaving directory '/usr/src/linux-headers-5.4.0-28-generic'
user@workstation:/tmp/mt76$
```

hoqogi, 2020-04-29:
I guess kernel 5.4 is not supported? Can you show me the last supported commit in the tree?

LorenzoBianconi, 2020-04-30:
@hoqogi can you please try to use this [branch](https://github.com/openwrt/mt76/commits/compat) and apply this [patch](https://github.com/openwrt/mt76/commit/c2a3cced36de8bd318ae662eb3f0333329d12991)?

hoqogi, 2020-04-30:
@LorenzoBianconi 

Here
```
user@workstation:/tmp/mt76$ git checkout compat 
Branch 'compat' set up to track remote branch 'compat' from 'origin'.
Switched to a new branch 'compat'

user@workstation:/tmp/mt76$ git cherry-pick c2a3cced36de8bd318ae662eb3f0333329d12991
[compat 96901d9] mt76: mt76x0: enable MCS 8 and MCS9
 Author: Lorenzo Bianconi <lorenzo@kernel.org>
 Date: Sat Apr 25 14:11:09 2020 +0200
 1 file changed, 5 insertions(+), 21 deletions(-)

user@workstation:/tmp/mt76$ make -C /lib/modules/`uname -r`/build M=$PWD
make: Entering directory '/usr/src/linux-headers-5.4.0-28-generic'
  CC [M]  /tmp/mt76/mt7603/pci.o
  CC [M]  /tmp/mt76/mt7603/soc.o
  CC [M]  /tmp/mt76/mt7603/main.o
  CC [M]  /tmp/mt76/mt7603/init.o
  CC [M]  /tmp/mt76/mt7603/mcu.o
  CC [M]  /tmp/mt76/mt7603/core.o
  CC [M]  /tmp/mt76/mt7603/dma.o
  CC [M]  /tmp/mt76/mt7603/mac.o
  CC [M]  /tmp/mt76/mt7603/eeprom.o
  CC [M]  /tmp/mt76/mt7603/beacon.o
  CC [M]  /tmp/mt76/mt7603/debugfs.o
  LD [M]  /tmp/mt76/mt7603/mt7603e.o
  CC [M]  /tmp/mt76/mt7615/pci.o
  CC [M]  /tmp/mt76/mt7615/init.o
  CC [M]  /tmp/mt76/mt7615/dma.o
  CC [M]  /tmp/mt76/mt7615/eeprom.o
  CC [M]  /tmp/mt76/mt7615/main.o
  CC [M]  /tmp/mt76/mt7615/mcu.o
  CC [M]  /tmp/mt76/mt7615/mac.o
  CC [M]  /tmp/mt76/mt7615/mmio.o
  CC [M]  /tmp/mt76/mt7615/debugfs.o
  CC [M]  /tmp/mt76/mt7615/trace.o
  LD [M]  /tmp/mt76/mt7615/mt7615e.o
  CC [M]  /tmp/mt76/mt76x0/usb.o
  CC [M]  /tmp/mt76/mt76x0/usb_mcu.o
  LD [M]  /tmp/mt76/mt76x0/mt76x0u.o
  CC [M]  /tmp/mt76/mt76x0/pci.o
  CC [M]  /tmp/mt76/mt76x0/pci_mcu.o
  LD [M]  /tmp/mt76/mt76x0/mt76x0e.o
  CC [M]  /tmp/mt76/mt76x0/init.o
  CC [M]  /tmp/mt76/mt76x0/main.o
  CC [M]  /tmp/mt76/mt76x0/eeprom.o
  CC [M]  /tmp/mt76/mt76x0/phy.o
  LD [M]  /tmp/mt76/mt76x0/mt76x0-common.o
  CC [M]  /tmp/mt76/mt76x2/eeprom.o
  CC [M]  /tmp/mt76/mt76x2/mac.o
  CC [M]  /tmp/mt76/mt76x2/init.o
  CC [M]  /tmp/mt76/mt76x2/phy.o
  CC [M]  /tmp/mt76/mt76x2/mcu.o
  LD [M]  /tmp/mt76/mt76x2/mt76x2-common.o
  CC [M]  /tmp/mt76/mt76x2/pci.o
  CC [M]  /tmp/mt76/mt76x2/pci_main.o
  CC [M]  /tmp/mt76/mt76x2/pci_init.o
  CC [M]  /tmp/mt76/mt76x2/pci_mcu.o
  CC [M]  /tmp/mt76/mt76x2/pci_phy.o
  LD [M]  /tmp/mt76/mt76x2/mt76x2e.o
  CC [M]  /tmp/mt76/mt76x2/usb.o
  CC [M]  /tmp/mt76/mt76x2/usb_init.o
  CC [M]  /tmp/mt76/mt76x2/usb_main.o
  CC [M]  /tmp/mt76/mt76x2/usb_mac.o
  CC [M]  /tmp/mt76/mt76x2/usb_mcu.o
  CC [M]  /tmp/mt76/mt76x2/usb_phy.o
  LD [M]  /tmp/mt76/mt76x2/mt76x2u.o
  AR      /tmp/mt76/built-in.a
  CC [M]  /tmp/mt76/mmio.o
  CC [M]  /tmp/mt76/util.o
  CC [M]  /tmp/mt76/trace.o
In file included from /tmp/mt76/trace.h:111,
                 from /tmp/mt76/trace.c:10:
./include/trace/define_trace.h:95:42: fatal error: ./trace.h: No such file or directory
   95 | #include TRACE_INCLUDE(TRACE_INCLUDE_FILE)
      |                                          ^
compilation terminated.
make[1]: *** [scripts/Makefile.build:275: /tmp/mt76/trace.o] Error 1
make: *** [Makefile:1719: /tmp/mt76] Error 2
make: Leaving directory '/usr/src/linux-headers-5.4.0-28-generic'

user@workstation:/tmp/mt76$ 
```

LorenzoBianconi, 2020-04-30:
you need add this patch to the Makefile

diff --git a/Makefile b/Makefile
index 0bc65a5..57623a9 100644
--- a/Makefile
+++ b/Makefile
@@ -17,6 +17,10 @@ mt76-$(CONFIG_PCI) += pci.o
 
 mt76-usb-y := usb.o usb_trace.o
 
+CFLAGS_trace.o := -I$(src)
+CFLAGS_usb_trace.o := -I$(src)
+CFLAGS_mt76x02_trace.o := -I$(src)
+
 mt76x02-lib-y := mt76x02_util.o mt76x02_mac.o mt76x02_mcu.o \
 		 mt76x02_eeprom.o mt76x02_phy.o mt76x02_mmio.o \
 		 mt76x02_txrx.o mt76x02_trace.o mt76x02_debugfs.o \

hoqogi, 2020-04-30:
Ok, so it builds now
compat branch + that your patch

But the situation is not improved, I still get crippled upload speed. 

When use dongle in windows machine (in the same location) it gets me full speed

So it's not about bag signal of defective module. Anything else I could try?

I can provide iw output if that might be of any help

LorenzoBianconi, 2020-04-30:
@hoqogi while transmitting traffic, could you please provide the output of:

cat /sys/kernel/debug/ieee80211/phy<x>/netdev:wlan<x>/stations/<ap-mac>/rc_stats

hoqogi, 2020-04-30:
Though I'm not sure, it might be still using module that comes with ubuntu kernel.

I did insmod .ko files in correct order before plugging the dongle.

How do I make sure that I use module I just built?

LorenzoBianconi, 2020-04-30:
dumping the table I pointed before, are you able to see mcs 8 and mcs9?

hoqogi, 2020-04-30:
@LorenzoBianconi 
done
https://pastebin.com/raw/Xd42VLT2

LorenzoBianconi, 2020-04-30:
@hoqogi ok, it is updated. Here you are transmitting at MCS5/SGI/HT40. Is the tx rate constant? what is the throughput you can get?

LorenzoBianconi, 2020-04-30:
I have no idea what does mcs or sgi mean.

> Yeah, I guess it's constant. I use speedtest.net to check speed, and
> every time I get roughly 3 or 4 mbits upload.
>
> Download is okay though, I get my 80-90 megabits down.
>
> <https://github.com/openwrt/mt76/issues/398#issuecomment-621873746>, or
> unsubscribe

- can you please paste the output of "while sleep 1; do iw dev wlan0 link |
grep tx ; done" while transmitting in uplink?

- could you please try to load the mt76-usb module with disable_usb_sg = 1 ?

hoqogi, 2020-04-30:
```tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 135.0 MBit/s VHT-MCS 6 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
	tx bitrate: 150.0 MBit/s VHT-MCS 7 40MHz short GI VHT-NSS 1
```

This is while performing upload test

hoqogi, 2020-04-30:
@LorenzoBianconi 
>could you please try to load the mt76-usb module with disable_usb_sg = 1 ?

Wow, it worked! Thank you so much!

hoqogi, 2020-04-30:
I didn't even need to build anything, intree module works fine now.
 
`echo 1 > /sys/module/mt76_usb/parameters/disable_usb_sg` 

And then I replugged the dongle and boom - full speed! 

Thank you for your support and time

LorenzoBianconi, 2020-04-30:
@hoqogi just out of curiosity:
- what cpu/motherboard are you using?
- are you using a usb3 port?

closed 2020-04-30 by hoqogi

hoqogi, 2020-04-30:
@LorenzoBianconi 
CPU is i3 6100T, I don't know about the board. It's an old OptiPlex 3050 micro.

Yes, it's usb3 port, but I had the same issue with usb2 ports

LorenzoBianconi, 2020-04-30:
> @LorenzoBianconi
> CPU is i3 6100T, I don't know about the board. It's an old OptiPlex 3050 micro.
> 
> Yes, it's usb3 port, but I had the same issue with usb2 ports

what is the output of 'lsusb -v'?

hoqogi, 2020-04-30:
https://pastebin.com/LbqfQ7HX

hoqogi, 2020-04-30:
Btw I've just tried the dongle on another device - a recent laptop. 

It is also running ubuntu and has the same uplink issues.

LorenzoBianconi, 2020-05-01:
@hoqogi I am currently connected using the same adapter, running standard fedora31 kernel (5.6.7-200.fc31) on i7 v7 laptop, and the device works fine without any change. I guess we should try a more updated kernel

hoqogi, 2020-05-01:
@LorenzoBianconi 
Try Ubuntu and it's kernels.

LorenzoBianconi, 2020-05-01:
@hoqogi I have just installed ubuntu20.04 in a vm and used qemu usb passthrough to move the usb dongle inside the guest. mt76x0u works fine so I think it is something related to the hw, probably the IOMMU driver (we had similar issue in the past)

hoqogi, 2020-05-02:
I wonder if I do actually have a faulty unit.

Now it dies randomly.

```
[12818.083038] mt76x0u 1-1.1:1.0: vendor request req:07 off:1718 failed:-110
[12821.282919] mt76x0u 1-1.1:1.0: vendor request req:07 off:1798 failed:-110
[12824.483232] mt76x0u 1-1.1:1.0: vendor request req:07 off:1718 failed:-110
```