openwrt/mt76 #399: Is it compatible with MT7615D wireless?

qiuzi, 2020-04-30:
MT7615E AC has been started in K2P, but 2.4G cannot be used

xuefer, 2020-05-08:
k2p 2.4G looks good now, but device got reboot when i enable dbdc

qiuzi, 2020-05-09:
@xuefer How to start DBDC?

xuefer, 2020-05-09:
`
echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/dbdc
`

xuefer, 2020-05-09:
are you sure it's 7615D? it seems to be 7615E in my K2P A1

qiuzi, 2020-05-09:
@xuefer  我从https://wikidevi.wi-cat.ru/MediaTek_MT7615 这里得知得

MediaTek MT7615D • Product page - 2x 2x2:2SS 802.11bgn/an+ac Wave2 MU-MIMO

xuefer, 2020-05-09:
https://www.acwifi.net/1954.html
I suppose you're almost right. It's MT7615DN

qiuzi, 2020-05-10:
@xuefer @nbd168 root@OpenWrt:~# echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/dbdc
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy0/mt76/dbdc
0
Seems unable to start? What is missing?

xuefer, 2020-05-12:
reference to #315

porentak, 2020-05-13:
DBDC is broken at least since commit e6a3f3ffe53a3716560f0dc9fc00774287e132c0

LorenzoBianconi, 2020-05-16:
@porentak @xuefer could you please try this [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/c6f649da8d09ca2c879a9f475a097c8c38b18963)?

LorenzoBianconi, 2020-05-16:
> @xuefer @nbd168 root@OpenWrt:~# echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/dbdc root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy0/mt76/dbdc
> 0
> Seems unable to start? What is missing?

I think the issue is fixed in this [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/c6f649da8d09ca2c879a9f475a097c8c38b18963)

porentak, 2020-05-16:
@LorenzoBianconi thanks. Your patch should be fine. I did the same modification as you suggested and it worked with 7615N. But I did not have time to test on 7615D yet.

porentak, 2020-05-16:
> I think the issue is fixed in this [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/c6f649da8d09ca2c879a9f475a097c8c38b18963)

One note from my side. In last OpenWRT master branch, if you do "wifi down", wlan interface stays in up state. And due to up state driver will not switch to DBDC.

I had to do "ifconfig wlan0 down" before enabling DBDC.

dengqf6, 2020-05-18:
@LorenzoBianconi It will be nice to have DBDC enabled by default with a property in DTS, like `mediatek,dbdc`

ryderlee1110, 2020-05-25:
> @xuefer How to start DBDC?

can we close this issue?

ryderlee1110, 2020-05-25:
> MT7615E AC has been started in K2P, but 2.4G cannot be used

Can we close this issue? I think he issue is fixed.

xuefer, 2020-05-25:
`
root@OpenWrt:/etc/config# wifi down
root@OpenWrt:/etc/config# dmesg
[  598.186606] device wlan0 left promiscuous mode
[  598.195715] br-lan: port 5(wlan0) entered disabled state
root@OpenWrt:/etc/config# echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/dbdc
root@OpenWrt:/etc/config# dmesg
[  598.186606] device wlan0 left promiscuous mode
[  598.195715] br-lan: port 5(wlan0) entered disabled state
[  604.010588] ------------[ cut here ]------------
[  604.019894] WARNING: CPU: 3 PID: 2862 at backports-5.4.27-1/net/wireless/core.c:849 wiphy_register+0xc24/0xc2c [cfg80211]
[  604.041773] Modules linked in: pppoe ppp_async ath5k ath pppox ppp_generic nf_conntrack_netlink mt7615e mt7615_common mt76 mac80211 lz4 iptable_nat ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_bpf xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ts_kmp ts_fsm ts_bm slhc nfnetlink nf_reject_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack_ftp nf_conncount lz4_decompress lz4_compress iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat act_connmark nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_heartbeat nf_log_ipv6 nf_log_common
[  604.042055]  ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb tun zram zsmalloc gpio_button_hotplug usbcore nls_base usb_common
[  604.243812] CPU: 3 PID: 2862 Comm: ash Tainted: G        W         5.4.34 #0
[  604.257824] Stack : 00000004 8006f488 00000000 80620000 80670000 8061b1ac 8061a5bc 863adbcc
[  604.274441]         807b0000 805b46ec 87e55ef4 80662c03 805b46ec 00000001 863adb70 a0954634
[  604.291057]         00000000 00000000 807f0000 00000000 00000030 000001e6 20202020 20202057
[  604.307671]         00000000 0000001d 00000000 0003b864 80000000 80670000 00000000 86d80f28
[  604.324285]         00000009 00000000 00000000 00000000 00000003 802e16f0 0000000c 807b000c
[  604.340900]         ...
[  604.345754] Call Trace:
[  604.350633] [<8000b600>] show_stack+0x30/0x100
[  604.359481] [<80502a88>] dump_stack+0xa4/0xdc
[  604.368153] [<8002870c>] __warn+0xc0/0x10c
[  604.376293] [<800287b4>] warn_slowpath_fmt+0x5c/0xac
[  604.386210] [<86d80f28>] wiphy_register+0xc24/0xc2c [cfg80211]
[  604.397946] [<86e00f70>] ieee80211_register_hw+0x4d0/0xc30 [mac80211]
[  604.410793] [<86d01f98>] mt76_register_phy+0x18/0x38 [mt76]
[  604.421893] [<86ce28cc>] mt7615_register_ext_phy+0x278/0x3c4 [mt7615_common]
[  604.435916] [<86ced778>] mt7615_init_debugfs+0xaf4/0xde8 [mt7615_common]
[  604.449633] ---[ end trace 3a0bfb59a9092f68 ]---
`
no reboot in k2p, but can't get it to work either. can't close this yet

kar200, 2020-06-08:
I am using a custom build based on latest master for my device (DIR-853 A3).
It uses the MT7615DN chipset. It seems to be working although It is used in the loungeroom as an AP connected to a opnsense box.

During first boot I do the following:
`
wifi down
rm /etc/config/wireless
echo "1" > /sys/kernel/debug/ieee80211/phy0/mt76/dbdc
wifi up
`
This creates a new wireless file with both network cards.
I then edit the wireless manually  "/etc/config/wireless" file as luci gives some weird options.

`
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option disabled '0'
        option htmode 'HT40'
        option channel '6'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'Openwrt'
        option encryption 'psk2+ccmp'
        option key 'password'

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11a'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0+1'
        option htmode 'VHT80'
        option channel '36'
        option disabled '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'Openwrt5'
        option encryption 'psk2+ccmp'
        option key 'password'
`

I added the DBDC in the startup through the web interface "LUCI"

`
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.
wifi down
echo "1" > /sys/kernel/debug/ieee80211/phy0/mt76/dbdc
wifi up
exit 0
`

This seems to make it work fine for me (although not extensively tested.

`root@OpenWrt:~# iw phy
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 543 bytes
        max # sched scan SSIDs: 10
        max # match sets: 16
        max # scan plans: 1
        max scan plan interval: 10
        max scan plan iterations: 0
        Retry short limit: 7
[... 217 lines cut ...]
                        * 5745 MHz [149] (11.0 dBm) (no IR)
                        * 5765 MHz [153] (11.0 dBm) (no IR)
                        * 5785 MHz [157] (11.0 dBm) (no IR)
                        * 5805 MHz [161] (11.0 dBm) (no IR)
                        * 5825 MHz [165] (11.0 dBm) (no IR)
        valid interface combinations:
                 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-G                                                                                                 O } <= 4,
                   total <= 4, #channels <= 1, STA/AP BI must match, radar detec                                                                                                 t widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 80+80 MHz, 160 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Supported extended features:
                * [ VHT_IBSS ]: VHT-IBSS
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
`

kar200, 2020-06-26:
> 
> no reboot in k2p, but can't get it to work either. can't close this yet
> 
> ```
> root@OpenWrt:/etc/config# cat /sys/kernel/debug/ieee80211/phy0/mt76/dbdc
> 0
> 
> root@OpenWrt:/etc/config# iw list | grep Band
>         Band 1:
> root@OpenWrt:/etc/config#
> ```
> 
> no secondary phy

Have you managed to make it work? I am using custom build on openwrt (DIR-853-A3) with master and it seems to work.
It has the same wireless card (MT7615DN).
I did a "wifi down" then deleted the "/etc/config/wireless" file. 
I then run

>#echo "1" > /sys/kernel/debug/ieee80211/phy0/mt76/dbdc

After doing a wifi up. A new wireless file with both interface gets created. I then manually edit the wireless file as some options are missing through Luci.

````
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option htmode 'HT40'
        option channel '6'
        option disabled '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'AP2'
        option encryption 'psk2+ccmp'
        option key 'password'
        option disabled '0'

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11a'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0+1'
        option htmode 'VHT80'
        option disabled '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'AP5'
        option encryption 'psk2+ccmp'
        option key 'password'
        option disabled '0'

````

I added (through luci-startup) the following (or edit /etc/rc.local).

````

# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.
wifi down
echo "1" > /sys/kernel/debug/ieee80211/phy0/mt76/dbdc
wifi up
echo e > /sys/class/net/eth0/queues/rx-0/rps_cpus
exit 0
````

It seems to be working fine for me. I did get the same error as yours but the wireless recovers after that.

````
[ 4205.750149] ------------[ cut here ]------------
[ 4205.759525] WARNING: CPU: 0 PID: 1275 at backports-5.7-rc3-1/net/mac80211/sta_info.c:1929 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[ 4205.786288] STA 60:f2:62:57:97:02 AC 2 txq pending airtime underflow: 4294967216, 80
[ 4205.786294] Modules linked in: ksmbd pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack_netlink nf_conntrack mt7615e mt7615_common mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY wireguard slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat ledtrig_usbport ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter
[ 4205.801944]  ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_udp_tunnel udp_tunnel nls_utf8 sha512_generic sha256_generic libsha256 seqiv jitterentropy_rng drbg md5 md4 hmac ghash_generic gf128mul gcm ecb des_generic libdes ctr cmac ccm arc4 leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[ 4206.036288] CPU: 0 PID: 1275 Comm: hostapd Not tainted 5.4.48 #0
[ 4206.048232] Stack : 8eae5e80 8007d788 00000001 00000001 806d0000 806d7d48 806d6e4c 8fc1fc24
[ 4206.064851]         808b0000 8fef7424 807230c7 8066a6fc 00000000 00000001 8fc1fbc8 4605b196
[ 4206.081464]         00000000 00000000 808f0000 00000000 00000030 000001a9 2e352064 38342e34
[ 4206.098076]         00000000 000000d1 00000000 00008dc0 80000000 80750000 00000000 00000789
[ 4206.114690]         00000009 00000000 00000000 8eae6000 00000003 80365da4 00000000 808b0000
[ 4206.131304]         ...
[ 4206.136156] Call Trace:
[ 4206.141034] [<8000c11c>] show_stack+0x30/0x100
[ 4206.149885] [<8059e210>] dump_stack+0xa4/0xdc
[ 4206.158554] [<80029504>] __warn+0xc0/0x120
[ 4206.166691] [<800295f0>] warn_slowpath_fmt+0x8c/0xac
[ 4206.176646] [<8e00b00c>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[ 4206.192088] [<8e00259c>] ieee80211_report_low_ack+0xf0/0x43c [mac80211]
[ 4206.205282] [<8e002900>] ieee80211_free_txskb+0x18/0x30 [mac80211]
[ 4206.217605] [<8ebf5b28>] mt76_tx_complete_skb+0xa8/0xbc [mt76]
[ 4206.229219] [<8eaaaa44>] mt7615_txp_skb_unmap+0x204/0x2b0 [mt7615_common]
[ 4206.243160] ---[ end trace 4b4252ee2883c01e ]---

ryderlee1110, 2020-06-26:
@kar200  @xuefer 
No need to manually enable dbdc with this patch - https://patchwork.kernel.org/patch/11622739/

kar200, 2020-06-27:
> @kar200 @xuefer
> No need to manually enable dbdc with this patch - https://patchwork.kernel.org/patch/11622739/

Thanks. I did apply the patch and indeed it is working .

porentak, 2020-07-14:
Before DBDC is enabled there is only one phy available (phy0) and command "iw phy0 info" shows Band 1 (2.4Ghz) and Band 2 (5GHz).

When you enable DBDC addition phy is created (phy1) and shows Band 2 (5GHz) but "iw phy0 info" still shows both bands.

Is this correct?

By my understanding phy0 should be 2.4GHz only, and phy1 should be 5Ghz.

Please correct me if I'm wrong.

kar200, 2020-07-14:
> Before DBDC is enabled there is only one phy available (phy0) and command "iw phy0 info" shows Band 1 (2.4Ghz) and Band 2 (5GHz).
>
> When you enable DBDC addition phy is created (phy1) and shows Band [...]

For me one stays with both bands and the other one has only 1 band (5Ghz I think).

kar200, 2020-07-14:
> Before DBDC is enabled there is only one phy available (phy0) and command "iw phy0 info" shows Band 1 (2.4Ghz) and Band 2 (5GHz).
>
> When you enable DBDC addition phy is created (phy1) and shows Band [...]

Ok Just deleted my previous comment as I pasted the wrong output.

For me this is what I get

````
root@OpenWrt:/# iw phy
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 543 bytes
        max # sched scan SSIDs: 10
        max # match sets: 16
        max # scan plans: 1
        max scan plan interval: 10
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports T-DLS.
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
        Band 2:
                Capabilities: 0x1ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
                VHT Capabilities (0x338001b2):
                        Max MPDU length: 11454
                        Supported Channel Width: neither 160 nor 80+80
                        RX LDPC
                        short GI (80 MHz)
                        TX STBC
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
[... 166 lines cut ...]
                        * 5745 MHz [149] (11.0 dBm) (no IR)
                        * 5765 MHz [153] (11.0 dBm) (no IR)
                        * 5785 MHz [157] (11.0 dBm) (no IR)
                        * 5805 MHz [161] (11.0 dBm) (no IR)
                        * 5825 MHz [165] (11.0 dBm) (no IR)
        valid interface combinations:
                 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 4,
                   total <= 4, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz, 80+80 MHz, 160 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Supported extended features:
                * [ VHT_IBSS ]: VHT-IBSS
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
````

porentak, 2020-07-31:
> Ok Just deleted my previous comment as I pasted the wrong output.
>
> For me this is what I get
>
> ```
> root@OpenWrt:/# iw phy
> Wiphy phy1
> ...
> ```

... phy0 should show only Band 1.

closed 2020-08-12 by nbd168

thelinuxdude, 2021-11-18:
On a Phicomm router based on a MT7621+MT77615D I can get the interfaces up.  But if I change the wireless to enable both phyX and run iwiinfo or wifi it blocks and gets stuck. (all from console).  If I startup with the phy disabled in the wireless file and after enable them in the wireless, run wifi or (/etc/init.d/network reload) the iwinfo commands work (do not block).  Anyone else experience its in 21.02?