openwrt/mt76 #400: not all devices can find this WIFI network

masterchop, 2020-05-02:
I have the Xiaomi MiWifi mini
Which was running pandora DDWRT version pretty well but it was too outdated.
So i decided to upgrade to: http://downloads.openwrt.org/releases/19.07.2/targets/ramips/mt7620/openwrt-19.07.2-ramips-mt7620-miwifi-mini-squashfs-sysupgrade.bin

which worked pretty well, but now, only 3 devices out of 50 cant reach the wifi network.
I dont understand what am i missing, i configure the same way as the old one, i think:

i used to have :
encryption: WPA-PSK/WAP2-PSK Mixed Mode
Cipher: Auto
Key: *******

I setup that in this new firmware and only my phone and 2 other devices can see it.
everything else like all my IoT cant reach it even the RaspberryPi3B
I ran a sudo iwlist wlan0 scan|grep ESSID and it doesnt even show in the list.
Some one said it was the  CRDA policy but not sure what he meant.

any help is welcome, i dont really want to rollback.

532910, 2020-05-02:
1. it's better to discuss it on https://forum.openwrt.org/
2. could you show your `/etc/config/wireless`

532910, 2020-05-02:
My miwifi mini works excellent with 19.07.2!

masterchop, 2020-05-02:
This is what i got in Pandora, I rolledback because i need it but i am looking for a soluion for sure:

config wifi-device 'rai0'
        option type 'mt7612'
        option hwmode '11a'
        option channel 'auto'
        option txpower '100'
        option htmode 'VHT80'
        option country 'CN'

config wifi-iface
        option device 'rai0'
        option mode 'ap'
        option ssid 'MiChop5G'
        option encryption 'psk-mixed'
        option key 'password'
        option network 'lan'

config wifi-device 'ra0'
        option type 'rt2860v2'
        option hwmode '11g'
        option channel 'auto'
        option txpower '100'
        option htmode 'HT40'
        option country 'CN'

config wifi-iface
        option device 'ra0'
        option network 'lan'
        option mode 'ap'
        option ssid 'MiChop'
        option encryption 'psk-mixed'
        option key 'password'

532910, 2020-05-02:
What is reason to show config that works? I mean show your openwrt confing that doesn't work.
Backtick it next time, please.

masterchop, 2020-05-02:
just upgrade and try to mirror the same but doesnt look the same :D

This is 19.07
config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
        option txpower '20'
        option channel 'auto'
        option htmode 'VHT40'
        option country 'CN'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option key 'password'
        option ssid 'MiChop'
        option encryption 'psk-mixed'

config wifi-device 'radio1'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path 'platform/10180000.wmac'
        option htmode 'HT20'
        option disabled '1'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'

masterchop, 2020-05-02:
what a complete dumbass, i was working with 5G that is why!!! 

532910 thank you for the follow up, it was my fault all the way. I can believe i spend so much time on this. 

Thanks again, without you i would probably stick to pandora old repos. Thanks again 👍 💯 🥇

closed 2020-05-02 by masterchop

532910, 2020-05-02:
https://en.wiktionary.org/wiki/backtick
https://github.github.com/gfm/#code-fence

2.4 GHz is **disabled**
try to set channel to '36' instead of 'auto'

masterchop, 2020-05-02:
backtick done :)
would it make a difference from auto to 36? everything just works now 👯

532910, 2020-05-02:
not all frequencies are good and I'd prefer to select it manually