openwrt/mt76 #401: MT7688AN and MT7610E transmit power throttled on Wavlink WL-WN570HA1

Andrew17856, 2020-05-04:
The Wavlink WL-WN570HA1 supports 27dBm and 25dBm on its 2.4Ghz and 5Ghz radios respectively.  See the commit page:

https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=a160f50d4f240019fafe95e80701ebb871c788fe

 - 2.4 GHz: 802.11b/g/n SoC, 1T1R, 27 dBm
 - 5 GHz: 802.11a/n/ac MT7610E, 1T1R, 25 dBm

Since upgrading to OpenWRT the signal strength is materially reduced (the radios cannot be set more than 24dBm and 18dBm respectively).  The reduced power relative to the stock firmware is confirmed using wireless clients (material reduction in signal strength).

Several people are having the problem.  See e.g. 
https://forum.openwrt.org/t/is-the-wavlink-wl-wn570ha1-fully-supported/59671

rimil0uz, 2020-05-14:
Hi. I have that device and im experiencing problems in the 5Ghz network with openwrt 19.07.2. Does not work at all. I'm really n00b using openwrt and linux so ill try to explain my self as good as I can.

Once you install openwrt 19.07.2 master, you can turn on the 5 ghz radio, in dafault it uses channel 36 and I can connect to the AP for a short period of time (1-3 min) then it starts failing. If you change channel even after a hardreset lets say to channel 100. The wifi don't even show anymore. You get a log like this:

`Thu Feb 27 21:12:22 2020 kern.info kernel: [  439.902903] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Thu Feb 27 21:12:22 2020 kern.info kernel: [  439.919586] br-lan: port 2(wlan1) entered blocking state
Thu Feb 27 21:12:22 2020 kern.info kernel: [  439.930215] br-lan: port 2(wlan1) entered disabled state
Thu Feb 27 21:12:22 2020 kern.info kernel: [  439.941134] device wlan1 entered promiscuous mode
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: Channel 100 (primary) not allowed for AP mode, flags: 0x10095b NO-IR RADAR
Thu Feb 27 21:12:22 2020 daemon.warn hostapd: wlan1: IEEE 802.11 Configured channel (100) not found from the channel list of current mode (2) IEEE 802.11a
Thu Feb 27 21:12:22 2020 daemon.warn hostapd: wlan1: IEEE 802.11 Hardware does not support configured channel
Thu Feb 27 21:12:22 2020 daemon.err hostapd: Could not select hw_mode and channel. (-3)
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->DISABLED
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: wlan1: AP-DISABLED
Thu Feb 27 21:12:22 2020 daemon.err hostapd: wlan1: Unable to setup interface.
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: wlan1: interface state DISABLED->DISABLED
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: wlan1: AP-DISABLED
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Thu Feb 27 21:12:22 2020 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1 wasn't started
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Thu Feb 27 21:12:22 2020 kern.info kernel: [  439.957085] device wlan1 left promiscuous mode
Thu Feb 27 21:12:22 2020 kern.info kernel: [  439.966103] br-lan: port 2(wlan1) entered disabled state
Thu Feb 27 21:12:22 2020 daemon.notice netifd: radio1 (2059): cat: can't open '/var/run/wifi-phy1.pid': No such file or directory
Thu Feb 27 21:12:22 2020 daemon.notice netifd: radio1 (2059): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Thu Feb 27 21:12:22 2020 daemon.notice netifd: radio1 (2059): Command failed: Invalid argument
Thu Feb 27 21:12:22 2020 daemon.notice netifd: radio1 (2059): Device setup failed: HOSTAPD_START_FAILED`

It was hard for me to find the original firmware but after some months I was able to revert back to original firmware. Although the firmware it self is buggy, the 5 ghz radio is stable, I can change to channel 100 w/o issue and the range is quite decent.

evilgreen, 2020-05-18:
I have WN570HA1 running 19.07.3, 5Ghz wi-fi Tx Power cannot be set higher than **13** dBm regardless of the country code used. Setting higher power manually using `iw phy phy1 set txpower fixed {value}` has no effect.

ThatMishakov, 2021-04-12:
Hello. I have Archer D2 (Lantiq ARX368 SoC + MT7610en 5Ghz radio) router which I'm trying to get supported by OpenWrt, and I've ran into the same issue, I can only set MT7610en transmission power to **11dBm** with mt76 driver on the OpenWrt master, and I can't even discover the network with my phone. Is there any way to fix it? The chip seems to initialize normally and reports to be working, but is obviously unusable. There are no issues with the OEM firmware/proprietary driver