openwrt/mt76 #402: mt7615 produces massive interrupt errors

tkso1997, 2020-05-04:
While the current build for my Xiaomi R3P runs very smoothly it shows a lot of interrupt errors caused my the mt7615 chips/drivers. Are these errors of any relevance or just a "cosmetic" thing?  

`root@OpenWrt:~# cat /proc/interrupts 
           CPU0       CPU1       CPU2       CPU3       
  8:   18994042   18994019   18993939   18993963  MIPS GIC Local   1  timer
  9:    1000669          0          0          0  MIPS GIC  63  IPI call
 10:          0     509858          0          0  MIPS GIC  64  IPI call
 11:          0          0    1116472          0  MIPS GIC  65  IPI call
 12:          0          0          0     557813  MIPS GIC  66  IPI call
 13:     774539          0          0          0  MIPS GIC  67  IPI resched
 14:          0    2137015          0          0  MIPS GIC  68  IPI resched
 15:          0          0    2696994          0  MIPS GIC  69  IPI resched
 16:          0          0          0    5773865  MIPS GIC  70  IPI resched
 17:          0          0          0          0  MIPS GIC  19  1e000600.gpio-bank0, 1e000600.gpio-bank1, 1e000600.gpio-bank2
 19:         12          0          0          0  MIPS GIC  33  ttyS0
 21:   40355768          0          0          0  MIPS GIC  10  1e100000.ethernet
 23:         58          0    7974871          0  MIPS GIC  11  mt7615e
 24:         38          0          0   11081282  MIPS GIC  31  mt7615e
 26:          0          0          0          0  1e000600.gpio  18  keys
ERR:     128647`

amk316, 2020-05-04:
i can say the same for my RE650, and the device connects wifi but no traffic go throw it!!!

root@OpenWrt:~# cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
  8:   23636426   23636622   23636626   23636612  MIPS GIC Local   1  timer
  9:      13089          0          0          0  MIPS GIC  63  IPI call
 10:          0       4061          0          0  MIPS GIC  64  IPI call
 11:          0          0       9031          0  MIPS GIC  65  IPI call
 12:          0          0          0       3452  MIPS GIC  66  IPI call
 13:      61588          0          0          0  MIPS GIC  67  IPI resched
 14:          0    1698071          0          0  MIPS GIC  68  IPI resched
 15:          0          0     329442          0  MIPS GIC  69  IPI resched
 16:          0          0          0     335847  MIPS GIC  70  IPI resched
 17:          0          0          0          0  MIPS GIC  19  1e000600.gpio-bank0, 1e000600.gpio-bank1, 1e000600.gpio-bank2
 19:         12          0          0          0  MIPS GIC  33  ttyS0
 21:     393474          0          0          0  MIPS GIC  10  1e100000.ethernet
 23:     943869          0          0          0  MIPS GIC  11  mt7615e
 24:    1212212          0          0          0  MIPS GIC  31  mt7615e
 26:          0          0          0          0  1e000600.gpio  18  keys
 27:          0          0          0          0  1e000600.gpio  25  keys
 30:          0          0          0          0  1e000600.gpio  30  keys
 31:          0          0          0          0  1e000600.gpio  31  keys
ERR:       2680

lantsem, 2020-05-04:
I am also worried about persistent interrupt errors. And when I used RPS for mt7615 queue, the growth rate of the error counter increased several times more from "normal".

Sixtiz, 2020-05-05:
The kernel trace above looks the same as reported in #385

tkso1997, 2020-05-12:
Anny news?

andrew-0xff, 2020-05-31:
Model	ASUS RT-AC65P
Architecture	MediaTek MT7621 ver:1 eco:3
Firmware Version	OpenWrt SNAPSHOT r13406-f166cf9ca0 / LuCI Master git-20.150.71479-704241d
Kernel Version	5.4.42

```
[23191.741595] ------------[ cut here ]------------
[23191.751047] WARNING: CPU: 0 PID: 0 at backports-5.7-rc3-1/net/mac80211/sta_info.c:1929 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[23191.777254] STA 6c:96:cf:d7:69:1f AC 2 txq pending airtime underflow: 4294967220, 76
[23191.777262] Modules linked in: ledtrig_heartbeat pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt7615e mt7615_common mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[23191.915564] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.42 #0
[23191.927333] Stack : 8ea85e80 80077280 806c0000 806bc31c 80720000 806bc2e4 806bb438 8fc0dc44
[23191.943953]         80860000 80707248 80706d83 806512c8 00000000 00000001 8fc0dbe8 9341354b
[23191.960566]         00000000 00000000 808a0000 00000000 00000030 00000675 342e3520 2032342e
[23191.977178]         00000000 00000481 00000000 000df86c 00000000 80720000 00000000 8ea08a84
[23191.993790]         00000009 00000000 00000000 00000000 00000002 8033d688 00000000 80860000
[23192.010402]         ...
[23192.015254] Call Trace:
[23192.020129] [<8000b72c>] show_stack+0x30/0x100
[23192.028990] [<805943e4>] dump_stack+0xa4/0xdc
[23192.037657] [<80028058>] __warn+0xc0/0x10c
[23192.045795] [<80028130>] warn_slowpath_fmt+0x8c/0xac
[23192.055774] [<8ea08a84>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[23192.071224] [<8ea0205c>] ieee80211_report_low_ack+0xf0/0x43c [mac80211]
[23192.084425] [<8ea023c0>] ieee80211_free_txskb+0x18/0x30 [mac80211]
[23192.096753] [<8e9e53cc>] mt76_tx_complete_skb+0xa8/0xbc [mt76]
[23192.108366] [<8e9d9eb8>] mt7615_txp_skb_unmap+0x1f0/0x22c [mt7615_common]
[23192.122126] ---[ end trace eb35a3a58c7e1e54 ]---
```

ryderlee1110, 2020-06-18:
can you revert 
https://github.com/openwrt/mt76/commit/05b23d7478feb4e96072c1cb3ef457c6d2e7d737
https://github.com/openwrt/mt76/commit/5f2f676b1f0131e007b19775a2e0378ac6cbf73c

and test again?

tkso1997, 2020-06-19:
I am using mt7621 (Xiaomi R3Pro).
I can't revert the patches as a lot has changed till the original commit.

lantsem, 2020-06-19:
@ryderlee1110 I saw this interrupts errors before this two commits with mt76 version 2020-03-17-2097f74f and RT-AC65P (mt7621). At that time ethernet driver for mt7530 with 4.14 kernel in openwrt also produces interrupts errors because of few bugs, but I run it with a bug fix patch, so interrupts errors counter increases only from running mt76.

ryderlee1110, 2020-06-19:
I cannot reproduce this issue on my MT7622(ARM CA53) platform.

tkso1997, 2020-06-19:
@ryderlee1110 I see these interrupt errors since the first time I tried mt76 about a year ago with any of the mt76 drivers on Xiaomi R3P and Xiaomi R3G (both MT7621). But not with the proprietary MTK drivers, even after weeks of usage. So apparently its MT7621 specific. 

Let me know if I can help with debugging (providing logs, testing, etc ... )