openwrt/mt76 #403: mt76x2u mac specific condition occurred

petrkalos, 2020-05-08:
I am trying to use a Netgear A6210 (mt7612u) usb dongle in `client mode` on an R7800 running vanilla `OpenWrt 19.07.2 r10947` with `kmod-mt76x2u` installed.

Very frequently (once every few hours) or in just a few seconds if the dongle is serving lots of traffic, it's getting dissasociated and I get the following in the logs
I am getting the following in dmesg on the following conditions
- Randomly every few hours
- Almost immediately if the dongle is service lots of traffic (5+MBps)
```
mt76x2u 4-1:1.0: mac specific condition occurred
```
- To rule out problematic cables I am plugging the dongle directly on the router
- To rule out the device is problematic I tried it on other devices (rpi/Windows) and didn't notice any issues

The problem is 100% reproducible hence I can provide more diagnostics.

LorenzoBianconi, 2020-05-10:
@petrkalos this conditions is used to reset the device in case of necessary (it is in the vendor sdk). You can try to comment it out and see what happen

petrkalos, 2020-05-10:
I moved to latest snapshot on Friday and it is running 247 since then without issues, I also stressed it and it seems rock solid and on top of that throughput seems very much improved.

LorenzoBianconi, 2020-05-11:
@petrkalos ok, thanks for testing. Please close the issue when you feel comfortable the issue is fixed now

closed 2020-05-17 by petrkalos