openwrt/mt76 #406: kernel warning in AP mode: STA AC 2 txq pending airtime underflow

lantsem, 2020-06-01:
Many users see this warning in AP mode with MT7615. ASUS RT-AC65P, OpenWrt master r13419-8d2c031f21, mt76 2020-05-30-85c51608-2:

```
[ 6209.357302] ------------[ cut here ]------------
[ 6209.366696] WARNING: CPU: 3 PID: 0 at backports-5.7-rc3-1/net/mac80211/sta_info.c:1929 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[ 6209.392881] STA 88:79:7e:d6:42:40 AC 2 txq pending airtime underflow: 4294967252, 44
[ 6209.392885] Modules linked in: mt7615e mt7615_common mt76 mac80211 iptable_nat ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY nf_reject_ipv4 nf_nat_ftp nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack_ftp nf_conncount iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables compat act_connmark nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[ 6209.580535] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 5.4.42 #0
[ 6209.592308] Stack : 8e881e80 80077050 80670000 806736b4 806d0000 8067367c 806727d0 8fc13c44
[ 6209.608934]         80810000 8fc3ccc4 806bad03 8060aa20 00000003 00000001 8fc13be8 9ad6bf41
[ 6209.625561]         00000000 00000000 80850000 00000000 00000030 0000015c 342e3520 2032342e
[ 6209.642190]         00000000 00000134 00000000 0008dbb7 00000000 806d0000 00000000 8e408a84
[ 6209.658810]         00000009 00000000 00000000 8e882000 00000002 80324ba4 0000000c 8081000c
[ 6209.675434]         ...
[ 6209.680290] Call Trace:
[ 6209.685172] [<8000b72c>] show_stack+0x30/0x100
[ 6209.694023] [<80552ee8>] dump_stack+0xa4/0xdc
[ 6209.702691] [<80028058>] __warn+0xc0/0x10c
[ 6209.710832] [<80028130>] warn_slowpath_fmt+0x8c/0xac
[ 6209.720835] [<8e408a84>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[ 6209.736352] [<8e40205c>] ieee80211_report_low_ack+0xf0/0x43c [mac80211]
[ 6209.749562] [<8e4023c0>] ieee80211_free_txskb+0x18/0x30 [mac80211]
[ 6209.761960] [<8ead5468>] mt76_tx_complete_skb+0xa8/0xbc [mt76]
[ 6209.773583] [<8e4aaa20>] mt7615_txp_skb_unmap+0x1f4/0x230 [mt7615_common]
[ 6209.787306] ---[ end trace 05dfd4ce6783b9a5 ]---
```

ryderlee1110, 2020-06-03:
I think this is mac80211 bug

xback, 2020-06-05:
@nbd168 
fyi

JFtico, 2020-06-06:
Also seeing similar on both 19.07.2, 19.07.3 and snapshot r12529-228bb84744 .
See kernel logs I posted in #375

lantsem, 2020-07-21:
Close it, since I no longer have a device and it doesn't look like a bug in the driver.

closed 2020-07-21 by lantsem