openwrt/mt76 #407: Kernel warning + ethernet errors - 19.07.3

JFtico, 2020-06-04:
Running on a newly installed (running for 3+ days) 19.07.3 build and when I looked at the kernel log, found the following:

`[34464.091643] ------------[ cut here ]------------
[34464.096303] WARNING: CPU: 2 PID: 0 at net/sched/sch_generic.c:320 dev_watchdog+0x1ac/0x324
[34464.104588] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[34464.104597] Modules linked in: pppoe ppp_async pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323
[34464.175343]  nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_cake nf_conntrack sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip_gre gre ifb ip_tunnel tun vfat fat nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci
[34464.246127]  xhci_mtk xhci_hcd ohci_platform ohci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 usbcore nls_base usb_common crc32c_generic
[34464.259663] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 4.14.180 #0
[34464.265730] Stack : 00000000 8ff6e840 80580000 800726fc 805b0000 80550510 00000000 00000000
[34464.274069]         8051be50 8fc0ddc4 8fc3c99c 8058a907 80516c40 00000001 8fc0dd68 ac07f5ad
[34464.282407]         00000000 00000000 806f0000 00006648 00000000 0000013a 00000005 00000000
[34464.290744]         00000000 80590000 0003f64f 00000000 00000000 805b0000 00000000 8054a5a0
[34464.299080]         80373590 00000140 00000002 8ff6e840 00000000 8029a9b8 00000008 806f0008
[34464.307418]         ...
[34464.309856] Call Trace:
[34464.312320] [<8000c7b0>] show_stack+0x58/0x100
[34464.316758] [<804581e4>] dump_stack+0xa4/0xe0
[34464.321104] [<8002f608>] __warn+0xe0/0x138
[34464.325187] [<8002f690>] warn_slowpath_fmt+0x30/0x3c
[34464.330142] [<80373590>] dev_watchdog+0x1ac/0x324
[34464.334858] [<8008940c>] call_timer_fn.isra.25+0x24/0x84
[34464.340147] [<800896c8>] run_timer_softirq+0x1bc/0x248
[34464.345281] [<80476090>] __do_softirq+0x128/0x2ec
[34464.349980] [<80033fb4>] irq_exit+0xac/0xc8
[34464.354172] [<8024d890>] plat_irq_dispatch+0xfc/0x138
[34464.359205] [<80007588>] except_vec_vi_end+0xb8/0xc4
[34464.364164] [<80008f50>] r4k_wait_irqoff+0x1c/0x24
[34464.369060] ---[ end trace b259e1aa47dc1253 ]---
[34464.373689] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[34464.379842] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[34464.379857] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0ebe0000, max=0, ctx=2089, dtx=2089, fdx=2088, next=2089
[34464.379869] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e770000, max=0, calc=2460, drx=2461
[34464.383954] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[34464.389240] mtk_soc_eth 1e100000.ethernet: PPE started
[72528.878735] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[100362.735600] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[100366.098553] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[100366.104839] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[100366.104856] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f120000, max=0, ctx=1358, dtx=1358, fdx=1357, next=1358
[100366.104868] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0f300000, max=0, calc=3773, drx=3774
[100366.108768] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[100366.114512] mtk_soc_eth 1e100000.ethernet: PPE started
[101346.098666] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[101346.104941] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[101346.104957] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0ea20000, max=0, ctx=2488, dtx=2488, fdx=2487, next=2488
[101346.104970] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e2b0000, max=0, calc=2064, drx=2065
[101346.108148] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[101346.113808] mtk_soc_eth 1e100000.ethernet: PPE started
[104822.111864] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
[106981.069253] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[106981.075526] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[106981.075542] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0e1a0000, max=0, ctx=3910, dtx=3910, fdx=3909, next=3910
[106981.075553] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0f3d0000, max=0, calc=1056, drx=1057
[106981.079018] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[106981.084469] mtk_soc_eth 1e100000.ethernet: PPE started
[113817.720706] mtk_soc_eth 1e100000.ethernet eth0: port 3 link down
[143641.063091] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[143641.069363] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[143641.069379] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f0a0000, max=0, ctx=3062, dtx=3062, fdx=3061, next=3062
[143641.069391] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0d870000, max=0, calc=565, drx=566
[143641.072683] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[143641.078316] mtk_soc_eth 1e100000.ethernet: PPE started
[154811.064268] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[154811.070559] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[154811.070576] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0e550000, max=0, ctx=717, dtx=717, fdx=716, next=717
[154811.070589] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0f3e0000, max=0, calc=1518, drx=1519
[154811.074108] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[154811.079996] mtk_soc_eth 1e100000.ethernet: PPE started
[155591.361211] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up`

The unit continues to function.

Proc Interrupts ERR is >2000 as well.

ryderlee1110, 2020-06-05:
please use the latest openwrt branch

amk316, 2020-06-05:
[    0.000000] Linux version 5.4.43 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r13495-5b9ba4a93e)) #0 SMP Thu Jun 4 20:21:42 2020

[  490.694247] ------------[ cut here ]------------
[  490.703619] WARNING: CPU: 0 PID: 0 at backports-5.7-rc3-1/net/mac80211/sta_info.c:1929 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[  490.729813] STA f8:94:c2:68:7c:d9 AC 0 txq pending airtime underflow: 4294965728, 1568
[  490.729821] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt7615e mt7615_common mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio gpio_button_hotplug
[  490.856467] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.43 #0
[  490.868234] Stack : 00000200 80077280 806c0000 806bd39c 80720000 806bd364 806bc4b8 87c0dbf4
[  490.884852]         80860000 80707248 80706d83 80652368 00000000 00000001 87c0db98 5a1261dc
[  490.901464]         00000000 00000000 808a0000 00000000 00000030 00000109 342e3520 2033342e
[  490.918076]         00000000 00000018 00000000 000d1193 00000000 80720000 00000000 86d08a84
[  490.934688]         00000009 00000288 60000211 86242480 00000002 8033d6bc 00000000 80860000
[  490.951301]         ...
[  490.956152] Call Trace:
[  490.961027] [<8000b72c>] show_stack+0x30/0x100
[  490.969890] [<80594728>] dump_stack+0xa4/0xdc
[  490.978556] [<80028058>] __warn+0xc0/0x10c
[  490.986693] [<80028130>] warn_slowpath_fmt+0x8c/0xac
[  490.996634] [<86d08a84>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[  491.012080] [<86d03a70>] ieee80211_tx_monitor+0x1044/0x126c [mac80211]
[  491.025307] ---[ end trace 79dc58a56d589e5a ]---

ryderlee1110, 2020-06-05:
see issue #406

JFtico, 2020-06-06:
Just adding that I'm now seeing 5Ghz firmware restarts as well.

`[297542.497919] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[297542.497942] mt76x2e 0000:01:00.0: Build: 1
[297542.497950] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[297542.516191] mt76x2e 0000:01:00.0: Firmware running!
[297542.526425] ieee80211 phy1: Hardware restart was requested
[297544.388085] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[297544.388107] mt76x2e 0000:01:00.0: Build: 1
[297544.388114] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[297544.406257] mt76x2e 0000:01:00.0: Firmware running!
[297544.416452] ieee80211 phy1: Hardware restart was requested
`

Disappointing that an official build released just a few weeks ago has this many issues.

As for running the latest master, well I can't sysupgrade to those due to massive changes.
I guess I'll try it on a test unit.

xyyx, 2020-08-06:
> please use the latest openwrt branch

I have same trouble with latest master branch too

azuwis, 2020-08-27:
I have the same problem (`eth0: transmit timed out`) using latest master (r14267-18fbb9aa21).

It can be easily reproduced by a speedtest, e.g. https://speedtest.net. The problem occurs when uploading, downloading is fine.

A workaround is installing sqm-scripts:
```
config queue 'wan'
	option qdisc 'fq_codel'
	option qdisc_advanced '0'
	option ingress_ecn 'ECN'
	option egress_ecn 'ECN'
	option qdisc_really_really_advanced '0'
	option itarget 'auto'
	option etarget 'auto'
	option linklayer 'none'
	option interface 'wan'
	option enabled '1'
	option download '0'
	option script 'piece_of_cake.qos'
	option upload '200000'
```

Maybe changing eth0 qdisc to fq_codel fixes the problem, I did not test it further.

fluffymadness, 2020-08-30:
Maybe it helps to narrow down this bug.....
got this on an ubiquity er-x.
This router has no wifi, and it got the internet shared from another router thats connected to the wan port of the er-x.
sqm is disabled, software offloading is disabled, but i have two active vlans on eth port 1 and 2

OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.136.49537-fb2f363
```
[54593.820302] mtk_soc_eth 1e100000.ethernet eth0: port 1 link down
[54606.539759] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[151734.775282] conntrack: generic helper won't handle protocol 47. Please consider loading the specific helper module.
[333222.268436] ------------[ cut here ]------------
[333222.277818] WARNING: CPU: 3 PID: 0 at net/sched/sch_generic.c:320 0x8038ba10
[333222.292068] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[333222.306104] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt sch_cake nf_conntrack sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit
[333222.447416]  act_mirred nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb leds_gpio gpio_button_hotplug
[333222.476537] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.14.180 #0
[333222.488823] Stack : 00000000 00000000 00000000 8ff51340 00000000 00000000 00000000 00000000
[333222.505625]         00000000 00000000 00000000 00000000 00000000 00000001 8fc0fd60 53261655
[333222.522432]         8fc0fdf8 00000000 00000000 00003c60 00000038 8049c0d8 00000008 00000000
[333222.539238]         00000000 80550000 00074579 70617773 8fc0fd40 00000000 00000000 8050ae44
[333222.556039]         8038ba10 00000140 00000003 8ff51340 00000008 802accb0 0000000c 806b000c
[333222.572840]         ...
[333222.577869] Call Trace:
[333222.577885] [<8049c0d8>] 0x8049c0d8
[333222.590004] [<8038ba10>] 0x8038ba10
[333222.597102] [<802accb0>] 0x802accb0
[333222.604202] [<8000c1a0>] 0x8000c1a0
[333222.611303] [<8000c1a8>] 0x8000c1a8
[333222.618401] [<80484f34>] 0x80484f34
[333222.625500] [<80071a90>] 0x80071a90
[333222.632601] [<8002e608>] 0x8002e608
[333222.639701] [<8038ba10>] 0x8038ba10
[333222.646799] [<8002e690>] 0x8002e690
[333222.653897] [<800550e8>] 0x800550e8
[333222.660998] [<8038ba10>] 0x8038ba10
[333222.668101] [<80099940>] 0x80099940
[333222.675206] [<8038b864>] 0x8038b864
[333222.682307] [<8008850c>] 0x8008850c
[333222.689407] [<8005f1fc>] 0x8005f1fc
[333222.696505] [<800887c8>] 0x800887c8
[333222.703604] [<800790f8>] 0x800790f8
[333222.710707] [<804a2ed8>] 0x804a2ed8
[333222.717807] [<80032fb4>] 0x80032fb4
[333222.724906] [<8025a2f0>] 0x8025a2f0
[333222.732008] [<80007488>] 0x80007488
[333222.739105] 
[333222.742319] ---[ end trace fd0bcdbc23326c74 ]---
[333222.751687] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[333222.764163] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[333222.776295] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0edc0000, max=0, ctx=3907, dtx=3907, fdx=3906, next=3907
[333222.798107] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0eb50000, max=0, calc=3629, drx=3630
[333222.821865] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[333222.841706] mtk_soc_eth 1e100000.ethernet: PPE started
[389475.769398] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[389475.781896] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[389475.794041] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0f2f0000, max=0, ctx=112, dtx=112, fdx=111, next=112
[389475.815161] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0d4d0000, max=0, calc=2242, drx=2243
[389475.836843] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[389475.856661] mtk_soc_eth 1e100000.ethernet: PPE started
```

azuwis, 2020-09-02:
It seems disabling `packet_steering` fixes the problem, in `/etc/config/network`:

```
config globals 'globals'
	option packet_steering '0'
```

and `reboot`, tested on zbt-wg3526-16m r14365-46abcb3ade.

It was enabled by default for mt7621 in https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=2c198ea162e900ad909f81cf6dabdf510d30fbe8, which is not included in 19.07.3, odd for this issue.

JFtico, 2020-09-03:
> It seems disabling `packet_steering` fixes the problem, in `/etc/config/network`:

Thanks for the tip, after 24hrs, both a 19.07.3 (setting absent) unit and one based on Snapshot r13978-d15b6e4895 (which did have packet_steering explicitly enabled), show no errors.

But then again, it was not always predictable, I had the 19.07.3 unit running for >6 days (as the primary home router) without triggering the issue before I applied this change.

BTW- NBD just committed a roll-back of the enabling of this setting to his staging tree: https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=commit;h=8a1dab11d19525377df8a12ce856e05ef312d76b

So it looks like packet_sterring off is the correct setting for the M76 platform.

JFtico, 2022-03-15:
Well, it is back (or still here) in 19.07.9

Even WITH packet_steering  set to disabled (as noted above), as that is the default on my builds. 

The following error occurred 2hrs after boot, having recently (2 minutes) changed SQM settings (I run SQM all the time), and this happened during or just after a speed test:

`
Tue Mar 15 14:18:44 2022 user.notice SQM: Stopping SQM on eth0.2
Tue Mar 15 14:18:45 2022 user.notice SQM: Starting SQM script: layer_cake.qos on eth0.2, in: 200000 Kbps, out: 24281 Kbps
Tue Mar 15 14:18:45 2022 user.notice SQM: Using generic sqm_start_default function.
Tue Mar 15 14:18:45 2022 user.notice SQM: layer_cake.qos was started on eth0.2 successfully

Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.103251] ------------[ cut here ]------------
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.107931] WARNING: CPU: 1 PID: 0 at net/sched/sch_generic.c:320 dev_watchdog+0x1ac/0x324
Tue Mar 15 14:20:41 2022 kern.info kernel: [ 8590.116206] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.116215] Modules linked in: pppoe ppp_async pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.186952]  nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_cake nf_conntrack sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip_gre gre ifb ip_tunnel tun vfat fat nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb mmc_block usb_storage mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.257731]  xhci_mtk xhci_hcd ohci_platform ohci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 usbcore nls_base usb_common crc32c_generic
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.271245] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.14.267 #0
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.277319] Stack : 00000000 8fe6ca40 80590000 8007304c 805b0000 80557224 00000000 00000000
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.285658]         80522890 8fc0bdc4 8fc3c33c 80592947 8051d600 00000001 8fc0bd68 ac07f5e6
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.293996]         00000000 00000000 806f0000 00006c10 00000000 0000014a 00000005 00000000
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.302331]         00000000 80590000 0004238d 00000000 00000000 805b0000 00000000 8055120c
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.310668]         803784e8 00000140 00000001 8fe6ca40 00000000 8029ecb0 00000004 806f0004
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.319014]         ...
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.321454] Call Trace:
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.323928] [<8000c7a0>] show_stack+0x58/0x100
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.328373] [<8045e014>] dump_stack+0xa4/0xe0
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.332721] [<8002f938>] __warn+0xe0/0x138
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.336802] [<8002f9c0>] warn_slowpath_fmt+0x30/0x3c
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.341762] [<803784e8>] dev_watchdog+0x1ac/0x324
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.346480] [<80089e78>] call_timer_fn.isra.25+0x24/0x84
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.351769] [<8008a134>] run_timer_softirq+0x1bc/0x248
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.356908] [<8047bfc0>] __do_softirq+0x128/0x2ec
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.361614] [<80034304>] irq_exit+0xac/0xc8
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.365812] [<802512d0>] plat_irq_dispatch+0xfc/0x138
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.370852] [<80007588>] except_vec_vi_end+0xb8/0xc4
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.375822] [<80008f50>] r4k_wait_irqoff+0x1c/0x24
Tue Mar 15 14:20:41 2022 kern.warn kernel: [ 8590.380761] ---[ end trace 37c1997030d849b2 ]---
Tue Mar 15 14:20:41 2022 kern.err kernel: [ 8590.385449] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
Tue Mar 15 14:20:41 2022 kern.info kernel: [ 8590.391699] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
Tue Mar 15 14:20:41 2022 kern.info kernel: [ 8590.391755] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0ebb0000, max=0, ctx=2692, dtx=2692, fdx=2691, next=2692
Tue Mar 15 14:20:41 2022 kern.info kernel: [ 8590.391787] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e740000, max=0, calc=960, drx=961
Tue Mar 15 14:20:41 2022 kern.info kernel: [ 8590.396307] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
Tue Mar 15 14:20:41 2022 kern.info kernel: [ 8590.405605] mtk_soc_eth 1e100000.ethernet: PPE started
`