openwrt/mt76 #408: mt7621a + mt7615e - partial connectivity loss after a day

quarkysg, 2020-06-05:
Hi all,

I'm currently running my own OpenWrt master branch (https://github.com/quarkysg/openwrt/tree/master-ramips) build for my Linksys EA7500 v2, which is a mt7621a board with 2 x mt7615e WiFi chip.  I have to say that WiFi connectivity is excellent for both 2G and 5G, both in terms of range and speed.  Well done dev!

The only minor issue I encounter is that I'm experiencing partial connectivity loss for my WiFi clients after about a day.  My WiFi clients will suddenly lose connectivity internally within the LAN.  I would not be able to connect to my WiFi nodes (e.g. ssh) when this happens.  It is as seems that host isolation is suddenly turned on, but it includes wired hosts as well.  To restore connectivity, I need to log in to Luci and restart the WiFi interfaces, without rebooting the router.

There are no exception or debug logs shown in the system or kernel logs.  Not sure if this is hostapd issue or the mt76 driver issue.

WiFi devices can still access Internet and the router Luci page when this happens.

I hope the dev can help look into this issue.

Thanks.

nbd168, 2020-06-05:
There were some issues with the switch driver as well, for which I pushed some fixes last night.
In my setup it was showing up as something that looked like WLAN clients couldn't communicate with random hosts on my LAN anymore. Maybe your case is similar.
Either way, latest OpenWrt master also contains a newer mt76 version, so you may want to give it a try

quarkysg, 2020-06-05:
I've built a new version of master with the DSA switch patches and have upgraded my Linksys EA7500 v2.  Will monitor and report back if I see the same issue again.

Thanks!

quarkysg, 2020-06-07:
I'm sorry to report that the issue I encountered is still unresolved, so it doesn't look like it's caused by the DSA switch.  I still have to restart the wireless interfaces via Luci to restore connectivity between wireless clients.

lantsem, 2020-06-07:
See #350. Also I am observing the other side of this bug in #385. This is currently the main driver stability issue from my point of view.

quarkysg, 2020-06-11:
@lantsem could be right with [#350](https://github.com/openwrt/mt76/issues/350).  I configured both wireless interfaces to 12 hours between GTK rekey intervals.  So far I do not see the connectivity issue for 2 days.  So looks like it is an issue with hostapd rather than the mt7615 driver.

Will continue to monitor.

quarkysg, 2020-06-14:
Looks like the mt76 driver for the mt7615 chipset is stable.  It’s been more than 5 days since my change to the GTK rekey intervals and I’ve not encounter the issue since. Likely a problem with hostapd instead of the mt76 driver. 

I’ll close the issue. 

Great work dev!

closed 2020-06-14 by quarkysg