openwrt/mt76 #409: Netgear A6210 (MT7612U) speed

dguerri, 2020-06-09:
Hello there,
more a question than an issue (for now).

what speed should I expect from a Netgear A6210 (MT7612U) in 802.11s or IBSS mode using AC and 80Mhz channel?

I performed some tests in ideal conditions (i.e. no significant interferences and antennas few meters apart) and I got ~240Mbps at most.

This result is quite stable and reproducible, but I expected to get 866 Mbps (minus overhead, of course).

My configuration is:

Channel 36, VHT80, 802.11s, WPA3 (SAE), no mesh forwarding.
I have been using iperf3 with the IPv6 link local address, no additional routing protocol or traffic of any sort on the link. 

Thanks!

dguerri, 2020-06-09:
FWIW, OpenWRT is telling me that RX/TX is as follow:
 `780.0 Mbit/s, 80 MHz, VHT-MCS 8, VHT-NSS 2`

dguerri, 2020-06-09:
```
root@rpi4:~# iw dev mesh0 station dump
Station 44:a5:6e:xx:xx:xx (on mesh0)
	inactive time:	96 ms
	rx bytes:	291178641
	rx packets:	189919
	tx bytes:	11243913
	tx packets:	95357
	tx retries:	1957
	tx failed:	1
	rx drop misc:	6
	signal:  	-32 [-32, -34] dBm
	signal avg:	-31 [-31, -34] dBm
	Toffset:	102297998 us
	tx bitrate:	702.0 MBit/s VHT-MCS 8 80MHz VHT-NSS 2
	tx duration:	1428372 us
	rx bitrate:	780.0 MBit/s VHT-MCS 8 80MHz short GI VHT-NSS 2
	rx duration:	38491 us
	airtime weight: 256
	expected throughput:	362.90Mbps
	mesh llid:	0
	mesh plid:	0
	mesh plink:	ESTAB
	mesh local PS mode:	ACTIVE
	mesh peer PS mode:	ACTIVE
	mesh non-peer PS mode:	ACTIVE
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		yes
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	connected time:	84 seconds
	associated at [boottime]:	556.637s
	associated at:	1591699679203 ms
	current time:	1591699763221 ms
```

ryderlee1110, 2020-06-11:
80Mhz VHT-MCS9 866 Mbps is just theoretical value, and the actual date rate you can get is 866 Mbps * 0.7x. 

tx bitrate:	702.0 MBit/s VHT-MCS 8 80MHz VHT-NSS 2
rx bitrate:	780.0 MBit/s VHT-MCS 8 80MHz short GI VHT-NSS 2

The data  is not reach MCS 9 short GI. Rate adaptation is the determination of the optimal data transmission rate most appropriate for current wireless channel conditions.

dguerri, 2020-06-11:
> 80Mhz VHT-MCS9 866 Mbps is just a theoretical value, and the actual date rate you can get is 866 Mbps * (0.7 ~0.8). 

Yep, I was expecting something around that value.

> The data didn't hit the ceiling - MCS 9 short GI. Rate adaptation is the determination of the optimal data transmission rate most appropriate for current wireless channel conditions.

That is very interesting. TIL, thanks.
As it turned out the channel was busier than expected. I moved to 149, which is free (see my screenshots) and I am still surprised by the low rate.
Keep in mind that radios were only ~1m apart.

What are the factors that can affect rate adaptation in my scenario?

LorenzoBianconi, 2020-06-11:
> Yep, I was expecting something around that value.
>
>
> That is very interesting. TIL, thanks.
> As it turned out the channel was busier than expected. I moved to 149, which is free (see my screenshots) [...]

You can look at tx rate statistics in
$cat /sys/kernel/debug/ieee80211/phy<x>/netdev:wlan<y>/stations/<mac>/rc_stats

dguerri, 2020-06-11:
FWIW, I can get to `VHT-MCS 9` now that I moved to channel 149, although it's not stable:

```
Station 44:a5:6e:3e:55:ec (on mesh0)
	inactive time:	1212 ms
	rx bytes:	36113921
	rx packets:	95034
	tx bytes:	1060521203
	tx packets:	346751
	tx retries:	7891
	tx failed:	1
	beacon loss:	1
	beacon rx:	2269
	rx drop misc:	0
	signal:  	-40 [-40, -40] dBm
	signal avg:	-40 [-40, -41] dBm
	beacon signal avg:	-40 dBm
	tx bitrate:	866.7 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 2
	tx duration:	11330313 us
	rx bitrate:	866.7 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 2
	rx duration:	784818 us
	airtime weight: 256
	expected throughput:	464.904Mbps
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		yes
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short slot time:yes
	connected time:	257 seconds
	associated at [boottime]:	27.990s
	associated at:	1591821656819 ms
	current time:	1591821910363 ms
```

These are the stats from debugfs, but I will have to do some research before I can read it correctly :)

```
root@hallway:~# cat /sys/kernel/debug/ieee80211/phy0/netdev:mesh0/stations/44\:a5\:6e\:3e\:55\:ec/rc_stats

              best   ____________rate__________    ____statistics___    _____last____    ______sum-of________
mode guard #  rate  [name   idx airtime  max_tp]  [avg(tp) avg(prob)]  [retry|suc|att]  [#success | #attempts]
VHT20 LGI 1         MCS0/1  170    1477     4.8       4.8      86.1       3     0 0            19   22
VHT20 LGI 1         MCS1/1  171     738     9.7       0.0       0.0       0     0 0             0   0
VHT20 LGI 1         MCS2/1  172     492    17.0      17.0     100.0       0     0 0             1   1
VHT20 LGI 1         MCS3/1  173     369    21.9       0.0       0.0       0     0 0             0   0
VHT20 LGI 1         MCS4/1  174     246    34.1       0.0       0.0       0     0 0             0   0
VHT20 LGI 1         MCS5/1  175     185    46.3       0.0       0.0       0     0 0             0   0
VHT20 LGI 1         MCS6/1  176     164    51.2       0.0       0.0       0     0 0             0   0
VHT20 LGI 1         MCS7/1  177     148    58.5       0.0       0.0       0     0 0             0   0
VHT20 LGI 1         MCS8/1  178     123    68.3       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS0/2  180     738     9.7       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS1/2  181     369    21.9       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS2/2  182     246    34.1      34.1     100.0       0     0 0             1   1
VHT20 LGI 2         MCS3/2  183     185    46.3       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS4/2  184     123    68.3       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS5/2  185      92    90.3       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS6/2  186      82   102.5       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS7/2  187      74   112.3       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS8/2  188      62   134.2     134.2     100.0       0     0 0             1   1
VHT20 SGI 1         MCS0/1  210    1329     4.8       0.0       0.0       0     0 0             0   0
VHT20 SGI 1         MCS1/1  211     665    12.2       0.0       0.0       0     0 0             0   0
VHT20 SGI 1         MCS2/1  212     443    19.5      19.5     100.0       0     0 0             1   1
VHT20 SGI 1         MCS3/1  213     332    24.4       0.0       0.0       0     0 0             0   0
VHT20 SGI 1         MCS4/1  214     222    39.0       0.0       0.0       0     0 0             0   0
VHT20 SGI 1         MCS5/1  215     166    51.2       0.0       0.0       0     0 0             0   0
[... 71 lines cut ...]
VHT80 LGI 2         MCS9/2  349      13   522.4     300.2      51.9       3     0 0          3586   4891
VHT80 SGI 1         MCS0/1  370     295    29.2       0.0       0.0       0     0 0             0   0
VHT80 SGI 1         MCS1/1  371     148    58.5       0.0       0.0       0     0 0             0   0
VHT80 SGI 1         MCS2/1  372      99    85.4      85.4     100.0       0     0 0             1   1
VHT80 SGI 1         MCS3/1  373      74   112.3       0.0       0.0       0     0 0             0   0
VHT80 SGI 1         MCS4/1  374      49   166.0     166.0     100.0       0     0 0            30   30
VHT80 SGI 1         MCS5/1  375      37   214.8     214.8     100.0       0     0 0            63   63
VHT80 SGI 1         MCS6/1  376      33   239.2     197.7      74.9       0     0 0            32   95
VHT80 SGI 1         MCS7/1  377      30   261.2     187.9      65.2       3     0 0           236   360
VHT80 SGI 1         MCS8/1  378      25   302.7     214.8      63.5       3     0 0           315   616
VHT80 SGI 1      P  MCS9/1  379      22   332.0     292.9      79.5       3     0 0           297   608
VHT80 SGI 2         MCS0/2  380     148    58.5      53.7      85.6       0     0 0            61   67
VHT80 SGI 2         MCS1/2  381      74   112.3     102.5      82.3       0     0 0           114   126
VHT80 SGI 2         MCS2/2  382      49   166.0     166.0      91.2       3     0 0           271   277
VHT80 SGI 2         MCS3/2  383      37   214.8     214.8      95.4       0     0 0           376   504
[... cut ...]
VHT80 SGI 2         MCS6/2  386      16   424.8     339.3      72.2       3     0 0          7652   9467
VHT80 SGI 2  A      MCS7/2  387      15   458.9     458.9      99.8       3   224 224       30158   33953
VHT80 SGI 2    C    MCS8/2  388      12   524.9     410.1      70.4       3     0 0         36130   43416
VHT80 SGI 2     D   MCS9/2  389      11   561.5     405.2      64.8       3     0 0         55352   64211

Total packet count::    ideal 441616      lookaround 81
Average # of aggregated frames per A-MPDU: 23.3
```

dguerri, 2020-06-11:
FWIW, I can get to `VHT-MCS 9` now that I moved to channel 149, although it's not stable:

```
Station 44:a5:6e:xx:xx:xx (on mesh0)
	inactive time:	1212 ms
	rx bytes:	36113921
	rx packets:	95034
	tx bytes:	1060521203
	tx packets:	346751
	tx retries:	7891
	tx failed:	1
	beacon loss:	1
	beacon rx:	2269
	rx drop misc:	0
	signal:  	-40 [-40, -40] dBm
	signal avg:	-40 [-40, -41] dBm
	beacon signal avg:	-40 dBm
	tx bitrate:	866.7 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 2
	tx duration:	11330313 us
	rx bitrate:	866.7 MBit/s VHT-MCS 9 80MHz short GI VHT-NSS 2
	rx duration:	784818 us
	airtime weight: 256
	expected throughput:	464.904Mbps
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		yes
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short slot time:yes
	connected time:	257 seconds
	associated at [boottime]:	27.990s
	associated at:	1591821656819 ms
	current time:	1591821910363 ms
```

These are the stats from debugfs, but I will have to do some research before I can read it correctly :)

```
root@hallway:~# cat /sys/kernel/debug/ieee80211/phy0/netdev:mesh0/stations/44\:a5\:6e\:xx\:xx\:xx/rc_stats

              best   ____________rate__________    ____statistics___    _____last____    ______sum-of________
mode guard #  rate  [name   idx airtime  max_tp]  [avg(tp) avg(prob)]  [retry|suc|att]  [#success | #attempts]
VHT20 LGI 1         MCS0/1  170    1477     4.8       4.8      86.1       3     0 0            19   22
VHT20 LGI 1         MCS1/1  171     738     9.7       0.0       0.0       0     0 0             0   0
VHT20 LGI 1         MCS2/1  172     492    17.0      17.0     100.0       0     0 0             1   1
VHT20 LGI 1         MCS3/1  173     369    21.9       0.0       0.0       0     0 0             0   0
VHT20 LGI 1         MCS4/1  174     246    34.1       0.0       0.0       0     0 0             0   0
VHT20 LGI 1         MCS5/1  175     185    46.3       0.0       0.0       0     0 0             0   0
VHT20 LGI 1         MCS6/1  176     164    51.2       0.0       0.0       0     0 0             0   0
VHT20 LGI 1         MCS7/1  177     148    58.5       0.0       0.0       0     0 0             0   0
VHT20 LGI 1         MCS8/1  178     123    68.3       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS0/2  180     738     9.7       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS1/2  181     369    21.9       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS2/2  182     246    34.1      34.1     100.0       0     0 0             1   1
VHT20 LGI 2         MCS3/2  183     185    46.3       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS4/2  184     123    68.3       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS5/2  185      92    90.3       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS6/2  186      82   102.5       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS7/2  187      74   112.3       0.0       0.0       0     0 0             0   0
VHT20 LGI 2         MCS8/2  188      62   134.2     134.2     100.0       0     0 0             1   1
VHT20 SGI 1         MCS0/1  210    1329     4.8       0.0       0.0       0     0 0             0   0
VHT20 SGI 1         MCS1/1  211     665    12.2       0.0       0.0       0     0 0             0   0
VHT20 SGI 1         MCS2/1  212     443    19.5      19.5     100.0       0     0 0             1   1
VHT20 SGI 1         MCS3/1  213     332    24.4       0.0       0.0       0     0 0             0   0
VHT20 SGI 1         MCS4/1  214     222    39.0       0.0       0.0       0     0 0             0   0
VHT20 SGI 1         MCS5/1  215     166    51.2       0.0       0.0       0     0 0             0   0
[... 71 lines cut ...]
VHT80 LGI 2         MCS9/2  349      13   522.4     300.2      51.9       3     0 0          3586   4891
VHT80 SGI 1         MCS0/1  370     295    29.2       0.0       0.0       0     0 0             0   0
VHT80 SGI 1         MCS1/1  371     148    58.5       0.0       0.0       0     0 0             0   0
VHT80 SGI 1         MCS2/1  372      99    85.4      85.4     100.0       0     0 0             1   1
VHT80 SGI 1         MCS3/1  373      74   112.3       0.0       0.0       0     0 0             0   0
VHT80 SGI 1         MCS4/1  374      49   166.0     166.0     100.0       0     0 0            30   30
VHT80 SGI 1         MCS5/1  375      37   214.8     214.8     100.0       0     0 0            63   63
VHT80 SGI 1         MCS6/1  376      33   239.2     197.7      74.9       0     0 0            32   95
VHT80 SGI 1         MCS7/1  377      30   261.2     187.9      65.2       3     0 0           236   360
VHT80 SGI 1         MCS8/1  378      25   302.7     214.8      63.5       3     0 0           315   616
VHT80 SGI 1      P  MCS9/1  379      22   332.0     292.9      79.5       3     0 0           297   608
VHT80 SGI 2         MCS0/2  380     148    58.5      53.7      85.6       0     0 0            61   67
VHT80 SGI 2         MCS1/2  381      74   112.3     102.5      82.3       0     0 0           114   126
VHT80 SGI 2         MCS2/2  382      49   166.0     166.0      91.2       3     0 0           271   277
VHT80 SGI 2         MCS3/2  383      37   214.8     214.8      95.4       0     0 0           376   504
[... cut ...]
VHT80 SGI 2         MCS6/2  386      16   424.8     339.3      72.2       3     0 0          7652   9467
VHT80 SGI 2  A      MCS7/2  387      15   458.9     458.9      99.8       3   224 224       30158   33953
VHT80 SGI 2    C    MCS8/2  388      12   524.9     410.1      70.4       3     0 0         36130   43416
VHT80 SGI 2     D   MCS9/2  389      11   561.5     405.2      64.8       3     0 0         55352   64211

Total packet count::    ideal 441616      lookaround 81
Average # of aggregated frames per A-MPDU: 23.3
```

dguerri, 2020-06-11:
A video with live stats: https://youtu.be/LCF49q1seQw

dguerri, 2020-06-11:
The "M" shaped signal on channels 52 - 64 is super-interesting... It always starts when I start heavy transfers on channel 149...