openwrt/mt76 #410: MT7610E: multicast broken when using WPA2/WPA3

blocktrron, 2020-06-11:
I've discovered today that when using mt76 (version from OpenWrt master as well as 19.07) with a MT7610E will result in no multicast TX when using encryption (WPA2-PASK / WPA3-SAE). It's very noticeable as IPv6 RAs won't be received by the clients.

When using an open AP, multicast TX works flawlessly. I've tested this on a Archer C20i as well as a Archer C2 and both show the same symptoms. It also does not matter whether or not 802.11w was enabled or disabled.

I've patched mt76 to disable hwcrypto for the MT7610E and multicast appears to be working w/o issues 

The same settings work flawlessly on MT7612E (Archer C50 v3).

Client tested: Nokia 6.1 (Android 10) ; Intel AX200 ; Qualcomm QCA6174A

blocktrron, 2020-06-11:
@LorenzoBianconi @nbd168 

Okay, got multicast working now on MT7610E. Can you provide me with feedback for this patch before i submit it to linux-wireless?

```
From a9bec5b7d480e83ab6605a433c8dc4b743d705bd Mon Sep 17 00:00:00 2001
From: David Bauer <mail@david-bauer.net>
Date: Thu, 11 Jun 2020 12:12:43 +0200
Subject: [PATCH] mt76: mt76x0: disable hardware crypto for group keys

---
 mt76x02_util.c | 4 ++++
 1 file changed, 4 insertions(+)

--- a/mt76x02_util.c
+++ b/mt76x02_util.c
@@ -421,6 +421,10 @@ int mt76x02_set_key(struct ieee80211_hw
 		return -EOPNOTSUPP;
 	}
 
+	/* MT7610 does not support hardware crypto offload for group keys */
+	if (is_mt76x0(dev) && !(key->flags & IEEE80211_KEY_FLAG_PAIRWISE))
+		return -EOPNOTSUPP;
+
 	/*
 	 * The hardware does not support per-STA RX GTK, fall back
 	 * to software mode for these.
```

spakka, 2020-07-14:
I see the same issue on mt7615e (on 19.07)

LorenzoBianconi, 2020-07-17:
> I've discovered today that when using mt76 (version from OpenWrt master as well as 19.07) with a MT7610E will result in no multicast TX when using encryption (WPA2-PASK / WPA3-SAE). It's very [...]

can you please provide your configuration? Are you running multiple-bss?

blocktrron, 2020-07-17:
It's a simple single-VAP configuration using PSK2 (CCMP).

I've sent you an email with more details regarding this issue.

spakka, 2020-07-17:
I found this problem also on MT7615E

LorenzoBianconi, 2020-07-22:
It's a simple single-VAP configuration using PSK2 (CCMP).

> I've sent you an email with more details regarding this issue.
>
> <https://github.com/openwrt/mt76/issues/410#issuecomment-660098641>, or
> unsubscribe

Hi David,

I have just tired and broadcast traffic from the AP (e.g. ARP)  seems
encrypted. Where did you exactly sniff the traffic? from a node into the
bss or from an unconnected device?

Regards,
Lorenzo

blocktrron, 2020-07-22:
It was from an unconnected node.

I've sent you wireless settings yesterday. When you were trying this, did you use an automatically assigned MAC address or did you set one from `/sys/class/ieee80211/phy0/addresses`? Broadcast encryption works for me when i don't set a specific MAC address and breaks when i set a specific one from sysfs.

I'm aware MediaTek chips have an internal hardware MAC filter. Could be the issue that a specific MAC has to be assigned a specific BSS-ID within the driver?

blocktrron, 2020-07-22:
Okay, it seems to only work when the VAP is assigned the first MAC address. As soon as i have a second VAP, encryption only seems to work for the first interface created.

LorenzoBianconi, 2020-07-22:
Okay, it seems to only work when the VAP is assigned the first MAC address.
As soon as i have a second VAP, encryption only seems to work for the first
interface created.

Ok, maybe I identified the issue. I will provide you a patch to test

> <https://github.com/openwrt/mt76/issues/410#issuecomment-662574553>, or
> unsubscribe

LorenzoBianconi, 2020-07-22:
@blocktrron can you please try this [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/3913b9321a2f70bf004eb2de64793a1b4f94623b)?

blocktrron, 2020-07-23:
Hi Lorenzo,

I've tried you suggested patch. The effect is now that i am able to exchange multicast data using my mobile phone, however it is still impossible for my Notebook to exchange data. When capturing the packets on the air, multicast traffic still seems to be unencrypted (maybe it is using a nullcipher?), but my mobile phone is happily sending enencrypted multicast messages to the AP.

Shall i send you pcaps or are you able to reproduce the issue?

LorenzoBianconi, 2020-07-23:
> Hi Lorenzo,
>
> I've tried you suggested patch. The effect is now that i am able to
> exchange multicast data using my mobile phone, however it is still
> impossible for my Notebook to exchange data. When [...]

What is the difference with respect to the upstream code? were you able to
connect with the mobile before? Was the mobile phone the first or the
second device connected?

> Shall i send you pcaps or are you able to reproduce the issue?
yes, please.

blocktrron, 2020-07-23:
Hi Lorenzo,

> were you able to connect with the mobile before?

Previously i was able to connect with the mobile only if the MAC address of the VAP was the first one from sysfs.

> Was the mobile phone the first or the second device connected?

It is the first one connected.

See this pcap: https://drive.google.com/file/d/1x4aliLZR9bS_f_JjYwPL0Sf9J43piuTD/view?usp=sharing

blocktrron, 2020-07-23:
Correction: Connection was possible previously, multicast traffic was only possible when the MAC address was the first one from sysfs. Now multicast traffic works to the phone, but it seems unencrypted.

Question here: As I'm still not able to receive multicast traffic using my notebook: Does the phones implementation seem to be broken?

LorenzoBianconi, 2020-07-23:
> Hi Lorenzo,
>
> were you able to connect with the mobile before?
>
> Previously i was able to connect with the mobile only if the MAC address
> of the VAP was the first one from sysfs.
>
> Was the mobile phone [...]

can you please add read permission for that?

What happens if you add multiple interfaces but without changing the mac
address?

blocktrron, 2020-07-23:
Read permissions added.

> What happens if you add multiple interfaces but without changing the mac
> address?

The situation does not change when i remove the mac address from the configuration.

LorenzoBianconi, 2020-07-23:
>
> Read permissions added.
>
> What happens if you add multiple interfaces but without changing the mac
> address?
>
> The situation does not change when i remove the mac address from the configuration.

ok, they are not-encrypted for sure.
IIUC you changed the AP mac address respect to the eeprom one, right?
If so, what happens if you do not change the address and add multiple
interfaces?

blocktrron, 2020-07-23:
Hi Lorenzo,

the situation does not change when i use the MAC address from eeprom.

LorenzoBianconi, 2020-07-24:
> Hi Lorenzo,
> 
> the situation does not change when i use the MAC address from eeprom.

are you using psk2 or owe as encryption method? are there any difference using psk2 or owe?

blocktrron, 2020-07-24:
I've tried both and neither of them worked.

blocktrron, 2020-08-13:
@LorenzoBianconi We're you able to reproduce the issue? In case you are lacking a MT7610 device, i might be able to sent you one (if you are based in EU).

I'm not sure how to otherwise progress this here. All documentation i have is a age-old vendor driver. As this is an information disclosure issue, from my POV falling back to software-crypto for multicast traffic on this chip is still preferred over having this unfixed.

If you agree, i will sent my patch to linux-wireless to push this forward (and maybe increase awareness of this bug).

nicklowe, 2020-12-19:
@nbd168 Are you aware of this?