openwrt/mt76 #411: MT7603E 2.4Ghz constant disconnects

vsilvar, 2020-06-20:
I'm seeing very bad performance and constant disconnects on several clients, making 2.4Ghz unusable.
The disconnect can be forced to happen instantly if a speed test is ran.

Any steps I can take to help solve/debug this?

**Device:** Xiaomi Router 3G
**Version:** 19.07.3

I don't know if this trace is related:
```
[42671.891042] br-lan: port 2(wlan0) entered blocking state
[42671.896398] br-lan: port 2(wlan0) entered forwarding state
[42747.721612] ------------[ cut here ]------------
[42747.726274] WARNING: CPU: 3 PID: 24 at /builder/shared-workdir/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2020-03-10-08054d5a/mt7603/mac.c:204 0x8f1a4320 [mt7603e@8f1a0000+0x9100]
[42747.744160] Modules linked in: pppoe ppp_async pppox ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet
[42747.816129]  ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 nf_nat nf_conntrack nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[42747.859438] CPU: 3 PID: 24 Comm: ksoftirqd/3 Not tainted 4.14.180 #0
[42747.865766] Stack : 00000000 00000000 00000000 00000001 00000000 00000000 00000000 00000000
[42747.874101]         00000000 00000000 00000000 00000000 00000000 00000001 8fc95b80 1cc28254
[42747.882437]         8fc95c18 00000000 00000000 00006c88 00000038 8049c0d8 00000008 00000000
[42747.890774]         00000000 80550000 000d1d2e 00000000 8fc95b60 00000000 00000000 8f1a8bb4
[42747.899121]         8f1a4320 000000cc 8f1b0000 00000001 00000003 802accb0 0000000c 806b000c
[42747.907459]         ...
[42747.909894] Call Trace:
[42747.909960] [<8049c0d8>] 0x8049c0d8
[42747.915875] [<8f1a4320>] 0x8f1a4320 [mt7603e@8f1a0000+0x9100]
[42747.921596] [<8f1b0000>] 0x8f1b0000 [mt76x2_common@8f1b0000+0x2a40]
[42747.927831] [<802accb0>] 0x802accb0
[42747.931316] [<8000c1a0>] 0x8000c1a0
[42747.934785] [<8000c1a8>] 0x8000c1a8
[42747.938253] [<80484f34>] 0x80484f34
[42747.941734] [<8f1a4320>] 0x8f1a4320 [mt7603e@8f1a0000+0x9100]
[42747.947452] [<80072a64>] 0x80072a64
[42747.950923] [<8002e608>] 0x8002e608
[42747.954413] [<8f1a4320>] 0x8f1a4320 [mt7603e@8f1a0000+0x9100]
[42747.960133] [<8002e6f0>] 0x8002e6f0
[42747.963623] [<8f1a4320>] 0x8f1a4320 [mt7603e@8f1a0000+0x9100]
[42747.969348] [<8f1a5738>] 0x8f1a5738 [mt7603e@8f1a0000+0x9100]
[42747.975071] [<8f1b8274>] 0x8f1b8274 [mt76@8f1b8000+0x79e0]
[42747.980558] [<8f1a4488>] 0x8f1a4488 [mt7603e@8f1a0000+0x9100]
[42747.986274] [<80017f6c>] 0x80017f6c
[42747.989769] [<8f1bc2ec>] 0x8f1bc2ec [mt76@8f1b8000+0x79e0]
[42747.995258] [<8f1a18d4>] 0x8f1a18d4 [mt7603e@8f1a0000+0x9100]
[42748.001029] [<8f1bbb80>] 0x8f1bbb80 [mt76@8f1b8000+0x79e0]
[42748.006521] [<8f1b8dec>] 0x8f1b8dec [mt76@8f1b8000+0x79e0]
[42748.011982] [<8f1b0000>] 0x8f1b0000 [mt76x2_common@8f1b0000+0x2a40]
[42748.018257] [<80363b90>] 0x80363b90
[42748.021745] [<804a2ed8>] 0x804a2ed8
[42748.025215] [<8049e340>] 0x8049e340
[42748.028694] [<8004fd10>] 0x8004fd10
[42748.032164] [<80032cc8>] 0x80032cc8
[42748.035644] [<8004feb8>] 0x8004feb8
[42748.039112] [<8049e458>] 0x8049e458
[42748.042626] [<8004bf38>] 0x8004bf38
[42748.046099] [<8004be08>] 0x8004be08
[42748.049576] [<8004be08>] 0x8004be08
[42748.053058] [<8004be08>] 0x8004be08
[42748.056527] [<80006f78>] 0x80006f78
[42748.060002] 
[42748.061657] ---[ end trace 678ba38ab904c798 ]---
[43649.331133] device wlan0 left promiscuous mode
```

Openwrtfunboy, 2020-06-24:
Same Newifi 3 D2.

9000000, 2020-07-03:
```
2.805935] libphy: Fixed MDIO Bus: probed
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.836815] libphy: mdio: probed
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.840278] mt7530 mdio-bus:1f: MT7530 adapts as multi-chip module
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.848119] mtk_soc_eth 1e100000.ethernet eth0: mediatek frame engine at 0xbe100000, irq 21
Fri Jul  3 19:11:42 2020 kern.err kernel: [    2.858732] mt7621-pci 1e140000.pcie: Parsing DT failed
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.866588] NET: Registered protocol family 10
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.872735] Segment Routing with IPv6
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.876555] NET: Registered protocol family 17
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.881377] 8021q: 802.1Q VLAN Support v1.8
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.888108] mt7530 mdio-bus:1f: MT7530 adapts as multi-chip module
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.904828] libphy: dsa slave smi: probed
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.909473] mt7530 mdio-bus:1f wan (uninitialized): PHY [dsa-0.0:01] driver [Generic PHY]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.919235] mt7530 mdio-bus:1f lan2 (uninitialized): PHY [dsa-0.0:02] driver [Generic PHY]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.928998] mt7530 mdio-bus:1f lan1 (uninitialized): PHY [dsa-0.0:03] driver [Generic PHY]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.939127] mt7530 mdio-bus:1f: configuring for fixed/rgmii link mode
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.950246] DSA: tree 0 setup
Fri Jul  3 19:11:42 2020 kern.err kernel: [    2.953577] rt2880-pinmux pinctrl: pcie is already enabled
Fri Jul  3 19:11:42 2020 kern.err kernel: [    2.959073] mt7621-pci 1e140000.pcie: Error applying setting, reverse things back
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.966679] mt7621-pci-phy 1e149000.pcie-phy: PHY for 0xbe149000 (dual port = 1)
Fri Jul  3 19:11:42 2020 kern.info kernel: [    2.974254] mt7621-pci-phy 1e14a000.pcie-phy: PHY for 0xbe14a000 (dual port = 0)
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.081690] mt7621-pci-phy 1e149000.pcie-phy: Xtal is 40MHz
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.087272] mt7621-pci-phy 1e14a000.pcie-phy: Xtal is 40MHz
Fri Jul  3 19:11:42 2020 kern.err kernel: [    3.192795] mt7621-pci 1e140000.pcie: pcie2 no card, disable it (RST & CLK)
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.199751] mt7621-pci 1e140000.pcie: PCIE0 enabled
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.204618] mt7621-pci 1e140000.pcie: PCIE1 enabled
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.209494] mt7621-pci 1e140000.pcie: PCI coherence region base: 0x60000000, mask/settings: 0xf0000002
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.218956] mt7621-pci 1e140000.pcie: PCI host bridge to bus 0000:00
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.225318] pci_bus 0000:00: root bus resource [io  0x1e160000-0x1e16ffff]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.232185] pci_bus 0000:00: root bus resource [mem 0x60000000-0x6fffffff]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.239048] pci_bus 0000:00: root bus resource [bus 00-ff]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.244563] pci 0000:00:00.0: [0e8d:0801] type 01 class 0x060400
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.250595] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.256853] pci 0000:00:00.0: reg 0x14: [mem 0x60400000-0x6040ffff]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.263183] pci 0000:00:00.0: supports D1
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.267189] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.273349] pci 0000:00:01.0: [0e8d:0801] type 01 class 0x060400
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.279389] pci 0000:00:01.0: reg 0x10: [mem 0x00000000-0x7fffffff]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.285648] pci 0000:00:01.0: reg 0x14: [mem 0x60410000-0x6041ffff]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.291966] pci 0000:00:01.0: supports D1
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.295972] pci 0000:00:01.0: PME# supported from D0 D1 D3hot
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.303309] pci 0000:01:00.0: [14c3:7603] type 00 class 0x028000
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.309370] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x000fffff]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.315767] pci 0000:01:00.0: PME# supported from D0 D3hot D3cold
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.323228] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.328467] pci 0000:00:00.0:   bridge window [io  0x0000-0x0fff]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.334553] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.341326] pci 0000:00:00.0:   bridge window [mem 0x60100000-0x601fffff pref]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.348538] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.355386] pci 0000:02:00.0: [14c3:7662] type 00 class 0x028000
Fri Jul  3 19:11:42 2020 kern.info kernel: [    3.361444] pci 0000:02:00.0: reg 0x10: [mem 0x00000000-0x000fffff 64bit]
[... cut ...]
Fri Jul  3 19:11:42 2020 kern.info kernel: [    4.968191] hctosys: unable to open rtc device (rtc0)
Fri Jul  3 19:11:42 2020 kern.info kernel: [    4.981482] VFS: Mounted root (squashfs filesystem) readonly on device 254:0.
Fri Jul  3 19:11:42 2020 kern.info kernel: [    4.992896] Freeing unused kernel memory: 1236K
Fri Jul  3 19:11:42 2020 kern.warn kernel: [    4.997451] This architecture does not have kernel memory protection.
Fri Jul  3 19:11:42 2020 kern.info kernel: [    5.003876] Run /sbin/init as init process
```

helicida, 2020-07-03:
Same with WR1200JS

9000000, 2020-07-12:
Xiaomi r3g v1 

```
Mon Jul 13 01:19:41 2020 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ENABLED
Mon Jul 13 01:19:41 2020 daemon.notice hostapd: wlan0: AP-ENABLED
Mon Jul 13 01:19:41 2020 daemon.warn omcproxy: mld: ignoring invalid MLD-message of type 143 from :: on 0
Mon Jul 13 01:19:41 2020 daemon.notice netifd: radio0 (8990): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Mon Jul 13 01:19:41 2020 daemon.notice netifd: radio0 (8990): Command failed: Invalid argument
Mon Jul 13 01:19:41 2020 daemon.warn omcproxy: mld: ignoring invalid MLD-message of type 143 from :: on 0
Mon Jul 13 01:19:42 2020 daemon.notice netifd: Network device 'wlan0' link is up
```

9000000, 2020-07-13:
```
Mon Jul 13 15:24:09 2020 daemon.notice hostapd: Remove interface 'wlan0'
Mon Jul 13 15:24:09 2020 daemon.notice hostapd: wlan0: interface state ENABLED->DISABLED
Mon Jul 13 15:24:09 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 34:ce:00:d3:c8:5c
Mon Jul 13 15:24:09 2020 daemon.notice hostapd: wlan0: AP-DISABLED
Mon Jul 13 15:24:09 2020 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Mon Jul 13 15:24:09 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Mon Jul 13 15:24:09 2020 kern.info kernel: [55493.164327] device wlan0 left promiscuous mode
Mon Jul 13 15:24:09 2020 kern.info kernel: [55493.169062] br-lan: port 3(wlan0) entered disabled state
Mon Jul 13 15:24:09 2020 daemon.notice netifd: Network device 'wlan0' link is down
Mon Jul 13 15:24:10 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Mon Jul 13 15:24:10 2020 kern.info kernel: [55493.913341] br-lan: port 3(wlan0) entered blocking state
Mon Jul 13 15:24:10 2020 kern.info kernel: [55493.918708] br-lan: port 3(wlan0) entered disabled state
```

nbd168, 2020-08-12:
Please try the latest master version

easyteacher, 2021-08-06:
> Please try the latest master version

I am using the latest version of mt76, but 2.4G WiFi is very likely to disappear after 4-5d uptime. 

I have set 2 APs in 2.4G, one for IoT devices and the other for other devices.

Neustradamus, 2022-05-04:
To follow this ticket

JFtico, 2022-05-09:
Now on 23.02.0.rc1 and still seeing this problem. Here is a recent crash of the 7603:

Box has MT7621 SoC with a 7612 + 7603e hardware.

Tue May  3 18:01:53 2022 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 16:3f:3d:0e:49:53
Tue May  3 18:01:53 2022 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.174 16:3f:3d:0e:49:53
Tue May  3 18:01:53 2022 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 16:3f:3d:0e:49:53
Tue May  3 18:01:53 2022 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.174 16:3f:3d:0e:49:53
Tue May  3 18:01:54 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 16:3f:3d:0e:49:53
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.214851] ------------[ cut here ]------------
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.219643] WARNING: CPU: 1 PID: 1538 at target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2022-03-15-053668ac/mt7603/mac.c:208 0x82724814 [mt7603e@3983f03d+0x9660]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.234312] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sch_cake nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.235513]  act_gact ledtrig_usbport ifb sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac mmc_block mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common crc32c_generic
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.344478] CPU: 1 PID: 1538 Comm: hostapd Not tainted 5.10.111 #0
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.350727] Stack : 00000000 8071d058 809a0000 80000000 00000000 00000000 00000000 00000000
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.359167]         00000000 00000000 00000000 00000000 00000000 00000001 82689928 a578131e
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.367605]         826899c0 00000000 00000000 826897d0 00000038 80398944 ffffffea 00000000
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.376041]         826897dc 0000022c 807c0838 ffffffff 82689908 806ec06c 80000000 806f0000
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.384476]         00000009 00000002 00003a98 80000000 00000000 8040235c 00000004 80980004
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.392912]         ...
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.395443] Call Trace:
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.395521] [<80398944>] 0x80398944
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.401609] [<8040235c>] 0x8040235c
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.405174] [<80007b08>] 0x80007b08
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.408738] [<80007b10>] 0x80007b10
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.412297] [<8037cd00>] 0x8037cd00
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.415879] [<82724814>] 0x82724814 [mt7603e@3983f03d+0x9660]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.421684] [<8007fb54>] 0x8007fb54
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.425250] [<82724814>] 0x82724814 [mt7603e@3983f03d+0x9660]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.431067] [<8002ff28>] 0x8002ff28
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.434636] [<82724814>] 0x82724814 [mt7603e@3983f03d+0x9660]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.440450] [<8002fff0>] 0x8002fff0
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.444011] [<806d20c8>] 0x806d20c8
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.447594] [<82724814>] 0x82724814 [mt7603e@3983f03d+0x9660]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.453425] [<8284a1d4>] 0x8284a1d4 [mac80211@afe9b814+0x81e80]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.459437] [<82721a70>] 0x82721a70 [mt7603e@3983f03d+0x9660]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.465283] [<82892f54>] 0x82892f54 [mt76@6b7daca7+0xafa0]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.470837] [<8284aa58>] 0x8284aa58 [mac80211@afe9b814+0x81e80]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.476860] [<828930c4>] 0x828930c4 [mt76@6b7daca7+0xafa0]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.482419] [<82807ef8>] 0x82807ef8 [mac80211@afe9b814+0x81e80]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.488436] [<828098bc>] 0x828098bc [mac80211@afe9b814+0x81e80]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.494420] [<80099bf4>] 0x80099bf4
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.498000] [<82809c5c>] 0x82809c5c [mac80211@afe9b814+0x81e80]
Tue May  3 18:01:55 2022 kern.warn kernel: [438087.503996] [<82793c5c>] 0x82793c5c [cfg80211@83ecf167+0x45e20]
[... cut ...]
Tue May  3 18:07:27 2022 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.174 16:3f:3d:0e:49:53
Tue May  3 18:07:27 2022 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 16:3f:3d:0e:49:53
Tue May  3 18:07:27 2022 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.174 16:3f:3d:0e:49:53
Tue May  3 18:07:28 2022 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.174 16:3f:3d:0e:49:53
Tue May  3 18:07:28 2022 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.174 16:3f:3d:0e:49:53