openwrt/mt76 #412: mt7612 crash on 7621

ptpt52, 2020-06-24:
master HEAD build

```
<1>[37947.153634] CPU 0 Unable to handle kernel paging request at virtual address 0000000c, epc == 80088620, ra == 8008936c
<4>[37947.164253] Oops[#1]:
<4>[37947.166525] CPU: 0 PID: 6036 Comm: kworker/u8:0 Tainted: G         C        5.4.45 #0
<4>[37947.174362] Workqueue: phy1 mt76x02_wdt_work [mt76x02_lib]
<4>[37947.179821] $ 0   : 00000000 00000001 00000000 00000001
<4>[37947.185024] $ 4   : 8e0b26c8 0004f0a8 00000004 00000001
<4>[37947.190225] $ 8   : fefefeff 00000018 80808080 fefefeff
<4>[37947.195427] $12   : 00000000 00000000 00000000 00000000
<4>[37947.200628] $16   : 8e0b26c8 00000008 8e0b2638 8e08c000
<4>[37947.205830] $20   : 00000000 8e0b1e8c 8e0b294c 80690000
<4>[37947.211031] $24   : bc08c42f 036be704                  
<4>[37947.216233] $28   : 8d430000 8d431dd8 8e06d2c0 8008936c
<4>[37947.221436] Hi    : 00000000
<4>[37947.224296] Lo    : 00000000
<4>[37947.227193] epc   : 80088620 hrtimer_active+0x4/0x74
<4>[37947.232134] ra    : 8008936c hrtimer_try_to_cancel+0x10/0xd0
<4>[37947.237761] Status: 11008403	KERNEL EXL IE 
<4>[37947.241925] Cause : 40800008 (ExcCode 02)
<4>[37947.245909] BadVA : 0000000c
<4>[37947.248770] PrId  : 0001992f (MIPS 1004Kc)
<4>[37947.252839] Modules linked in: ksmbd qcserial pppoe ppp_async option l2tp_ppp cdc_mbim usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY wireguard usbserial usbnet usblp ts_fsm ts_bm slhc sch_cake r8152 nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nf_tproxy_ipv6 nf_tproxy_ipv4
<4>[37947.253039]  nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables crc_ccitt compat_xtables compat cdc_wdm br_netfilter asn1_decoder natflow natcap fuse tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet
<4>[37947.340414]  ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip6_gre ip_gre gre ifb sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip6_tunnel tunnel6 tunnel4 ip_tunnel tun vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac ghash_generic gf128mul gcm ecb des_generic libdes ctr cmac ccm arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ohci_platform ohci_hcd softdog fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat(C) usbcore nls_base usb_common mii crc32c_generic
<4>[37947.512624] Process kworker/u8:0 (pid: 6036, threadinfo=26550c6d, task=8be035a8, tls=00000000)
<4>[37947.521189] Stack : 00000003 8e0b0c20 00000000 00000000 8e0b1044 00000001 8e0b26c8 80089444
<4>[37947.529508]         8e06d2c0 8e13e738 8e0b1e80 0000000f 8e0b2640 803c8a44 8e0b2a58 8e0b32c8
<4>[37947.537827]         00000000 8fc08000 8e0b32c8 00000000 8e0b1fa4 8e0a58f0 00000000 0000ffff
<4>[37947.546145]         8e0b3098 0000000f 00703ff3 8e0b2970 8e0b1fa4 8e0b2488 8e0b32c8 8de5cb80
<4>[37947.554464]         8fc08000 8e08c000 00000000 00000100 00000000 80690000 8fc08000 80040ef4
<4>[37947.562783]         ...
<4>[37947.565216] Call Trace:
<4>[37947.567677] [<80088620>] hrtimer_active+0x4/0x74
<4>[37947.572276] Code: 03e00008  27bd0028  8c820024 <8c43000c> 30650001  14a00011  00000000  0000000f  90850028 
<4>[37947.581986] 
<4>[37947.584529] ---[ end trace 17fce6e4d186caba ]---

```

ptpt52, 2020-06-24:
some more log on xiaomi miwifi-r3 (mt7620)

```
<6>[47775.602319] pppoe-xwan01: renamed from ppp1
<4>[47778.432455] warning: (zone_netdev_event:346)dev=pppoe-xwan01 set zone=127 type=0
<6>[47854.649116] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
<6>[47854.654872] mt76x2e 0000:01:00.0: Build: 1
<6>[47854.659137] mt76x2e 0000:01:00.0: Build Time: 201507311614____
<6>[47854.676532] mt76x2e 0000:01:00.0: Firmware running!
<1>[47854.682238] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 86ad0a90, ra == 86ad0a88
<4>[47854.693160] Oops[#1]:
<4>[47854.695488] CPU: 0 PID: 26882 Comm: kworker/u2:0 Not tainted 4.14.180 #0
<4>[47854.702362] Workqueue: phy0 mt76x02_wdt_work [mt76x02_lib]
<4>[47854.707961] task: 87c21680 task.stack: 87ef8000
<4>[47854.712577] $ 0   : 00000000 00000001 00000000 00000000
<4>[47854.717915] $ 4   : 00000000 00000001 01080020 00000000
<4>[47854.723250] $ 8   : 00000000 00000002 0000001f 00000000
<4>[47854.728588] $12   : 00000000 3fff0000 ffffffff 40000000
<4>[47854.733924] $16   : 86b71e18 0000001e 86b71da0 86b7268c
<4>[47854.739261] $20   : 0000000a 86b71dac 86b726e4 86b726cc
<4>[47854.744599] $24   : 054d0800 000001e0                  
<4>[47854.749939] $28   : 87ef8000 87ef9df8 86b72dbc 86ad0a88
<4>[47854.755278] Hi    : 0000001f
<4>[47854.758211] Lo    : 00000000
<4>[47854.761171] epc   : 86ad0a90 mt76_dma_attach+0x5d8/0xeec [mt76]
<4>[47854.767218] ra    : 86ad0a88 mt76_dma_attach+0x5d0/0xeec [mt76]
<4>[47854.773250] Status: 1100f403	KERNEL EXL IE 
<4>[47854.777526] Cause : 4080000c (ExcCode 03)
<4>[47854.781612] BadVA : 00000000
<4>[47854.784548] PrId  : 00019650 (MIPS 24KEc)
<4>[47854.788631] Modules linked in: natcap ksmbd rt2800soc rt2800mmio rt2800lib qcserial pppoe ppp_async option l2tp_ppp cdc_mbim usb_wwan sierra_net sierra rt2x00soc rt2x00mmio rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY wireguard
<4>[47854.861198]  usbserial usbnet usblp ts_fsm ts_bm slhc r8152 nft_set_rbtree nft_set_hash nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_meta nft_log nft_limit nft_hash nft_fwd_netdev nft_exthdr nft_dup_netdev nft_ct nft_counter nft_chain_route_ipv6 nft_chain_route_ipv4 nf_tables_netdev nf_tables_ipv6 nf_tables_ipv4 nf_tables_inet nf_tables_bridge nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_rtcache
<4>[47854.933974]  nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda macvlan iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables crc_ccitt compat_xtables compat cdc_wdm br_netfilter natflow fuse sch_cake tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_conntrack_ipv6
<4>[47855.006839]  nf_defrag_ipv6 nf_nat_ipv6 ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 nf_nat nf_conntrack ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip6_gre ip_gre gre ifb sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip6_tunnel tunnel6 tunnel4 ip_tunnel tun vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha512_generic sha256_generic sha1_generic seqiv drbg md5 md4 hmac ghash_generic gf128mul gcm ecb des_generic ctr cmac ccm arc4 uas usb_storage leds_gpio ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat usbcore nls_base usb_common mii aead crypto_null cryptomgr crc32c_generic crypto_hash [last unloaded: natcap]
<4>[47855.080685] Process kworker/u2:0 (pid: 26882, threadinfo=87ef8000, task=87c21680, tls=00000000)
<4>[47855.089552] Stack : 86b72fb4 00000000 86b71da0 0000000a 0000000a 86b72fb4 00000003 86b71da0
<4>[47855.098091]         00000003 86ab5768 00000000 0000ffff 87c21680 8003d5a0 86b72aa8 86b7d1f0
<4>[47855.106627]         00703ff3 86b726f0 86b71ebc 86b72358 86b72fb4 86c46300 00000040 86408100
<4>[47855.115164]         00000000 00000004 80510000 fffffffe 80500000 8003be00 87c05000 80510000
<4>[47855.123701]         87c05014 87c05000 80510000 fffffffe 86c46300 87c05000 86c46318 80510000
<4>[47855.132240]         ...
<4>[47855.134744] Call Trace:
<4>[47855.137252] [<86ad0a90>] mt76_dma_attach+0x5d8/0xeec [mt76]
<4>[47855.142961] [<86ad0a88>] mt76_dma_attach+0x5d0/0xeec [mt76]
<4>[47855.148651] Code: 02402025  8e030898  8e020874 <ac430000> 8e030888  8e020874  ac430004  8e020874  8c42000c 
<4>[47855.158618] 
<4>[47855.161001] ---[ end trace 4b75f0437df72879 ]---

```

9000000, 2020-06-27:
My device does not know what the error is 
```
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30323.954622] [<8000b72c>] show_stack+0x30/0x100
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30323.959049] [<80553df0>] dump_stack+0xa4/0xdc
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30323.963390] [<80028058>] __warn+0xc0/0x10c
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30323.967466] [<80028100>] warn_slowpath_fmt+0x5c/0xac
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30323.972445] [<8eb33990>] ieee80211_proberesp_get+0x9c4/0xa14 [mac80211]
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30323.979265] ---[ end trace 2bbe4d4a17412334 ]---
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30323.983905] ------------[ cut here ]------------
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30323.988581] WARNING: CPU: 0 PID: 14742 at backports-5.7-rc3-1/net/mac80211/tx.c:949 ieee80211_proberesp_get+0x9c4/0xa14 [mac80211]
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.000289] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY wireguard slhc sch_cake nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.000452]  leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.097253] CPU: 0 PID: 14742 Comm: kworker/u8:2 Tainted: G        W         5.4.48 #0
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.105180] Workqueue: phy0 ieee80211_ibss_leave [mac80211]
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.110723] Stack : 805fa718 8fd3da24 80670000 806c0000 8e805000 8060bc04 8eb33990 00000009
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.119042]         8fd3dbd0 8fdb4000 8ea44c20 800772c0 00000000 00000001 8fd3d9e0 ce53ef69
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.127361]         00000000 00000000 00000000 00000000 0000005d 00000493 7362695f 656c5f73
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.135678]         00000000 000005e4 00000000 00019adc 00000000 806d0000 00000000 8eb33990
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.143997]         00000009 8fd3dbd0 8fdb4000 8ea44c20 00000002 80324ebc 00000000 80810000
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.152314]         ...
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.154747] Call Trace:
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.157190] [<8000b72c>] show_stack+0x30/0x100
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.161619] [<80553df0>] dump_stack+0xa4/0xdc
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.165960] [<80028058>] __warn+0xc0/0x10c
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.170042] [<80028100>] warn_slowpath_fmt+0x5c/0xac
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.175023] [<8eb33990>] ieee80211_proberesp_get+0x9c4/0xa14 [mac80211]
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.181771] ---[ end trace 2bbe4d4a17412335 ]---
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.186420] ------------[ cut here ]------------
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.191087] WARNING: CPU: 0 PID: 14742 at backports-5.7-rc3-1/net/mac80211/tx.c:949 ieee80211_proberesp_get+0x9c4/0xa14 [mac80211]
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.202793] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY wireguard slhc sch_cake nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.202955]  leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.299760] CPU: 0 PID: 14742 Comm: kworker/u8:2 Tainted: G        W         5.4.48 #0
Sat Jun 27 16:24:23 2020 kern.warn kernel: [30324.307704] Workqueue: phy0 ieee80211_ibss_leave [mac80211]
[... cut ...]
Sat Jun 27 16:24:27 2020 kern.warn kernel: [30328.948035] [<80028058>] __warn+0xc0/0x10c
Sat Jun 27 16:24:27 2020 kern.warn kernel: [30328.952127] [<80028100>] warn_slowpath_fmt+0x5c/0xac
Sat Jun 27 16:24:27 2020 kern.warn kernel: [30328.957174] [<8eb33990>] ieee80211_proberesp_get+0x9c4/0xa14 [mac80211]
Sat Jun 27 16:24:27 2020 kern.warn kernel: [30328.964040] ---[ end trace 2bbe4d4a1741234b ]---
Sat Jun 27 17:08:00 2020 daemon.err uhttpd[2018]: luci: accepted login on / for root from 192.168.7.162
```

ptpt52, 2020-07-03:
I post some more log here:
```
Oops#1 Part1
<4>[10097.115433]  [last unloaded: natcap]
<4>[10097.206143] CPU: 3 PID: 13823 Comm: kworker/u8:1 Tainted: G        WC        5.4.48 #0
<4>[10097.214145] Workqueue: phy1 ieee80211_ibss_leave [mac80211]
<4>[10097.219694] Stack : 805cffa8 8cff3a14 80650000 80690000 8db31a80 805e0228 8d533990 00000009
<4>[10097.228028]         8cff3bc0 8d7d10cc 8d470c20 8006ed48 00000003 00000001 8cff39d0 71b3315a
<4>[10097.236360]         00000000 00000000 00000000 00000000 0000005d 000011fc 7362695f 656c5f73
<4>[10097.244689]         00000000 000001f6 00000000 00034481 00000000 806b0000 00000000 8d533990
<4>[10097.253018]         00000009 8cff3bc0 8d7d10cc 8d470c20 00000002 802f7100 0000000c 807f000c
<4>[10097.261350]         ...
<4>[10097.263788] Call Trace:
<4>[10097.266250] [<8000b5fc>] show_stack+0x30/0x100
<4>[10097.270696] [<8052b368>] dump_stack+0xa4/0xdc
<4>[10097.275050] [<80027c54>] __warn+0xc0/0x10c
<4>[10097.279136] [<80027cfc>] warn_slowpath_fmt+0x5c/0xac
<4>[10097.284142] [<8d533990>] ieee80211_proberesp_get+0x9c4/0xa14 [mac80211]
<4>[10097.290938] ---[ end trace 719fd59c40b95065 ]---
<4>[10097.295581] ------------[ cut here ]------------
<4>[10097.300270] WARNING: CPU: 3 PID: 13823 at backports-5.7-rc3-1/net/mac80211/tx.c:949 ieee80211_proberesp_get+0x9c4/0xa14 [mac80211]
<4>[10097.311985] Modules linked in: ksmbd natcap qcserial pppoe ppp_async option l2tp_ppp cdc_mbim usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY wireguard usbserial usbnet usblp ts_fsm ts_bm slhc sch_cake r8152 nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nf_tproxy_ipv6
<4>[10097.312287]  nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables crc_ccitt compat_xtables compat cdc_wdm br_netfilter asn1_decoder natflow tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet
<4>[10097.399205]  ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip6_gre ip_gre gre ifb sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip6_tunnel tunnel6 tunnel4 ip_tunnel tun vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha512_generic sha256_generic libsha256 sha1_generic seqiv fuse jitterentropy_rng drbg md5 md4 hmac ghash_generic gf128mul gcm ecb des_generic libdes ctr cmac ccm arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ohci_platform ohci_hcd softdog fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat(C) usbcore nls_base usb_common mii crc32c_generic
<4>[10097.485450]  [last unloaded: natcap]
<4>[10097.576161] CPU: 3 PID: 13823 Comm: kworker/u8:1 Tainted: G        WC        5.4.48 #0
<4>[10097.584195] Workqueue: phy1 ieee80211_ibss_leave [mac80211]
<4>[10097.589745] Stack : 805cffa8 8cff3a14 80650000 80690000 8db31a80 805e0228 8d533990 00000009
<4>[10097.598078]         8cff3bc0 8d7d10cc 8d470c20 8006ed48 00000003 00000001 8cff39d0 71b3315a
<4>[10097.606410]         00000000 00000000 00000000 00000000 0000005d 00001211 7362695f 656c5f73
<4>[10097.614739]         00000000 000001f6 00000000 0008ea03 00000000 806b0000 00000000 8d533990
<4>[10097.623070]         00000009 8cff3bc0 8d7d10cc 8d470c20 00000002 802f7100 0000000c 807f000c
<4>[10097.631401]         ...
<4>[10097.633839] Call Trace:
<4>[10097.636304] [<8000b5fc>] show_stack+0x30/0x100
<4>[10097.640750] [<8052b368>] dump_stack+0xa4/0xdc
<4>[10097.645104] [<80027c54>] __warn+0xc0/0x10c
<4>[10097.649190] [<80027cfc>] warn_slowpath_fmt+0x5c/0xac
<4>[10097.654203] [<8d533990>] ieee80211_proberesp_get+0x9c4/0xa14 [mac80211]
<4>[10097.661157] ---[ end trace 719fd59c40b95066 ]---
<4>[10097.665819] ------------[ cut here ]------------
[... cut ...]
<4>[10453.515724]         8fc0d9f0 8fc0d9f0 00000000 00000000 8d470c20 8dbaf480 8d7d7000 8ccdcc00
<4>[10453.524051]         00600307 00000002 8dbaf480 8e8c6900 8d470c20 8d7d7000 00000000 8d537244
<4>[10453.532374]         ...
<4>[10453.534821] Call Trace:
<4>[10453.537376] [<8d532798>] ieee80211_reserve_tid+0xd64/0x13e0 [mac80211]
<4>[10453.543931] Code: ae430000  ae420004  ae320018 <ac520000> 8e22001c  2612009c  24420001  ae22001c  8e1100c8 
<4>[10453.553648] 
<4>[10453.555914] ---[ end trace 719fd59c40b95083 ]---
```

ptpt52, 2020-07-03:
and more:
```
<4>[ 1665.111439] CPU: 1 PID: 170 Comm: kworker/u8:2 Tainted: G    B   WC        5.4.48 #0
<4>[ 1665.119272] Workqueue: phy1 ieee80211_ba_session_work [mac80211]
<4>[ 1665.125250] Stack : 805cffa8 8ff29a34 80650000 80690000 8fd49f80 805e0228 8d533990 00000009
<4>[ 1665.133574]         8ff29be0 8e88e0cc 8dbb8c20 8006ed48 00000001 00000001 8ff299f0 b34d6451
<4>[ 1665.141897]         00000000 00000000 00000000 00000000 0000005d 000009c7 616d5b20 32303863
<4>[ 1665.150218]         00000000 00000052 00000000 0001d1e8 00000000 806b0000 00000000 8d533990
<4>[ 1665.158540]         00000009 8ff29be0 8e88e0cc 8dbb8c20 00000003 802f7100 00000004 807f0004
<4>[ 1665.166861]         ...
<4>[ 1665.169298] Call Trace:
<4>[ 1665.171753] [<8000b5fc>] show_stack+0x30/0x100
<4>[ 1665.176193] [<8052b368>] dump_stack+0xa4/0xdc
<4>[ 1665.180545] [<80027c54>] __warn+0xc0/0x10c
<4>[ 1665.184627] [<80027cfc>] warn_slowpath_fmt+0x5c/0xac
<4>[ 1665.189633] [<8d533990>] ieee80211_proberesp_get+0x9c4/0xa14 [mac80211]
<4>[ 1665.196433] ---[ end trace cfa5a3d1c20f6ad1 ]---
<1>[ 1665.201654] CPU 2 Unable to handle kernel paging request at virtual address 00000000, epc == 8db822d0, ra == 8db81504
<4>[ 1665.212300] Oops[#1]:
<4>[ 1665.214571] CPU: 2 PID: 20 Comm: ksoftirqd/2 Tainted: G    B   WC        5.4.48 #0
<4>[ 1665.222106] $ 0   : 00000000 00000001 00000000 00000004
<4>[ 1665.227321] $ 4   : 8dbba710 00000890 8dbba710 00000000
<4>[ 1665.232529] $ 8   : 00000020 8054597c 0000000a 00000122
<4>[ 1665.237733] $12   : 00000100 00000402 80696de0 00000040
<4>[ 1665.242937] $16   : 8dbba000 8d59b050 8f031000 0000000c
<4>[ 1665.248141] $20   : 00000038 00000000 00080038 8dbba000
<4>[ 1665.253352] $24   : 00000000 80055cdc                  
<4>[ 1665.258562] $28   : 8fc88000 8fc89da0 8dbb9e80 8db81504
<4>[ 1665.263772] Hi    : 000000fe
<4>[ 1665.266636] Lo    : 00000000
<4>[ 1665.269535] epc   : 8db822d0 mt76_rx+0x44/0x74 [mt76]
<4>[ 1665.274567] ra    : 8db81504 mt76_dma_cleanup+0x750/0x103c [mt76]
<4>[ 1665.280637] Status: 11007c03	KERNEL EXL IE 
<4>[ 1665.284809] Cause : 4080000c (ExcCode 03)
<4>[ 1665.288801] BadVA : 00000000
<4>[ 1665.291670] PrId  : 0001992f (MIPS 1004Kc)
<4>[ 1665.295742] Modules linked in: ksmbd natcap qcserial pppoe ppp_async option l2tp_ppp cdc_mbim usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY wireguard usbserial usbnet usblp ts_fsm ts_bm slhc sch_cake r8152 nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nf_tproxy_ipv6
<4>[ 1665.295939]  nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables crc_ccitt compat_xtables compat cdc_wdm br_netfilter asn1_decoder natflow tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet
<4>[ 1665.382608]  ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip6_gre ip_gre gre ifb sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip6_tunnel tunnel6 tunnel4 ip_tunnel tun vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha512_generic sha256_generic libsha256 sha1_generic seqiv fuse jitterentropy_rng drbg md5 md4 hmac ghash_generic gf128mul gcm ecb des_generic libdes ctr cmac ccm arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ohci_platform ohci_hcd softdog fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat(C) usbcore nls_base usb_common mii crc32c_generic
<4>[ 1665.468559]  [last unloaded: natcap]
<4>[ 1665.559101] Process ksoftirqd/2 (pid: 20, threadinfo=2e5efcc2, task=e8b129aa, tls=00000000)
<4>[ 1665.567408] Stack : 00000000 00000000 00000000 00000000 ffffffff 8dbba970 40000000 00000000
[... cut ...]
<4>[ 1666.556289] [<8006fd88>] generic_handle_irq+0x40/0x58
<4>[ 1666.561341] [<8054ca84>] do_IRQ+0x1c/0x2c
<4>[ 1666.565340] [<80298c6c>] plat_irq_dispatch+0x64/0x104
<4>[ 1666.570379] [<80006de8>] except_vec_vi_end+0xb8/0xc4
<4>[ 1666.575334] [<8000b870>] die+0xc8/0x114
<4>[ 1666.579176] [<800175e0>] do_page_fault+0x4a8/0x4b8
<4>[ 1666.583958] [<8001c6d8>] tlb_do_page_fault_1+0x118/0x120
<4>[ 1666.589282] [<8db81504>] mt76_dma_cleanup+0x750/0x103c [mt76]
<4>[ 1666.595012] ---[ end trace cfa5a3d1c20f6ad3 ]---

```

ptpt52, 2020-07-03:
```
 924 static ieee80211_tx_result debug_noinline
 925 ieee80211_tx_h_fragment(struct ieee80211_tx_data *tx)
 926 {
 927     struct sk_buff *skb = tx->skb;
 928     struct ieee80211_tx_info *info = IEEE80211_SKB_CB(skb);
 929     struct ieee80211_hdr *hdr = (void *)skb->data;
 930     int frag_threshold = tx->local->hw.wiphy->frag_threshold;
 931     int hdrlen;
 932     int fragnum;
 933 
 934     /* no matter what happens, tx->skb moves to tx->skbs */
 935     __skb_queue_tail(&tx->skbs, skb);
 936     tx->skb = NULL;
 937 
 938     if (info->flags & IEEE80211_TX_CTL_DONTFRAG)
 939         return TX_CONTINUE;
 940 
 941     if (ieee80211_hw_check(&tx->local->hw, SUPPORTS_TX_FRAG))
 942         return TX_CONTINUE;
 943 
 944     /*
 945      * Warn when submitting a fragmented A-MPDU frame and drop it.
 946      * This scenario is handled in ieee80211_tx_prepare but extra
 947      * caution taken here as fragmented ampdu may cause Tx stop.
 948      */
 949     if (WARN_ON(info->flags & IEEE80211_TX_CTL_AMPDU))
 950         return TX_DROP;
```

ptpt52, 2020-07-03:
This warn raise with mt7612
```
WARN_ON(info->flags & IEEE80211_TX_CTL_AMPDU)
```
what may cause this?

@nbd168  ping

ptpt52, 2020-07-25:
same problem on mt76 version before 20200715
related to https://github.com/openwrt/mt76/issues/427

```
<4>[37927.774115] ------------[ cut here ]------------
<4>[37927.783492] WARNING: CPU: 1 PID: 0 at backports-5.7-rc3-1/net/mac80211/sta_info.c:1929 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
<4>[37927.809710] STA 98:e7:f5:c1:72:e9 AC 2 txq pending airtime underflow: 4294967064, 232
<4>[37927.809722] Modules linked in: ksmbd natcap qcserial pppoe ppp_async option l2tp_ppp cdc_mbim usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY wireguard usbserial usbnet usblp ts_fsm ts_bm slhc sch_cake r8152 nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nf_tproxy_ipv6
<4>[37927.825524]  nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables crc_ccitt compat_xtables compat cdc_wdm br_netfilter asn1_decoder natflow fuse tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet
<4>[37927.999014]  ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip6_gre ip_gre gre ifb sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip6_tunnel tunnel6 tunnel4 ip_tunnel tun vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac ghash_generic gf128mul gcm ecb des_generic libdes ctr cmac ccm arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ohci_platform ohci_hcd softdog fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat usbcore nls_base usb_common mii crc32c_generic
<4>[37928.171741]  [last unloaded: natcap]
<4>[37928.351705] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 5.4.52 #0
<4>[37928.363492] Stack : 00000000 8006eb90 80650000 8064c760 806b0000 8064c728 8064b88c 87c0dcb4
<4>[37928.380137]         807f0000 87c3beb4 80694c03 805e1298 00000001 00000001 87c0dc58 2aa32be6
<4>[37928.396765]         00000000 00000000 80830000 00000000 00000030 0000018c 342e3520 2032352e
<4>[37928.413393]         00000000 0000075e 00000000 00055dd9 00000000 806b0000 00000000 85d08a84
<4>[37928.430022]         00000009 00000000 000000ad ffffffff 00000002 802f71a0 00000004 807f0004
<4>[37928.446650]         ...
<4>[37928.451511] Call Trace:
<4>[37928.456397] [<8000b5fc>] show_stack+0x30/0x100
<4>[37928.465250] [<8052b8e0>] dump_stack+0xa4/0xdc
<4>[37928.473926] [<80027c54>] __warn+0xc0/0x10c
<4>[37928.482074] [<80027d2c>] warn_slowpath_fmt+0x8c/0xac
<4>[37928.492107] [<85d08a84>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
<4>[37928.507597] [<85d0205c>] ieee80211_report_low_ack+0xf0/0x43c [mac80211]
<4>[37928.520838] [<85d023c0>] ieee80211_free_txskb+0x18/0x30 [mac80211]
<4>[37928.533194] [<85c25b08>] mt76_tx_complete_skb+0xa8/0xbc [mt76]
<4>[37928.544811] [<85c20ae8>] mt76_dma_attach+0x59c/0x868 [mt76]
<4>[37928.556228] ---[ end trace 704c686bd67fb08c ]---
<6>[69407.097391] ralink_soc_eth 1e100000.ethernet eth0: port 1 link up
<6>[69409.594850] ralink_soc_eth 1e100000.ethernet eth0: port 0 link up
<6>[128097.568290] TCP: request_sock_TCP: Possible SYN flooding on port 5000. Sending cookies.  Check SNMP counters.
<4>[129651.257310] {natflow}:natflow_netdev_event(): catch NETDEV_UNREGISTER event for dev=pppoe-wan
<4>[129651.274827] warning: catch unregister event for dev=pppoe-wan
<4>[129651.670017] {natflow}:natflow_netdev_event(): catch NETDEV_UNREGISTER event for dev=eth0.2
<4>[129651.686883] warning: catch unregister event for dev=eth0.2
<4>[129651.753427] warning: (zone_netdev_event:346)dev=eth0.2 set zone=127 type=0
<4>[129651.767384] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=eth0.2, add ingress hook
[... cut ...]
<4>[136035.578886]         85e08dfc 8011cdfc 85dc6580 80150004 00003730 00000003 87cdc000 83427000
<4>[136035.595686]         808954e0 00000000 00000000 00000000 00000000 00000000 00000000 00000000
<4>[136035.612485]         ...
<4>[136035.617518] Call Trace:
<4>[136035.622558] [<85c2507c>] mt76_eeprom_init+0x220/0x278 [mt76]
<4>[136035.633990] Code: 8e1008c0  8e020004  8e030000 <ac620004> ac430000  24020100  ae020000  24020122  0c153283 
<4>[136035.653566] 
<4>[136035.658217] ---[ end trace 704c686bd67fb08d ]---

```