openwrt/mt76 #413: MT7628 just have one antenna work in mt76 driver

wurobinson, 2020-06-25:
Hi,
I did an experiment on MT7628 router whitch has two antennas.  The steps are

1.Burn an OpenWrt firmware into MT7628 router. 
2. Start the OpenWrt then turn on the radio of MT7628 by OpenWrt.
3.  Disconnet one of physical antenna connections, just reserve one  physical antenna connection. 
4. Keep a certain distance  between WLAN card and MT7628 antenna, test and record the RSSI which is sent from MT7628 by the WLAN card. 
5. Reconnect the previously disconnected antenna, disconnect the previously connected antenna. Repeat step 4.
6. Disconnec t all antennas, Repeat step 4.

The result is when connect "antenna 0", the RSSI will be good, assumption is -25dbm, while connect "antenna 1" is -75dbm, and when disconnect all antenna the RSSI is -75dbm too. This means whether connect "antenna 1" or not make no difference to siginal, "antenna 1" not work!  I think here is an issue in this mt76 driver,  that result in "antenna 1" in off state.

I had tried  OpenWrt firmware release  “17.01.x”, "18.06.x", "19.07.x" , target profile is "HiWiFi HC5861B" by HiWiFi HC5861B router and some other MT7628 routers. The test results are the same.

This issue may be hard to discover, because the MT7628 routers usually always have two antenna connections, and in most case here is no big difference between just one antenna work and two antennas work. When some people burn the OpenWrt firmware into their MT7628 routers, and their wireless terminals have a good signal, they will take for granted that the antennas works. 

Here is some debug messages

[   10.378544] Loading modules backported from Linux version v4.19.98-0-gd183c8e2647a
[   10.386304] Backport generated by backports.git v4.19.98-1-0-g8204eb99
[   10.412027] ip_tables: (C) 2000-2006 Netfilter Core Team
[   10.437855] nf_conntrack version 0.5.0 (2048 buckets, 8192 max)
[   10.568603] xt_time: kernel timezone is -0000
[   10.772717] mt76_wmac 10300000.wmac: ASIC revision: 76280001
[   10.873431] urngd: v1.0.2 started.
[   11.076609] random: crng init done
[   11.080101] random: 6 urandom warning(s) missed due to ratelimiting
[   11.811475] mt76_wmac 10300000.wmac: Firmware Version: 20151201
[   11.817498] mt76_wmac 10300000.wmac: Build Time: 20151201183641
[   11.859424] mt76_wmac 10300000.wmac: firmware init done
[   12.147870] PPP generic driver version 2.4.2
[   12.160473] NET: Registered protocol family 24

Here is the wireless configuration

config wifi-device 'radio0'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path 'platform/10300000.wmac'
        option htmode 'HT20'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

root@OpenWrt:/# wifi
root@OpenWrt:/# [  115.791994] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  115.803450] br-lan: port 2(wlan0) entered blocking state
[  115.808854] br-lan: port 2(wlan0) entered disabled state
[  115.814737] device wlan0 entered promiscuous mode
[  115.832208] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  115.838890] br-lan: port 2(wlan0) entered blocking state
[  115.844356] br-lan: port 2(wlan0) entered forwarding state
[  125.991158] device wlan0 left promiscuous mode
[  125.995811] br-lan: port 2(wlan0) entered disabled state
[  126.606147] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  126.635667] br-lan: port 2(wlan0) entered blocking state
[  126.641150] br-lan: port 2(wlan0) entered disabled state
[  126.646930] device wlan0 entered promiscuous mode
[  126.651888] br-lan: port 2(wlan0) entered blocking state
[  126.657281] br-lan: port 2(wlan0) entered forwarding state
[  126.716902] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

sanitariu, 2020-08-02:
Maybe the problem is from firmware to choose which antenna is working. I have devices like tp-link 840 version 14 or version 5 which have exactly the same issue. Very low signal with openwrt. I think second antenna is not setup correctly on mediatek 7628

closed 2020-11-03 by wurobinson

mythfish, 2022-03-11:
@wurobinson is there any solution for this issue?