openwrt/mt76 #416: mt766u module not recognized as wifi module

lucabianchi91, 2020-06-26:
Hi,
we have some problems with a usb wifi module `HC-M7662BU1` with MTK chipset `MT7662U`.
The device is recognized as usb (i.e. `lsusb` shows `Bus 001 Device 007: ID 0e8d:7632 MediaTek Inc. 
`) but, even though `mt76x2u` driver is loaded, the module is not recognized as a wireless interface.
We also checked that the firmwares `mt7662.bin` and `mt7662_rom_patch.bin` are present in `/lib/firmware` and dmesg shows only the following:
```
[   72.219555] usb 1-1.4: new high-speed USB device number 7 using dwc_otg
[   72.249720] usb 1-1.4: New USB device found, idVendor=0e8d, idProduct=7632, bcdDevice= 1.00
[   72.263982] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   72.274412] usb 1-1.4: Product: BT+WIFI
[   72.281279] usb 1-1.4: Manufacturer: MediaTek Inc.
[   72.289069] usb 1-1.4: SerialNumber: 000000000
``` 

Any idea of what could be the issue?

LorenzoBianconi, 2020-06-26:
> Hi,
> we have some problems with a usb wifi module `HC-M7662BU1` with MTK chipset `MT7662U`.
> The device is recognized as usb (i.e. `lsusb` shows `Bus 001 Device 007: ID 0e8d:7632 MediaTek Inc. `) but, [...]

Could you please try this [patch](https://github.com/LorenzoBianconi/wireless-drivers-next/commit/03bd7f3588a706a9a7585177b60d2310680ae0b9)?

lucabianchi91, 2020-06-29:
Hi Lorenzo,
the patch worked (we see the module as wireless interface and we can ping/iperf through it) but  we notice the following in `dmesg`:
```
[    7.771465] usb 1-1.2: reset high-speed USB device number 4 using dwc_otg
[    7.800973] mt76x2u: probe of 1-1.2:1.0 failed with error -22
[    7.887449] usb 1-1.2: reset high-speed USB device number 4 using dwc_otg
[    7.917023] mt76x2u: probe of 1-1.2:1.1 failed with error -22
[    8.003424] usb 1-1.2: reset high-speed USB device number 4 using dwc_otg
[    8.033247] mt76x2u 1-1.2:1.2: ASIC revision: 76620044
[    8.075614] mt76x2u 1-1.2:1.2: ROM patch build: 20141115060606a
[    8.263026] mt76x2u 1-1.2:1.2: Firmware Version: 0.0.00
[    8.269725] mt76x2u 1-1.2:1.2: Build: 1
[    8.275460] mt76x2u 1-1.2:1.2: Build Time: 201507311614____
[    8.468088] mt76x2u 1-1.2:1.2: error: MCU resp evt:9 seq:1-0
[    8.474804] mt76x2u 1-1.2:1.2: error: MCU resp evt:9 seq:1-0
[    8.481507] mt76x2u 1-1.2:1.2: error: MCU resp evt:9 seq:1-0
[    9.096417] ieee80211 phy2: Selected rate control algorithm 'minstrel_ht'
[    9.097749] usbcore: registered new interface driver mt76x2u
``` 
so instead of using `1-1.2:1.0` and being mounted as `wlan1`, it uses `1-1.2:1.2` and `wlan2`. 
We will investigate if this is a power issue. Should I close this issue for now?
Thank you

LorenzoBianconi, 2020-06-29:
the patch worked (we see the module as wireless interface and we can
ping/iperf through it) but we notice the following in dmesg:

> [    7.771465] usb 1-1.2: reset high-speed USB device number 4 using dwc_otg
> [    7.800973] mt76x2u: probe of 1-1.2:1.0 failed with error -22
> [    7.887449] usb 1-1.2: reset high-speed USB device number 4 using dwc_otg
> [    7.917023] mt76x2u: probe of 1-1.2:1.1 failed with error -22
> [    8.003424] usb 1-1.2: reset high-speed USB device number 4 using dwc_otg
> [    8.033247] mt76x2u 1-1.2:1.2: ASIC revision: 76620044
> [    8.075614] mt76x2u 1-1.2:1.2: ROM patch build: 20141115060606a
> [    8.263026] mt76x2u 1-1.2:1.2: Firmware Version: 0.0.00
> [    8.269725] mt76x2u 1-1.2:1.2: Build: 1
> [    8.275460] mt76x2u 1-1.2:1.2: Build Time: 201507311614____
> [    8.468088] mt76x2u 1-1.2:1.2: error: MCU resp evt:9 seq:1-0
> [    8.474804] mt76x2u 1-1.2:1.2: error: MCU resp evt:9 seq:1-0
> [    8.481507] mt76x2u 1-1.2:1.2: error: MCU resp evt:9 seq:1-0
> [    9.096417] ieee80211 phy2: Selected rate control algorithm 'minstrel_ht'
> [    9.097749] usbcore: registered new interface driver mt76x2u
>
> so instead of using 1-1.2:1.0 and being mounted as wlan1, it uses
> 1-1.2:1.2 and wlan2.
> We will investigate if this is a power issue. Should I close this issue
> for now?
> Thank you
>
> <https://github.com/openwrt/mt76/issues/416#issuecomment-651209714>, or
> unsubscribe

Hi Luca,

After the initial error, is the dongle working fine?

Regards,
Lorenzo

lucabianchi91, 2020-06-29:
Yes, it's working fine

lucabianchi91, 2020-06-29:
In the meanwhile, we checked again and we can say that the `mt76x2u: probe of 1-1.2:1.1 failed with error -22` errors are not related to electrical problems

pgqz53, 2020-09-26:
If this has been fixed, can we close the issue now? @lucabianchi91 

On an unrelated note, does Bluetooth on this module work for you? I can't seem to find any MediaTek USB Bluetooth drivers in the Linux kernel unfortunately. Would you mind posting the VID&PID of the Bluetooth part as well?

lucabianchi91, 2020-09-28:
We didn't try new versions of the driver since we didn't know they were targeting this issue (did they?)
On June,29 the module was working but only after the reported errors (that prevent a stable configuration of devices).
If needed we can try again (as soon as we can access our offices)

pgqz53, 2020-10-02:
Unfortunately the USB drivers don't seem to receive much attention these days, as I wasn't able to find any relevant fixes when skimming through the commit history. It's a pity because `mt76` is actually still the only viable solution to get 802.11ac over USB without DKMS drivers, but mt76x0u is completely non-functional ( #454 ), and judging from this issue, mt76x2u doesn't seem to play nice with chips newer than MT7612U either. 

Additionally, while I can't source the MT7662U module you mentioned right now, I do have some MT7632U modules on hand. MediaTek doesn't seem to have MT7632U on their website anymore, but I happen to know the USB PID & VID is also `0e8d 7632`, which makes me suspect MT7662U is more or less a rebadged version of MT7632U with some minor changes. I'll post my experiences with said MT7632U module and see if I can reproduce this issue later this week.

pgqz53, 2020-10-09:
@lucabianchi91 Are you sure the module is actually MT7662U instead of MT7632U? According to the ID combination, what you have seem to be an MT7632U chip instead. I'm aware the ID can be changed in the eeprom/efuse, but I see very little reason of having a MT7662U pretending to be a MT7632U.

I have tested two (almost) identical MT7632U modules, and I don't seem to get the probing errors as you did. Maybe this is somehow related to the dwc USB controller? It might be a good idea if you could connect this to something with a different USB host controller (on a computer, or try a different SoC) just to further isolate this problem.

lucabianchi91, 2020-10-09:
Yes, we are sure the module is mt7662bu.
We have updated the driver to the 31 May commit and now the interface is behaving well  - it's loaded as and remains as wlan0.
This is the new kernel log `dmesg`:
```
[   10.263182] usb 2-1: reset high-speed USB device number 2 using ehci-platform
[   10.297434] mt76x2u: probe of 2-1:1.0 failed with error -22
[   10.439148] usb 2-1: reset high-speed USB device number 2 using ehci-platform
[   10.473368] mt76x2u: probe of 2-1:1.1 failed with error -22
[   10.611143] usb 2-1: reset high-speed USB device number 2 using ehci-platform
[   10.645439] mt76x2u 2-1:1.2: ASIC revision: 76620044
[   10.706014] mt76x2u 2-1:1.2: ROM patch build: 20141115060606a
[   10.900336] mt76x2u 2-1:1.2: Firmware Version: 0.0.00
[   10.905400] mt76x2u 2-1:1.2: Build: 1
[   10.909070] mt76x2u 2-1:1.2: Build Time: 201507311614____
[   11.125187] mt76x2u 2-1:1.2: error: MCU resp evt:9 seq:1-0
[   11.130811] mt76x2u 2-1:1.2: error: MCU resp evt:9 seq:1-0
[   11.136435] mt76x2u 2-1:1.2: error: MCU resp evt:9 seq:1-0
[   11.752619] ieee80211 phy2: Selected rate control algorithm 'minstrel_ht'
```
We have no problems with mt76x0 and mt76x2 in IBSS mode.

closed 2020-10-09 by lucabianchi91

dsseng, 2024-02-01:
mt7662tu module fails to start (after adding an ID manually). Is it expected to be supported by mt76x2u? Note the different revision (100 vs 044). I am trying to repurpose an [LG TV wireless module](https://fccid.io/BEJLGSBWAC72). BT part works fine, both pairing and mSBC voice communication works before I try to probe the WLAN driver.

[image]