openwrt/mt76 #418: mt7615e 5ghz wifi after some time will stop working and result in dnsmasq spam

jazzrz86, 2020-06-30:
after about 10-11 hours, the 5ghz wifi will stop working and logs show a spam of errors on dnsmasq. The devices are still connected, but no longer have access to internet:

`Tuee Jun 30 14:35:40 2020 daemon.info dnsmasq-dhcp[10520]: DHCPREQUEST(br-lan) 192.168.1.111 98:de:d0:57:ff:1d
Tue Jun 30 14:35:40 2020 daemon.info dnsmasq-dhcp[10520]: DHCPACK(br-lan) 192.168.1.111 98:de:d0:57:ff:1d DESKTOP-IK9GMJA
Tue Jun 30 14:54:32 2020 daemon.info dnsmasq-dhcp[10520]: DHCPREQUEST(br-lan) 192.168.1.174 b4:b5:2f:14:a5:3b
Tue Jun 30 14:54:32 2020 daemon.info dnsmasq-dhcp[10520]: DHCPACK(br-lan) 192.168.1.174 b4:b5:2f:14:a5:3b HP078CF3
Tue Jun 30 14:55:23 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:55:27 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:55:31 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:55:35 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:55:39 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:55:44 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:55:48 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:55:52 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:55:56 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:56:00 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:56:04 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:56:08 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:56:12 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:56:16 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:56:20 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:56:24 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:56:28 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:56:32 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:56:36 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable
Tue Jun 30 14:56:40 2020 daemon.err dnsmasq[10520]: failed to send packet: Resource temporarily unavailable`

The only way to fix this is to restart the 5ghz wifi, after which it will work again for another 10 to 11 hours.

Using master and latest mt76 (as of june 22). This also happens on the snapshot mt76 (may 30).

I initially thought this was related to #406 and its related issues, but it is not, as the kernel error can occur before or after dnsmasq errors.

kernel log:
`[    0.000000] Linux version 5.4.48 (jjubuntu@jjubuntu-GA-MA785GM-US2H) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r13527-61307544d1)) #0 SMP Mon Jun 29 13:55:55 2020
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router 3 Pro
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s26672 r8192 d22480 u57344
[    0.000000] pcpu-alloc: s26672 r8192 d22480 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00014fcf
[    0.000000] Readback ErrCtl register=00014fcf
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 511088K/524288K available (5286K kernel code, 202K rwdata, 560K rodata, 1236K init, 226K bss, 13200K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[... 311 lines cut ...]
[   27.964603] mt7530 mdio-bus:1f lan2: Link is Up - 100Mbps/Full - flow control rx/tx
[   27.972471] br-lan: port 2(lan2) entered blocking state
[   27.977750] br-lan: port 2(lan2) entered forwarding state
[   28.011761] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   29.120478] mt7530 mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control rx/tx
[   29.128306] br-lan: port 3(lan3) entered blocking state
[   29.133610] br-lan: port 3(lan3) entered forwarding state
[   30.153291] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control rx/tx
[   31.122199] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[   31.155694] br-lan: port 4(wlan0) entered blocking state
[   31.161155] br-lan: port 4(wlan0) entered disabled state
[   31.167249] device wlan0 entered promiscuous mode
[   31.499315] do_page_fault(): sending SIGSEGV to rrdtool queue for invalid read access from 00000021
[   31.508508] epc = 77e57cf4 in libc.so[77e0e000+97000]
[   31.513717] ra  = 77e58194 in libc.so[77e0e000+97000]
[   32.182974] pppoe-wan: renamed from ppp0
[   33.242905] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   33.250070] br-lan: port 4(wlan0) entered blocking state
[   33.255545] br-lan: port 4(wlan0) entered forwarding state
[   34.623320] br-lan: port 5(wlan1) entered blocking state
[   34.628715] br-lan: port 5(wlan1) entered disabled state
[   34.635097] device wlan1 entered promiscuous mode
[   37.415657] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   37.422661] br-lan: port 5(wlan1) entered blocking state
[   37.428064] br-lan: port 5(wlan1) entered forwarding state`

nbd168, 2020-08-12:
Please try the latest version

jazzrz86, 2020-08-13:
@nbd168 unfortunately this is still happening, after 10 hrs the same result. Restarting 5ghz wifi still works.
Also still seeing massive amounts of interrupt errors, not sure of that is related. Performance wise, it is very good if only this teething issue could be debugged. Is there anything I can help you to find this problem? So far there are no kernel logs or errors when this occurrs.

jazzrz86, 2020-08-26:
This is still happening as of the latest version (b36d7ae096a3d8c7d6a8a246f2e8a471a467041e). The interrupt errors have reduced greatly, and thus seems unrelated to this issue.

fluffymadness, 2020-08-29:
I wonder if there is any change when enabling sqm.
The openwrt wiki says that anything regarding flow offload (software + hardware offload) will be disabled and the cpu does everything when enabling sqm
https://github.com/openwrt/mt76/issues/407#issuecomment-681844696

marcussacana, 2023-12-10:
any hints to solve this problem? I'm not sure if is fault of 5ghz in my case but my system flood that same error message and the internet DNS just stop from responding and after a while the router became complete frozen.