openwrt/mt76 #419: Request to rewrite driver for mt76 platform

Openwrtfunboy, 2020-06-30:
The fact is that a lot of time has passed, a lot of patches have come out, and the driver is still frankly bad: disconnecting clients, missing Internet, kernel warning and kernel panic.

The driver in the current implementation is terrible, it is not comfortable to use it. I am ready to help developers by becoming a tester, like many active participants here. The driver must be rewrite.

Hope for understanding.

ryderlee1110, 2020-06-30:
which chips? I think at least mt7615 is relative stable on mt7622 platform, but I know that there're still some issues happens on mt7621.

jazzrz86, 2020-07-01:
@ryderlee1110 do you think the issues we are seeing on mt7615 with mt7621 is not actually due to the mt7615 drivers but due to the mt7621 soc drivers? esp with the interrupt error spam. I reversed those 2 MSI patches to test but still getting interrupt errors. However, not seeing the dnsmasq spam yet although its early still.

ryderlee1110, 2020-07-01:
> @ryderlee1110 do you think the issues we are seeing on mt7615 with mt7621 is not actually due to the mt7615 drivers but due to the mt7621 soc drivers? esp with the interrupt error spam. I reversed [...]

I actually don't know. I've never seen interrupt errors on mt7622 platform.

tkso1997, 2020-07-01:
@ryderlee1110 actually I don't think that the cpu is to weak. It performs really good in a fully wired environment or with proprietary drivers. 
Would it help if we donate a mt7621 device?

Openwrtfunboy, 2020-07-02:
> @ryderlee1110 actually I don't think that the cpu is too weak. It performs really good in a fully wired environment or with proprietary drivers.
> Would it help if we donate a mt7621 device?

Hm. Maybe you have instruction to install proprietary driver?

boecko, 2020-07-15:
With the current mt76-HEAD (aa97be8e02de14cc5418e467cfd91b5184515e59) my TP-Link RE650 performs quite nicely with iperf3.
MacBook <-5GHz-> TP-Link RE650 <-GBIT Ethernet-> RaspberryPi4
```
[ ID] Interval           Transfer     Bitrate
[ 10]   0.00-1.00   sec  66.9 MBytes   561 Mbits/sec
[ 10]   1.00-2.00   sec  61.4 MBytes   516 Mbits/sec
[ 10]   2.00-3.00   sec  65.0 MBytes   545 Mbits/sec
[ 10]   3.00-4.00   sec  68.3 MBytes   573 Mbits/sec
[ 10]   4.00-5.00   sec  71.1 MBytes   596 Mbits/sec
[ 10]   5.00-6.00   sec  69.9 MBytes   587 Mbits/sec
[ 10]   6.00-7.00   sec  45.6 MBytes   383 Mbits/sec
[ 10]   7.00-8.00   sec  58.5 MBytes   491 Mbits/sec
[ 10]   8.00-9.00   sec  64.6 MBytes   542 Mbits/sec
[ 10]   9.00-10.00  sec  70.0 MBytes   587 Mbits/sec
```

Didn't test it in the long term, but the results are promising.

tkso1997, 2020-07-15:
Could you check the interrupt error thing by running ‘cat /proc/interrupts’

jazzrz86, 2020-07-16:
> Could you check the interrupt error thing by running ‘cat /proc/interrupts’

I just checked it is still happening. Looks like the closed source driver is still the way to go

boecko, 2020-07-16:
> Could you check the interrupt error thing by running ‘cat /proc/interrupts’
```
root@kingslanding:~# uptime
 18:49:14 up 1 day, 5 min,  load average: 0.09, 0.04, 0.00
root@kingslanding:~# cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
  8:   21678461   21678561   21678124   21678517  MIPS GIC Local   1  timer
  9:     849260          0          0          0  MIPS GIC  63  IPI call
 10:          0    1556802          0          0  MIPS GIC  64  IPI call
 11:          0          0   48638430          0  MIPS GIC  65  IPI call
 12:          0          0          0    1052777  MIPS GIC  66  IPI call
 13:    4871846          0          0          0  MIPS GIC  67  IPI resched
 14:          0    6800992          0          0  MIPS GIC  68  IPI resched
 15:          0          0    1559336          0  MIPS GIC  69  IPI resched
 16:          0          0          0    3993761  MIPS GIC  70  IPI resched
 17:          0          0          0          0  MIPS GIC  19  1e000600.gpio-bank0, 1e000600.gpio-bank1, 1e000600.gpio-bank2
 19:         14          0          0          0  MIPS GIC  33  ttyS0
 21:   90858880          0          0          0  MIPS GIC  10  1e100000.ethernet
 23:        108          0   26421930          0  MIPS GIC  11  mt7615e
 24:         50          0          0   90060396  MIPS GIC  31  mt7615e
 26:          0          0          0          0  1e000600.gpio  18  keys
 27:          0          0          0          0  1e000600.gpio  25  keys
 30:          0          0          0          0  1e000600.gpio  30  keys
 31:          0          0          0          0  1e000600.gpio  31  keys
ERR:     587490
```

tkso1997, 2020-07-17:
:(

tkso1997, 2020-07-17:
Is it appart from these spooky interrupt errors stable?

boecko, 2020-07-17:
5GHz is stable. I had no disconnects over the night while running a timemachine backup.

boecko, 2020-07-17:
> > Could you check the interrupt error thing by running ‘cat /proc/interrupts’
> 
> I just checked it is still happening. Looks like the closed source driver is still the way to go

The closed source driver is just for the 4.x kernel, isn't it?

tkso1997, 2020-07-17:
Nope it runs fine on 5.x

boecko, 2020-07-18:
> Nope it runs fine on 5.x
> 
> I am using this:
> https://github.com/coolsnowwolf/lede/tree/master/package/lean/mt

Puh .. I compiled it against openwrt/master and tried it out.
This might be a good alternative **if**
- you are fluent in mandarin and can accept non standard luci-wifi interface
- CFG80211 would work (it doesn't)

Openwrtfunboy, 2020-07-19:
> Nope it runs fine on 5.x
> 
> I am using this:
> https://github.com/coolsnowwolf/lede/tree/master/package/lean/mt

How can i install proprietary driver to 19.07.3?

boecko, 2020-07-23:
Interrupt errors count is lower with the current HEAD (b5df0fbb1847a51f0a11f03a3fa8dc9d98d4df82) 
Maybe because of 
> mt7615: update firmware to version 20200629
?

```
root@kingslanding:~# uptime
 09:49:30 up  9:11,  load average: 0.00, 0.00, 0.00
root@kingslanding:~# cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
  8:    8273664    8273504    8273605    8273675  MIPS GIC Local   1  timer
  9:     222956          0          0          0  MIPS GIC  63  IPI call
 10:          0     514656          0          0  MIPS GIC  64  IPI call
 11:          0          0     902170          0  MIPS GIC  65  IPI call
 12:          0          0          0     346162  MIPS GIC  66  IPI call
 13:    2547198          0          0          0  MIPS GIC  67  IPI resched
 14:          0    1044332          0          0  MIPS GIC  68  IPI resched
 15:          0          0     260185          0  MIPS GIC  69  IPI resched
 16:          0          0          0    1506429  MIPS GIC  70  IPI resched
 17:          0          0          0          0  MIPS GIC  19  1e000600.gpio-bank0, 1e000600.gpio-bank1, 1e000600.gpio-bank2
 19:         14          0          0          0  MIPS GIC  33  ttyS0
 21:    3027266          0          0          0  MIPS GIC  10  1e100000.ethernet
 23:        142          0    3967511          0  MIPS GIC  11  mt7615e
 24:         60          0          0    4056906  MIPS GIC  31  mt7615e
 26:          0          0          0          0  1e000600.gpio  18  keys
 27:          0          0          0          0  1e000600.gpio  25  keys
 30:          0          0          0          0  1e000600.gpio  30  keys
 31:          0          0          0          0  1e000600.gpio  31  keys
ERR:      48295
```

jazzrz86, 2020-07-23:
> Interrupt errors count is lower with the current HEAD ([b5df0fb](https://github.com/openwrt/mt76/commit/b5df0fbb1847a51f0a11f03a3fa8dc9d98d4df82))
> Maybe because of
> 
> > mt7615: update firmware to version 20200629
> > ?
> 
> ```
> root@kingslanding:~# uptime
>  09:49:30 up  9:11,  load average: 0.00, 0.00, 0.00
> root@kingslanding:~# cat /proc/interrupts
>            CPU0       CPU1       CPU2       CPU3
>   8:    8273664    8273504    8273605    8273675  MIPS GIC Local   1  timer
>   9:     222956          0          0          0  MIPS GIC  63  IPI call
>  10:          0     514656          0          0  MIPS GIC  64  IPI call
>  11:          0          0     902170          0  MIPS GIC  65  IPI call
>  12:          0          0          0     346162  MIPS GIC  66  IPI call
>  13:    2547198          0          0          0  MIPS GIC  67  IPI resched
>  14:          0    1044332          0          0  MIPS GIC  68  IPI resched
>  15:          0          0     260185          0  MIPS GIC  69  IPI resched
>  16:          0          0          0    1506429  MIPS GIC  70  IPI resched
>  17:          0          0          0          0  MIPS GIC  19  1e000600.gpio-bank0, 1e000600.gpio-bank1, 1e000600.gpio-bank2
>  19:         14          0          0          0  MIPS GIC  33  ttyS0
>  21:    3027266          0          0          0  MIPS GIC  10  1e100000.ethernet
>  23:        142          0    3967511          0  MIPS GIC  11  mt7615e
>  24:         60          0          0    4056906  MIPS GIC  31  mt7615e
>  26:          0          0          0          0  1e000600.gpio  18  keys
>  27:          0          0          0          0  1e000600.gpio  25  keys
>  30:          0          0          0          0  1e000600.gpio  30  keys
>  31:          0          0          0          0  1e000600.gpio  31  keys
> ERR:      48295
> ```

To be honest this is about where it was at the start of june. and still the driver will start to show issues around 1 to 2 days uptime.  as always, restarting the 5ghz wifi will fix it again for another 1 to 2 days. this never happens when its 2.4ghz though

namidairo, 2020-07-23:
> Interrupt errors count is lower with the current HEAD ([b5df0fb](https://github.com/openwrt/mt76/commit/b5df0fbb1847a51f0a11f03a3fa8dc9d98d4df82))
> Maybe because of
> 
> > mt7615: update firmware to version 20200629
> > ?

From linux-firmware... The delta from 20200131 looks fairly minimal too, so it's probably just the one change.

```
Update to 20200629111339 version to aligh SDK. Mainly fix DFS false alarm.
```

nbd168, 2020-08-12:
Please try the latest version, it should be much more stable now

boecko, 2020-08-13:
2.4GHz is more stable now, too. Thank you
Still a lot of interrupt errors, though.

nbd168, 2020-08-23:
If your version was older than 37615174f5b360bfbcf5f11813500599e3b8196d (from August 12), you need to test a newer version, because there was a critical stability fix in there

bricco1981, 2020-08-28:
yes, last version of mt76 looks much better and stable.

Openwrtfunboy, 2020-08-30:
> If your version was older than 37615174f5b360bfbcf5f11813500599e3b8196d (from August 12), you need to test a newer version, because there was a critical stability fix in there

Hi @nbd168 
I try snapshot from 30 august(i have newifi3 d2) and i still have problem with loss internet.

Openwrtfunboy, 2020-08-30:
I had 7 devices of 2.4 and 3 devices of 5. PS4 was downloading the game at that moment and was connected at 2.4, the computer was downloading a torrent and was similarly connected at 2.4, a 5 GHz TV was playing 4k content ( YouTube).

> > If your version was older than 37615174f5b360bfbcf5f11813500599e3b8196d (from August 12), you need to test a newer version, because there was a critical stability fix in there
> 
> Hi @nbd168
> I try snapshot from 30 august(i have newifi3 d2) and i still have problem with loss internet.
What i do before one device (phone) lost internet. 
I had 7 devices of 2.4 and 3 devices of 5. PS4 was downloading the game at that moment and was connected at 2.4, the computer was downloading a torrent and was similarly connected at 2.4, a 5 GHz TV was playing 4k content (YouTube).

stevenxed, 2020-08-31:
Since the Aug 25th commit, devices on 2.4ghz are losing packets and have an average ping time of 25-30ms to the access point with few dips back down to 1-2ms.

5ghz is relatively stable until a day or two later. I've noticed a decrease in false positive deauths via inactivity timer, but otherwise same disconnects/reconnects.

Openwrtfunboy, 2020-08-31:
> Since the Aug 25th commit, devices on (MT7603E) 2.4ghz are losing packets and have an average ping time of 25-30ms to the access point with few dips back down to 1-2ms.
> 
> (MT7615E) 5ghz is relatively stable until a day or two later. I've noticed a decrease in false positive deauths via inactivity timer, but otherwise same disconnects/reconnects.

Notice same with loss packets when i play in online game.

ryderlee1110, 2020-08-31:
Are we talking about MT7615 or MT7603? Did you guys use the latest openwrt master?

mbc07, 2020-08-31:
FWIW the mt76 always ran very stable for me on a D-Link DIR-882 (MT7621AT + dual MT7615N radios). The only real issue I had from time to time was some devices under the 5 GHz band momentarily dropping to very low TX rates (less than 1 Mbps, RX rate was unaffected) for about half a minute before ramping up to the usual rates, but that got fixed in the OpenWrt snapshot from August 12th. As far as I remember I got a uptime of nearly 9 days on that snapshot, without experiencing any kind of stability or connection loss issue in any band.

That uptime got interrupted when I updated to the OpenWrt snapshot from August 23th, which significantly reduced the number of IRQ errors. Current uptime on my DIR-882 is close to reaching full 7 days at the time of this post and the wireless of both bands remain rock solid and without issues just like the August 12th snapshot I was using earlier...

jazzrz86, 2020-08-31:
@mbc07 Interesting, I have a mir3p and its basically the same hardware but the 5ghz wifi loses dns always after about 12 hrs. Can i see what config options you put in your snapshot? what addons do you use?

stevenxed, 2020-08-31:
@ryderlee1110 Yes I use the compiled snapshots and still on the Aug 25th build (18fbb9aa21). Regarding the disconnects/reconnects, it happens with both bands, so both MT7615 and MT7603 chips. The packet loss and 25ms ping issue is only noticeable with 2.4ghz. Using a nanoHD and unfortunately the stock firmware is giving me the same disconnecting issues, though not surprising since it seems to be based on LEDE.

mbc07, 2020-08-31:
@jazzrz86 nothing really special, just used the image builder with the following additional packages:
`make image PROFILE="dlink_dir-882-a1" PACKAGES="kmod-fs-ext4 kmod-fs-vfat kmod-usb-storage-uas block-mount ca-bundle collectd-mod-disk collectd-mod-uptime dosfstools e2fsprogs kmod-fs-antfs antfs-mount ipset libustream-openssl20200215 luci luci-app-samba4 luci-app-statistics luci-app-upnp wpad-openssl -wpad-basic-wolfssl"`

My ISP modem is configured in bridge mode and automatically serves public IPv4 and IPv6 addresses and DNS servers to the DIR-882 through DHCP, so nearly all OpenWrt settings were left on the default values, except for the wireless network settings and the settings related to the additional packages I bundled on the image (UPnP, Samba, RRDTool)...

ayavilevich, 2020-10-06:
Hi, I am running OpenWRT on a DIR-882 router with a MT7621 SoC.

My kernel log says:

```
[   12.474753] mt7621-pci 1e140000.pcie: bus=1 slot=0 irq=23
[   12.496889] mt7615e 0000:01:00.0: enabling device (0000 -> 0002)
[   12.525405] mt7621-pci 1e140000.pcie: bus=2 slot=1 irq=24
[   12.537527] mt7615e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
[   12.537527] 
[   12.547591] mt7615e 0000:02:00.0: enabling device (0000 -> 0002)
[   12.598890] mt7615e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
[   12.598890] 
[   12.858757] mt7615e 0000:01:00.0: N9 Firmware Version: _reserved_, Build Time: 20200814163649
[   12.875916] mt7615e 0000:02:00.0: N9 Firmware Version: _reserved_, Build Time: 20200814163649
[   12.947121] mt7615e 0000:02:00.0: CR4 Firmware Version: _reserved_, Build Time: 20190121161307
[   12.947132] mt7615e 0000:01:00.0: CR4 Firmware Version: _reserved_, Build Time: 20190121161307
```

Does it mean I am on the latest driver version? How does one update drivers?
What does each "built time" value above referrers to?

Thanks!

Espionage724, 2020-12-08:
> Nope it runs fine on 5.x
> 
> I am using this:
> https://github.com/coolsnowwolf/lede/tree/master/package/lean/mt

How would I go about having my OpenWRT use those drivers instead of `mt79`? I compiled OpenWRT from that source for my router (NETGEAR R6260), but it still seems to use `mt79` for `MT7615E`.