openwrt/mt76 #421: missing 2.4g in ipTIME A8004T + mt7615e

d3m3vilurr, 2020-07-05:
similar #381 issues, an driver init step was failed, it has been only enabled 5g network.
```
Sun Jul  5 15:51:10 2020 user.info kernel: [   10.274136] procd: - init -
Sun Jul  5 15:51:10 2020 user.info kernel: [   10.668362] kmodloader: loading kernel modules from /etc/modules.d/*
Sun Jul  5 15:51:10 2020 kern.info kernel: [   10.694459] Loading modules backported from Linux version v5.7-rc3-0-g6a8b55ed4056
Sun Jul  5 15:51:10 2020 kern.info kernel: [   10.709694] Backport generated by backports.git v5.7-rc3-1-0-gc0c7d2bb
Sun Jul  5 15:51:10 2020 kern.info kernel: [   10.739997] xt_time: kernel timezone is -0000
Sun Jul  5 15:51:10 2020 kern.info kernel: [   10.797735] mt7621-pci 1e140000.pcie: bus=1 slot=0 irq=23
Sun Jul  5 15:51:10 2020 kern.info kernel: [   10.808564] pci 0000:00:00.0: enabling device (0006 -> 0007)
Sun Jul  5 15:51:10 2020 kern.info kernel: [   10.819861] mt7615e 0000:01:00.0: enabling device (0000 -> 0002)
Sun Jul  5 15:51:10 2020 kern.debug kernel: [   10.844112] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
Sun Jul  5 15:51:10 2020 kern.err kernel: [   10.847573] mt7615e 0000:01:00.0: Firmware is not ready for download
Sun Jul  5 15:51:10 2020 kern.info kernel: [   10.848171] mt7621-pci 1e140000.pcie: bus=2 slot=1 irq=24
Sun Jul  5 15:51:10 2020 kern.info kernel: [   10.871066] pci 0000:00:01.0: enabling device (0006 -> 0007)
Sun Jul  5 15:51:10 2020 kern.info kernel: [   10.882388] mt7615e 0000:02:00.0: enabling device (0000 -> 0002)
Sun Jul  5 15:51:10 2020 user.info kernel: [   10.900044] urngd: v1.0.2 started.
Sun Jul  5 15:51:10 2020 kern.debug kernel: [   10.909400] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
Sun Jul  5 15:51:10 2020 kern.info kernel: [   10.927721] mt7615e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
Sun Jul  5 15:51:10 2020 kern.info kernel: [   10.927721]
Sun Jul  5 15:51:10 2020 kern.info kernel: [   10.933767] PPP generic driver version 2.4.2
Sun Jul  5 15:51:10 2020 kern.info kernel: [   10.957199] NET: Registered protocol family 24
Sun Jul  5 15:51:10 2020 user.info kernel: [   10.981535] kmodloader: done loading kernel modules from /etc/modules.d/*
Sun Jul  5 15:51:10 2020 kern.notice kernel: [   11.159702] random: crng init done
Sun Jul  5 15:51:10 2020 kern.info kernel: [   11.235768] mt7615e 0000:02:00.0: N9 Firmware Version: 2.0, Build Time: 20200131181812
Sun Jul  5 15:51:10 2020 kern.info kernel: [   11.298068] mt7615e 0000:02:00.0: CR4 Firmware Version: _reserved_, Build Time: 20190121161307
Sun Jul  5 15:51:11 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
```

d3m3vilurr, 2020-07-07:
#316 is pretty same issue. kernel would get the firmware data after hard reset.

dolhastelian, 2021-08-30:
Hello,
same issue here
Device ipTime A8004T

Openwrt version: 
 - 21.02.0-rc4 (r16256-2d5ee43dc6) - Kernel + Sysupgrade

First time after configuring wireless both 2.4 and 5Ghz work properly but after restart 2.4G device is not working anymore

kern.err kernel: [   10.847573] mt7615e 0000:01:00.0: Firmware is not ready for download <- same error as above

+ trying to start the interface manually from cmd line gives I/O error.

Any fix possible ?

This is a very cheap but capable router with Openwrt

d3m3vilurr, 2021-08-30:
sadly, I always uses hard reboot (unplug&replug power cable)