openwrt/mt76 #422: MT7615D testmode

porentak, 2020-07-14:
Great to see that testmode has been added for MT7615 device.

Is there a plan to support testmode also for DBDC mode? Currently it is not possible to send frames on two bands concurrently.

It there are already some early patches for that, please let me know.

ryderlee1110, 2021-01-09:
https://github.com/openwrt/mt76/commit/a897a69769f51d1c3473c2ecf2d4d19f1d4ea954

porentak, 2021-01-10:
@ryderlee1110 

Some time ago, I took patches from linux-wireless and tried it. Testmode data was per-phy but I could not get both band concurrently in the air - only 2.4Ghz or 5Ghz at same time.

Tomorrow I will have all the equipment to test it again. Just in case I've missed something. If it won't work, I will update this thread with all the steps I have done.

Thanks!

csyuanc, 2021-01-10:
For the moment, per-phy works, but both phy concurrently cannot work yet.

thelinuxdude, 2021-11-17:
Is there any update on con-current 2.4 5 support for the 7615D?