openwrt/mt76 #424: mt7603 error when wmm mode is turned off

9000000, 2020-07-16:
I want to turn off wmm to use the 2.4 GHz band more stable, but the download speed is too low, the download speed is only 6 Mbp

@nbd168 Can you fix this error?

I use the latest build on xiaomi router mir3g v1

nbd168, 2020-08-12:
Disabling WMM is incompatible with 802.11n.

closed 2020-08-12 by nbd168

9000000, 2020-08-12:
i used the latest version but still got the error, i know wmm socks won't support n band, But I need the above stability in g band rather than a flicker connection on n band, I switch to g band but the download speed is too low only 6MBit, then want the maximum speed on g band is 54Mb Can you fix that, thanks for your contributions very much
@nbd168