openwrt/mt76 #427: mt7615 crash on lastest HEAD code

ptpt52, 2020-07-25:
```
[  594.200025] device wlan1 left promiscuous mode
[  594.204796] br-lan: port 2(wlan1) entered disabled state
[  595.415519] br-lan: port 2(wlan1) entered blocking state
[  595.420903] br-lan: port 2(wlan1) entered disabled state
[  595.426992] device wlan1 entered promiscuous mode
[  595.432240] br-lan: port 2(wlan1) entered blocking state
[  595.437594] br-lan: port 2(wlan1) entered forwarding state
[  595.604245] br-lan: port 2(wlan1) entered disabled state
[  597.516543] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  597.523474] br-lan: port 2(wlan1) entered blocking state
[  597.528872] br-lan: port 2(wlan1) entered forwarding state
[  650.621401] ------------[ cut here ]------------
[  650.626203] WARNING: CPU: 2 PID: 0 at backports-5.8-rc2-1/net/mac80211/sta_info.c:1929 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[  650.639296] STA ce:76:f9:4d:9c:70 AC 2 txq pending airtime underflow: 4294967224, 72
[  650.639300] Modules linked in: pppoe ppp_async l2tp_ppp pptp pppox ppp_mppe ppp_generic mt7615e mt7615_common mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY wireguard ts_fsm ts_bm slhc sch_cake nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc
[  650.647203]  nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables crc_ccitt compat_xtables compat br_netfilter asn1_decoder natcap tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6
[  650.734078]  nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_gre ip_gre gre ifb sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip6_tunnel tunnel6 tunnel4 ip_tunnel tun nls_utf8 nls_iso8859_1 nls_base sha1_generic ecb arc4 leds_gpio softdog gpio_button_hotplug [last unloaded: natflow]
[  650.859191] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.4.52 #0
[  650.865086] Stack : 8611de60 8006eb90 80650000 8064c760 806b0000 8064c728 8064b88c 87c11c44
[  650.873415]         807f0000 87c3c414 80694c03 805e1298 00000002 00000001 87c11be8 1d06bbe7
[  650.881741]         00000000 00000000 80830000 00000000 00000030 0000016c 342e3520 2032352e
[  650.890066]         00000000 00000020 00000000 000d1c37 00000000 806b0000 00000000 86208b44
[  650.898386]         00000009 00000000 00000000 8611e000 00000002 802f71a0 00000008 807f0008
[  650.906707]         ...
[  650.909144] Call Trace:
[  650.911601] [<8000b5fc>] show_stack+0x30/0x100
[  650.916049] [<8052b910>] dump_stack+0xa4/0xdc
[  650.920398] [<80027c54>] __warn+0xc0/0x10c
[  650.924489] [<80027d2c>] warn_slowpath_fmt+0x8c/0xac
[  650.929559] [<86208b44>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[  650.937326] [<862020ec>] ieee80211_report_low_ack+0xf0/0x470 [mac80211]
[  650.943996] [<86202484>] ieee80211_free_txskb+0x18/0x30 [mac80211]
[  650.950222] [<86125c18>] mt76_tx_complete_skb+0xa8/0xbc [mt76]
[  650.956054] [<862abeb4>] mt7615_txp_skb_unmap+0x1f4/0x230 [mt7615_common]
[  650.962909] ---[ end trace b6b4fcced51928e6 ]---
```

prabuselva, 2020-07-25:
I too confirm the same thing on 7615 crash on Xiaomi R3P (snapshot build/Jul 25)
```
[    0.000000] Linux version 5.4.52 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r13927-b2f7355eec)) #0 SMP Fri Jul 24 17:44:31 2020
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router 3 Pro
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] percpu: Embedded 14 pages/cpu s26704 r8192 d22448 u57344
[    0.000000] pcpu-alloc: s26704 r8192 d22448 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00041440
[    0.000000] Readback ErrCtl register=00041440
.....
.....
[   26.077882] br-lan: port 3(lan3) entered forwarding state
[   26.086646] br-lan: port 4(wlan0) entered disabled state
[   26.092534] br-lan: port 1(lan1) entered blocking state
[   26.097794] br-lan: port 1(lan1) entered forwarding state
[   26.103748] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   26.682233] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control rx/tx
[   27.452807] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[   27.459677] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   27.466372] br-lan: port 4(wlan0) entered blocking state
[   27.471720] br-lan: port 4(wlan0) entered forwarding state
[   28.837572] br-lan: port 5(wlan1) entered blocking state
[   28.842971] br-lan: port 5(wlan1) entered disabled state
[   28.849168] device wlan1 entered promiscuous mode
[  104.878175] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  104.884910] br-lan: port 5(wlan1) entered blocking state
[  104.890276] br-lan: port 5(wlan1) entered forwarding state
[  326.569006] ------------[ cut here ]------------
[  326.573811] WARNING: CPU: 3 PID: 0 at backports-5.8-rc2-1/net/mac80211/sta_info.c:1929 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[  326.586951] STA ac:d1:b8:62:35:83 AC 2 txq pending airtime underflow: 4294967248, 48
[  326.586958] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt7615e mt7615_common mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[  326.657599] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 5.4.52 #0
[  326.663493] Stack : 8eb5de60 80077338 806c0000 806bfbfc 80720000 806bfbc4 806bed18 8fc13c44
[  326.671819]         80860000 8fc3ccc8 8070ad83 80654758 00000003 00000001 8fc13be8 bb2b85e0
[  326.680145]         00000000 00000000 808a0000 00000000 00000030 00000154 342e3520 2032352e
[  326.688470]         00000000 00000010 00000000 000a08bf 00000000 80720000 00000000 8ed08b44
[  326.696790]         00000009 00000000 00000000 8eb5dfe0 00000002 8033f0dc 0000000c 8086000c
[  326.705115]         ...
[  326.707553] Call Trace:
[  326.710011] [<8000b72c>] show_stack+0x30/0x100
[  326.714446] [<805967e0>] dump_stack+0xa4/0xdc
[  326.718796] [<80028058>] __warn+0xc0/0x10c
[  326.722878] [<80028130>] warn_slowpath_fmt+0x8c/0xac
[  326.727938] [<8ed08b44>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[  326.735702] [<8ed020ec>] ieee80211_report_low_ack+0xf0/0x470 [mac80211]
[... 5 lines cut ...]
[  466.200858] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:448 dev_watchdog+0x2fc/0x304
[  466.209126] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[... cut ...]
[  466.345613] [<80028130>] warn_slowpath_fmt+0x8c/0xac
[  466.350555] [<80460638>] dev_watchdog+0x2fc/0x304
[  466.355246] [<800904f8>] call_timer_fn.isra.34+0x20/0x90
[  466.360535] [<80090740>] run_timer_softirq+0x1d8/0x230
[  466.365652] [<805b734c>] __do_softirq+0x16c/0x334
[  466.370343] [<8002c80c>] irq_exit+0x98/0xb0
[  466.374517] [<802de670>] plat_irq_dispatch+0x64/0x104
[  466.379545] [<80006de8>] except_vec_vi_end+0xb8/0xc4
[  466.384485] [<805b6a00>] r4k_wait_irqoff+0x1c/0x24
[  466.389487] ---[ end trace b0ff0aa78968b695 ]---
```

UAb5eSMn, 2020-07-26:
Same on Edimax RG21S with todays snapshot. Complete log can be provided if needed.
```
[    0.000000] Linux version 5.4.52 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r13938-15f585afc5)) #0 SMP Sun Jul 26 00:14:45 2020
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Edimax RG21S
...
[ 3706.306331] ------------[ cut here ]------------
[ 3706.315724] WARNING: CPU: 3 PID: 1327 at backports-5.8-rc2-1/net/mac80211/sta_info.c:1929 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[ 3706.342455] STA d8:50:e6:89:8a:56 AC 2 txq pending airtime underflow: 4294967144, 152
[ 3706.342461] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt7615e mt7615_common mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio gpio_button_hotplug
[ 3706.468977] CPU: 3 PID: 1327 Comm: hostapd Not tainted 5.4.52 #0
[ 3706.480922] Stack : 8e8d9e60 80077338 806c0000 806bfbfc 80720000 806bfbc4 806bed18 8fc13c44
[ 3706.497547]         80860000 8fe25268 8070ad83 80654758 00000003 00000001 8fc13be8 ababbb31
[ 3706.514169]         00000000 00000000 808a0000 00000000 00000030 00000184 2e352064 32352e34
[ 3706.530791]         00000000 000000b8 00000000 000727f1 80000000 80720000 00000000 8e908b44
[ 3706.547415]         00000009 00000000 00000000 8e8d9fe0 00000003 8033f0dc 0000000c 8086000c
[ 3706.564040]         ...
[ 3706.568901] Call Trace:
[ 3706.573787] [<8000b72c>] show_stack+0x30/0x100
[ 3706.582635] [<805967e0>] dump_stack+0xa4/0xdc
[ 3706.591310] [<80028058>] __warn+0xc0/0x10c
[ 3706.599457] [<80028130>] warn_slowpath_fmt+0x8c/0xac
[ 3706.609447] [<8e908b44>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[ 3706.624934] [<8e9020ec>] ieee80211_report_low_ack+0xf0/0x470 [mac80211]
[ 3706.638173] [<8e902484>] ieee80211_free_txskb+0x18/0x30 [mac80211]
[ 3706.650539] [<8e815478>] mt76_tx_complete_skb+0xa8/0xbc [mt76]
[ 3706.662173] [<8e86aa40>] mt7615_txp_skb_unmap+0x1f4/0x230 [mt7615_common]
[ 3706.676002] ---[ end trace 5b2112acd4709ac4 ]---
```

thegroove, 2020-07-28:
Same here, not 100% sure it's the same message because the system instantly reboots before I can read it. Also slightly different chip, 7612:

```
04:00.0 Network controller: MEDIATEK Corp. Device 7612
        Subsystem: MEDIATEK Corp. Device 7612
        Flags: bus master, fast devsel, latency 0, IRQ 17
        Memory at fe600000 (64-bit, non-prefetchable) [size=1M]
        Expansion ROM at fe700000 [disabled] [size=64K]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Device Serial Number 00-00-00-00-00-00-00-00
        Capabilities: [158] Latency Tolerance Reporting
        Capabilities: [160] L1 PM Substates
        Kernel driver in use: mt76x2e04:00.0 Network controller: MEDIATEK Corp. Device 7612
        Subsystem: MEDIATEK Corp. Device 7612
        Flags: bus master, fast devsel, latency 0, IRQ 17
        Memory at fe600000 (64-bit, non-prefetchable) [size=1M]
        Expansion ROM at fe700000 [disabled] [size=64K]
        Capabilities: [40] Power Management version 3
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [148] Device Serial Number 00-00-00-00-00-00-00-00
        Capabilities: [158] Latency Tolerance Reporting
        Capabilities: [160] L1 PM Substates
        Kernel driver in use: mt76x2e
```

namidairo, 2020-07-29:
Quite a few duplicates on this.

As noted in one of the other issues opened about this, it might be a mac80211 issue?

@ptpt52 Try https://patchwork.kernel.org/patch/11684897/ and maybe https://patchwork.kernel.org/patch/11684095/ ?

ptpt52, 2020-07-29:
again, with latest code and patch from @namidairo ,  seems work but stream crash:
```
daemon.info hostapd: wlan1: STA ce:76:f9:4d:9c:70 IEEE 802.11: disconnected due to excessive missing ACKs
```

my phone get kick when I run iperf3

ptpt52, 2020-07-29:
update:
I try this https://openwrt.org/faq/disconnected_due_to_excessive_missing_acks
seems work now.

ptpt52, 2020-08-06:
```
<1>[94810.967555] CPU 0 Unable to handle kernel paging request at virtual address 00000104, epc == 85ce50bc, ra == 85ce50a0
<4>[94810.988733] Oops[#1]:
<4>[94810.993252] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.52 #0
<4>[94811.005018] $ 0   : 00000000 00000001 00000122 00000100
<4>[94811.015416] $ 4   : 863e5f58 00000000 00000000 87cf279c
<4>[94811.025811] $ 8   : 0000000c 8054ac10 863e5ea0 85c1c480
<4>[94811.036205] $12   : 00000fff 85c1cce8 bd08d464 35650000
<4>[94811.046603] $16   : 834d86a0 863e5f58 863e5e60 00000002
<4>[94811.056998] $20   : 863e5e60 85dc7984 87c0bbac 87cf279c
<4>[94811.067393] $24   : 0d800000 7c000000                  
<4>[94811.077787] $28   : 80686000 87c0bb28 00000002 85ce50a0
<4>[94811.088183] Hi    : 000002ff
<4>[94811.093899] Lo    : ef9db4cd
<4>[94811.099650] epc   : 85ce50bc mt76_eeprom_init+0x220/0x278 [mt76]
<4>[94811.111601] ra    : 85ce50a0 mt76_eeprom_init+0x204/0x278 [mt76]
<4>[94811.123539] Status: 11008403	KERNEL EXL IE 
<4>[94811.131857] Cause : 4080000c (ExcCode 03)
<4>[94811.139821] BadVA : 00000104
<4>[94811.145537] PrId  : 0001992f (MIPS 1004Kc)
<4>[94811.153670] Modules linked in: natcap ksmbd qcserial pppoe ppp_async option l2tp_ppp cdc_mbim usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY wireguard usbserial usbnet usblp ts_fsm ts_bm slhc sch_cake r8152 nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nf_tproxy_ipv6
<4>[94811.153980]  nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables crc_ccitt compat_xtables compat cdc_wdm br_netfilter asn1_decoder natflow fuse tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet
<4>[94811.327220]  ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip6_gre ip_gre gre ifb sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip6_tunnel tunnel6 tunnel4 ip_tunnel tun vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac ghash_generic gf128mul gcm ecb des_generic libdes ctr cmac ccm arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ohci_platform ohci_hcd softdog fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat usbcore nls_base usb_common mii crc32c_generic
<4>[94811.499831]  [last unloaded: natcap]
<4>[94811.679431] Process swapper/0 (pid: 0, threadinfo=b7f8f70d, task=0e2d6161, tls=00000000)
<4>[94811.695515] Stack : 863e4c00 85c1c480 87cf2000 8621f000 8a11040f 8345bc00 863e5e60 85ce643c
<4>[94811.712143]         85d9f182 85c1c000 84c6b700 806963b0 806f45fa 80690000 8345bc00 8345bc00
<4>[94811.728770]         863e5e60 85ce172c 00000000 00000000 00000000 00006000 8345b480 00000000
<4>[94811.745396]         84c6b700 806f45f8 85dc7980 00000000 85c1c000 00000000 84c6b700 806963b0
<4>[94811.762021]         806f45fa 00000000 00000000 00000000 00000000 00000000 00000000 00000000
<4>[94811.778645]         ...
<4>[94811.783506] Call Trace:
<4>[94811.788373] [<85ce50bc>] mt76_eeprom_init+0x220/0x278 [mt76]
<4>[94811.799733] Code: 8e1008e0  8e020004  8e030000 <ac620004> ac430000  24020100  ae020000  24020122  0c1545cb 
<4>[94811.819138] 
<4>[94811.823307] ---[ end trace f0a1b1b682b00204 ]---
```

nbd168, 2020-08-12:
Please try the latest version

closed 2020-08-25 by nbd168

UAb5eSMn, 2020-09-12:
No errors for over a month on my device, so it looks good to me.