openwrt/mt76 #430: MT7621 - Clients disconnecting

Camis, 2020-08-05:
```
Wed Aug  5 13:20:29 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 90:78:****
Wed Aug  5 13:20:29 2020 daemon.info hostapd: wlan0: STA 90:78:**** RADIUS: starting accounting session ACBA497F6AD16C46
Wed Aug  5 13:20:29 2020 daemon.info hostapd: wlan0: STA 90:78:**** WPA: pairwise key handshake completed (RSN)
Wed Aug  5 13:20:29 2020 daemon.info hostapd: wlan1: STA 90:78:**** IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Wed Aug  5 13:20:35 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 90:78:****
Wed Aug  5 13:20:35 2020 daemon.info hostapd: wlan0: STA 90:78:**** IEEE 802.11: disassociated
Wed Aug  5 13:20:35 2020 daemon.notice hostapd: wlan1: STA-OPMODE-N_SS-CHANGED 90:78:**** 1
Wed Aug  5 13:20:35 2020 daemon.info hostapd: wlan1: STA 90:78:**** IEEE 802.11: associated (aid 1)
Wed Aug  5 13:20:35 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 90:78:****
Wed Aug  5 13:20:35 2020 daemon.info hostapd: wlan1: STA 90:78:**** RADIUS: starting accounting session B30F2D4C310A41EA
Wed Aug  5 13:20:35 2020 daemon.info hostapd: wlan1: STA 90:78:**** WPA: pairwise key handshake completed (RSN)
Wed Aug  5 13:20:36 2020 daemon.info hostapd: wlan0: STA 90:78:**** IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
```

```
Model	Netgear R6220
Architecture	MediaTek MT7621
OpenWrt 19.07.3 r11063-85e04e9f46 
```

```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'HT20'
	option country 'PL'
	option channel '11'
	option legacy_rates '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid '*****'
	option key '**********'
	option ft_over_ds '1'
	option ieee80211w '1'
	option ft_psk_generate_local '1'
	option ieee80211r '1'
	option mobility_domain '27DB'
	option encryption 'sae'
	option disassoc_low_ack '0'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
	option htmode 'VHT80'
	option country 'PL'
	option channel '128'
	option legacy_rates '0'

config wifi-iface 'wifinet3'
	option ssid '*****-IoT'
	option encryption 'psk2+ccmp'
	option device 'radio0'
	option mode 'ap'
	option ieee80211w '1'
	option key '**********'
	option ft_over_ds '1'
	option ft_psk_generate_local '1'
	option ieee80211r '1'
	option mobility_domain '285D'
	option disassoc_low_ack '0'

config wifi-iface 'wifinet4'
	option ssid '*****_5GHz'
	option device 'radio1'
	option ieee80211r '1'
	option ft_over_ds '1'
	option key '**********'
	option network 'lan'
	option mode 'ap'
	option ieee80211w '1'
	option ft_psk_generate_local '1'
	option mobility_domain '27DB'
	option encryption 'sae'
	option disassoc_low_ack '0'

config wifi-iface 'wifinet6'
	option ssid '*****_5GHz-IoT'
	option encryption 'psk2+ccmp'
	option device 'radio1'
	option ieee80211r '1'
	option ft_over_ds '1'
	option key '**********'
	option mode 'ap'
	option ieee80211w '1'
	option ft_psk_generate_local '1'
	option mobility_domain '285D'
	option disassoc_low_ack '0'

config wifi-iface 'wifinet5'
	option ssid '*****-Guest'
	option encryption 'psk2+ccmp'
	option device 'radio0'
	option mode 'ap'
	option ieee80211w '1'
	option ft_over_ds '1'
	option mobility_domain '28B1'
	option ft_psk_generate_local '1'
	option ieee80211r '1'
	option network 'lan'
	option key '**********'
	option disassoc_low_ack '0'

config wifi-iface 'wifinet7'
	option ssid '*****_5GHz-Guest'
	option encryption 'psk2+ccmp'
	option device 'radio1'
	option mode 'ap'
	option ieee80211w '1'
	option ft_over_ds '1'
	option ft_psk_generate_local '1'
	option ieee80211r '1'
	option mobility_domain '28B1'
	option network 'lan'
	option key '**********'
	option disassoc_low_ack '0'
```

db260179, 2020-11-22:
In your https://bugs.openwrt.org/index.php?do=details&task_id=3275

I've identified an unknown issue at the moment with setting the 'option country' in the wireless settings.

If you remove this settings and let the driver set itself, then your clients will not keep disconnecting.

danpawlik, 2023-08-16:
Got similar issue on Xiaomi 4a Gigabit version (as dumb ap) and Samsung tv qe55q67bau. No meter if I use cryptographic lib  wpad-mbedtls, wpad-openssl or wpad-wolfss issue is still the same - TV can connect, but after 1 sec of streaming, it just freeze. Very high utilization on the dumb ap, which is not providing any interesting info:

```
Tue Aug 15 21:59:35 2023 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
Tue Aug 15 21:59:35 2023 daemon.debug hostapd: phy1-ap0: STA xx:xx:xx:xx:xx:xx WPA: event 3 notification
Tue Aug 15 21:59:35 2023 daemon.debug hostapd: phy1-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.1X: unauthorizing port
Tue Aug 15 21:59:35 2023 daemon.debug hostapd: phy1-ap0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated
Tue Aug 15 21:59:35 2023 daemon.debug hostapd: phy1-ap0: STA xx:xx:xx:xx:xx:xx MLME: MLME-DEAUTHENTICATE.indication(xx:xx:xx:xx:xx:xx, 3)
Tue Aug 15 21:59:35 2023 daemon.debug hostapd: phy1-ap0: STA xx:xx:xx:xx:xx:xx MLME: MLME-DELETEKEYS.request(xx:xx:xx:xx:xx:xx)
```

I found a workaround for that issue: force swith encryption algorithm for WPA2 to TKIP (just TKIP not CCMP + TKIP).

What is odd, it does not happen on Xiaomi ax3200.

lukasz1992, 2023-08-16:
Forcing TKIP disables 802.11n (on 2.4GHz or 5GHz). Try setting fragmentation threshold to 2346 instead, reboot the router and tell us if it helps.

danpawlik, 2023-08-17:
With fragmentation set to `2346`  TV is working normally, but the film from time to time play bad quality, like the bandwidth is not constant to it. Also I check with `rts threshold` set to  `2347` and situation was same. Going back to solution without changing fragmentation/rts and check how it will be working without "country" set in the config file.

danpawlik, 2023-08-18:
After a while, it seems that the problem is...in `option country "PL"`. I changed the country to US + added `option noscan '1'` and it works well. Finally I don't have any issue with MT7621 and Samsung TV.

My current config looks like:
```
  config wifi-device 'radio1'
      option type 'mac80211'
      option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
      option band '5g'
      option htmode 'VHT80'
      option log_level '0'
      option channel '140'
      option txpower '18'
      option country 'US'
      option cell_density '0'
      option noscan '1'
```

Hope it would help someone.