openwrt/mt76 #432: mt7615 speed test get low resault

ptpt52, 2020-08-09:
for version 
commit b5df0fbb1847a51f0a11f03a3fa8dc9d98d4df82 (mt7615: update firmware to version 20200629)
I build and run on mi-router-ac2100, mt7615 iperf3 test get 500Mbps

but for the latest version HEAD in master
I get 350Mbps only

ptpt52, 2020-08-09:
@nbd168  Please notice this

nbd168, 2020-08-09:
I can't reproduce the performance difference in my test. Could you please use git bisect to track this down?

ptpt52, 2020-08-09:
drop this commit ae4757a0ae900aabe6eeb21674ab8323b5a34e4f (mt76: mt7615: reschedule runtime-pm receiving a tx interrupt)
the speed test become better and stable
but not as fast as 500Mbps, for now I cannot reproduce that speed

nbd168, 2020-08-09:
That's really odd. These changes should be pretty much no-op for MT7615

ptpt52, 2020-08-10:
for now, I update to latest HEAD, and test speed is 100-200Mbps only

nbd168, 2020-08-12:
I've added some more fixes, please test.

ptpt52, 2020-08-12:
Compared with the last time, there is some improvement, up to 340 Mbps

garyttirn, 2020-08-13:
I just rebuild my DIR-882 image with [latest mt76](https://github.com/openwrt/mt76/commit/efc8669db5f9ff1bf1d7487e0f762ab1bbe8cd85) and I also see lower RX throughput, however rather inconsistently.

Station RX :
```
kari@zenbook:~$ iperf3 -R -c 192.168.254.7
Connecting to host 192.168.254.7, port 5201
Reverse mode, remote host 192.168.254.7 is sending
[  5] local 192.168.254.42 port 51848 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  67.6 MBytes   567 Mbits/sec                  
[  5]   1.00-2.00   sec  70.4 MBytes   591 Mbits/sec                  
[  5]   2.00-3.00   sec  69.7 MBytes   585 Mbits/sec                  
[  5]   3.00-4.00   sec  67.1 MBytes   563 Mbits/sec                  
[  5]   4.00-5.00   sec  44.1 MBytes   370 Mbits/sec                  
[  5]   5.00-6.00   sec  55.4 MBytes   465 Mbits/sec                  
[  5]   6.00-7.00   sec  69.7 MBytes   585 Mbits/sec                  
[  5]   7.00-8.00   sec  70.2 MBytes   589 Mbits/sec                  
[  5]   8.00-9.00   sec  70.1 MBytes   588 Mbits/sec                  
[  5]   9.00-10.00  sec  67.4 MBytes   565 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   661 MBytes   555 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   652 MBytes   547 Mbits/sec                  receiver

kari@zenbook:~$ iperf3 -R -c 192.168.254.7
Connecting to host 192.168.254.7, port 5201
Reverse mode, remote host 192.168.254.7 is sending
[  5] local 192.168.254.42 port 52018 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  34.3 MBytes   288 Mbits/sec                  
[  5]   1.00-2.00   sec  34.5 MBytes   290 Mbits/sec                  
[  5]   2.00-3.00   sec  30.7 MBytes   258 Mbits/sec                  
[  5]   3.00-4.00   sec  38.6 MBytes   323 Mbits/sec                  
[  5]   4.00-5.00   sec  36.1 MBytes   303 Mbits/sec                  
[  5]   5.00-6.00   sec  25.8 MBytes   216 Mbits/sec                  
[  5]   6.00-7.00   sec  32.1 MBytes   270 Mbits/sec                  
[  5]   7.00-8.00   sec  32.5 MBytes   272 Mbits/sec                  
[  5]   8.00-9.00   sec  19.3 MBytes   161 Mbits/sec                  
[  5]   9.00-10.00  sec  21.3 MBytes   178 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   316 MBytes   265 Mbits/sec  400             sender
[  5]   0.00-10.00  sec   305 MBytes   256 Mbits/sec                  receiver
```
Station TX :
```
iperf Done.
kari@zenbook:~$ iperf3 -c 192.168.254.7
Connecting to host 192.168.254.7, port 5201
[  5] local 192.168.254.42 port 51610 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  74.9 MBytes   628 Mbits/sec    0   2.06 MBytes       
[  5]   1.00-2.00   sec  73.8 MBytes   619 Mbits/sec    0   2.21 MBytes       
[  5]   2.00-3.00   sec  76.2 MBytes   640 Mbits/sec    0   2.19 MBytes       
[  5]   3.00-4.00   sec  73.8 MBytes   619 Mbits/sec    0   2.73 MBytes       
[  5]   4.00-5.00   sec  70.0 MBytes   587 Mbits/sec    0   2.03 MBytes       
[  5]   5.00-6.00   sec  73.8 MBytes   619 Mbits/sec    0   2.06 MBytes       
[  5]   6.00-7.00   sec  72.5 MBytes   608 Mbits/sec    0   2.23 MBytes       
[  5]   7.00-8.00   sec  73.8 MBytes   619 Mbits/sec    0   2.93 MBytes       
[  5]   8.00-9.00   sec  61.2 MBytes   514 Mbits/sec    0   2.04 MBytes       
[  5]   9.00-10.00  sec  70.0 MBytes   587 Mbits/sec    0   2.01 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   720 MBytes   604 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   718 MBytes   603 Mbits/sec                  receiver
```

Seem to me the station RX can quickly jump between MCS3 and MSC9 and seems more prone to drop to lower MCS index than before. Station TX seems to be relatively stable and also gives more consistent results and higher throughput. 

Back [in July](https://forum.openwrt.org/t/odds-of-timely-d-link-dir-882-or-878-support-in-2018/10725/49?u=garyttirn) under very similar test conditions, I was getting 

```
kari@zenbook:~$ iperf3 -R -c 192.168.254.7
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec   521 MBytes   437 Mbits/sec  111             sender
[  4]   0.00-10.00  sec   512 MBytes   429 Mbits/sec                  receiver

kari@zenbook:~$ iperf3 -c 192.168.254.7
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-10.00  sec   664 MBytes   557 Mbits/sec    0             sender
[  4]   0.00-10.00  sec   661 MBytes   554 Mbits/sec                  receiver
```

nbd168, 2020-08-25:
Please try the latest version

ptpt52, 2020-08-26:
Thanks for your update, at this moment, the test result is:
```
 iperf3 -c 192.168.17.1 -R
Connecting to host 192.168.17.1, port 5201
Reverse mode, remote host 192.168.17.1 is sending
[  5] local 192.168.15.218 port 49371 connected to 192.168.17.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  47.7 MBytes   400 Mbits/sec                  
[  5]   1.00-2.00   sec  51.0 MBytes   428 Mbits/sec                  
[  5]   2.00-3.00   sec  46.1 MBytes   387 Mbits/sec                  
[  5]   3.00-4.00   sec  42.6 MBytes   358 Mbits/sec                  
[  5]   4.00-5.00   sec  45.2 MBytes   379 Mbits/sec                  
[  5]   5.00-6.00   sec  42.7 MBytes   358 Mbits/sec                  
[  5]   6.00-7.00   sec  39.8 MBytes   334 Mbits/sec                  
[  5]   7.00-8.00   sec  45.4 MBytes   381 Mbits/sec                  
[  5]   8.00-9.00   sec  45.7 MBytes   383 Mbits/sec                  
[  5]   9.00-10.00  sec  47.6 MBytes   399 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   457 MBytes   383 Mbits/sec  171             sender
[  5]   0.00-10.00  sec   454 MBytes   381 Mbits/sec                  receiver

iperf Done.
```

ptpt52, 2020-08-26:
```
iperf3 -c 192.168.17.1 -R
Connecting to host 192.168.17.1, port 5201
Reverse mode, remote host 192.168.17.1 is sending
[  5] local 192.168.15.218 port 49440 connected to 192.168.17.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  48.0 MBytes   402 Mbits/sec                  
[  5]   1.00-2.00   sec  44.3 MBytes   372 Mbits/sec                  
[  5]   2.00-3.00   sec  47.6 MBytes   399 Mbits/sec                  
[  5]   3.00-4.00   sec  50.1 MBytes   420 Mbits/sec                  
[  5]   4.00-5.00   sec  54.0 MBytes   453 Mbits/sec                  
[  5]   5.00-6.00   sec  50.1 MBytes   421 Mbits/sec                  
[  5]   6.00-7.00   sec  51.0 MBytes   428 Mbits/sec                  
[  5]   7.00-8.00   sec  52.3 MBytes   439 Mbits/sec                  
[  5]   8.00-9.00   sec  54.0 MBytes   453 Mbits/sec                  
[  5]   9.00-10.00  sec  47.5 MBytes   399 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   502 MBytes   421 Mbits/sec  156             sender
[  5]   0.00-10.00  sec   499 MBytes   419 Mbits/sec                  receiver

iperf Done.
```

garyttirn, 2020-08-27:
Thanks for the updates. 

Tested with [b36d7ae](https://github.com/openwrt/mt76/commit/b36d7ae096a3d8c7d6a8a246f2e8a471a467041e) and latest OpenWRT master.

Seem to me the station RX is now stable and doesn't seems as prone to drop to lower MCS index as before.
However this doesn't translate to higher throughput, now it is consistently around 300-350Mbps.

Station TX is as stable as it was and gives higher throughput.

Station RX :
```
kari@zenbook:~$ iperf3 -R -c 192.168.254.7
Connecting to host 192.168.254.7, port 5201
Reverse mode, remote host 192.168.254.7 is sending
[  5] local 192.168.254.42 port 46198 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  38.1 MBytes   319 Mbits/sec                  
[  5]   1.00-2.00   sec  41.5 MBytes   348 Mbits/sec                  
[  5]   2.00-3.00   sec  38.0 MBytes   319 Mbits/sec                  
[  5]   3.00-4.00   sec  37.9 MBytes   318 Mbits/sec                  
[  5]   4.00-5.00   sec  36.4 MBytes   306 Mbits/sec                  
[  5]   5.00-6.00   sec  38.2 MBytes   321 Mbits/sec                  
[  5]   6.00-7.00   sec  39.0 MBytes   327 Mbits/sec                  
[  5]   7.00-8.00   sec  36.6 MBytes   307 Mbits/sec                  
[  5]   8.00-9.00   sec  38.9 MBytes   326 Mbits/sec                  
[  5]   9.00-10.00  sec  37.0 MBytes   311 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   391 MBytes   328 Mbits/sec  226             sender
[  5]   0.00-10.00  sec   382 MBytes   320 Mbits/sec                  receiver
```

Station TX :
```
kari@zenbook:~$ iperf3 -c 192.168.254.7
Connecting to host 192.168.254.7, port 5201
[  5] local 192.168.254.42 port 46462 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  75.4 MBytes   632 Mbits/sec    0   2.70 MBytes       
[  5]   1.00-2.00   sec  72.5 MBytes   608 Mbits/sec    0    865 KBytes       
[  5]   2.00-3.00   sec  72.5 MBytes   608 Mbits/sec    0   1.02 MBytes       
[  5]   3.00-4.00   sec  75.0 MBytes   629 Mbits/sec    0   1.34 MBytes       
[  5]   4.00-5.00   sec  72.5 MBytes   608 Mbits/sec    0   1.08 MBytes       
[  5]   5.00-6.00   sec  65.0 MBytes   545 Mbits/sec    0   1.12 MBytes       
[  5]   6.00-7.00   sec  73.8 MBytes   619 Mbits/sec    0   1.10 MBytes       
[  5]   7.00-8.00   sec  73.8 MBytes   619 Mbits/sec    0   1.08 MBytes       
[  5]   8.00-9.00   sec  72.5 MBytes   608 Mbits/sec    0   1.33 MBytes       
[  5]   9.00-10.00  sec  73.8 MBytes   619 Mbits/sec    0   1.43 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   727 MBytes   610 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   724 MBytes   608 Mbits/sec                  receiver

iperf Done.
```

Interrupt errors all all but gone.
```
root@wlan-1:~# cat /proc/interrupts | grep ERR
ERR:        126
```

nzer, 2020-09-01:
Did you check logs?
It looks like wifi stalls occasionally under load and it may look like speed degradation. I use recent snapshot with mt76 5.4.61+2020-08-25-b36d7ae0-2 on redmi ac2100.
I'll build from git today and see if it is fixed.
Log:
`Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.136795] ------------[ cut here ]------------
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.141478] WARNING: CPU: 1 PID: 0 at net/sched/sch_generic.c:448 dev_watchdog+0x2fc/0x304
Tue Sep  1 12:27:17 2020 kern.info kernel: [50629.149744] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.156699] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt7615e mt7615_common mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY wireguard slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.156903]  nf_reject_ipv6 ip6_udp_tunnel udp_tunnel leds_gpio gpio_button_hotplug
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.251773] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 5.4.61 #0
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.257668] Stack : ffffffff 8007d458 806d0000 806d407c 80740000 806d4044 806d3198 87c0fdb4
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.265994]         80880000 87c3c188 8071ed63 80667edc 00000001 00000001 87c0fd58 5909d490
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.274317]         00000000 00000000 808c0000 00000000 00000030 000001b6 342e3520 2031362e
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.282640]         00000000 000009d7 00000000 0003d77d 00000000 80740000 00000000 80471e1c
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.290962]         00000009 00000001 00200000 00000122 00000002 8034f8e0 00000004 80880004
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.299284]         ...
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.301722] Call Trace:
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.304180] [<8000b72c>] show_stack+0x30/0x100
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.308634] [<805a940c>] dump_stack+0xa4/0xdc
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.312990] [<8002bea0>] __warn+0xc0/0x10c
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.317072] [<8002bf78>] warn_slowpath_fmt+0x8c/0xac
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.322036] [<80471e1c>] dev_watchdog+0x2fc/0x304
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.326735] [<80096284>] call_timer_fn.isra.34+0x20/0x90
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.332026] [<800964cc>] run_timer_softirq+0x1d8/0x230
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.337166] [<805c9f34>] __do_softirq+0x16c/0x334
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.341863] [<8003060c>] irq_exit+0x98/0xb0
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.346036] [<802eeee0>] plat_irq_dispatch+0x64/0x104
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.351070] [<80006de8>] except_vec_vi_end+0xb8/0xc4
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.356017] [<805c95e8>] r4k_wait_irqoff+0x1c/0x24
Tue Sep  1 12:27:17 2020 kern.warn kernel: [50629.360937] ---[ end trace d02399f8670bd8db ]---`

fluffymadness, 2020-09-01:
@nzer I'm not sure if this is really because of wifi or something else...
While my build is old 19.07.3 
I got a similiar eth0 transmit queue timeout on an er-x that has no wifi at all.
https://github.com/openwrt/mt76/issues/407#issuecomment-683418024

https://github.com/openwrt/openwrt/pull/2798#issuecomment-636651471
https://forum.openwrt.org/t/mtk-soc-eth-watchdog-timeout-after-r11573/50000/246

stevenxed, 2020-09-17:
Updated from Sept 9 snapshot to Sept 17 snapshot on a nanoHD.
Station Tx went down from around 500Mbit/s to <100Mbit/s. Phone barely gets 30Mbit/s.
Station Rx is fine.

garyttirn, 2020-09-20:
DIR-882 A1, 20th September openwrt snapshot, this [commit ](https://github.com/openwrt/openwrt/commit/e18534f90558882bd9cbaa55bd64d775035c6556)should be included in the build as far as I know.

Reported rate RX/TX : 866.7 Mbit/s, 80 MHz, VHT-MCS 9, VHT-NSS 2, Short GI.
160Mhz channel width configured, channel 100.

Station RX is very low and is prone to drop to lower MCS index (as low as 3). Station TX is still stable.

Station RX :
```
zenbook:~$ iperf3 -R -c 192.168.254.7
Connecting to host 192.168.254.7, port 5201
Reverse mode, remote host 192.168.254.7 is sending
[  5] local 192.168.254.42 port 39698 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   304 KBytes  2.49 Mbits/sec                  
[  5]   1.00-2.00   sec   223 KBytes  1.83 Mbits/sec                  
[  5]   2.00-3.00   sec  1.14 MBytes  9.59 Mbits/sec                  
[  5]   3.00-4.00   sec  2.67 MBytes  22.4 Mbits/sec                  
[  5]   4.00-5.00   sec  1.58 MBytes  13.3 Mbits/sec                  
[  5]   5.00-6.00   sec  2.25 MBytes  18.8 Mbits/sec                  
[  5]   6.00-7.00   sec  4.45 MBytes  37.4 Mbits/sec                  
[  5]   7.00-8.00   sec  5.84 MBytes  49.0 Mbits/sec                  
[  5]   8.00-9.00   sec  8.35 MBytes  70.0 Mbits/sec                  
[  5]   9.00-10.00  sec  7.10 MBytes  59.6 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  40.5 MBytes  33.9 Mbits/sec  1635             sender
[  5]   0.00-10.00  sec  33.9 MBytes  28.4 Mbits/sec                  receiver

iperf Done.

zenbook:~$ iperf3 -R -c 192.168.254.7
Connecting to host 192.168.254.7, port 5201
Reverse mode, remote host 192.168.254.7 is sending
[  5] local 192.168.254.42 port 40198 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   173 KBytes  1.42 Mbits/sec                  
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   7.00-8.00   sec  4.54 MBytes  38.1 Mbits/sec                  
[  5]   8.00-9.00   sec  1.67 MBytes  14.0 Mbits/sec                  
[  5]   9.00-10.00  sec  5.84 MBytes  49.0 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  16.8 MBytes  14.1 Mbits/sec  926             sender
[  5]   0.00-10.00  sec  12.2 MBytes  10.2 Mbits/sec                  receiver
```

Station TX :
```
zenbook:~$ iperf3  -c 192.168.254.7
Connecting to host 192.168.254.7, port 5201
[  5] local 192.168.254.42 port 39694 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  53.2 MBytes   446 Mbits/sec    0   2.34 MBytes       
[  5]   1.00-2.00   sec  63.8 MBytes   535 Mbits/sec    0   2.44 MBytes       
[  5]   2.00-3.00   sec  68.8 MBytes   577 Mbits/sec    0   2.26 MBytes       
[  5]   3.00-4.00   sec  71.2 MBytes   598 Mbits/sec  276   1.94 MBytes       
[  5]   4.00-5.00   sec  65.0 MBytes   545 Mbits/sec    0   2.43 MBytes       
[  5]   5.00-6.00   sec  66.2 MBytes   556 Mbits/sec    0   2.26 MBytes       
[  5]   6.00-7.00   sec  65.0 MBytes   545 Mbits/sec    0   2.04 MBytes       
[  5]   7.00-8.00   sec  70.0 MBytes   587 Mbits/sec    0   1.57 MBytes       
[  5]   8.00-9.00   sec  65.0 MBytes   545 Mbits/sec    0   1.16 MBytes       
[  5]   9.00-10.00  sec  62.5 MBytes   524 Mbits/sec    0   1.10 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   651 MBytes   546 Mbits/sec  276             sender
[  5]   0.00-10.00  sec   648 MBytes   543 Mbits/sec                  receiver
```

Switching to 80Mhz channel width makes signal to drop down to 324.0 Mbit/s, 40 MHz, VHT-MCS 8, VHT-NSS 2.

namidairo, 2020-09-21:
I had to revert a couple of the recent mac80211 patches locally to get station RX back to normal speeds, otherwise it would sputter along at around 30mbit with a bunch of retries in iperf.

mcaptur, 2020-09-21:
Confirmed reverting 
mac80211: do not allow bigger VHT MPDUs than the hardware supports
and
mac80211: allow bigger A-MSDU sizes in VHT, even if HT is limited

fixes performance - tested on DIr-878 and RedmiAC2100

ptpt52, 2020-09-21:
> Confirmed reverting
> mac80211: do not allow bigger VHT MPDUs than the hardware supports
> and
> mac80211: allow bigger A-MSDU sizes in VHT, even if HT is limited
> 
> fixes performance - tested on DIr-878 and RedmiAC2100

I will try this out

namidairo, 2020-09-21:
On a bit of a hunch I changed the capability declared in mt7615/init.c at https://github.com/openwrt/mt76/blob/master/mt7615/init.c#L493 to IEEE80211_VHT_CAP_MAX_MPDU_LENGTH_7991 and the performance regression is gone even with the two mac80211 commits in?

Anyone know if the hardware actually supports 11k mpdu lengths? The only reference I can find to it (on wikidevi mirrors/offshoots brief description of MT7615, presumably copy pasted from somewhere) suggests that it does 8k max and the 7915 init uses the same value. 

> MPDU up to 8K, A-MPDU up to 1M

mcaptur, 2020-09-23:
> On a bit of a hunch I changed the capability declared in mt7615/init.c at https://github.com/openwrt/mt76/blob/master/mt7615/init.c#L493 to IEEE80211_VHT_CAP_MAX_MPDU_LENGTH_7991 and the performance [...]

Yes tested this out it fixes poor performance

`--- a/mt7615/init.c	2020-09-23 04:14:53.256018081 +0200
+++ b/mt7615/init.c	2020-09-23 04:16:34.067261275 +0200
@@ -490,7 +490,7 @@
 	dev->mphy.sband_2g.sband.ht_cap.cap |= IEEE80211_HT_CAP_LDPC_CODING;
 	dev->mphy.sband_5g.sband.ht_cap.cap |= IEEE80211_HT_CAP_LDPC_CODING;
 	dev->mphy.sband_5g.sband.vht_cap.cap |=
-			IEEE80211_VHT_CAP_MAX_MPDU_LENGTH_11454 |
+			IEEE80211_VHT_CAP_MAX_MPDU_LENGTH_7991 |
 			IEEE80211_VHT_CAP_MAX_A_MPDU_LENGTH_EXPONENT_MASK;
 	mt7615_cap_dbdc_disable(dev);
 	dev->phy.dfs_state = -1;
`

nbd168, 2020-09-23:
Pushed the fix to master, thanks for tracking this down. @ptpt52, please test the latest version and let me know if performance is fixed for you as well.

ptpt52, 2020-09-23:
@nbd168 
I test the latest code, and it is up to 300Mbps seems not the ideal speed

test
```
iperf3 -c 192.168.17.1 -R
```

garyttirn, 2020-09-24:
Low station RX has been fixed in latest snapshot. 

```
zenbook:~$ iperf3 -R -c 192.168.254.7
Connecting to host 192.168.254.7, port 5201
Reverse mode, remote host 192.168.254.7 is sending
[  5] local 192.168.254.42 port 59434 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  37.3 MBytes   313 Mbits/sec                  
[  5]   1.00-2.00   sec  39.8 MBytes   334 Mbits/sec                  
[  5]   2.00-3.00   sec  39.8 MBytes   334 Mbits/sec                  
[  5]   3.00-4.00   sec  40.5 MBytes   340 Mbits/sec                  
[  5]   4.00-5.00   sec  39.1 MBytes   328 Mbits/sec                  
[  5]   5.00-6.00   sec  39.8 MBytes   334 Mbits/sec                  
[  5]   6.00-7.00   sec  39.9 MBytes   335 Mbits/sec                  
[  5]   7.00-8.00   sec  40.7 MBytes   341 Mbits/sec                  
[  5]   8.00-9.00   sec  44.1 MBytes   370 Mbits/sec                  
[  5]   9.00-10.00  sec  41.2 MBytes   346 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   413 MBytes   347 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   402 MBytes   337 Mbits/sec                  receiver

```

mcaptur, 2020-09-25:
so this seems fixed on clients such as pocofone and s2, however a samsung tab a and xiaomi mi tv stick can barely connect and when connected get 900kbps! this on mt7615, i have two other mt 7603 (xiaomi mir3g routers) which behave fine.. how can i help troublshoot?

Robot-DaneelOlivaw, 2020-09-27:
I'm running OpenWrt SNAPSHOT r14572-4682d4d770 on Xiaomi R3P. It still has low station RX result:
```
root@Xiaomi-R3P:~# iperf3 -c 192.168.31.166
Connecting to host 192.168.31.166, port 5201
[  5] local 192.168.31.129 port 46088 connected to 192.168.31.166 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  7.13 MBytes  59.6 Mbits/sec  116   49.5 KBytes       
[  5]   1.00-2.00   sec  15.9 MBytes   133 Mbits/sec  111   38.2 KBytes       
[  5]   2.00-3.00   sec  12.4 MBytes   104 Mbits/sec   66   82.0 KBytes       
[  5]   3.00-4.00   sec  14.0 MBytes   118 Mbits/sec  118   73.5 KBytes       
[  5]   4.00-5.00   sec  9.76 MBytes  81.9 Mbits/sec  189   35.4 KBytes       
[  5]   5.00-6.00   sec  14.7 MBytes   123 Mbits/sec   49    143 KBytes       
[  5]   6.00-7.00   sec  9.45 MBytes  79.3 Mbits/sec  185   63.6 KBytes       
[  5]   7.00-8.00   sec  12.1 MBytes   102 Mbits/sec  101   1.41 KBytes       
[  5]   8.00-9.01   sec  8.39 MBytes  69.7 Mbits/sec  312    267 KBytes       
[  5]   9.01-10.00  sec  14.3 MBytes   121 Mbits/sec  326   17.0 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   118 MBytes  99.1 Mbits/sec  1573             sender
[  5]   0.00-10.01  sec   117 MBytes  98.1 Mbits/sec                  receiver

iperf Done.

root@Xiaomi-R3P:~# iperf3 -c 192.168.31.166
Connecting to host 192.168.31.166, port 5201
[  5] local 192.168.31.129 port 46096 connected to 192.168.31.166 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  14.4 MBytes   121 Mbits/sec  494   69.3 KBytes       
[  5]   1.00-2.00   sec  11.7 MBytes  98.0 Mbits/sec  132   29.7 KBytes       
[  5]   2.00-3.00   sec   382 KBytes  3.13 Mbits/sec   19   1.41 KBytes       
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  5]   4.00-5.00   sec  2.92 MBytes  24.5 Mbits/sec   92   39.6 KBytes       
[  5]   5.00-6.01   sec  16.7 MBytes   139 Mbits/sec   18    158 KBytes       
[  5]   6.01-7.00   sec  10.9 MBytes  92.0 Mbits/sec  192   65.0 KBytes       
[  5]   7.00-8.00   sec  9.82 MBytes  82.4 Mbits/sec  170   48.1 KBytes       
[  5]   8.00-9.00   sec  12.4 MBytes   104 Mbits/sec  109   76.4 KBytes       
[  5]   9.00-10.00  sec  12.4 MBytes   104 Mbits/sec  120   94.7 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  91.7 MBytes  76.9 Mbits/sec  1347             sender
[  5]   0.00-10.21  sec  91.7 MBytes  75.3 Mbits/sec                  receiver

iperf Done.
```

Even ping is not stable:
```
root@Xiaomi-R3P:~# ping 192.168.31.166
PING 192.168.31.166 (192.168.31.166): 56 data bytes
64 bytes from 192.168.31.166: seq=0 ttl=64 time=2.205 ms
64 bytes from 192.168.31.166: seq=1 ttl=64 time=13.845 ms
64 bytes from 192.168.31.166: seq=2 ttl=64 time=143.740 ms
64 bytes from 192.168.31.166: seq=3 ttl=64 time=163.839 ms
64 bytes from 192.168.31.166: seq=4 ttl=64 time=183.595 ms
64 bytes from 192.168.31.166: seq=5 ttl=64 time=49.941 ms
64 bytes from 192.168.31.166: seq=6 ttl=64 time=18.766 ms
64 bytes from 192.168.31.166: seq=7 ttl=64 time=38.449 ms
64 bytes from 192.168.31.166: seq=8 ttl=64 time=59.033 ms
64 bytes from 192.168.31.166: seq=9 ttl=64 time=78.838 ms
^C
--- 192.168.31.166 ping statistics ---
10 packets transmitted, 10 packets received, 0% packet loss
round-trip min/avg/max = 2.205/75.225/183.595 ms
```

mcaptur, 2020-09-27:
Thats got MT7615 also if i recall correctly.. seems issue with that chip..

jesusdf, 2020-09-28:
I've also found something similar with a Netgear WAC124 and this chip:

https://forum.openwrt.org/t/strange-behaviour-on-5g-wifi-with-netgear-wac124/74771/6

`This is what I found:

Samsung Galaxy S4: Speed ok (the maximum it can provide) (40mbps I think)
Samsung Galaxy S8: Poor speed. (3 mbps)
Xiaomi Mi A1: the maximum it can provide (148 mbps)
Two Intel Wifi6 AX200 in two separate computers: Poor speed. (4 mbps)
Intel AC 8265: the maximum speed it can provide (200mbps)

What they have in common is that newer devices that support Wifi 6 are prone to have speed issues with this router. If I change the channel width to HT40 (wifi N), speed is ok for that bandwidth, so it seems that something is wrong with wifi AC and newer devices.`

Robot-DaneelOlivaw, 2020-09-29:
> Thats got MT7615 also if i recall correctly.. seems issue with that chip..

I flashed a Xiaomi Mini(SoC MT7620A, 802.11nac MT76x2E) with OpenWrt 19.07.4 r11208-ce6496d796. The route ran perfectly when I first loaded OpenWrt for it years ago. Yet it has the same 5 GHz issue as Xiaomi R3P(SoC MT7621AT, 802.11nac MT7615E) now. So apart from the 2017 MacBook Pro that I've tested on, I ran iperf3 on a Surface Laptop as well. Basically every setting is kept default(Channel 36 5180 MHz, Bandwidth 80MHz).

The Station RX speed of Surface is much faster on both routers, at about 250 Mbits/sec with the Xiaomi R3P, though ping is still not ideal.

<details>
<summary>Xiaomi R3P - Surface Laptop</summary>

```
RX rate: 866.7 Mbit/s, 80 MHz, VHT-MCS 9, VHT-NSS 2, Short GI

root@Xiaomi-R3P:~# iperf3 -c 192.168.31.66
Connecting to host 192.168.31.66, port 5201
[  5] local 192.168.31.210 port 43414 connected to 192.168.31.66 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.01   sec  29.3 MBytes   243 Mbits/sec    0    221 KBytes
[  5]   1.01-2.00   sec  28.2 MBytes   239 Mbits/sec    0    221 KBytes
[  5]   2.00-3.00   sec  27.1 MBytes   227 Mbits/sec    0    221 KBytes
[  5]   3.00-4.00   sec  27.3 MBytes   229 Mbits/sec    0    221 KBytes
[  5]   4.00-5.00   sec  27.4 MBytes   230 Mbits/sec    0    221 KBytes
[  5]   5.00-6.00   sec  29.0 MBytes   242 Mbits/sec    0    221 KBytes
[  5]   6.00-7.00   sec  30.0 MBytes   251 Mbits/sec    0    221 KBytes
[  5]   7.00-8.00   sec  30.0 MBytes   251 Mbits/sec    0    221 KBytes
[  5]   8.00-9.01   sec  29.3 MBytes   245 Mbits/sec    0    221 KBytes
[  5]   9.01-10.00  sec  29.2 MBytes   245 Mbits/sec    0    221 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   287 MBytes   240 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   286 MBytes   240 Mbits/sec                  receiver

iperf Done.
root@Xiaomi-R3P:~# iperf3 -c 192.168.31.66
Connecting to host 192.168.31.66, port 5201
[  5] local 192.168.31.210 port 43418 connected to 192.168.31.66 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  29.2 MBytes   245 Mbits/sec    0    221 KBytes
[  5]   1.00-2.00   sec  30.9 MBytes   260 Mbits/sec    0    221 KBytes
[  5]   2.00-3.02   sec  32.0 MBytes   264 Mbits/sec    0    221 KBytes
[  5]   3.02-4.01   sec  31.5 MBytes   266 Mbits/sec    0    221 KBytes
[  5]   4.01-5.01   sec  31.1 MBytes   261 Mbits/sec    0    221 KBytes
[  5]   5.01-6.00   sec  30.8 MBytes   260 Mbits/sec    0    221 KBytes
[  5]   6.00-7.00   sec  30.8 MBytes   259 Mbits/sec    0    221 KBytes
[  5]   7.00-8.00   sec  30.9 MBytes   259 Mbits/sec    0    221 KBytes
[  5]   8.00-9.00   sec  31.3 MBytes   262 Mbits/sec    0    221 KBytes
[  5]   9.00-10.01  sec  30.8 MBytes   258 Mbits/sec    0    221 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec   309 MBytes   259 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   309 MBytes   259 Mbits/sec                  receiver

iperf Done.

root@Xiaomi-R3P:~# ping 192.168.31.66
PING 192.168.31.66 (192.168.31.66): 56 data bytes
64 bytes from 192.168.31.66: seq=0 ttl=128 time=198.258 ms
64 bytes from 192.168.31.66: seq=1 ttl=128 time=13.610 ms
64 bytes from 192.168.31.66: seq=2 ttl=128 time=33.612 ms
64 bytes from 192.168.31.66: seq=3 ttl=128 time=53.682 ms
64 bytes from 192.168.31.66: seq=4 ttl=128 time=72.153 ms
64 bytes from 192.168.31.66: seq=5 ttl=128 time=91.858 ms
64 bytes from 192.168.31.66: seq=6 ttl=128 time=112.042 ms
64 bytes from 192.168.31.66: seq=7 ttl=128 time=131.809 ms
64 bytes from 192.168.31.66: seq=8 ttl=128 time=49.596 ms
64 bytes from 192.168.31.66: seq=9 ttl=128 time=69.429 ms
64 bytes from 192.168.31.66: seq=10 ttl=128 time=89.809 ms
64 bytes from 192.168.31.66: seq=11 ttl=128 time=109.470 ms
64 bytes from 192.168.31.66: seq=12 ttl=128 time=129.552 ms
64 bytes from 192.168.31.66: seq=13 ttl=128 time=151.310 ms
64 bytes from 192.168.31.66: seq=14 ttl=128 time=68.949 ms
[... 335 lines cut ...]
64 bytes from 192.168.31.166: seq=4 ttl=64 time=197.760 ms
64 bytes from 192.168.31.166: seq=5 ttl=64 time=216.060 ms
64 bytes from 192.168.31.166: seq=6 ttl=64 time=35.520 ms
64 bytes from 192.168.31.166: seq=7 ttl=64 time=58.120 ms
64 bytes from 192.168.31.166: seq=8 ttl=64 time=82.320 ms
64 bytes from 192.168.31.166: seq=9 ttl=64 time=106.440 ms
64 bytes from 192.168.31.166: seq=10 ttl=64 time=130.620 ms
64 bytes from 192.168.31.166: seq=11 ttl=64 time=154.120 ms
64 bytes from 192.168.31.166: seq=12 ttl=64 time=177.920 ms
64 bytes from 192.168.31.166: seq=13 ttl=64 time=201.500 ms
64 bytes from 192.168.31.166: seq=14 ttl=64 time=20.520 ms
64 bytes from 192.168.31.166: seq=15 ttl=64 time=44.240 ms
64 bytes from 192.168.31.166: seq=16 ttl=64 time=68.200 ms
64 bytes from 192.168.31.166: seq=17 ttl=64 time=91.400 ms
64 bytes from 192.168.31.166: seq=18 ttl=64 time=114.720 ms
64 bytes from 192.168.31.166: seq=19 ttl=64 time=139.180 ms
64 bytes from 192.168.31.166: seq=20 ttl=64 time=163.200 ms
^C
--- 192.168.31.166 ping statistics ---
21 packets transmitted, 21 packets received, 0% packet loss
round-trip min/avg/max = 20.520/120.881/216.060 ms
```
</details>

With the original firmware, everything is stable, except ping is still high on the MacBook. Now I'm confused. Is the issue related to the routers or the laptops? If it's router related, Surface is performing well. If it's laptop related, MacBook is stable with the original firmware. Anyway, it doesn't look like it has something to do with mt7615.

langit7, 2020-10-10:
tested with SNAPSHOT r14660-5423d9d27e

<details>
  <summary>Xiaomi R3P-TP Link USB</summary>

  ```
`root@XiaomiR3P:~# ping 192.168.7.53
PING 192.168.7.53 (192.168.7.53): 56 data bytes
64 bytes from 192.168.7.53: seq=0 ttl=64 time=78.935 ms
64 bytes from 192.168.7.53: seq=1 ttl=64 time=2.553 ms
64 bytes from 192.168.7.53: seq=2 ttl=64 time=123.854 ms
64 bytes from 192.168.7.53: seq=3 ttl=64 time=39.009 ms
64 bytes from 192.168.7.53: seq=4 ttl=64 time=58.321 ms
64 bytes from 192.168.7.53: seq=5 ttl=64 time=80.014 ms
64 bytes from 192.168.7.53: seq=6 ttl=64 time=86.058 ms
64 bytes from 192.168.7.53: seq=7 ttl=64 time=123.775 ms
64 bytes from 192.168.7.53: seq=8 ttl=64 time=36.423 ms
64 bytes from 192.168.7.53: seq=9 ttl=64 time=56.618 ms
64 bytes from 192.168.7.53: seq=10 ttl=64 time=76.774 ms
64 bytes from 192.168.7.53: seq=11 ttl=64 time=86.086 ms
64 bytes from 192.168.7.53: seq=12 ttl=64 time=9.221 ms
64 bytes from 192.168.7.53: seq=13 ttl=64 time=2.655 ms
64 bytes from 192.168.7.53: seq=14 ttl=64 time=184.795 ms
64 bytes from 192.168.7.53: seq=15 ttl=64 time=104.881 ms`
  ```
</details>

WiredLife, 2020-10-11:
Is vht160 fixed for xiaomi/redmi ac2100?
I only get very slow speeds with vht160 so i must use vht80 instead.

garyttirn, 2020-10-18:
Results with lastest snapshot (17.10.2020) with DIR-882 and Intel 9260
(Signal 866.7 Mbit/s, 80 MHz, VHT-MCS 9, VHT-NSS 2, Short GI. Channel 100)
RX seem to be about as good as it was around July 2020, fairly consistently above 400Mbps.
There is still small disparity between RX/TX as TX is consistently over 500Mbps.

Station RX:
```
root@zenbook:~# iperf3 -R -c 192.168.254.7 
Connecting to host 192.168.254.7, port 5201
Reverse mode, remote host 192.168.254.7 is sending
[  5] local 192.168.254.42 port 58588 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  41.2 MBytes   345 Mbits/sec                  
[  5]   1.00-2.00   sec  45.2 MBytes   379 Mbits/sec                  
[  5]   2.00-3.00   sec  49.8 MBytes   418 Mbits/sec                  
[  5]   3.00-4.00   sec  45.6 MBytes   383 Mbits/sec                  
[  5]   4.00-5.00   sec  50.9 MBytes   427 Mbits/sec                  
[  5]   5.00-6.00   sec  49.0 MBytes   411 Mbits/sec                  
[  5]   6.00-7.00   sec  51.4 MBytes   431 Mbits/sec                  
[  5]   7.00-8.00   sec  52.1 MBytes   437 Mbits/sec                  
[  5]   8.00-9.00   sec  52.6 MBytes   442 Mbits/sec                  
[  5]   9.00-10.00  sec  48.9 MBytes   411 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   497 MBytes   417 Mbits/sec   66             sender
[  5]   0.00-10.00  sec   487 MBytes   408 Mbits/sec                  receiver

iperf Done.
```

Station TX:
```
root@zenbook:~# iperf3 -c 192.168.254.7 
Connecting to host 192.168.254.7, port 5201
[  5] local 192.168.254.42 port 58592 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  63.6 MBytes   533 Mbits/sec   76   1.81 MBytes       
[  5]   1.00-2.00   sec  71.2 MBytes   598 Mbits/sec    0   1.83 MBytes       
[  5]   2.00-3.00   sec  66.2 MBytes   556 Mbits/sec   46   1.60 MBytes       
[  5]   3.00-4.00   sec  66.2 MBytes   556 Mbits/sec  140   1.82 MBytes       
[  5]   4.00-5.00   sec  66.2 MBytes   556 Mbits/sec    0   1.47 MBytes       
[  5]   5.00-6.00   sec  63.8 MBytes   535 Mbits/sec    0   1.23 MBytes       
[  5]   6.00-7.00   sec  61.2 MBytes   514 Mbits/sec    0   1.47 MBytes       
[  5]   7.00-8.00   sec  57.5 MBytes   482 Mbits/sec    0   1.26 MBytes       
[  5]   8.00-9.00   sec  63.8 MBytes   535 Mbits/sec  134   1.31 MBytes       
[  5]   9.00-10.00  sec  63.8 MBytes   535 Mbits/sec    0   1.99 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   644 MBytes   540 Mbits/sec  396             sender
[  5]   0.00-10.00  sec   642 MBytes   539 Mbits/sec                  receiver
```

garyttirn, 2021-02-20:
Results with 21.02 snapshot (21.02-SNAPSHOT r15839-ef47bc424c) with DIR-882 and Intel 9260 
(Signal 866.7 Mbit/s, 80 MHz, VHT-MCS 9, VHT-NSS 2, Short GI. Channel 100)
The small disparity between RX/TX is gone.
Great work, thanks.

Station RX:
<pre>
@zenbook:~$ iperf3 -R -c 192.168.254.7 
Connecting to host 192.168.254.7, port 5201
Reverse mode, remote host 192.168.254.7 is sending
[  5] local 192.168.254.25 port 37718 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  70.2 MBytes   589 Mbits/sec                  
[  5]   1.00-2.00   sec  73.5 MBytes   617 Mbits/sec                  
[  5]   2.00-3.00   sec  74.9 MBytes   628 Mbits/sec                  
[  5]   3.00-4.00   sec  74.5 MBytes   625 Mbits/sec                  
[  5]   4.00-5.00   sec  73.2 MBytes   614 Mbits/sec                  
[  5]   5.00-6.00   sec  75.0 MBytes   629 Mbits/sec                  
[  5]   6.00-7.00   sec  74.4 MBytes   624 Mbits/sec                  
[  5]   7.00-8.00   sec  74.1 MBytes   621 Mbits/sec                  
[  5]   8.00-9.00   sec  61.1 MBytes   512 Mbits/sec                  
[  5]   9.00-10.00  sec  72.2 MBytes   606 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   734 MBytes   616 Mbits/sec    0             sender
[  5]   0.00-10.00  sec   723 MBytes   607 Mbits/sec                  receiver
</pre>

Station TX:
<pre>
@zenbook:~$ iperf3 -c 192.168.254.7 
Connecting to host 192.168.254.7, port 5201
[  5] local 192.168.254.25 port 37722 connected to 192.168.254.7 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  70.0 MBytes   587 Mbits/sec    0   1.49 MBytes       
[  5]   1.00-2.00   sec  68.8 MBytes   577 Mbits/sec    0   1.49 MBytes       
[  5]   2.00-3.00   sec  73.8 MBytes   619 Mbits/sec  234   1.93 MBytes       
[  5]   3.00-4.00   sec  75.0 MBytes   629 Mbits/sec    0   2.47 MBytes       
[  5]   4.00-5.00   sec  72.5 MBytes   608 Mbits/sec    0   1.83 MBytes       
[  5]   5.00-6.00   sec  70.0 MBytes   587 Mbits/sec   96   1.21 MBytes       
[  5]   6.00-7.00   sec  70.0 MBytes   587 Mbits/sec    0    996 KBytes       
[  5]   7.00-8.00   sec  67.5 MBytes   566 Mbits/sec    0    871 KBytes       
[  5]   8.00-9.00   sec  67.5 MBytes   566 Mbits/sec    0   1.37 MBytes       
[  5]   9.00-10.00  sec  68.8 MBytes   577 Mbits/sec    0   1.18 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   704 MBytes   590 Mbits/sec  330             sender
[  5]   0.00-10.00  sec   701 MBytes   588 Mbits/sec                  receiver
</pre>

I at least would be in favor of closing the issue as this is rather likely as good as performance is going to get.

langit7, 2021-02-20:
> 9, VHT-NSS 2, Short GI. Channel 1

Could you test   5G wifi range and also ping test

garyttirn, 2021-02-24:
> > 9, VHT-NSS 2, Short GI. Channel 1
> 
> Could you test 5G wifi range and also ping test

Just a quick check and not very scientific one at that. 

Under normal conditions I have about -50dBm signal (Channel 100, power: 26 dBm) with straight line of sight to AP at about 3 meter distance and it gives the above results with ping 3.01 ms and jitter 0.17 ms

At about -60dBm (mostly VHT-MCS7) signal downlink is able to deliver ~350 - 400Mbps while Uplink is still pushing closer to 450 - 500Mbps. This is about 8 meters from the AP with no direct line of sight with ping 3.19 ms and jitter 0.90 ms.

In my opinion what I'm seeing is rather fitting to DIR-882 results with stock firmware made by [smallnetbuilder.com](https://www.smallnetbuilder.com/wireless/wireless-reviews/33144-d-link-dir-882-ac2600-mu-mimo-wi-fi-router-reviewed?start=3)

Going beyond 8m means going outside the house and then client is already Fast Roaming to 2,4Ghz.

ptpt52, 2021-02-25:
for the latest code test:

top:
client---wifi---router----server

download: 500-600Mbps

upload: 20-80Mbps

seems something wrong in upload speed.

ptpt52, 2021-03-25:
with latest code this could be close.

closed 2021-03-25 by ptpt52