openwrt/mt76 #436: INTERNET LOSS MT7603E | THE LAST PROBLEM THAT REMAINS

Openwrtfunboy, 2020-08-13:
When a total of more than 8 clients are connected and someone starts downloading (sometimes it is not even necessary to download something), then someone or even several users have completely lost the Internet before reconnecting.

I had 5 devices connected to a 2.4 network, 4 to 5 GHz network and my Internet started to disappear constantly when I started downloading on another device (ps 4) from 2.4

Tried the latest stable 19.07.3 and latest SNAPSHOTS

Hostname
OpenWrt
Model
Newifi-D2
Architecture
MediaTek MT7621 ver:1 eco:3
Firmware Version
OpenWrt SNAPSHOT r14024-80a6d3d4a2 / LuCI Master git-20.216.62430-9acc7d8
Kernel Version
5.4.52

Openwrtfunboy, 2020-08-13:
about this problem also type @JFtico

Do you still have same problem?

Openwrtfunboy, 2020-08-14:
@nbd168 can we get fix, please?

Do you need any more information?

JFtico, 2020-08-15:
> about this problem also type @JFtico
> 
> Do you still have same problem?

I still see it on 19.07.3 based units. Regular loss of Internet.

But one based on OpenWrt SNAPSHOT r13978-d15b6e4895  seems to be fine, with about 5 to 7 2.4 devices, 2 of which are monitors with constant (but small-sized) traffic. And the 5Ghz has 4 to 6 devices.
Both logs are clean, but I still see a large number of Interrupt errors, here's the relevant output from:
cat /proc/interrupts;uptime

ERR:       9690
 16:14:17 up 2 days,  3:25,  load average: 0.00, 0.00, 0.00

Openwrtfunboy, 2020-08-15:
> I still see it on 19.07.3 based units. Regular loss of Internet.
>
> But one based on OpenWrt SNAPSHOT r13978-d15b6e4895 seems to be fine, with about 5 to 7 2.4 devices, 2 of which are monitors with [...]
Oh great if it true. I can test.
Can you drop here file with this snapshot? or link to it.
Also, what date is this SNAPESHOT?

fluffymadness, 2020-09-03:
@Openwrtfunboy 
Did you only loose internet connection on the 2.4 ghz band? Maybe your router is also affected by this?
https://github.com/openwrt/mt76/pull/442.

erenkan, 2020-09-14:
same here. Xiaomi Mir3G. When i started the download, it goes disconnect after the 10min

JFtico, 2020-09-18:
Noting that on 19.07.4 just released, I still see this issue, mostly with iOS devices, where the device shows a solid connection, but does not have Internet access.

I also see a lot of connect/disconnect sequences from those devices

`
Tue Sep  8 01:04:13 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:04:37 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:04:37 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:07:12 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 00:bb:3a:5e:a2:0d
Tue Sep  8 01:07:13 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 00:bb:3a:5e:a2:0d
Tue Sep  8 01:08:36 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:08:36 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:08:39 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 00:51:ed:f2:cc:96
Tue Sep  8 01:08:39 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 00:51:ed:f2:cc:96
Tue Sep  8 01:10:12 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 00:51:ed:f2:cc:96
Tue Sep  8 01:10:12 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 00:51:ed:f2:cc:96
Tue Sep  8 01:19:08 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:19:08 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:19:24 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:19:24 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:27:27 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:27:27 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:29:05 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:29:05 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:34:22 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:39:32 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 00:51:ed:f2:cc:96
Tue Sep  8 01:39:32 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 00:51:ed:f2:cc:96
Tue Sep  8 01:41:07 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED b6:c9:00:54:be:e7
Tue Sep  8 01:48:19 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 00:51:ed:f2:cc:96
Tue Sep  8 01:48:19 2020 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 00:51:ed:f2:cc:96
`
Note: the date was actually today (2020-09-18), for some reason time was off on the unit.