openwrt/mt76 #438: MT7622 issues

graphine27, 2020-08-18:
I get this error quite often after a boot, I have to keep rebooting until it works.
mt7622-wmac 18000000.wmac: driver own failed

ryderlee1110, 2020-08-18:
Disable BTIF node in dts or make it load as module.

graphine27, 2020-08-18:
I already did: https://github.com/graphine27/openwrt/commit/26d43226c0e2c9216c59403214a5a5bc01a24d6b

ryderlee1110, 2020-09-03:
Try this - https://github.com/ryderlee1110/wireless/commit/cb509c1427ec58880c4aadf458022973b50d0378

ryderlee1110, 2020-09-05:
Try this https://github.com/LorenzoBianconi/wireless-drivers-next/commit/ee00e9f28a3fcd8081c0ce464b1e9e28b211cb53

graphine27, 2020-09-05:
@ryderlee1110  Thanks.

I was having better results by changing MT7615_DRV_OWN_RETRY_COUNT to 600.

Then I  tried the changes you suggested and it was no longer working. So I tried to modify the patch a little to also see how many tries it takes. After doing that it seems to have started working, not sure what happen. 

kern.err kernel: [   16.430708] mt7622-wmac 18000000.wmac: It took 1 tries
My patch: [mt7622.txt](https://github.com/openwrt/mt76/files/5178604/mt7622.txt)

A few questions if you don't mind:

Is MT7622 same sillicon as MT7615 2.4 Ghz?

Are there multiple versions of EEPROM/firmware for MT7622? Where is the most recent one?

I’ve connected 4 antennas to the board, how does the board decides which device uses which antenna?

Doing a speed test gives about 100mbps for each device if ran separately. If I run a speed test on 2 devices at the same time one device gets 40, another one gets 20, which looks like both devices use the same antenna.

Shouldn’t they bot get 100mbps and both use different antennas?

ryderlee1110, 2020-09-05:
It looks like you just added a printk then 100% of the time it worked fine, right?

1. Similar
2. Just one version.
3. Did not get this question. Are using both mt7622 and mt7615 at the same time and just having 4 antennas totally?

graphine27, 2020-09-06:
Yes, also I’ve set n_retry to 1000 but it works on the first try according to the log, so that change is useless.
I no longer get driver own failed after many reboots and power offs, seems very stable.

I only have MT7622, I will get MT7915 card when I will find one for sale.

ryderlee1110, 2020-09-07:
Could you please try this https://github.com/LorenzoBianconi/wireless-drivers-next/commit/eb5d01ab78e6abf8ab980dd6c24d8dc86ec1b242 ?

graphine27, 2020-09-07:
This works, even better than the previous patch. I would get driver own failed after sysupgrade on previous patch.
With this patch, no driver own failed yet.

ryderlee1110, 2020-09-07:
Thanks. Can we close this ticket now?

adriandumitru, 2020-09-07:
Yes, thanks.

closed 2020-09-07 by graphine27