openwrt/mt76 #440: MT7621DN "DBDC"

kar200, 2020-08-28:
Hi there,

I am having issues with the MT7621DN. Since the last few changes (sorry I did not keep track of when it stopped).

1 - The automatic setup of the wireless + Luci do not provide the correct details. Although now the /etc/config/wireless file gets generated with all the details in it except it sets up both wireless cards as a 802.11ac but the command "iw phy" shows that the phy0 does not have 5ghz capabilty.

2 - this is the hard one for me to try and bypass and it's started to happen lately where if the cards are set to disabled in the wifi config file I can start them while the router is still on. If I set the phy0 as enabled there is no networking. I can still use normal commands like "ls" from the serial console but there is no networking. Also it seems the folder /sys/kernel/debug/ieee80211/ has only phy0 in it (when the cards are set to enabled). This does not happen if I set phy1 to enabled though.

This is the console output + logread of both cases.

Fresh install or card disabled
````
3: System Boot system code via Flash.
## Booting image at bc180000 ...
   Image Name:   MIPS OpenWrt Linux-5.4.60
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2483706 Bytes =  2.4 MB
   Load Address: 80001000
   Entry Point:  80001000
......................................   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK
No initrd
## Transferring control to Linux (at address 80001000) ...
## Giving linux memsize in MB, 256

Starting kernel ...

[    0.000000] Linux version 5.4.60 (karim@WIN-MTB) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r14257-e12ac40552)) #0 SMP Fri Aug 28 09:44:29 2020
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is D-Link DIR-853 A3
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] percpu: Embedded 18 pages/cpu s43344 r8192 d22192 u73728
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00022500
[    0.000000] Readback ErrCtl register=00022500
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 250084K/262144K available (5938K kernel code, 318K rwdata, 1332K rodata, 1360K init, 239K bss, 12060K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x3b8/0x5c8 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.015484] Calibrating delay loop... 583.68 BogoMIPS (lpj=1167360)
[    0.055834] pid_max: default: 32768 minimum: 301
[    0.065193] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.079599] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.097691] rcu: Hierarchical SRCU implementation.
[    0.107834] smp: Bringing up secondary CPUs ...
[    0.129766] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.129778] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[... 647 lines cut ...]
Fri Aug 28 09:45:02 2020 daemon.info dnsmasq[2052]: reading /tmp/resolv.conf.d/resolv.conf.auto
Fri Aug 28 09:45:02 2020 daemon.info dnsmasq[2052]: using only locally-known addresses for domain test
Fri Aug 28 09:45:02 2020 daemon.info dnsmasq[2052]: using only locally-known addresses for domain onion
Fri Aug 28 09:45:02 2020 daemon.info dnsmasq[2052]: using only locally-known addresses for domain localhost
Fri Aug 28 09:45:02 2020 daemon.info dnsmasq[2052]: using only locally-known addresses for domain local
Fri Aug 28 09:45:02 2020 daemon.info dnsmasq[2052]: using only locally-known addresses for domain invalid
Fri Aug 28 09:45:02 2020 daemon.info dnsmasq[2052]: using only locally-known addresses for domain bind
Fri Aug 28 09:45:02 2020 daemon.info dnsmasq[2052]: using only locally-known addresses for domain lan
Fri Aug 28 09:45:02 2020 daemon.info dnsmasq[2052]: using nameserver 192.168.20.1#53
Fri Aug 28 09:45:02 2020 user.notice firewall: Reloading firewall due to ifup of wan (wan)
Fri Aug 28 09:45:02 2020 daemon.debug dnsmasq[2052]: listening on br-lan(#10): fe80::eead:e0ff:fee8:11cd%br-lan port 53
Fri Aug 28 09:45:02 2020 daemon.debug dnsmasq[2052]: listening on wan(#7): fe80::eead:e0ff:fee8:11d0%wan port 53
Fri Aug 28 09:45:03 2020 daemon.info dnsmasq[2052]: read /etc/hosts - 4 addresses
Fri Aug 28 09:45:03 2020 daemon.info dnsmasq[2052]: read /tmp/hosts/odhcpd - 1 addresses
Fri Aug 28 09:45:03 2020 daemon.info dnsmasq[2052]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Fri Aug 28 09:45:03 2020 daemon.info dnsmasq-dhcp[2052]: read /etc/ethers - 0 addresses
Fri Aug 28 20:52:20 2020 daemon.info dnsmasq-dhcp[2052]: DHCPDISCOVER(br-lan) 00:e0:4c:77:3e:ac
Fri Aug 28 20:52:20 2020 daemon.info dnsmasq-dhcp[2052]: DHCPOFFER(br-lan) 192.168.1.207 00:e0:4c:77:3e:ac
Fri Aug 28 20:52:20 2020 daemon.info dnsmasq-dhcp[2052]: DHCPREQUEST(br-lan) 192.168.1.207 00:e0:4c:77:3e:ac
Fri Aug 28 20:52:20 2020 daemon.info dnsmasq-dhcp[2052]: DHCPACK(br-lan) 192.168.1.207 00:e0:4c:77:3e:ac WIN-MTB
Fri Aug 28 20:52:23 2020 daemon.warn dnsmasq[2052]: possible DNS-rebind attack detected: mobile.pipe.aria.microsoft.com
Fri Aug 28 20:52:38 2020 daemon.warn dnsmasq[2052]: possible DNS-rebind attack detected: mobile.pipe.aria.microsoft.com
Fri Aug 28 20:53:10 2020 daemon.warn dnsmasq[2052]: possible DNS-rebind attack detected: mobile.pipe.aria.microsoft.com
Fri Aug 28 20:54:14 2020 daemon.warn dnsmasq[2052]: possible DNS-rebind attack detected: mobile.pipe.aria.microsoft.com
````

kar200, 2020-08-28:
Sorry forgot to attach the follwoing 

When the card is enabled

````
Starting kernel ...

[    0.000000] Linux version 5.4.60 (karim@WIN-MTB) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r14257-e12ac40552)) #0 SMP Fri Aug 28 09:44:29 2020
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is D-Link DIR-853 A3
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] percpu: Embedded 18 pages/cpu s43344 r8192 d22192 u73728
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00022500
[    0.000000] Readback ErrCtl register=00022500
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 250084K/262144K available (5938K kernel code, 318K rwdata, 1332K rodata, 1360K init, 239K bss, 12060K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x3b8/0x5c8 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.015485] Calibrating delay loop... 583.68 BogoMIPS (lpj=1167360)
[    0.055836] pid_max: default: 32768 minimum: 301
[    0.065197] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.079603] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.097730] rcu: Hierarchical SRCU implementation.
[    0.107883] smp: Bringing up secondary CPUs ...
[    0.129832] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.129843] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.129856] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.129957] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.145086] Synchronize counters for CPU 1: done.
[    0.220930] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.220939] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.220947] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.221007] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.243662] Synchronize counters for CPU 2: done.
[    0.316178] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.316186] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.316194] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.316257] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.331272] Synchronize counters for CPU 3: done.
[    0.390887] smp: Brought up 1 node, 4 CPUs
[    0.403610] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.422916] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.436859] pinctrl core: initialized pinctrl subsystem
[    0.448669] NET: Registered protocol family 16
[    0.485711] random: fast init done
[    0.485879] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[... 546 lines cut ...]
Fri Aug 28 20:56:06 2020 kern.info kernel: [   18.829479] br-lan: port 4(lan4) entered blocking state
Fri Aug 28 20:56:06 2020 kern.info kernel: [   18.839980] br-lan: port 4(lan4) entered disabled state
Fri Aug 28 20:56:06 2020 kern.info kernel: [   18.852160] device lan4 entered promiscuous mode
Fri Aug 28 20:56:06 2020 daemon.debug dnsmasq[1084]: listening on lo(#1): 127.0.0.1 port 53
Fri Aug 28 20:56:06 2020 daemon.debug dnsmasq[1084]: listening on lo(#1): ::1 port 53
Fri Aug 28 20:56:06 2020 daemon.notice netifd: Interface 'loopback' is enabled
Fri Aug 28 20:56:06 2020 daemon.notice netifd: Interface 'loopback' is setting up now
Fri Aug 28 20:56:06 2020 daemon.notice netifd: Interface 'loopback' is now up
Fri Aug 28 20:56:06 2020 kern.info kernel: [   18.884260] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
Fri Aug 28 20:56:06 2020 kern.info kernel: [   18.898306] 8021q: adding VLAN 0 to HW filter on device wan
[... cut ...]
Fri Aug 28 20:56:07 2020 daemon.info procd: - init complete -
Fri Aug 28 20:56:07 2020 daemon.err odhcpd[1270]: Failed to send to ff02::1%lan@br-lan (Address not available)
Fri Aug 28 20:56:07 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Fri Aug 28 20:56:10 2020 kern.info kernel: [   22.841950] mt7530 mdio-bus:1f lan3: Link is Up - 1Gbps/Full - flow control off
Fri Aug 28 20:56:10 2020 kern.info kernel: [   23.001928] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control rx/tx
````

ptpt52, 2020-09-07:
set radio0 to `11a` and `VHT80`
set radio1 to `11g`  and `HT20`
that may work

kar200, 2020-09-07:
> set radio0 to `11a` and `VHT80`
> set radio1 to `11g` and `HT20`
> that may work

Thanks for tip. In fact my wireless cards both work if I use them with x-wrt. I use the opposite Actually where radio0 has to be 11g and radio1 11a.

With x-wrt I am not compiling with DSA and I kept swconfig. With openwrt and dsa for some reason the issue appear. It seems to be related to later commits (I cannot tell which one).

If I set the cards to disabled in the wireless config file the networking works fine. 
If I set the cards to enabled and restart then the issue appear and I have no networking.
If I boot with the cards off and then set them to on after booting then it works fine. 

I think the issue is to do with the time and the way the DBDC detection happen but unfortunately my knowlegde is quite limited on this.
Given the limited amount of devices that use DBDC I don't think this will be a priority on the to do list.

vj4git, 2020-10-19:
I have the exact same issue - having the MT7615DN card. I have my radio0 as N (with allow legacy rates checked) and radio1 as AC (allow legacy rates unchecked). Works fine, but if I reboot I will lose all networking and cannot connect to the AP at all. 

What I do is start the wifi after reboot and then ssh and change the /etc/config/wireless
`Option disabled '1'`
for both radios, so it has network after inadvertent reboot. I hope someone has a fix or better workaround.

kar200, 2020-10-19:
> I have the exact same issue - having the MT7615DN card. I have my radio0 as N (with allow legacy rates checked) and radio1 as AC (allow legacy rates unchecked). Works fine, but if I reboot I will [...]

I am not using openwrt at the moment but this is one option i was using
Another option would be to revert the automatic DBDC commit ([this one](https://github.com/openwrt/mt76/commit/99804560372bfb830de28a41ecc4736747c2ae60) and use the startup script to enable the second inteface - Note I have not tried this one either.
A third option which I did use it to compile from x-wrt and using swconfig instead of DSA.

porentak, 2020-12-15:
@kar200 

> Fri Aug 28 20:56:06 2020 daemon.notice netifd: radio1 (1544): Could not find PHY for device 'radio1'

Due to this error, second wireless interface is not started at boot time. Second phy interface (phy1) registers to late (approx 7s after first one).

This issue does not happen if MT7621 runs in single core mode.

With this patch:

`--- a/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh`
`+++ b/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh`
`@@ -903,7 +903,7 @@ drv_mac80211_setup() {`
` `
`        find_phy || {`
`                echo "Could not find PHY for device '$1'"`
`-               wireless_set_retry 0`
`+               wireless_set_retry 1`
`                return 1`
`        }`

it seems it works just fine but I'm not sure if this is right approach for this issue.

kar200, 2020-12-16:
@porentak 

Thanks a lot I will give it a try (I actually didn't compile openwrt for a while) and I will let you know if it works. 
An other option maybe would be to revert the commit for the automatic detection of DBDC and do it manually (I have not tried it though).

K

porentak, 2020-12-17:
@kar200 
Actually you don't have to compile OpenWrt, you can just modify file on the target and reboot the system.

`/lib/netifd/wireless/mac80211.sh`

kar200, 2020-12-18:
> Actually you don't have to compile OpenWrt, you can just modify file on the target and reboot the system.
> 
> `/lib/netifd/w`

Thanks for your reply and I did try with changing the value from 0 to 1 on the actual file and it did not work (in /lib/netifd/wireless/mac80211.sh).

The router still looses network connectivity with enabling both wireless network interfaces in /etc/config/wireless

porentak, 2020-12-18:
@kar200 

> Thanks for your reply and I did try with changing the value from 0 to 1 on the actual file and it did not work (in /lib/netifd/wireless/mac80211.sh).

> The router still looses network connectivity with enabling both wireless network interfaces in /etc/config/wireless

Can you share wireless config and logread dump including client association event?

kar200, 2020-12-19:
@porentak 

Ok weird. I was using the official master and did not update the mt76 to the latest git. I pulled again master today which includes all the mt76 changes and your workaround worked this time. 

As you said I don't know if this is the right way of doing it but at least for me it works. I am closing this ticket since it works for me and also there is a duplicate one in the list somewhere. 

Thanks again for you help.

K

simontretter, 2020-12-23:
Same issue here on the GL-MT1300 (also using the MT7615DN).

kar200, 2020-12-23:
@simontretter I just reopened the issue here for tracking purposes.

@porentak I was wrong and i still have the exact same issue.
i have no networking even on the ethernet ports. I can run the "logread" comand for example but anything related to networking commands would just hang there.

If I set both radios to disabled or just radio0 to disabled it works. If I set both radio to enabled then the problem happens.
If I use other forks of openwrt that don't use DSA it works as well and the issue does not occur.

My wireless config
`
root@OpenWrt:/# cat /etc/config/wireless

config wifi-device 'radio0'
        option type 'mac80211'
        option channel '3'
        option hwmode '11g'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
        option htmode 'HT20'
        option disabled '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt2'
        option encryption 'none'

config wifi-device 'radio1'
        option type 'mac80211'
        option channel '36'
        option hwmode '11a'
        option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0+1'
        option htmode 'VHT80'
        option disabled '0'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt5'
        option encryption 'none'
`

logread when both card are set to enabled
'
root@OpenWrt:/# logread
Wed Dec 23 05:40:42 2020 kern.notice kernel: [    0.000000] Linux version 5.4.85 (karim@) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r15237-fca0eb2d92)) #0 SMP Tue Dec 22 23:45:19 2020
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000] MIPS: machine is mediatek,mt7621-soc
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000] VPE topology {2,2} total 4
Wed Dec 23 05:40:42 2020 kern.warn kernel: [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
Wed Dec 23 05:40:42 2020 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Wed Dec 23 05:40:42 2020 kern.warn kernel: [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000] Zone ranges:
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000]   HighMem  empty
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000] Movable zone start for each node
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000] Early memory node ranges
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Wed Dec 23 05:40:42 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Wed Dec 23 05:40:42 2020 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Wed Dec 23 05:40:42 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Wed Dec 23 05:40:42 2020 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Wed Dec 23 05:40:42 2020 kern.info kernel: [    0.000000] percpu: Embedded 18 pages/cpu s43344 r8192 d22192 u73728
Wed Dec 23 05:40:42 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s43344 r8192 d22192 u73728 alloc=18*4096
Wed Dec 23 05:40:42 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[... 319 lines cut ...]
Wed Dec 23 05:40:49 2020 daemon.err odhcpd[1475]: Failed to send to ff02::1%lan@br-lan (Address not available)
Wed Dec 23 05:40:49 2020 daemon.debug dnsmasq[1198]: listening on eth0(#2): fe80::f68c:ebff:fea8:f065%eth0 port 53
Wed Dec 23 05:40:49 2020 daemon.notice netifd: radio1 (2267): Could not find PHY for device 'radio1'
Wed Dec 23 05:40:49 2020 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Wed Dec 23 05:40:49 2020 daemon.notice netifd: radio1 (2331): WARNING: Variable 'data' does not exist or is not an array/object
Wed Dec 23 05:41:05 2020 daemon.err odhcpd[1475]: Failed to send to ff02::1%lan@br-lan (Address not available)
Wed Dec 23 05:41:21 2020 daemon.err odhcpd[1475]: Failed to send to ff02::1%lan@br-lan (Address not available)
Wed Dec 23 05:41:23 2020 daemon.info dnsmasq[1198]: exiting on receipt of SIGTERM
Wed Dec 23 05:41:37 2020 daemon.err odhcpd[1475]: Failed to send to ff02::1%lan@br-lan (Address not available)
Wed Dec 23 05:41:53 2020 daemon.err odhcpd[1475]: Failed to send to ff02::1%lan@br-lan (Address not available)
Wed Dec 23 05:42:09 2020 daemon.err odhcpd[1475]: Failed to send to ff02::1%lan@br-lan (Address not available)
Wed Dec 23 05:42:25 2020 daemon.err odhcpd[1475]: Failed to send to ff02::1%lan@br-lan (Address not available)
Wed Dec 23 05:42:41 2020 daemon.err odhcpd[1475]: Failed to send to ff02::1%lan@br-lan (Address not available)
[... cut ...]
Wed Dec 23 05:44:33 2020 daemon.err odhcpd[1475]: Failed to send to ff02::1%lan@br-lan (Address not available)
Wed Dec 23 05:44:49 2020 daemon.err odhcpd[1475]: Failed to send to ff02::1%lan@br-lan (Address not available)
Wed Dec 23 05:45:05 2020 daemon.err odhcpd[1475]: Failed to send to ff02::1%lan@br-lan (Address not available)
Wed Dec 23 05:45:21 2020 daemon.err odhcpd[1475]: Failed to send to ff02::1%lan@br-lan (Address not available)
Wed Dec 23 05:45:37 2020 daemon.err odhcpd[1475]: Failed to send to ff02::1%lan@br-lan (Address not available)
'

vj4git, 2020-12-28:
@kar200 @porentak  this is what I did to getting DBDC working without loss of network connectivity - within /etc/config/wireless add option macaddr '<your2g_or_5g>' to BOTH config wifi-device AS WELL AS config wifi-iface for BOTH frequencies. Something like this and it then survives across boot.

```
config wifi-device 'radio0'
option type 'mac80211'
option macaddr 'my2g-radio-MAC'
option channel '6'
option hwmode '11g'
option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
option htmode 'HT20'
option disabled '0'

config wifi-iface 'default_radio0'
option device 'radio0'
option macaddr 'my2g-radio-MAC'
..
..
config wifi-device 'radio1'
option type 'mac80211'
option macaddr 'my5g-radio-MAC'
option channel '36'
option hwmode '11a'
option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0+1'
option htmode 'VHT80'
option disabled '0'

config wifi-iface 'default_radio1'
option device 'radio1'
option macaddr 'my5g-radio-MAC'

```
However, even though both radios are active and accept connections, my 5g speed somehow is not great. I am getting 2g around 110-140mbits, but 5g hovers around 80mbits. I tried switching 2g to 5g paths but then one of the radios don't start. I am sure I have the order correct, similar to @kar200 - iw list shows phy0 with both 2g/5g bands while phy1 shows only 5g bands.

simontretter, 2021-01-01:
Specifying the macaddr like this does not resolve the loss of network connectivity on reboot for the GL-MT1300.

kar200, 2021-01-05:
@vj4git 
I tried your config and it sort of like worked. Not sure how yours work but for me the first interface does not come up at boot which is probably why it worked. I have to restart the the wifi down/wifi up which makes both interfaces work.

I did not test outside of just enabling the wireless. As I said previously if I disable the first interface at boot (2g) and only enable the 5g one then the router boots fine.

On the other hand when using swconfig through [https://github.com/x-wrt/x-wrt](url) then both interfaces boot normally at boot time.

kar200, 2021-01-15:
Hi all,

I compiled again today from master with all the latest changes and the router does not freeze anymore.
I still have to add "wifi down"+"wifi up" to bring both wireless cards up at boot (just a noob way to fix the issue) but since it is not related to this issue i am closing this.

closed 2021-01-15 by kar200