openwrt/mt76 #441: ieee80211_sta_update_pending_airtime kernel crash

mjsir911, 2020-08-30:
Hello,

I'm having an issue where after days/weeks my DIR-878's kernel seems to crash. This prevents wlan routing & causes a bunch of other issues, but I still have internet access even after this happens.

Temporary fix is easy enough, restarting the router works

```
[193350.515184] ------------[ cut here ]------------
[193350.524767] WARNING: CPU: 0 PID: 0 at backports-5.7-rc3-1/net/mac80211/sta_info.c:1929 ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[193350.551139] STA 40:40:a7:9a:49:8f AC 2 txq pending airtime underflow: 4294967252, 44
[193350.551143] Modules linked in: pppoe ppp_async iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mt7615e mt7615_common mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio gpio_button_hotplug
[193350.676737] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.43 #0
[193350.688678] Stack : 872a1e80 80077050 80670000 8066cc34 806d0000 8066cbfc 8066bd50 87c0dc44
[193350.705471]         80810000 806b31c4 806b2d03 80604380 00000000 00000001 87c0dbe8 2b4125d3
[193350.722256]         00000000 00000000 80850000 00000000 00000030 0000015a 2e352064 33342e34
[193350.739041]         00000000 00002593 00000000 70617773 00000000 806d0000 00000000 86c08a84
[193350.755826]         00000009 00000000 00000000 872a2000 00000003 80323c30 00000000 80810000
[193350.772610]         ...
[193350.777636] Call Trace:
[193350.782683] [<8000b72c>] show_stack+0x30/0x100
[193350.791696] [<8054cfdc>] dump_stack+0xa4/0xdc
[193350.800535] [<80028058>] __warn+0xc0/0x10c
[193350.808846] [<80028130>] warn_slowpath_fmt+0x8c/0xac
[193350.818963] [<86c08a84>] ieee80211_sta_update_pending_airtime+0x22c/0x234 [mac80211]
[193350.834583] [<86c0205c>] ieee80211_report_low_ack+0xf0/0x43c [mac80211]
[193350.847957] [<86c023c0>] ieee80211_free_txskb+0x18/0x30 [mac80211]
[193350.860458] [<87325468>] mt76_tx_complete_skb+0xa8/0xbc [mt76]
[193350.872247] [<86ceaa20>] mt7615_txp_skb_unmap+0x1f4/0x230 [mt7615_common]
[193350.886171] ---[ end trace d420dc140c7f437c ]---
```

mjsir911, 2020-08-30:
seems to be a duplicate of #427 (which is closed)

ryderlee1110, 2020-08-31:
Did you try the latest openwrt master?