openwrt/mt76 #444: Firmware Resets

Mushoz, 2020-09-14:
I have upgraded my DIR-860L device from 19.07.1 to 19.07.4 (which means the mt76 driver received this update: https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=9da31d0fb402f5ee3f3973ee07625942634a38fb) and now my kernel logs are getting spammed by this, over and over and over again:

```
[303621.090349] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[303621.101702] mt76x2e 0000:01:00.0: Build: 1
[303621.110451] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[303621.138812] mt76x2e 0000:01:00.0: Firmware running!
[303621.158800] ieee80211 phy0: Hardware restart was requested
[305405.459830] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[305405.471014] mt76x2e 0000:01:00.0: Build: 1
[305405.479389] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[305405.508163] mt76x2e 0000:01:00.0: Firmware running!
[305405.528274] ieee80211 phy0: Hardware restart was requested
[305441.699606] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[305441.710760] mt76x2e 0000:01:00.0: Build: 1
[305441.719123] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[305441.747991] mt76x2e 0000:01:00.0: Firmware running!
[305441.768070] ieee80211 phy0: Hardware restart was requested
[305541.329064] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[305541.340234] mt76x2e 0000:01:00.0: Build: 1
[305541.348868] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[305541.377404] mt76x2e 0000:01:00.0: Firmware running!
[305541.397528] ieee80211 phy0: Hardware restart was requested
[312081.304855] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[312081.316019] mt76x2e 0000:01:00.0: Build: 1
[312081.324483] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[312081.352923] mt76x2e 0000:01:00.0: Firmware running!
[312081.373055] ieee80211 phy0: Hardware restart was requested
```

I experience small hiccups in the my connection, supposedly due to the firmware resetting. I did see these message very rarely on 19.07.1 as well, but not as often as I do on 19.07.4. If you need any additional information for debugging this, please let me know!

ptpt52, 2020-09-15:
same problem on my 7621 or 7621 devices

Mushoz, 2020-09-16:
@ptpt52 there are many different mt7621 devices, but many different WiFi chips. With which devices are you having issues? Might help us pinpoint the issue.

dchard, 2020-09-17:
Same problem here too (mt7621, D-Link 860L). Only difference is that my PPPoE connection is also dropping.

This is the WIFI issue (only affecting 2.4GHz btw):

```
 72.414731] ieee80211 phy1: Hardware restart was requested
[   73.458692] mt76x2e 0000:02:00.0: MCU message 2 (seq 15) timed out
[   73.827737] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[   73.838746] mt76x2e 0000:02:00.0: Build: 1
[   73.846963] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[   73.874621] mt76x2e 0000:02:00.0: Firmware running!
[   73.886742] ieee80211 phy1: Hardware restart was requested
[   74.930615] mt76x2e 0000:02:00.0: MCU message 2 (seq 3) timed out
[   75.299752] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[   75.310734] mt76x2e 0000:02:00.0: Build: 1
[   75.318920] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[   75.346629] mt76x2e 0000:02:00.0: Firmware running!
[   75.358757] ieee80211 phy1: Hardware restart was requested
[   76.402616] mt76x2e 0000:02:00.0: MCU message 2 (seq 6) timed out
[   76.771890] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[   76.782952] mt76x2e 0000:02:00.0: Build: 1
[   76.791166] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[   76.818650] mt76x2e 0000:02:00.0: Firmware running!
[   76.830887] ieee80211 phy1: Hardware restart was requested
```

This is the PPPoE:

```
[ 6654.729554] mt7530 mdio-bus:1f wan: Link is Down
[ 6654.744496] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[ 6654.758093] 8021q: adding VLAN 0 to HW filter on device wan
[ 6654.868874] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[ 6654.882354] 8021q: adding VLAN 0 to HW filter on device wan
[ 6658.978879] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[ 6674.357221] pppoe-wan: renamed from ppp0
[ 6977.846481] mt7530 mdio-bus:1f wan: Link is Down
[ 6977.863373] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[ 6977.877022] 8021q: adding VLAN 0 to HW filter on device wan
[ 6978.014972] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[ 6978.028593] 8021q: adding VLAN 0 to HW filter on device wan
[ 6982.114198] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[ 6987.476699] pppoe-wan: renamed from ppp0
[ 7417.007646] mt7530 mdio-bus:1f wan: Link is Down
[ 7417.022711] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[ 7417.036385] 8021q: adding VLAN 0 to HW filter on device wan
[ 7417.147297] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[ 7417.160884] 8021q: adding VLAN 0 to HW filter on device wan
[ 7421.249310] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[ 7436.616196] pppoe-wan: renamed from ppp0
[ 7712.923390] mt7530 mdio-bus:1f wan: Link is Down
[ 7712.939600] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[ 7712.953338] 8021q: adding VLAN 0 to HW filter on device wan
[ 7713.097398] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[ 7713.110920] 8021q: adding VLAN 0 to HW filter on device wan
[ 7717.216718] mt7530 mdio-bus:1f wan: Link is Up - 1Gbps/Full - flow control off
[ 7722.577165] pppoe-wan: renamed from ppp0
```

dchard, 2020-09-19:
Tried with today's master, and the issue is the same: the 2.4GHz wifi cannot be enabled, otherwise this happens (every second, repeatedly):

```
[  929.185194] mt76x2e 0000:02:00.0: MCU message 2 (seq 15) timed out
[  929.554328] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  929.565341] mt76x2e 0000:02:00.0: Build: 1
[  929.573601] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[  929.601191] mt76x2e 0000:02:00.0: Firmware running!
[  929.613344] ieee80211 phy1: Hardware restart was requested
```

vgmz95, 2020-09-23:
Same problem on Xiaomi R3G router with openwrt 19.07.4

JFtico, 2020-09-24:
Adding a comment that I do NOT see this error on a ZBT-WE3526 (7621A/7603) on 19.07.3
I USED to see it regularly  (a few times a week) on 19.07.3, so the new code on the same hardware with the same clients seems better (in terms of this error) with .4

Maybe something in how the ROM data is being interpreted on some devices?

garyttirn, 2020-09-27:
I'm also seeing this on DIR-860L with snaphot r14572-4682d4d770. Works fine up to 19.07.3.
```
[  768.364765] ieee80211 phy1: Hardware restart was requested
[  769.408626] mt76x2e 0000:02:00.0: MCU message 2 (seq 12) timed out
[  769.777815] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  769.788791] mt76x2e 0000:02:00.0: Build: 1
[  769.796981] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[  769.824617] mt76x2e 0000:02:00.0: Firmware running!
[  769.836817] ieee80211 phy1: Hardware restart was requested
[  770.880625] mt76x2e 0000:02:00.0: MCU message 2 (seq 15) timed out
[  771.249762] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  771.260733] mt76x2e 0000:02:00.0: Build: 1
[  771.268924] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[  771.296616] mt76x2e 0000:02:00.0: Firmware running!
[  771.308751] ieee80211 phy1: Hardware restart was requested
```

headnail, 2020-11-21:
Same MCU time outs on 860L B1 here as well, occurring on both dd-wrt and openrwrt, appears to only affect 5GHz radio.
Stock firmware runs without any radio issues - eliminating hardware fault hypothesis.

Borromini, 2020-12-06:
I just bumped to a recent master build (d346beb08c3a7867497000dc382635ee8ea0eedb) and I'm bumping into the hardware restart issue as well.

helicida, 2021-01-06:
Newifi-D2 with stable 19.07.5, same here. 5G WiFi not working at all:

Sun Dec  6 08:27:24 2020 kern.info kernel: [ 1812.411269] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Sun Dec  6 08:27:24 2020 kern.info kernel: [ 1812.416754] mt76x2e 0000:01:00.0: Build: 1
Sun Dec  6 08:27:24 2020 kern.info kernel: [ 1812.420953] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Sun Dec  6 08:27:24 2020 kern.info kernel: [ 1812.449560] mt76x2e 0000:01:00.0: Firmware running!
Sun Dec  6 08:27:24 2020 kern.info kernel: [ 1812.459721] ieee80211 phy1: Hardware restart was requested

siwind, 2021-02-04:
My 2.4G wifi alse couldn't start.  
Log is :
```
[   13.628773] wireguard: WireGuard 1.0.20200611 loaded. See www.wireguard.com for information.
[   13.637245] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[   13.661392] xt_time: kernel timezone is -0000
[   13.735279] bus=0x1, slot = 0x0, irq=0xff
[   13.739327] PCI: Enabling device 0000:00:00.0 (0000 -> 0002)
[   13.773087] mt7603e 0000:01:00.0: Refused to change power state, currently in D3
[   13.780649] mt7603e 0000:01:00.0: ASIC revision: 0000
[   13.786804] ------------[ cut here ]------------
[   13.791455] WARNING: CPU: 0 PID: 583 at /home/wang/data/OpenSource/openwrt-19.07.6/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2021-02-01-4ba1709c/mt7603/eeprom.c:26 mt7603_eeprom_init+0x16c/0x514 [mt7603e]
[   13.811024] Modules linked in: mt7603e(+) mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_socket xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD wireguard ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink macvlan iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net
[   13.881384]  ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 nf_nat nf_conntrack nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip6_udp_tunnel udp_tunnel tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat usbcore nls_base usb_common crc32c_generic
[   13.942911] CPU: 0 PID: 583 Comm: kmodloader Not tainted 4.14.215 #0
[   13.949234] Stack : 00000000 000f0000 82e10000 8007297c 805b0000 8054c5e8 00000000 00000000
[   13.957570]         80517f6c 824f5a64 8393e31c 80586907 80512d00 00000001 824f5a08 ac07f5ce
[   13.965906]         00000000 00000000 806f0000 000054f0 00000000 00000103 00000007 00000000
[   13.974239]         00000000 80590000 000e633f 00000000 80000000 805b0000 00000000 82e08c4c
[   13.982571]         82e076ec 0000001a 83b27880 000f0000 00000003 8029c2a8 00000000 806f0000
[   13.990903]         ...
[   13.993341] Call Trace:
[   13.995799] [<8000c7b0>] show_stack+0x58/0x100
[   14.000228] [<80454be4>] dump_stack+0xa4/0xe0
[   14.004575] [<8002f798>] __warn+0xe0/0x138
[   14.008652] [<8002f880>] warn_slowpath_null+0x1c/0x2c
[   14.013702] [<82e076ec>] mt7603_eeprom_init+0x16c/0x514 [mt7603e]
[   14.019788] [<82e01da0>] mt7603_register_device+0x150/0xba4 [mt7603e]
[   14.026219] [<82e00188>] init_module+0x18188/0x19270 [mt7603e]
[   14.032103] ---[ end trace eb764efd0c3d03b3 ]---
[   16.083866] mt7603e 0000:01:00.0: Firmware Version: _e1_drv_tv
[   16.089695] mt7603e 0000:01:00.0: Build Time: 20140313215548
[   17.112836] mt7603e 0000:01:00.0: Timeout waiting for ROM code to become ready
[   17.120232] mt7603e: probe of 0000:01:00.0 failed with error -5
[   17.136939] bus=0x2, slot = 0x1, irq=0xff
[   17.141152] mt76x2e 0000:02:00.0: ASIC revision: 76120044
[   17.803509] mt76x2e 0000:02:00.0: ROM patch build: 20141115060606a
[   17.813346] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[   17.818824] mt76x2e 0000:02:00.0: Build: 1
[   17.822941] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[   17.842899] mt76x2e 0000:02:00.0: Firmware running!
[   17.851018] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   17.858162] PPP generic driver version 2.4.2
[   17.864246] NET: Registered protocol family 24
```

The mt76 driver is:
```
#opkg list-installed | grep mt76

kmod-mt76-core - 4.14.215+2021-02-01-4ba1709c-1
kmod-mt7603 - 4.14.215+2021-02-01-4ba1709c-1
kmod-mt76x02-common - 4.14.215+2021-02-01-4ba1709c-1
kmod-mt76x2 - 4.14.215+2021-02-01-4ba1709c-1
kmod-mt76x2-common - 4.14.215+2021-02-01-4ba1709c-1
```

rolandvans, 2021-04-14:
> 
> 
> Same MCU time outs on 860L B1 here as well, occurring on both dd-wrt and openrwrt, appears to only affect 5GHz radio.
> Stock firmware runs without any radio issues - eliminating hardware fault hypothesis.

I see the same in the log file. 5GHz is not stable in 19.07.x:

 Model:D-Link DIR-860L B1
Architecture: MediaTek MT7621 ver:1 eco:3
Software: OpenWrt 19.07.7 r11306-c4a6851c72 / LuCI openwrt-19.07 branch git-21.086.32701-7456e2a

Errors in log:
Mon Apr 5 19:54:00 2021 kern.err kernel: [ 8536.871237] mt76x2e 0000:01:00.0: MCU message 31 (seq 8) timed out
Mon Apr 5 19:54:00 2021 kern.info kernel: [ 8536.932817] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Mon Apr 5 19:54:00 2021 kern.info kernel: [ 8536.932836] mt76x2e 0000:01:00.0: Build: 1
Mon Apr 5 19:54:00 2021 kern.info kernel: [ 8536.932844] mt76x2e 0000:01:00.0: Build Time: 201507311614 Mon Apr 5 19:54:00 2021 kern.info kernel: [ 8536.951281] mt76x2e 0000:01:00.0: Firmware running!
Mon Apr 5 19:54:00 2021 kern.info kernel: [ 8536.961405] ieee80211 phy0: Hardware restart was requested
Mon Apr 5 20:28:47 2021 kern.err kernel: [10624.060218] mt76x2e 0000:01:00.0: MCU message 31 (seq 1) timed out
Mon Apr 5 20:28:47 2021 kern.info kernel: [10624.121807] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Mon Apr 5 20:28:47 2021 kern.info kernel: [10624.121826] mt76x2e 0000:01:00.0: Build: 1
Mon Apr 5 20:28:47 2021 kern.info kernel: [10624.121834] mt76x2e 0000:01:00.0: Build Time: 201507311614 Mon Apr 5 20:28:47 2021 kern.info kernel: [10624.150387] ieee80211 phy0: Hardware restart was requested
Mon Apr 5 20:28:47 2021 kern.info kernel: [10624.140242] mt76x2e 0000:01:00.0: Firmware running!

Borromini, 2021-04-14:
At this point, with 21.02 being around the corner, I'd advise trying that. Has a much newer mt76 version.

Builds can be grabbed here: https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/ramips/mt7621/ (include LuCI if you're worried about that).

robho, 2021-06-24:
I have tried 21.02.0-rc3 on a Netgear R6220 and I see the firmware resets too:
`[13527.715355] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[13527.726533] mt76x2e 0000:01:00.0: Build: 1
[13527.734913] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[13527.766324] mt76x2e 0000:01:00.0: Firmware running!
[13527.782783] ieee80211 phy1: Hardware restart was requested
[14133.839264] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[14133.850442] mt76x2e 0000:01:00.0: Build: 1
[14133.858685] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[14133.894093] mt76x2e 0000:01:00.0: Firmware running!
[14133.906315] ieee80211 phy1: Hardware restart was requested
`
phy1 is the 5 GHz radio.

Borromini, 2021-07-06:
@robho Maybe something similar to https://github.com/openwrt/openwrt/commit/af1b6799c6ec9af7a30d63a5ddfed20f443b991c could help you (need to adapt your device DTS and recompile).

noctux, 2021-09-08:
This issue seems to persist in 21.02.0 for the DIR-860L:

```
Wed Sep  8 15:13:21 2021 kern.info kernel: [102481.248692] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Wed Sep  8 15:13:21 2021 kern.info kernel: [102481.259854] mt76x2e 0000:01:00.0: Build: 1
Wed Sep  8 15:13:21 2021 kern.info kernel: [102481.268247] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Wed Sep  8 15:13:21 2021 kern.info kernel: [102481.295534] mt76x2e 0000:01:00.0: Firmware running!
Wed Sep  8 15:13:21 2021 kern.info kernel: [102481.307691] ieee80211 phy0: Hardware restart was requested
Wed Sep  8 15:14:01 2021 kern.info kernel: [102520.956179] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Wed Sep  8 15:14:01 2021 kern.info kernel: [102520.967339] mt76x2e 0000:01:00.0: Build: 1
Wed Sep  8 15:14:01 2021 kern.info kernel: [102520.975703] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Wed Sep  8 15:14:01 2021 kern.info kernel: [102521.003034] mt76x2e 0000:01:00.0: Firmware running!
Wed Sep  8 15:14:01 2021 kern.info kernel: [102521.015263] ieee80211 phy0: Hardware restart was requested
Wed Sep  8 15:15:00 2021 kern.info kernel: [102580.819460] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Wed Sep  8 15:15:00 2021 kern.info kernel: [102580.830618] mt76x2e 0000:01:00.0: Build: 1
Wed Sep  8 15:15:00 2021 kern.info kernel: [102580.838982] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Wed Sep  8 15:15:00 2021 kern.info kernel: [102580.866277] mt76x2e 0000:01:00.0: Firmware running!
Wed Sep  8 15:15:00 2021 kern.info kernel: [102580.878503] ieee80211 phy0: Hardware restart was requested
```

nquest, 2021-09-25:
Confirmed on Newifi-D2.
```log
Fri Sep 24 12:18:56 2021 kern.info kernel: [88785.900774] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Fri Sep 24 12:18:56 2021 kern.info kernel: [88785.906262] mt76x2e 0000:01:00.0: Build: 1
Fri Sep 24 12:18:56 2021 kern.info kernel: [88785.910412] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Fri Sep 24 12:18:56 2021 kern.info kernel: [88785.939466] mt76x2e 0000:01:00.0: Firmware running!
Fri Sep 24 12:19:01 2021 kern.info kernel: [88790.570742] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Fri Sep 24 12:19:01 2021 kern.info kernel: [88790.576227] mt76x2e 0000:01:00.0: Build: 1
Fri Sep 24 12:19:01 2021 kern.info kernel: [88790.580384] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Fri Sep 24 12:19:01 2021 kern.info kernel: [88790.609510] mt76x2e 0000:01:00.0: Firmware running!
Fri Sep 24 12:21:33 2021 kern.info kernel: [88943.000822] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Fri Sep 24 12:21:33 2021 kern.info kernel: [88943.006305] mt76x2e 0000:01:00.0: Build: 1
Fri Sep 24 12:21:33 2021 kern.info kernel: [88943.010499] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Fri Sep 24 12:21:33 2021 kern.info kernel: [88943.039463] mt76x2e 0000:01:00.0: Firmware running!
```

rizalwidhi, 2021-12-27:
Same issue seems to persist in OpenWrt 21.02.1 r16325-88151b8303 for Xiaomi MiWiFi Mini : 
This WIFI issue (only affecting 5 GHz)

Mon Dec 27 06:52:47 2021 kern.info kernel: [ 2025.470732] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Mon Dec 27 06:52:47 2021 kern.info kernel: [ 2025.476398] mt76x2e 0000:01:00.0: Build: 1
Mon Dec 27 06:52:47 2021 kern.info kernel: [ 2025.480623] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Mon Dec 27 06:52:47 2021 kern.info kernel: [ 2025.497791] mt76x2e 0000:01:00.0: Firmware running!
Mon Dec 27 06:52:47 2021 kern.info kernel: [ 2025.504631] ieee80211 phy0: Hardware restart was requested

> This issue seems to persist in 21.02.0 for the DIR-860L:
>
> ```
> Wed Sep  8 15:13:21 2021 kern.info kernel: [102481.248692] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
> Wed Sep  8 15:13:21 2021 [...]

weimdall, 2022-01-07:
Same issue for me too, on OpenWrt 21.02.1 r16325-88151b8303 (only 5Ghz)

> Thu Jan  6 20:11:07 2022 kern.info kernel: [ 6370.319507] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
Thu Jan  6 20:11:07 2022 kern.info kernel: [ 6370.325017] mt76x2e 0000:02:00.0: Build: 1
Thu Jan  6 20:11:07 2022 kern.info kernel: [ 6370.329199] mt76x2e 0000:02:00.0: Build Time: 201507311614____
Thu Jan  6 20:11:07 2022 kern.info kernel: [ 6370.350322] mt76x2e 0000:02:00.0: Firmware running!
Thu Jan  6 20:11:07 2022 kern.info kernel: [ 6370.358439] ieee80211 phy1: Hardware restart was requested
Thu Jan  6 20:15:35 2022 kern.info kernel: [ 6638.456983] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
Thu Jan  6 20:15:35 2022 kern.info kernel: [ 6638.462496] mt76x2e 0000:02:00.0: Build: 1
Thu Jan  6 20:15:35 2022 kern.info kernel: [ 6638.466673] mt76x2e 0000:02:00.0: Build Time: 201507311614____
Thu Jan  6 20:15:35 2022 kern.info kernel: [ 6638.487764] mt76x2e 0000:02:00.0: Firmware running!
Thu Jan  6 20:15:35 2022 kern.info kernel: [ 6638.496018] ieee80211 phy1: Hardware restart was requested
Thu Jan  6 20:16:20 2022 kern.info kernel: [ 6683.581268] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
Thu Jan  6 20:16:20 2022 kern.info kernel: [ 6683.586785] mt76x2e 0000:02:00.0: Build: 1
Thu Jan  6 20:16:20 2022 kern.info kernel: [ 6683.590923] mt76x2e 0000:02:00.0: Build Time: 201507311614____
Thu Jan  6 20:16:20 2022 kern.info kernel: [ 6683.611988] mt76x2e 0000:02:00.0: Firmware running!
Thu Jan  6 20:16:20 2022 kern.info kernel: [ 6683.620232] ieee80211 phy1: Hardware restart was requested

jxph1123, 2022-01-10:
With the version below, no more firmware reset issues.
PKG_SOURCE_DATE:=2018-11-09
PKG_SOURCE_VERSION:=ffccb48c5082b2c6f43a0235b77b62ca388810f0
PKG_MIRROR_HASH:=bff0063db1a1b1353f59aad16abad4dd93ceeb4c672811ad864cebe0e39c7d8d

weimdall, 2022-01-10:
> The mt76 uses the version below and no more firmware reset issues. 
PKG_SOURCE_DATE:=2018-11-09
PKG_SOURCE_VERSION:=ffccb48c5082b2c6f43a0235b77b62ca388810f0 PKG_MIRROR_HASH:=bff0063db1a1b1353f59aad16abad4dd93ceeb4c672811ad864cebe0e39c7d8d

Well, currently I have the version :
PKG_SOURCE_DATE:=2021-12-03
PKG_SOURCE_VERSION:=678071ef70297b7264661c356ddb3c8cf7f3c87b
PKG_MIRROR_HASH:=b1f8613f7c65ca6a893f83ed9efc3f7ce72b9b4904fd11b89264f57f4f2a3b5e

But going back to a version from 2018 doesn't sound like a great idea to me,
therefore I'd have to compile it against Linux 5.4.154 mips .. not sure it will be compatible ?

jxph1123, 2022-01-10:
> > The mt76 uses the version below and no more firmware reset issues.
> > PKG_SOURCE_DATE:=2018-11-09
> > PKG_SOURCE_VERSION:=ffccb48c5082b2c6f43a0235b77b62ca388810f0 PKG_MIRROR_HASH:=bff0063db1a1b1353f59aad16abad4dd93ceeb4c672811ad864cebe0e39c7d8d
> 
> Well, currently I have the version : PKG_SOURCE_DATE:=2021-12-03 PKG_SOURCE_VERSION:=678071ef70297b7264661c356ddb3c8cf7f3c87b PKG_MIRROR_HASH:=b1f8613f7c65ca6a893f83ed9efc3f7ce72b9b4904fd11b89264f57f4f2a3b5e
> 
> But going back to a version from 2018 sound like a bad idea to me, therefore I'd have to compile it against Linux 5.4.154 mips .. not sure it will be compatible ?

I compiled open-19.07, Linux 4.14.236, no problem, Linux 5.4.154 not tested.

jxph1123, 2022-01-10:
https://forum.openwrt.org/t/resolved-d-link-dir-860l-mt7621-5ghz-wifi-issues-openwrt-19-07-5/84641/4
Based on this discussion, I rolled back the version, and there was no firmware restart.

ptpt52, 2022-01-11:
Hi all
Please upgrade the new version of firmware bin and test it

1. download the firmware https://raw.githubusercontent.com/ptpt52/mt76/master/firmware/mt7662_firmware_e3_v1.9.bin
2. rename it to `mt7662.bin` and replace file `/lib/firmware/mt7662.bin`
3. reboot

ptpt52, 2022-01-11:
new version fw boot log:
```
dmesg | grep mt76x2e
[   28.218503] mt76x2e 0000:02:00.0: enabling device (0000 -> 0002)
[   28.230735] mt76x2e 0000:02:00.0: ASIC revision: 76120044
[   29.071944] mt76x2e 0000:02:00.0: ROM patch build: 20141115060606a
[   29.192558] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[   29.203585] mt76x2e 0000:02:00.0: Build: 1
[   29.211846] mt76x2e 0000:02:00.0: Build Time: 201706151016____
[   29.237211] mt76x2e 0000:02:00.0: Firmware running!
```

jxph1123, 2022-01-11:
> Hi all Please upgrade the new version of firmware bin and test it
> 
> 1. download the firmware https://raw.githubusercontent.com/ptpt52/mt76/master/firmware/mt7662_firmware_e3_v1.9.bin
> 2. rename it to `mt7662.bin` and replace file `/lib/firmware/mt7662.bin`
> 3. reboot

Can you provide the source code and compile it yourself, thank you.

ptpt52, 2022-01-11:
for clarity
Wifi Firmware bin is come from mtk SDK and no source code is possible for us

the old version in mt76 (https://raw.githubusercontent.com/openwrt/mt76/master/firmware/mt7662_firmware_e3_v1.9.bin) also no source

ptpt52, 2022-01-11:
@jxph1123  hi you may do more test to get the result
there is no likely reason the old version of mt76 fix this issue

Borromini, 2022-01-11:
> Can you provide the source code and compile it yourself, thank you.

Firmware blobs are always binary. Even if manufacturers would be willing to provide you source code for it, that would violate FCC regulations and whatnot.

jxph1123, 2022-01-11:
> 

Thanks, I will try.

myapple, 2022-01-12:
@ptpt52 
hi，How about your base test result?  do WDS, STA, MESH and AP all work OK ？

myapple, 2022-01-12:
@ptpt52 I am afraid this firmware may be not a good update , for testing AP mode with heavy load, it hangs quickly, and can not recover automaticlly

noctux, 2022-01-12:
@ptpt52: Thanks! Just out of curiosity: on my router, `/lib/firmware` looks like this:

```
ls /lib/firmware/
mt7662.bin            mt7662_rom_patch.bin  regulatory.db
```

Just extrapolating from the name, I'd assume `mt7662_rom_patch.bin` to be some binary patch on the original firmware? Doesn't that one have to be updated as well when switching firmwares? Or is that completely unrelated? In your repository, it appears to point to a 7 year old file, which probably targets the old firmware revision?

ptpt52, 2022-01-12:
> @ptpt52 I am afraid this firmware may be not a good update , for testing AP mode with heavy load, it hangs quickly, and can not recover automaticlly

could you post the log?
```
dmesg
logread
```

my case: 7612, replace the firmware and test on AP, STA, 80211s mesh, all works.

ptpt52, 2022-01-12:
> @ptpt52: Thanks! Just out of curiosity: on my router, `/lib/firmware` looks like this:
>
> ```
> ls /lib/firmware/
> mt7662.bin            mt7662_rom_patch.bin  regulatory.db
> ```
>
> Just extrapolating from [...]

just update and replace the `mt7622.bin` 
and the `mt7662_rom_patch.bin` is the same, no need to update

jxph1123, 2022-01-12:
My 7621 has been tested for a day, and the firmware has not restarted yet, everything is normal, continue to test.

JFtico, 2022-01-17:
Very interesting, thanks. Testing on a ZBT WE3526 with 19.07.6 and so far so good aster a few hours.

noctux, 2022-01-17:
I have been running the firmware file linked by @ptpt52 for some time now, and so far, I've not seen any firmware restarts being reported in dmesg. Subjectively, 5Ghz might have become a bit more stable, but as I'm running a dual-band SSID and I have no hard metrics, it might as well be purly wishful thinking.

However, I have noticed this kernel crash in dmesg; it does concern mt76, but might as well be unrelated.

```
[161950.425958] ------------[ cut here ]------------
[161950.435400] WARNING: CPU: 3 PID: 951 at backports-5.10.68-1/net/mac80211/airtime.c:457 0x86857060 [mac80211@571e22cd+0x7d550]
[161950.458181] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY slhc sch_cake nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore
[161950.458368]  nls_base usb_common
[161950.639485] CPU: 3 PID: 951 Comm: mt76-tx phy1 Not tainted 5.4.154 #0
[161950.652465] Stack : 00000000 80850000 00000000 8007e0d8 00000000 00000000 00000000 00000000
[161950.669260]         00000000 00000000 00000000 00000000 00000000 00000001 86a09c30 51e67459
[161950.686050]         86a09cc8 00000000 00000000 00000000 00000038 805ebea4 00000018 35206465
[161950.702852]         00000000 00001f79 00000000 746d203a 00000000 86a09c10 00000000 86857060
[161950.719645]         00000009 00000014 00000004 00000002 00000008 8035cbf4 0000000c 8082000c
[161950.736438]         ...
[161950.741467] Call Trace:
[161950.741481] [<8007e0d8>] 0x8007e0d8
[161950.753651] [<805ebea4>] 0x805ebea4
[161950.760776] [<86857060>] 0x86857060 [mac80211@571e22cd+0x7d550]
[161950.772720] [<8035cbf4>] 0x8035cbf4
[161950.779832] [<8000b05c>] 0x8000b05c
[161950.786938] [<8000b064>] 0x8000b064
[161950.794040] [<805d17d8>] 0x805d17d8
[161950.801143] [<8007ab80>] 0x8007ab80
[161950.808247] [<8002c3d0>] 0x8002c3d0
[161950.815359] [<86857060>] 0x86857060 [mac80211@571e22cd+0x7d550]
[161950.827303] [<8002c478>] 0x8002c478
[161950.834428] [<86857060>] 0x86857060 [mac80211@571e22cd+0x7d550]
[161950.846387] [<86857248>] 0x86857248 [mac80211@571e22cd+0x7d550]
[161950.858353] [<86857328>] 0x86857328 [mac80211@571e22cd+0x7d550]
[161950.870310] [<8041e98c>] 0x8041e98c
[161950.877444] [<86ee2a24>] 0x86ee2a24 [mt76x02_lib@844a4414+0x9de0]
[161950.889744] [<86f20530>] 0x86f20530 [mt76@633da2ff+0x96e0]
[161950.900859] [<86f20530>] 0x86f20530 [mt76@633da2ff+0x96e0]
[161950.911944] [<86ee5004>] 0x86ee5004 [mt76x02_lib@844a4414+0x9de0]
[161950.924238] [<86f20530>] 0x86f20530 [mt76@633da2ff+0x96e0]
[161950.935334] [<86f205d4>] 0x86f205d4 [mt76@633da2ff+0x96e0]
[161950.946418] [<8004b6d4>] 0x8004b6d4
[161950.953524] [<805ed83c>] 0x805ed83c
[161950.960642] [<8004bbe4>] 0x8004bbe4
[161950.967747] [<8004baa4>] 0x8004baa4
[161950.974859] [<8004baa4>] 0x8004baa4
[161950.981968] [<8004baa4>] 0x8004baa4
[161950.989071] [<80006718>] 0x80006718
[161950.996178] 
[161950.999794] ---[ end trace b2767dd2d11e78c7 ]---
```

Still, the firmware update seems like a net improvement. @ptpt53 What's the source of this firmwarefile? Teh linux-firmware upstream at kernel.org still provides the same version that openwrt ships, and I was unable to find this particular file (based on various checksum values) anywhere on the internet, and mediathek does not seem to provide any firmware downloads for this particular platform. Do you happen to know what the licensing situation is for that file?

JFtico, 2022-01-18:
@ptpt52 I did see a firmware restart using the new driver.

Context: test router with one connected laptop via the 5Ghz (7612e) radio, guessing it performed a large set of App downloads post OS update on the laptop. No other wired or wireless clients on the unit.

`
Mon Jan 17 19:42:50 2022 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED a4:83:e7:c0:97:e8
Mon Jan 17 19:42:55 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED a4:83:e7:c0:97:e8
Mon Jan 17 21:29:18 2022 kern.info kernel: [24197.424818] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Mon Jan 17 21:29:18 2022 kern.info kernel: [24197.424841] mt76x2e 0000:01:00.0: Build: 1
Mon Jan 17 21:29:18 2022 kern.info kernel: [24197.424849] mt76x2e 0000:01:00.0: Build Time: 201706151016____
Mon Jan 17 21:29:18 2022 kern.info kernel: [24197.452964] mt76x2e 0000:01:00.0: Firmware running!
Mon Jan 17 21:29:18 2022 kern.info kernel: [24197.463191] ieee80211 phy1: Hardware restart was requested
Mon Jan 17 21:29:48 2022 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED a4:83:e7:c0:97:e8
Mon Jan 17 21:29:53 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED a4:83:e7:c0:97:e8

`

Mushoz, 2022-01-19:
@LorenzoBianconi @nbd168 this seems to be an extremely common issue from the amount of replies, and also a long-standing one with the earliest reports dating back more than a year. Is there anything that you can share with us on this topic? Eg, is this most likely a firmware issue, that cannot be fixed due to the firmware blob being closed source? Or is this probably a driver issue, where the root cause simply hasn't been uncovered yet? Is there any progress being made with fixing this issue?

My apologies if this sounded entitled, because you guys definitely don't owe us anything. And I think I can speak for everywhere here that we very much appreciate all the work that is being done for this otherwise wonderful and performant driver. However, some form of communication on this topic would very much be appreciated if you can find the time to do so :) Much appreciated!

marcoferrara10, 2022-01-23:
Good Morning Everyone,I've the same issue of yours and I've installed the firmware version provided by @ptpt52 . 
Well at first this firmware seems to be better than the previous, because by testing the wifi5g speed i have really good results, like the (by using a speedtest utility, like ookla) download at 150Mbps (previous was around 7 bps) and the uplink at 100 Mbps(previous was at ~0.25 Mbps). 
But... i still have a lot of hardware restart....
`[  688.349514] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[  688.355055] mt76x2e 0000:01:00.0: Build: 1
[  688.359252] mt76x2e 0000:01:00.0: Build Time: 201706151016____
[  688.380175] mt76x2e 0000:01:00.0: Firmware running!
[  688.388375] ieee80211 phy1: Hardware restart was requested
[  695.757410] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[  695.762925] mt76x2e 0000:01:00.0: Build: 1
[  695.767094] mt76x2e 0000:01:00.0: Build Time: 201706151016____
[  695.788099] mt76x2e 0000:01:00.0: Firmware running!
[  695.796419] ieee80211 phy1: Hardware restart was requested
[  872.275699] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[  872.281261] mt76x2e 0000:01:00.0: Build: 1
[  872.285436] mt76x2e 0000:01:00.0: Build Time: 201706151016____
[  872.306387] mt76x2e 0000:01:00.0: Firmware running!
[  872.314590] ieee80211 phy1: Hardware restart was requested
[  943.358995] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[  943.364579] mt76x2e 0000:01:00.0: Build: 1
[  943.368749] mt76x2e 0000:01:00.0: Build Time: 201706151016____
[  943.389689] mt76x2e 0000:01:00.0: Firmware running!
[  943.397897] ieee80211 phy1: Hardware restart was requested
[  947.754918] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[  947.760456] mt76x2e 0000:01:00.0: Build: 1
[  947.764629] mt76x2e 0000:01:00.0: Build Time: 201706151016____
[  947.785649] mt76x2e 0000:01:00.0: Firmware running!
[  947.793888] ieee80211 phy1: Hardware restart was requested
[  952.034855] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[  952.040395] mt76x2e 0000:01:00.0: Build: 1
[  952.044576] mt76x2e 0000:01:00.0: Build Time: 201706151016____
[  952.065542] mt76x2e 0000:01:00.0: Firmware running!
[  952.073734] ieee80211 phy1: Hardware restart was requested
[  956.210777] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[  956.216330] mt76x2e 0000:01:00.0: Build: 1
[  956.220491] mt76x2e 0000:01:00.0: Build Time: 201706151016____
[  956.241465] mt76x2e 0000:01:00.0: Firmware running!
[  956.249705] ieee80211 phy1: Hardware restart was requested
[  964.754644] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[  964.760514] mt76x2e 0000:01:00.0: Build: 1
[  964.764698] mt76x2e 0000:01:00.0: Build Time: 201706151016____
[  964.785400] mt76x2e 0000:01:00.0: Firmware running!
[  964.793541] ieee80211 phy1: Hardware restart was requested
[  967.266590] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[  967.272130] mt76x2e 0000:01:00.0: Build: 1
[  967.276270] mt76x2e 0000:01:00.0: Build Time: 201706151016____
[  967.297312] mt76x2e 0000:01:00.0: Firmware running!
[  967.305558] ieee80211 phy1: Hardware restart was requested`

This is my experience ... for sure I will let you know once i have new information.

lukasz1992, 2022-01-24:
19.07 branch has ancient version of this driver

DruvsEgo, 2022-01-24:
MTC WR1201 1200M

[161950.425958] ------------[ cut here ]------------
[161950.435400] WARNING: CPU: 3 PID: 951 at backports-5.10.68-1/net/mac80211/airtime.c:457 0x86857060 [mac80211@571e22cd+0x7d550]
[161950.458181] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY slhc sch_cake nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore
[161950.458368]  nls_base usb_common
[161950.639485] CPU: 3 PID: 951 Comm: mt76-tx phy1 Not tainted 5.4.154 #0
[161950.652465] Stack : 00000000 80850000 00000000 8007e0d8 00000000 00000000 00000000 00000000
[161950.669260]         00000000 00000000 00000000 00000000 00000000 00000001 86a09c30 51e67459
[161950.686050]         86a09cc8 00000000 00000000 00000000 00000038 805ebea4 00000018 35206465
[161950.702852]         00000000 00001f79 00000000 746d203a 00000000 86a09c10 00000000 86857060
[161950.719645]         00000009 00000014 00000004 00000002 00000008 8035cbf4 0000000c 8082000c
[161950.736438]         ...
[161950.741467] Call Trace:
[161950.741481] [<8007e0d8>] 0x8007e0d8
[161950.753651] [<805ebea4>] 0x805ebea4
[161950.760776] [<86857060>] 0x86857060 [mac80211@571e22cd+0x7d550]
[161950.772720] [<8035cbf4>] 0x8035cbf4
[161950.779832] [<8000b05c>] 0x8000b05c
[161950.786938] [<8000b064>] 0x8000b064
[161950.794040] [<805d17d8>] 0x805d17d8
[161950.801143] [<8007ab80>] 0x8007ab80
[161950.808247] [<8002c3d0>] 0x8002c3d0
[161950.815359] [<86857060>] 0x86857060 [mac80211@571e22cd+0x7d550]
[161950.827303] [<8002c478>] 0x8002c478
[161950.834428] [<86857060>] 0x86857060 [mac80211@571e22cd+0x7d550]
[161950.846387] [<86857248>] 0x86857248 [mac80211@571e22cd+0x7d550]
[161950.858353] [<86857328>] 0x86857328 [mac80211@571e22cd+0x7d550]
[161950.870310] [<8041e98c>] 0x8041e98c
[161950.877444] [<86ee2a24>] 0x86ee2a24 [mt76x02_lib@844a4414+0x9de0]
[161950.889744] [<86f20530>] 0x86f20530 [mt76@633da2ff+0x96e0]
[161950.900859] [<86f20530>] 0x86f20530 [mt76@633da2ff+0x96e0]
[161950.911944] [<86ee5004>] 0x86ee5004 [mt76x02_lib@844a4414+0x9de0]
[161950.924238] [<86f20530>] 0x86f20530 [mt76@633da2ff+0x96e0]
[161950.935334] [<86f205d4>] 0x86f205d4 [mt76@633da2ff+0x96e0]
[161950.946418] [<8004b6d4>] 0x8004b6d4
[161950.953524] [<805ed83c>] 0x805ed83c
[161950.960642] [<8004bbe4>] 0x8004bbe4
[161950.967747] [<8004baa4>] 0x8004baa4
[161950.974859] [<8004baa4>] 0x8004baa4
[161950.981968] [<8004baa4>] 0x8004baa4
[161950.989071] [<80006718>] 0x80006718
[161950.996178] 
[161950.999794] ---[ end trace b2767dd2d11e78c7 ]---

[ 688.355055] mt76x2e 0000:01:00.0: Build: 1
[ 688.359252] mt76x2e 0000:01:00.0: Build Time: 201706151016____
[ 688.380175] mt76x2e 0000:01:00.0: Firmware running!
[ 688.388375] ieee80211 phy1: Hardware restart was requested
[ 695.757410] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00

Sorry google translate.
there were both errors, nothing helped until I removed the router away from the HDD. Three days the log is clean. Perhaps you have interference near the router.

Mushoz, 2022-01-31:
@nbd168 just replied in a duplicate issue with a potential fix: https://github.com/openwrt/mt76/issues/457#issuecomment-1025779340

Anyone interested in giving it a go? :)

gstammw, 2022-04-05:
Issue persists on Xiaomi Mi Router 4a Gigabit Edition running 21.02.2.
Very annoying as  wifi-connectivity on 5Ghz gets interrupted. I did not see the problem on phy0 which is 2.4 GHz, but there is less utilization of the 2.4 GHz-band in my environment anyway.

Full log, error-message at the end, occuring multiple times:

> [    0.000000] Linux version 5.4.179 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16495-bf0c965af0)) #0 SMP Wed Feb 16 20:29:10 2022
> [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
> [    0.000000] printk: bootconsole [early0] enabled
> [    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
> [    0.000000] MIPS: machine is Xiaomi Mi Router 4A Gigabit Edition
> [    0.000000] Initrd not found or empty - disabling initrd
> [    0.000000] VPE topology {2,2} total 4
> [    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
> [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
> [    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
> [    0.000000] Zone ranges:
> [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
> [    0.000000]   HighMem  empty
> [    0.000000] Movable zone start for each node
> [    0.000000] Early memory node ranges
> [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
> [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
> [    0.000000] On node 0 totalpages: 32768
> [    0.000000]   Normal zone: 288 pages used for memmap
> [    0.000000]   Normal zone: 0 pages reserved
> [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
> [    0.000000] percpu: Embedded 14 pages/cpu s26768 r8192 d22384 u57344
> [    0.000000] pcpu-alloc: s26768 r8192 d22384 u57344 alloc=14*4096
> [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
> [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
> [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
> [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
> [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
> [    0.000000] Writing ErrCtl register=00045800
> [    0.000000] Readback ErrCtl register=00045800
> [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
> [    0.000000] Memory: 120968K/131072K available (6095K kernel code, 204K rwdata, 748K rodata, 1260K init, 238K bss, 10104K reserved, 0K cma-reserved, 0K highmem)
> [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
> [    0.000000] rcu: Hierarchical RCU implementation.
> [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
> [    0.000000] NR_IRQS: 256
> [    0.000000] random: get_random_bytes called from 0x806e5a3c with crng_init=0
> [    0.000000] CPU Clock: 880MHz
> [    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
> [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
> [    0.000008] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
> [    0.007804] Calibrating delay loop... 583.68 BogoMIPS (lpj=1167360)
> [    0.041991] pid_max: default: 32768 minimum: 301
> [    0.046755] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
> [    0.053978] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
> [    0.064251] rcu: Hierarchical SRCU implementation.
> [    0.069640] smp: Bringing up secondary CPUs ...
> [    3.760990] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
> [    3.761001] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
> [    3.761012] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
> [    3.761110] CPU1 revision is: 0001992f (MIPS 1004Kc)
> [    0.102353] Synchronize counters for CPU 1: done.
> [    3.828273] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
> [    3.828282] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
> [    3.828289] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
> [    3.828347] CPU2 revision is: 0001992f (MIPS 1004Kc)
> [    0.166988] Synchronize counters for CPU 2: done.
> [    3.883577] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
> [    3.883584] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
> [    3.883592] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
> [    3.883655] CPU3 revision is: 0001992f (MIPS 1004Kc)
> [    0.224840] Synchronize counters for CPU 3: done.
> [    0.254715] smp: Brought up 1 node, 4 CPUs
> [    0.263174] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
> [    0.272846] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[... 199 lines cut ...]
> [  110.709992] mt76x2e 0000:01:00.0: Build: 1
> [  110.714132] mt76x2e 0000:01:00.0: Build Time: 201507311614____
> [  110.735333] mt76x2e 0000:01:00.0: Firmware running!
> [  110.743455] ieee80211 phy1: Hardware restart was requested
> [  114.792415] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
> [  114.797923] mt76x2e 0000:01:00.0: Build: 1
> [  114.802078] mt76x2e 0000:01:00.0: Build Time: 201507311614____
> [  114.823287] mt76x2e 0000:01:00.0: Firmware running!
> [  114.831526] ieee80211 phy1: Hardware restart was requested
[... cut ...]
> [  124.713928] mt76x2e 0000:01:00.0: Build: 1
> [  124.718097] mt76x2e 0000:01:00.0: Build Time: 201507311614____
> [  124.739172] mt76x2e 0000:01:00.0: Firmware running!
> [  124.747342] ieee80211 phy1: Hardware restart was requested
> [  127.768258] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
> [  127.773762] mt76x2e 0000:01:00.0: Build: 1
> [  127.777917] mt76x2e 0000:01:00.0: Build Time: 201507311614____
> [  127.799153] mt76x2e 0000:01:00.0: Firmware running!
> [  127.807349] ieee80211 phy1: Hardware restart was requested

lukasz1992, 2022-04-05:
21.02 branch has not updated mt76. Try 22.03 or master branch, should be better

imago54, 2022-06-15:
Issue is even more frequent on OpenWrt 22.03.0-rc4 (Xiaomi Mi Router 3G). 5GHz (phy1) doesn't survive even one full speedtest.net run.

[ 122.054413] mt76x2e 0000:02:00.0: MCU message 1f (seq 1) timed out
[ 122.115646] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[ 122.121174] mt76x2e 0000:02:00.0: Build: 1
[ 122.125620] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[ 122.154440] mt76x2e 0000:02:00.0: Firmware running!
[ 122.164688] ieee80211 phy1: Hardware restart was requested

HiGarfield, 2022-06-16:
try this

https://github.com/HiGarfield/mt76/commit/8b37f81861c3b5acc62ecc2f8b0a2795ea3b1a59

lukasz1992, 2022-06-25:
Or this https://github.com/openwrt/mt76/commit/ba39ed3b44f1a0b3bd87c72452a261908afba754

Kuznetsovkit, 2022-08-31:
Worked on this problem and may be found something that can help fix it.
The essence of the problem is that after several resets (`MCU message 1f (seq 1) timed out`), all traffic from the client stops being transmitted. Except for OSPF messages. The problem is fixed - service network restart.

There are several boards with mt7621 and with two wifi chips mt7612e on them. Wlan configured as adhoc, 2.4GHz, width 40 MHz. I use quagga (1.2.4) to set up routing and testing bandwidth with iperf3. If configure mesh, then it works longer (~12h) before the traffic is lose than adhoc. That's why i use adhoc. Ap also loses traffic.

Client problem when testing iperf3 looks like this:

```
[  5] 175.00-176.00 sec  4.91 MBytes  41.2 Mbits/sec    0    438 KBytes
[  5] 176.00-177.00 sec  5.53 MBytes  46.4 Mbits/sec    0    467 KBytes
[  5] 177.00-178.00 sec  5.65 MBytes  47.4 Mbits/sec    0    508 KBytes
[  5] 178.00-179.00 sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes
[  980.125497] mt76x2e 0000:02:00.0: MCU message 1f (seq 5) timed out
[  980.170990] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  980.176549] mt76x2e 0000:02:00.0: Build: 1
[  980.180725] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[  980.201517] mt76x2e 0000:02:00.0: Firmware running!
[  980.209639] ieee80211 phy1: Hardware restart was requested
[  5] 179.00-180.00 sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes
[  5] 180.00-181.00 sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes
[  5] 181.00-182.00 sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes
[  5] 182.00-183.00 sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes
```

After the traffic is lose, i ping from the server:
tcpdump server log (10.87.254.44):

```
05:49:23.260947 IP 10.87.254.43 > 224.0.0.5: OSPFv2, Hello, length 48
05:49:23.494748 IP 10.87.254.44 > 224.0.0.5: OSPFv2, Hello, length 48
05:49:23.807103 ARP, Request who-has 10.87.254.43 tell 10.87.255.44, length 28
05:49:24.830398 ARP, Request who-has 10.87.254.43 tell 10.87.254.44, length 28
05:49:25.270844 IP 10.87.254.43 > 224.0.0.5: OSPFv2, Hello, length 48
05:49:25.498754 IP 10.87.254.44 > 224.0.0.5: OSPFv2, Hello, length 48
05:49:25.854409 ARP, Request who-has 10.87.254.43 tell 10.87.254.44, length 28
```

tcpdump client log (10.87.254.43):

```
02:57:41.176570 IP 10.87.254.43 > 224.0.0.5: OSPFv2, Hello, length 48
02:57:41.401442 IP 10.87.254.44 > 224.0.0.5: OSPFv2, Hello, length 48
02:57:41.802756 ARP, Request who-has 10.87.254.43 tell 10.87.254.44, length 28
02:57:41.802893 ARP, Reply 10.87.254.43 is-at 92:a0:37:00:0d:00 (oui Unknown), length 28
02:57:42.825348 ARP, Request who-has 10.87.254.43 tell 10.87.254.44, length 28
02:57:42.825475 ARP, Reply 10.87.254.43 is-at 92:a0:37:00:0d:00 (oui Unknown), length 28
02:57:43.186536 IP 10.87.254.43 > 224.0.0.5: OSPFv2, Hello, length 48
02:57:43.405492 IP 10.87.254.44 > 224.0.0.5: OSPFv2, Hello, length 48
02:57:44.872640 ARP, Request who-has 10.87.254.43 tell 10.87.254.44, length 28
02:57:44.872777 ARP, Reply 10.87.254.43 is-at 92:a0:37:00:0d:00 (oui Unknown), length 28
```

That is, the client receives requests from the server, sends a reply, but the reply does not reach the server. Only OSPF messages from the client are visible. Statistics output does not show an increase in transmitted or droped packets. Only transmitted packets from quagga are counted.

I also tracked the receipt of skb from the queue in the mt76_txq_dequeue function (tx.c file). Only skb containing ospf messages can be retrieved from the queue. Although before that i received ping messages.
It turns out that tcpdump see messages, but driver cannot get them from queue. And why this is happening, I can not understand. And can this help fix the problem.
There should be no problems with the board, because i tested driver from mediatek and there iperf3 testing took about 3 days, until the place in MgmtRing ran out.
Does anyone have any ideas what can be checked or fixed?

Testing:
in Openwrt v21.02.3 with mt76 version [678071e](https://github.com/openwrt/mt76/commit/678071ef70297b7264661c356ddb3c8cf7f3c87b)
in Openwrt v22.03.0-rc6 with mt76 version [5ec78e1](https://github.com/openwrt/mt76/commit/5ec78e1ec43d1e39edfea1efb9fd4541fa004af0)

spamsool, 2024-06-17:
I am still having this problem of firmware resets on D-Team Newifi D2 (Newifi3) running latest official OpenWRT 23.05.3
Is there any solution available?

olnet93, 2024-09-15:
I have a similar problem. 
The problem is only when connected via 5GHz to a laptop with an Intel(R) Dual Band Wireless-N 7260 network card. 
There is no problem with all the phones I have. 

The router:
Model Netgear WAC104
Architecture MediaTek MT7621 ver:1 eco:3Firmware Version OpenWrt 23.05.4 r24012-d8dd03c46f / LuCI openwrt-23.05 branch git-24.086.45142-09d5a38
Kernel Version 5.15.162

In the logs:
Sun Sep 15 12:08:14 2024 kern.info kernel: [799643.963789] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Sun Sep 15 12:08:14 2024 kern.info kernel: [799643.975275] mt76x2e 0000:01:00.0: Build: 1
Sun Sep 15 12:08:14 2024 kern.info kernel: [799643.983652] mt76x2e 0000:01:00.0: Build Time: 201607111443____
Sun Sep 15 12:08:14 2024 kern.info kernel: [799644.012715] mt76x2e 0000:01:00.0: Firmware running!
Sun Sep 15 12:08:14 2024 kern.info kernel: [799644.024814] ieee80211 phy1: Hardware restart was requested
Sun Sep 15 12:09:17 2024 kern.info kernel: [799706.933174] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Sun Sep 15 12:09:17 2024 kern.info kernel: [799706.944356] mt76x2e 0000:01:00.0: Build: 1
Sun Sep 15 12:09:17 2024 kern.info kernel: [799706.952981] mt76x2e 0000:01:00.0: Build Time: 201607111443____
Sun Sep 15 12:09:17 2024 kern.info kernel: [799706.982141] mt76x2e 0000:01:00.0: Firmware running!
Sun Sep 15 12:09:17 2024 kern.info kernel: [799706.993605] ieee80211 phy1: Hardware restart was requested
Sun Sep 15 12:09:53 2024 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 80:19:34:4c:5f:b3
Sun Sep 15 12:09:58 2024 daemon.info hostapd: phy1-ap0: STA 80:19:34:4c:5f:b3 IEEE 802.11: authenticated
Sun Sep 15 12:09:58 2024 daemon.info hostapd: phy1-ap0: STA 80:19:34:4c:5f:b3 IEEE 802.11: associated (aid 3)
Sun Sep 15 12:09:58 2024 daemon.notice hostapd: phy1-ap0: STA-OPMODE-SMPS-MODE-CHANGED 80:19:34:4c:5f:b3 off
Sun Sep 15 12:09:58 2024 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 80:19:34:4c:5f:b3 auth_alg=open
Sun Sep 15 12:09:58 2024 daemon.info hostapd: phy1-ap0: STA 80:19:34:4c:5f:b3 WPA: pairwise key handshake completed (RSN)
Sun Sep 15 12:09:58 2024 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 80:19:34:4c:5f:b3