openwrt/mt76 #446: ETSI EN 300328 certification - Adaptivity requirements (Acsip AI7688H / Mediatek MT7688)

knn, 2020-09-16:
We are currently developing an industrial IoT device using the Acsip AI7688H Wifi module which is based on a Mediatek MT7688 SoC. We have created our own build of OpenWrt 19.07.2 which uses "MediaTek LinkIt Smart 7688" as the target profile. This OpenWrt build uses the open-source mt76 WiFi driver (lsmod shows 'mt76' / 'mt7603e' drivers loaded.)

We now want to have our device certified to the requirements of ETSI EN 300328. Because our device uses a transmit power larger than 10 dBm, the ETSI standard requires it to be an adaptive device, i.e. it needs to employ an adaptivity mechanism such as DAA (Detect and Avoid) or LBT (Listen before Talk; load- or frame-based) if the Wifi channel is in use.

The vanilla mt7603e driver does not apply such mechanisms in the standard configuration, so we have failed the certification. However, we've noticed that the driver contains various functions regarding EDCCA (Energy Detect / Clear Channel Assessment) and we've enabled this functionality using the patch 001-mt76-mt7603-driver-enable-adaptivity.patch (listed below).

Our testing laboratory has repeated the adaptivity test with this patched driver and has confirmed that the Wifi interface does indeed behave differently if the Wifi channel is in use; however, it still uses a max duty cycle of 12% in an evaluation window of 50 ms (measured during continous transmission using iperf3).

The evaluation of the duty cycle depends on the type of packet:
- If Short Control Signaling (SCS) transmissions are used, a duty cycle of 10% is allowed. (The testing laboratory assumes that the measured packets are SCS transmissions.)
- If SCS is not used, a duty cycle of 0% is allowed. The device has to wait until the channel is not used anymore before transmitting again.

The testing laboratory recommended to check if we can influence these SCS transmissions in software, but we couldn't find anything related to it in the code of the mt7603e driver (there are matches in different drivers, though), or in the mac80211 interface of the kernel. Also, we've tried to understand the EDCCA functions in detail; however, because we do not have the complete datasheet of the Wifi peripheral, this task is unfeasible.

Can someone help us with the adaptivity functionality?

Thanks in advance for any help.

**001-mt76-mt7603-driver-enable-adaptivity.patch** 
``` Index: mt76-2020-03-10-08054d5a/mt7603/init.c
===================================================================
--- mt76-2020-03-10-08054d5a.orig/mt7603/init.c
+++ mt76-2020-03-10-08054d5a/mt7603/init.c
@@ -543,6 +543,9 @@ int mt7603_register_device(struct mt7603
 	dev->sensitivity_limit = 28;
 	dev->dynamic_sensitivity = true;
 
+	dev->ed_monitor_enabled = true;
+	dev_info(dev->mt76.dev, "Enabled adaptivity using CCA-ED\n");
+
 	ret = mt7603_init_hardware(dev);
 	if (ret)
 		return ret;
```

rettichschnidi, 2020-09-17:
Non-technical input: Seems like you are in the same boat as I was a while ago when our product (MT7688 running  the mt76 driver) failed the Wi-Fi Alliance certification. What worked very well for me was getting @nbd168 hired via https://embedd.com/. All problems have been resolved within a few weeks.

knn, 2020-09-21:
@rettichschnidi Thanks for the tip! We will contact the company as you suggested.

knn, 2020-12-07:
We successfully passed the ETSI certification with the following patches by @nbd168:

- 35e3cd1db479d26a5bdafcc4ee90c4a1c18f4402 mt76: mt7603: fix ED/CCA monitoring with single-stream devices
- [openwrt/mac80211: minstrel: remove deferred sampling code](https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/kernel/mac80211/patches/subsys/334-mac80211-minstrel-remove-deferred-sampling-code.patch;h=e3a877fb94bb94abb5708885120e608d6690f15c;hb=HEAD)
- [openwrt/mac80211: minstrel: fix tx status processing corner case](https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/kernel/mac80211/patches/subsys/335-mac80211-minstrel-fix-tx-status-processing-corner-ca.patch;h=b1c339a1e60b8d32150ab4c658a76a1cd63babc5;hb=HEAD)

closed 2020-12-07 by knn