openwrt/mt76 #448: K2P with DBDC stuck at "link up"

xuefer, 2020-09-19:
Device Info: https://openwrt.org/toh/hwdata/phicomm/phicomm_k2p
```
WLAN Hardware: MediaTek MT7615D
WLAN 2.4GHz: b/g/n
WLAN 5.0GHz: a/n/ac
```

```
root@OpenWrt:/# dmesg | grep 7615
[    1.806137] pci 0000:01:00.0: [14c3:7615] type 00 class 0x000280
```

since changeset 99804560372bfb830de28a41ecc4736747c2ae60 we don't need ```echo 1 > /sys/kernel/debug/ieee80211/phy0/mt76/dbdc``` any more, but it's not enough for K2P MT7615D

```ip a``` command (and all script to bring wlan up) freeze during/after a clean reboot
this is what happen after loading mt7615e module, with a few extra log added by me
```
[   14.406743] mt7621-pci 1e140000.pcie: bus=1 slot=0 irq=23
[   14.417549] pci 0000:00:00.0: enabling device (0004 -> 0007)
[   14.428838] mt7615e 0000:01:00.0: enabling device (0000 -> 0002)
[   14.463379] mt7615_register_device: has_2g=1 n_rates=12
[   14.819124] mt7615e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20180518100604a
[   14.819124] 
[   14.824876] NET: Registered protocol family 24
[   14.866768] kmodloader: done loading kernel modules from /etc/modules.d/*
[   15.389394] mt7615e 0000:01:00.0: N9 Firmware Version: _reserved_, Build Time: 20200814163649
[   15.448834] mt7615e 0000:01:00.0: CR4 Firmware Version: _reserved_, Build Time: 20190121161307
[   16.368333] zram0: detected capacity change from 0 to 61865984
[   21.002884] mtk_soc_eth 1e100000.ethernet eth0: Link is Down
[   21.031289] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[   21.047348] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   21.067379] mtk_soc_eth 1e100000.ethernet eth0: Link is Down
[   21.228767] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[   21.244892] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   21.248259] mt7530 mdio-bus:1f lan1: configuring for phy/gmii link mode
[   21.275728] 8021q: adding VLAN 0 to HW filter on device lan1
[   21.290667] br-wan: port 1(lan1) entered blocking state
[   21.301162] br-wan: port 1(lan1) entered disabled state
[   21.312526] device lan1 entered promiscuous mode
[   21.321792] device eth0 entered promiscuous mode
[   21.348247] mt7530 mdio-bus:1f lan2: configuring for phy/gmii link mode
[   21.362221] 8021q: adding VLAN 0 to HW filter on device lan2
[   21.377620] br-wan: port 2(lan2) entered blocking state
[   21.388201] br-wan: port 2(lan2) entered disabled state
[   21.400190] device lan2 entered promiscuous mode
[   21.418789] mt7530 mdio-bus:1f lan3: configuring for phy/gmii link mode
[   21.432652] 8021q: adding VLAN 0 to HW filter on device lan3
[   21.447444] br-wan: port 3(lan3) entered blocking state
[   21.458071] br-wan: port 3(lan3) entered disabled state
[   21.469990] device lan3 entered promiscuous mode
[   21.489244] mt7530 mdio-bus:1f lan4: configuring for phy/gmii link mode
[   21.503200] 8021q: adding VLAN 0 to HW filter on device lan4
[   21.517682] br-wan: port 4(lan4) entered blocking state
[   21.528245] br-wan: port 4(lan4) entered disabled state
[   21.540657] device lan4 entered promiscuous mode
[   21.556942] mt7530 mdio-bus:1f wan: configuring for phy/gmii link mode
[   21.570619] 8021q: adding VLAN 0 to HW filter on device wan
[   21.585077] br-wan: port 5(wan) entered blocking state
[   21.595446] br-wan: port 5(wan) entered disabled state
[   21.607353] device wan entered promiscuous mode
[   23.072402] mt7615 DBDC: 1
[   23.077930] mt76_check_sband: band=0 n_channels=14
[   23.087589] mt76_check_sband: band=1 n_channels=27
[   23.097165] mt7615_register_ext_phy
[   25.610530] mt7530 mdio-bus:1f lan4: Link is Up - 1Gbps/Full - flow control rx/tx
```

```
  541 root         0 SWN  [jffs2_gcd_mtd6]
  626 root      1252 S    /sbin/ubusd
  627 root      1256 S    /bin/ash --login
  644 root      1044 S    /sbin/urngd
  682 root         0 IW   [kworker/3:2-eve]
  778 root         0 IW<  [cfg80211]
  795 root         0 SW   [mt76-tx phy0]
  800 root         0 IW<  [kworker/u9:1-na]
  995 root      1264 S    /sbin/logd -S 64
 1032 root      2080 S    /sbin/rpcd -s /var/run/ubus.sock -t 30
 1074 root         0 IW<  [kworker/1:2H]
 1127 dnsmasq   1436 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c
 1168 root      1148 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p
 1235 root      1900 D    /usr/sbin/hostapd -s -g /var/run/hostapd/global
 1236 root      1872 S    /usr/sbin/wpa_supplicant -n -s -g /var/run/wpa_suppl
 1277 root      1696 S    /sbin/netifd
 1312 root      1488 S    /usr/sbin/odhcpd
 1351 root      1384 S    /bin/sh /etc/rc.common /etc/rc.d/S50qos boot
 1447 root         0 DW   [kworker/0:2+eve]
 1481 root      1292 S    /bin/sh /sbin/hotplug-call iface
 1482 root      1616 S    /bin/sh ./mac80211.sh mac80211 setup radio0 {"config
 1543 root      1292 S    /bin/sh /sbin/hotplug-call iface
 1546 root      1372 S    /bin/sh /etc/rc.common /etc/init.d/qos enabled
 1583 root      1248 S    flock 1000
 1623 root         0 IW   [kworker/3:3-eve]
 1911 root      1248 S    /bin/sh /usr/bin/qos-start
 1913 root      1248 S    /bin/sh /usr/bin/qos-stop
 1915 root      1248 S    /bin/sh /usr/bin/qos-stop
 1917 root      1624 D    tc qdisc show
 1919 root      1248 S    grep -E (hfsc|ingress)
 1921 root      1252 S    awk {print $5}
 1987 root      1248 D    ip link set dev wlan0 up
 1988 root         0 IW   [kworker/0:3-eve]
 1989 root      1248 R    ps
```

If I disable DBDC (in source code) in mt*.ko it won't stuck at all, and rmmod + reloading DBDC enabled mt*.ko won't stuck until after reboot as reported above

kar200, 2020-09-25:
Same issue here. Yours seems to be worded better. Not sure which ones the devs want to keep.

https://github.com/openwrt/mt76/issues/440

kar200, 2021-01-15:
@xuefer 

I compiled today with the latest changes and it seems the issue has been fixed (for me anyway).

I still have to add (wifi down + wifi up) at boot for both cards to come up.

kar200, 2021-01-17:
After changing settings in the wireless file it still freezes.