openwrt/mt76 #450: Packets dropped on mt7615 for A-MSDU transmissions

drandyhaas, 2020-09-21:
Hi, 
In trying to figure out what's going on in issue 387: https://github.com/openwrt/mt76/issues/387
I found that packets are dropped by the mt7615 driver (at least on my winstars 538a6) when sent from another router as an A-MSDU aggregated packet. Large pings (with -s 2000 for instance) are replied to by the mt7615 router when sent as "fragmented ip" packets, but not when sent as "a-msdu" packets. See wireshark sniff below (from a 3rd machine monitoring traffic in the middle).

[image]

Is A-MSDU (and A-MPDU) supported by mt76? At least for reception (rx)? The driver advertises that it is (in "iw phy0 info").

nbd168, 2020-09-21:
A-MSDU itself is supported. Could you please send me a .pcap of a monitor mode capture that shows the issue?

drandyhaas, 2020-09-21:
Here is a pcap...

[ath2mt.pcap.zip](https://github.com/openwrt/mt76/files/5255583/ath2mt.pcap.zip)

nbd168, 2020-09-21:
What kind of router and firmware are you using on the A-MSDU transmitter side?

drandyhaas, 2020-09-21:
On the transmitter side it's a linksys ea8300, running openwrt 19.07.3, with ath10k-ct drivers.
From the dmesg: "ath10k 4.19 driver, optimized for CT firmware"

nbd168, 2020-09-21:
Are you using 802.11s? Or is this a normal 4-address mode (WDS) link?

drandyhaas, 2020-09-21:
It's 802.11s. No encryption (for testing purposes).

nbd168, 2020-09-21:
I just spent some time reading the mac80211 A-MSDU parsing code and the 802.11 standard. If I'm reading things correctly, it seems that there may be a 802.11s specific bug in the mac80211 A-MSDU decap code. I'll work on a patch for you to test.

drandyhaas, 2020-09-21:
Ok, I look forward to giving it a try. Thanks!

drandyhaas, 2020-09-23:
Antin else I can do to help here? Maybe point me where to look in the code?

nbd168, 2020-09-23:
The problem is that the bug comes from a wrong assumptions in the rx processing code. The code assumes that the mesh header comes after the 802.11 header before A-MSDU subframes, when in fact the mesh header is part of each individual subframe.
Fixing this flawed assumption will require some significant rework. I will work on it, but I need more time to get it done.

drandyhaas, 2020-09-23:
Ok, thanks! Just let me know if there's anything else I can do to help.

drandyhaas, 2020-09-30:
Any progress?

drandyhaas, 2020-10-08:
Is there some estimate of the time this could take to fix? If it's going to be long I'll have to try to hack it myself.. Thanks.

drandyhaas, 2020-10-14:
Can you please at least point me to the place in the code where this is done? As far as I see, the 802.11s decap is all fine in the standard Linux code, so I'm not sure where to look in the mt76 code. Thanks.

nbd168, 2020-10-15:
The issue is in the Linux stack rx code, not mt76.
ieee80211_data_to_8023_exthdr in net/wireless/util.c assumes that in mesh mode, the mesh header immediately follows the 802.11 header and applies to all subframes.
According to the 802.11-2016 standard, chapter 9.3.2.2.2, figure 9-56, the mesh header is part of each A-MSDU subframe.
Taking this into account will need some significant rework of the rx path and cfg80211 utility functions, including the mesh forwarding code.
The A-MSDU decap code needs to be adjusted to leave the mesh header in place, and mesh forwarding needs to happen later in the rx processing chain.

drandyhaas, 2020-10-15:
Thanks very much for the info. 
What confuses me here is how other drivers, like ath10k etc manage to work just fine, if the problem is upstream in a piece of shared linux decap code. I guess mt76 is the only ones using this piece of linux code?

nbd168, 2020-10-15:
ath10k has hardware A-MSDU decapsulation, so it bypasses the software part.

drandyhaas, 2020-10-15:
I see. Does anyone have the datahseet (in English prefereably) of the MT7615? I'd be surprised if it doesn't have hardware offload capabilities too.

nbd168, 2020-10-15:
MTK told me it can't offload Mesh A-MSDU.

drandyhaas, 2020-10-15:
OK. Note that mesh forwarding at the 802.11s level is not so important to get right for now. Usually people use BATMAN to do forwarding at layer 2 ethernet, and explicitly turn off mesh forwarding. So it may be possible to get the basic mesh a-msdu working fine for now without getting mesh forwarding working.

drandyhaas, 2020-11-18:
Any hope of finding time to work on this? I could try, but it would be challenging...

drandyhaas, 2020-11-30:
Until a real fix for this is available, it works OK to simply disable mstu.
I comment out this line
`ieee80211_hw_set(hw, SUPPORTS_AMSDU_IN_AMPDU);`
here
https://github.com/openwrt/mt76/blob/master/mac80211.c#L307
and I get reasonable performance of about 150Mb/s in both directions over 80211s.

dhewg, 2021-05-12:
> Until a real fix for this is available, it works OK to simply disable mstu.
> I comment out this line
> `ieee80211_hw_set(hw, SUPPORTS_AMSDU_IN_AMPDU);`
> here
> https://github.com/openwrt/mt76/blob/master/mac80211.c#L307
> and I get reasonable performance of about 150Mb/s in both directions over 80211s.

Thanks for that!

If anyone cares, here's a patch to apply to openwrt to do exactly that:
[0001-mt76-HACK-disable-A-MSDU.patch.txt](https://github.com/openwrt/mt76/files/6466494/0001-mt76-HACK-disable-A-MSDU.patch.txt)

Without patch:
```
[  5] local 192.168.0.2 port 5201 connected to 192.168.0.1 port 41296
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   501 KBytes  4.10 Mbits/sec                  
[  5]   1.00-2.01   sec   600 KBytes  4.88 Mbits/sec                  
[  5]   2.01-3.01   sec   679 KBytes  5.56 Mbits/sec                  
[  5]   3.01-4.01   sec   455 KBytes  3.73 Mbits/sec                  
[  5]   4.01-5.00   sec   513 KBytes  4.23 Mbits/sec                  
[  5]   5.00-6.00   sec   492 KBytes  4.03 Mbits/sec                  
[  5]   6.00-7.00   sec   547 KBytes  4.48 Mbits/sec                  
[  5]   7.00-8.01   sec   615 KBytes  5.00 Mbits/sec                  
[  5]   8.01-9.01   sec   496 KBytes  4.07 Mbits/sec                  
[  5]   9.01-10.01  sec   532 KBytes  4.36 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.01  sec  5.30 MBytes  4.45 Mbits/sec                  receiver
```

With patch:
```
[  5] local 192.168.0.2 port 5201 connected to 192.168.0.1 port 41300
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  5.44 MBytes  45.5 Mbits/sec                  
[  5]   1.00-2.00   sec  6.12 MBytes  51.4 Mbits/sec                  
[  5]   2.00-3.00   sec  7.08 MBytes  59.4 Mbits/sec                  
[  5]   3.00-4.00   sec  4.75 MBytes  39.8 Mbits/sec                  
[  5]   4.00-5.00   sec  7.24 MBytes  60.7 Mbits/sec                  
[  5]   5.00-6.00   sec  7.38 MBytes  62.0 Mbits/sec                  
[  5]   6.00-7.00   sec  7.23 MBytes  60.7 Mbits/sec                  
[  5]   7.00-8.00   sec  7.50 MBytes  62.9 Mbits/sec                  
[  5]   8.00-9.00   sec  7.65 MBytes  64.2 Mbits/sec                  
[  5]   9.00-10.00  sec  7.89 MBytes  66.2 Mbits/sec                  
[  5]  10.00-10.07  sec   557 KBytes  69.8 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.07  sec  68.8 MBytes  57.4 Mbits/sec                  receiver
```

ryddim, 2021-12-03:
Hi, I am a noob and I do not know how to patch the openwrt firmware or comment out the line as mentioned above. Could you please shed some light on how this can be done. I have also been facing the same speed issues by using 802.11s.

dhewg, 2022-09-28:
@blocktrron came up with a cleaner way to work around the issue, attached a patch for the openwrt tree:
[0001-mac80211-mask-nested-A-MSDU-support-for-mesh.patch.txt](https://github.com/openwrt/mt76/files/9661690/0001-mac80211-mask-nested-A-MSDU-support-for-mesh.patch.txt)
I'm successfully running this on a mesh setup with one mt76 enabled host and one without it since july.

Djfe, 2023-05-09:
He's working on it 🥳
https://www.spinics.net/lists/linux-wireless/msg232408.html

shown19, 2023-06-16:
> Hi, I am a noob and I do not know how to patch the openwrt firmware or comment out the line as mentioned above. Could you please shed some light on how this can be done. I have also been facing the same speed issues by using 802.11s.

Hi may I know, if you managed to patch yours? I wanted to try patching the mac80211.c also in hope that this will fix the packet loss and large ping issue in my MT7621 Chipset, device: Newifi D2. I already watched tutorial in Youtube but only in how to build the firmware not the applying the patch method. Would be glad if someone could help me, I'm willing to learn and be guided. Thank you.

sayap, 2023-06-16:
@shown19 can try this:
```
cd /path/to/openwrt/source/
wget https://github.com/openwrt/mt76/files/9661690/0001-mac80211-mask-nested-A-MSDU-support-for-mesh.patch.txt -O package/kernel/mt76/patches/0001-mac80211-mask-nested-A-MSDU-support-for-mesh.patch
```
This will place the patch shared above under package/kernel/mt76/patches/. You can then run `make` to build the firmware again.

shown19, 2023-06-16:
> 

Thank you for the response man but noob question, do you need to create this directory first /path/to/openwrt/source/ , right?

I did create the directory manually, and then invoke the wget command but I received, "No such file or directory" . Sorry man, I'm still learning though I already have an idea by following this guide: https://openwrt.org/docs/guide-developer/toolchain/use-patches-with-buildsystem but seems like I failed or need to understand it deeper.

shown19, 2023-06-16:
> @shown19 can try this:
>
> ```
> cd /path/to/openwrt/source/
> wget https://github.com/openwrt/mt76/files/9661690/0001-mac80211-mask-nested-A-MSDU-support-for-mesh.patch.txt -O [...]

Thank you for the response man but noob question, do you need to create this directory first /path/to/openwrt/source/ , right?

I did create the directory manually, and then invoke the wget command but I received, "No such file or directory" . Sorry man, I'm still learning though I already have an idea by following this guide: https://openwrt.org/docs/guide-developer/toolchain/use-patches-with-buildsystem but seems like I failed or need to understand it deeper.

Djfe, 2023-06-16:
> Thank you for the response man but noob question, do you need to create this directory first `/path/to/openwrt/source/` , right?

the path is an example. just make sure the working directory of your terminal is in the root of the openwrt folder that you cloned with git (or extracted from a zip)

cd into that folder wherever it is (probably a sub folder of your home directory)
(cd = change directory)

there is probably no folder that is called source (yours might've been empty)
be in the folder that has the subfolders/file structure like in the openwrt repository (and a folder called packages)

shown19, 2023-06-18:
Hi guys it's me again, may I know if how to verify if the patch is certainly applied or working on my build? Just wanna make sure that the changes is actually loaded other than invoking `cat /sys/kernel/debug/ieee80211/phy0/mwlwifi/tx_amsdu` since this doesn't exist/work on my device.

kwanchen, 2023-07-01:
> 

Any luck in applying the patch and trying it out? Does it fix the packet loss and ping issues?

Also are you trying to use 802.11s between two MT7621 devices? I'm noticing tremendous packet loss through iperf3 between two mt7621 devices using the MASTER snapshots. Heard its fixed in kernel 6.1, but most of openwrt is still on 5.15.

Djfe, 2023-07-01:
mt76 (mt7915 in particular) was hit by a few bugs recently. should be solved by now (in master), but I don't know when you pulled last (might be solved by the next snapshot build)

Djfe, 2023-07-01:
oops, kind of left out what I wanted to say:
blocktrron's patch fixes the packetloss (so yes to your question) but it's also not ideal performance wise, so we still need the actual fix/patch from nbd

my guess (last post) was that your issues weren't related to this bug, but issues with the mt76 update in master, that were quickly resolved because many people noticed right away :)
one of those bugs was related to meshing
https://github.com/openwrt/openwrt/issues/13003

kwanchen, 2023-07-01:
> openwrt/openwrt#13003 <https://github.com/openwrt/openwrt/issues/13003>

My last build was 2 weeks ago, kernel 5.15.17. Will try a new sometime this
weekend. Thanks for letting me know. Not being too knowledgeable about this
stuff, I probably erroneously attributed the poor mesh performance to this
bug. I was never able to get it to work so I've been relying on WDS, but
eager to try out the 802.11s mesh since I'm planning to add another router
(1 main router, 2 wireless extenders) to the mix. Wanted to see if 802.11s
would work better than WDS.