openwrt/mt76 #451: MT7615 stopped functioning (Redmi AC2100, Dumb AP, OpenWrt SNAPSHOT r14538-d643b10a76)

morhimi, 2020-09-22:
Hi all,

I was advised to this repo after posting on the OpenWrt forum here:
https://forum.openwrt.org/t/new-xiaomi-router-ac2100/48101/865?u=morhimi

I'm running 4 Dumb APs, all the same hardware - Redmi AC2100.
In the last 2 months, they were pretty stable and 5ghz worked mostly flawlessly.

I usually update them to latest master once a week, starting with 1 AP and if everything goes well continue with the rest.

In the last 7-10 days, I 4 different snapshots have 5ghz completely broken for me, the latest snapshot that is running fine for me is: `OpenWrt SNAPSHOT r14423-be9694aaa2`

basically clients connect successfully and stay connected, but there is no traffic at all, the 2.4ghz works fine.

let me know how can I assist with identifying the regression
my wireless config:

root@OpenWrt_RM2100_Kids:~# cat /etc/config/wireless

config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'HT20'
	option channel 'auto'
	option country 'IL'
	option legacy_rates '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'StayAway'
	option key '********'
	option ieee80211r '1'
	option ft_over_ds '1'
	option ft_psk_generate_local '1'
	option nasid '88C3973DE8AE'
	option ieee80211k '1'
	option bss_transition '1'
	option time_advertisement '2'
	option time_zone 'IST-2IDT,M3.4.4/26,M10.5.0'
	option ieee80211v '1'
	option wnm_sleep_mode '1'
	option encryption 'psk2'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11a'
	option path '1e140000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option legacy_rates '0'
	option country 'IL'
	option htmode 'VHT80'
	option channel '52'
	option disabled '1'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'StayAway'
	option key '********'
	option ieee80211r '1'
	option nasid '88C3973DE8AF'
	option ft_over_ds '1'
	option ft_psk_generate_local '1'
	option ieee80211k '1'
	option bss_transition '1'
	option time_advertisement '2'
	option time_zone 'IST-2IDT,M3.4.4/26,M10.5.0'
	option ieee80211v '1'
	option wnm_sleep_mode '1'
	option encryption 'psk2'
	option disabled '1'

morhimi, 2020-09-22:
root@OpenWrt_RM2100_Kids:~# dmesg
[    0.000000] Linux version 5.4.66 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r14538-d643b10a76)) #0 SMP Mon Sep 21 19:27:56 2020
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Redmi Router AC2100
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] percpu: Embedded 14 pages/cpu s26768 r8192 d22384 u57344
[    0.000000] pcpu-alloc: s26768 r8192 d22384 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00068400
[    0.000000] Readback ErrCtl register=00068400
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 120584K/131072K available (5930K kernel code, 206K rwdata, 1272K rodata, 1284K init, 238K bss, 10488K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x340/0x558 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000008] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007804] Calibrating delay loop... 583.68 BogoMIPS (lpj=1167360)
[    0.042001] pid_max: default: 32768 minimum: 301
[    0.046776] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.054003] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.064447] rcu: Hierarchical SRCU implementation.
[    0.069892] smp: Bringing up secondary CPUs ...
[    2.311728] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    2.311739] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    2.311751] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    2.311853] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.102665] Synchronize counters for CPU 1: done.
[    2.373061] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    2.373070] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    2.373079] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    2.373139] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.167394] Synchronize counters for CPU 2: done.
[    2.434459] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    2.434467] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    2.434476] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    2.434538] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.225260] Synchronize counters for CPU 3: done.
[    0.255142] smp: Brought up 1 node, 4 CPUs
[    0.263552] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.273243] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.280318] pinctrl core: initialized pinctrl subsystem
[    0.287222] NET: Registered protocol family 16
[    0.302798] FPU Affinity set after 4688 emulations
[    0.319064] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[... 226 lines cut ...]
[   24.214549] br-lan: port 1(lan1) entered blocking state
[   24.219811] br-lan: port 1(lan1) entered forwarding state
[   24.226489] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   24.233429] br-lan: port 4(wlan1) entered blocking state
[   24.238825] br-lan: port 4(wlan1) entered disabled state
[   24.244868] device wlan1 entered promiscuous mode
[   25.401452] br-lan: port 5(wlan0) entered blocking state
[   25.406988] br-lan: port 5(wlan0) entered disabled state
[   25.413181] device wlan0 entered promiscuous mode
[   32.933967] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   32.940583] br-lan: port 5(wlan0) entered blocking state
[... cut ...]
[ 6637.505827] br-lan: port 4(wlan1) entered disabled state
[ 6638.239206] device wlan0 left promiscuous mode
[ 6638.243886] br-lan: port 5(wlan0) entered disabled state
[ 6643.317438] br-lan: port 4(wlan0) entered blocking state
[ 6643.322974] br-lan: port 4(wlan0) entered disabled state
[ 6643.328875] device wlan0 entered promiscuous mode
[ 6650.736871] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 6650.743501] br-lan: port 4(wlan0) entered blocking state
[ 6650.748846] br-lan: port 4(wlan0) entered forwarding state

stevenxed, 2020-09-23:
Try the latest snapshot uploaded today. It has this fix included: https://github.com/openwrt/mt76/commit/73301065ac320b691e074b230a05ce24b26794f3

morhimi, 2020-09-24:
10x, looks like the issue is indeed resolved for me:
[image]

closed 2020-09-24 by morhimi