openwrt/mt76 #457: MT7612EN Constant hardware restart requested on b22977c2

JosiahWhite, 2020-09-29:
When this happens, all wifi traffic stops for multiple seconds while the hardware restarts
Chip: MT7612EN
Device: Xiaomi MiWiFi 4A gigabit edition

Using three of these devices as AP-repeaters with the backhaul on the same radio in WDS mode. All three are experiencing the same issue to differing degrees, some only have the message printed once per hour, others multiple times per minute. Frequency seems to depend on how many clients are connected and active. The device the log below was from a quiet AP with only 2 clients connected. If it matters, the other side of the backhaul is an Archer C7 v2 with a QCA9880-BR4A radio.

All of the devices are running the latest OpenWrt snapshot which is using this master branch at commit ID b22977c

Dmesg output follows:
```
[ 1105.247542] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[ 1105.253078] mt76x2e 0000:01:00.0: Build: 1
[ 1105.257250] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[ 1105.278354] mt76x2e 0000:01:00.0: Firmware running!
[ 1105.286566] ieee80211 phy1: Hardware restart was requested
[ 2382.541609] mt76x2e 0000:01:00.0: MCU message 31 (seq 6) timed out
[ 2382.586767] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[ 2382.592296] mt76x2e 0000:01:00.0: Build: 1
[ 2382.596478] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[ 2382.617593] mt76x2e 0000:01:00.0: Firmware running!
[ 2382.625861] ieee80211 phy1: Hardware restart was requested
[ 2570.690347] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[ 2570.695877] mt76x2e 0000:01:00.0: Build: 1
[ 2570.700035] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[ 2570.721149] mt76x2e 0000:01:00.0: Firmware running!
[ 2570.729397] ieee80211 phy1: Hardware restart was requested
[ 3302.287051] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[ 3302.292558] mt76x2e 0000:01:00.0: Build: 1
[ 3302.296706] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[ 3302.317843] mt76x2e 0000:01:00.0: Firmware running!
[ 3302.330030] ieee80211 phy1: Hardware restart was requested
[ 3326.257359] mt76x2e 0000:01:00.0: MCU message 31 (seq 3) timed out
[ 3326.302482] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[ 3326.307988] mt76x2e 0000:01:00.0: Build: 1
[ 3326.312155] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[ 3326.333352] mt76x2e 0000:01:00.0: Firmware running!
[ 3326.341610] ieee80211 phy1: Hardware restart was requested
```

Kevinjil, 2020-10-05:
Similar behaviour with a MT7621AT based Newifi-D2 running OpenWrt 19.07.4.

```
Mon Oct  5 17:35:57 2020 kern.info kernel: [489806.464298] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Mon Oct  5 17:35:57 2020 kern.info kernel: [489806.469869] mt76x2e 0000:01:00.0: Build: 1
Mon Oct  5 17:35:57 2020 kern.info kernel: [489806.474137] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Mon Oct  5 17:35:57 2020 kern.info kernel: [489806.502477] mt76x2e 0000:01:00.0: Firmware running!
Mon Oct  5 17:35:57 2020 kern.info kernel: [489806.512600] ieee80211 phy1: Hardware restart was requested
Mon Oct  5 17:36:06 2020 kern.info kernel: [489815.924024] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Mon Oct  5 17:36:06 2020 kern.info kernel: [489815.929595] mt76x2e 0000:01:00.0: Build: 1
Mon Oct  5 17:36:06 2020 kern.info kernel: [489815.933833] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Mon Oct  5 17:36:06 2020 kern.info kernel: [489815.962189] mt76x2e 0000:01:00.0: Firmware running!
Mon Oct  5 17:36:06 2020 kern.info kernel: [489815.972375] ieee80211 phy1: Hardware restart was requested
Mon Oct  5 17:37:18 2020 kern.info kernel: [489887.941970] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Mon Oct  5 17:37:18 2020 kern.info kernel: [489887.947540] mt76x2e 0000:01:00.0: Build: 1
Mon Oct  5 17:37:18 2020 kern.info kernel: [489887.951821] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Mon Oct  5 17:37:18 2020 kern.info kernel: [489887.980063] mt76x2e 0000:01:00.0: Firmware running!
Mon Oct  5 17:37:18 2020 kern.info kernel: [489887.990240] ieee80211 phy1: Hardware restart was requested
Mon Oct  5 17:37:21 2020 kern.info kernel: [489890.711825] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Mon Oct  5 17:37:21 2020 kern.info kernel: [489890.717395] mt76x2e 0000:01:00.0: Build: 1
Mon Oct  5 17:37:21 2020 kern.info kernel: [489890.721669] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Mon Oct  5 17:37:21 2020 kern.info kernel: [489890.750033] mt76x2e 0000:01:00.0: Firmware running!
Mon Oct  5 17:37:21 2020 kern.info kernel: [489890.760117] ieee80211 phy1: Hardware restart was requested
Mon Oct  5 17:37:25 2020 kern.info kernel: [489894.391767] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Mon Oct  5 17:37:25 2020 kern.info kernel: [489894.397338] mt76x2e 0000:01:00.0: Build: 1
Mon Oct  5 17:37:25 2020 kern.info kernel: [489894.401657] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Mon Oct  5 17:37:25 2020 kern.info kernel: [489894.429880] mt76x2e 0000:01:00.0: Firmware running!
Mon Oct  5 17:37:25 2020 kern.info kernel: [489894.440081] ieee80211 phy1: Hardware restart was requested
```

db260179, 2020-12-22:
Its worth reducing the transmit power setting to 20 dBm - depending on country limits etc.
I've noticed setting above that level, this issue occurs.
Possible that gpio pins need to be set high to work at high levels?

Borromini, 2020-12-27:
For what it's worth, I've seen a similar issue on my DIR-860L B1 (PHY1 is the N radio here though). With kernel 5.4, the radio often completely disappeared. I brought this up in the pending [5.10 PR for ramips](https://github.com/openwrt/openwrt/pull/3693) and one of the people who worked on the mt7621-pcie suggested the GPIO pins related to the PCI-E ports needed [to be reset explicitly](https://github.com/openwrt/openwrt/pull/3693#issuecomment-751194094). This needs to be done through the DTS. These pins are board specific, so you need to find those out first, but it might be worth a try.

ptpt52, 2020-12-31:
I notice this issue in many devices of mt7621

Vladdrako, 2021-01-02:
@Borromini How to find the needed GPIO pin? I have the same problem on mir3g.

Borromini, 2021-01-02:
@Vladdrako See https://github.com/openwrt/openwrt/pull/3693#issuecomment-750782970 seems it could be 19, 8 and 7. Try modifying the PCI-E section in your DTS and recompile:

```
 &pcie {
        pinctrl-names = "default";
        pinctrl-0 = <&pcie_pins>;

       reset-gpios = <&gpio 19 GPIO_ACTIVE_LOW>,
                       <&gpio 8 GPIO_ACTIVE_LOW>,
                       <&gpio 7 GPIO_ACTIVE_LOW>;

        status = "okay";
 };
```

I checked the Mi Router 3G DTS, it just says `status = "okay"` like mine (DIR-860L rev B1) did. You might see some extra errors in dmesg, check the comment section of the PR for more insights on it.

Vladdrako, 2021-01-06:
@Borromini gpio8 used for Front LED on my board. Sadly, resetting gpio 19 and 7 didn't fix MCU error. Anyway, thanks for the help.

Mushoz, 2021-01-06:
Is there anything that we can do to help debug this issue further? Also, what versions are affected by this bug? I've had this issue for a while on the 19.07 branch, including the latest 19.07.5. It didn't happen on the earlier versions of 19.07 if I remember correctly. And from what I am reading, it's also happening on devices on recent master branch images?

Borromini, 2021-01-06:
@Mushoz Don't you have a DIR-860L B1 as well? (You can PM me on the OpenWrt forum if you'd like).

infunx, 2021-01-28:
Hi guys.
Just want you to know and try it.
I had the same problem with Newifi 3 D2.
And I fixed it :)
The reason was the HDD connected to my router and lying next to my router. I have used a longer cable and moved HDD 0.5m+ (about 70cm, ok) far from the router and everything is ok now.
Dmesg is clean :)
Try it. 
Or you can try to turn HDD off first, if you do not have long cable.
If you do not have HDD near your router then look for other electromagnetic sources.

Btw, when I put my HDD near router again - errors come back. So, it is hardware problem.

Hope it will help.

gstammw, 2021-09-08:
See https://bugs.openwrt.org/index.php?do=details&task_id=3989. But still exists in  OpenWrt 21.02.0 FINAL on the Xiaomi Mi Router 4A Gigabit Edition.

Borromini, 2021-09-08:
@gstammw Did you try fiddling with the GPIO resets in the DTS to see if that fixes it for you?

-------- Oorspronkelijk bericht --------

iAbadia, 2021-10-05:
I'm experiencing the same issue on:

Xiaomi Mi Router 3G
MediaTek MT7621 ver:1 eco:3
OpenWrt 21.02.0 r16279-5cc0535800 / LuCI openwrt-21.02 branch git-21.231.26241-422c175

Same on and on in the Kernel log:
```
[1213184.312296] ieee80211 phy1: Hardware restart was requested
[1213221.697213] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[1213221.702897] mt76x2e 0000:02:00.0: Build: 1
[1213221.707216] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[1213221.728069] mt76x2e 0000:02:00.0: Firmware running!
[1213221.736223] ieee80211 phy1: Hardware restart was requested
```

aero, 2021-10-09:
My xiaomi miwifi mini also reboots automatically often on openwrt 21.02.0
I rollbacked the firmware to openwrt 19.07.7.

HiGarfield, 2021-11-04:
Same issue on Newifi-D2:

`[19561.719469] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[19561.724954] mt76x2e 0000:01:00.0: Build: 1
[19561.729168] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[19561.758190] mt76x2e 0000:01:00.0: Firmware running!
[19561.768250] ieee80211 phy1: Hardware restart was requested
[19606.318380] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[19606.323864] mt76x2e 0000:01:00.0: Build: 1
[19606.328067] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[19606.357122] mt76x2e 0000:01:00.0: Firmware running!
[19606.367217] ieee80211 phy1: Hardware restart was requested
[19643.547575] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[19643.553061] mt76x2e 0000:01:00.0: Build: 1
[19643.557242] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[19643.586232] mt76x2e 0000:01:00.0: Firmware running!
[19643.596304] ieee80211 phy1: Hardware restart was requested
[19675.826701] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[19675.832380] mt76x2e 0000:01:00.0: Build: 1
[19675.836595] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[19675.865448] mt76x2e 0000:01:00.0: Firmware running!
[19675.875483] ieee80211 phy1: Hardware restart was requested
[20040.597919] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[20040.603407] mt76x2e 0000:01:00.0: Build: 1
[20040.607659] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[20040.636281] mt76x2e 0000:01:00.0: Firmware running!
[20040.646353] ieee80211 phy1: Hardware restart was requested
[20050.877247] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[20050.882732] mt76x2e 0000:01:00.0: Build: 1
[20050.886947] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[20050.916022] mt76x2e 0000:01:00.0: Firmware running!
[20050.926077] ieee80211 phy1: Hardware restart was requested
[22142.691835] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[22142.697329] mt76x2e 0000:01:00.0: Build: 1
[22142.701517] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[22142.730636] mt76x2e 0000:01:00.0: Firmware running!
[22142.740691] ieee80211 phy1: Hardware restart was requested`

noctux, 2021-11-07:
@Borromini Do you still run the patched DTS on a DIR 860L B1? If so, do you experience the issue with the released 21.01 version? With the unpatched release builds, I do still experience this issue, so I was wondering if it is worth it to give it a shot.

Borromini, 2021-11-08:
@noctux It should be part of 21.02 since it was merged well before. I just set up my DIR-860L B1 somewhere else recently after it being unused for a long time and it seems to work fine.

noctux, 2021-11-08:
@Borromini Thank you for your answer! Interesting. I assume that you are correct: [https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=target/linux/ramips/dts/mt7621_dlink_dir-860l-b1.dts;hb=HEAD]() indicates that 19 and 8 are reset on current 860l builds (probably the two pins out of the 3 options that might be in use on the DIR 860L B1). However, I still see firmware reset-warnings and experience instable wifi (even though 21.01.1 has improved the situation slightly, but not definitely):

```
Mon Nov  8 11:58:38 2021 kern.info kernel: [575851.537080] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Mon Nov  8 11:58:38 2021 kern.info kernel: [575851.548446] mt76x2e 0000:01:00.0: Build: 1
Mon Nov  8 11:58:38 2021 kern.info kernel: [575851.556984] mt76x2e 0000:01:00.0: Build Time: 201507311614____
Mon Nov  8 11:58:38 2021 kern.info kernel: [575851.583791] mt76x2e 0000:01:00.0: Firmware running!
Mon Nov  8 11:58:38 2021 kern.info kernel: [575851.595971] ieee80211 phy0: Hardware restart was requested
```

So unfortunately, it might not be the missing DTS-pins alone :/ But I'm glad (and a bit surpised) that your B1 works alight :D Maybe I have strange clients on my network...

Borromini, 2021-11-09:
@noctux You might want to try pin 7, it seems 19, 8 and 7 are the common ones used on MT7621. But I lack the technical background, so I'm going out on a limb.

jxph1123, 2022-01-10:
With the version below, no more firmware reset issues.
PKG_SOURCE_DATE:=2018-11-09
PKG_SOURCE_VERSION:=ffccb48c5082b2c6f43a0235b77b62ca388810f0
PKG_MIRROR_HASH:=bff0063db1a1b1353f59aad16abad4dd93ceeb4c672811ad864cebe0e39c7d8d

nbd168, 2022-01-31:
Please try the latest mt76 master version included in my staging tree at https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=summary

Mushoz, 2022-01-31:
Cross-posted a link to your message in the following issue: https://github.com/openwrt/mt76/issues/444

This ticket seems to have been opened ~16 days after the other one, and is a duplicate. We might want to keep the conversation in one ticket to keep things clear :)

DruvsEgo, 2022-02-03:
> Please try the latest mt76 master version included in my staging tree at https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=summary
Thu Feb  3 06:38:21 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3

 15.213230] mt7621-pci 1e140000.pcie: bus=1 slot=0 irq=23
Thu Feb  3 06:38:21 2022 kern.info kernel: [   15.224075] pci 0000:00:00.0: enabling device (0004 -> 0007)
Thu Feb  3 06:38:21 2022 kern.info kernel: [   15.235373] mt76x2e 0000:01:00.0: enabling device (0000 -> 0002)
Thu Feb  3 06:38:21 2022 kern.info kernel: [   15.247575] mt76x2e 0000:01:00.0: ASIC revision: 76120044
Thu Feb  3 06:38:21 2022 kern.info kernel: [   15.912934] mt76x2e 0000:01:00.0: ROM patch build: 20141115060606a
Thu Feb  3 06:38:21 2022 kern.info kernel: [   15.965907] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
Thu Feb  3 06:38:21 2022 kern.info kernel: [   15.976891] mt76x2e 0000:01:00.0: Build: 1
Thu Feb  3 06:38:21 2022 kern.info kernel: [   15.985085] mt76x2e 0000:01:00.0: Build Time: 201706151016____
Thu Feb  3 06:38:21 2022 kern.info kernel: [   16.011057] mt76x2e 0000:01:00.0: Firmware running!
Thu Feb  3 06:38:21 2022 kern.debug kernel: [   16.022621] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
Thu Feb  3 06:38:21 2022 kern.info kernel: [   16.028208] mt7621-pci 1e140000.pcie: bus=2 slot=1 irq=24
Thu Feb  3 06:38:21 2022 kern.info kernel: [   16.039129] pci 0000:00:01.0: enabling device (0004 -> 0007)
Thu Feb  3 06:38:21 2022 kern.info kernel: [   16.050455] mt76x2e 0000:02:00.0: enabling device (0000 -> 0002)
Thu Feb  3 06:38:21 2022 kern.info kernel: [   16.062696] mt76x2e 0000:02:00.0: ASIC revision: 76020044
Thu Feb  3 06:38:21 2022 kern.info kernel: [   16.731178] mt76x2e 0000:02:00.0: ROM patch already applied
Thu Feb  3 06:38:21 2022 kern.info kernel: [   16.743415] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
Thu Feb  3 06:38:21 2022 kern.info kernel: [   16.754400] mt76x2e 0000:02:00.0: Build: 1
Thu Feb  3 06:38:21 2022 kern.info kernel: [   16.762598] mt76x2e 0000:02:00.0: Build Time: 201706151016____
Thu Feb  3 06:38:21 2022 kern.info kernel: [   16.791235] mt76x2e 0000:02:00.0: Firmware running!
Thu Feb  3 06:38:21 2022 kern.debug kernel: [   16.803169] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
firmware version

Thu Feb  3 07:25:38 2022 kern.err kernel: [  749.178989] mt76x2e 0000:01:00.0: MAC error detected
at a distance errors are not often, Under load

Thu Feb  3 15:31:23 2022 kern.err kernel: [29893.867412] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:31:26 2022 kern.err kernel: [29896.977375] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:32:15 2022 kern.err kernel: [29945.856770] mt76x2e 0000:01:00.0: MAC error detected
at a distance errors are not often,Under load

Thu Feb  3 15:46:59 2022 kern.err kernel: [30829.764956] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:47:00 2022 kern.err kernel: [30831.005029] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:47:14 2022 kern.err kernel: [30845.555142] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:47:16 2022 kern.err kernel: [30846.795164] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:47:18 2022 kern.err kernel: [30848.695189] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:47:31 2022 kern.err kernel: [30862.135387] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:47:46 2022 kern.err kernel: [30877.145494] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:47:47 2022 kern.err kernel: [30878.385533] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:47:49 2022 kern.err kernel: [30880.285532] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:48:07 2022 kern.err kernel: [30898.135841] mt76x2e 0000:01:00.0: MAC error detected

Very small distance to hdd, many errors, Under load 
Thu Feb  3 15:49:27 2022 kern.info kernel: [30978.376070] mt7530 mdio-bus:1f lan1: Link is Up - 100Mbps/Full - flow control off
PC from wifi to ethernet

Thu Feb  3 15:50:48 2022 kern.err kernel: [31059.063773] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:51:03 2022 kern.err kernel: [31074.383483] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:51:04 2022 kern.err kernel: [31075.623469] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:51:07 2022 kern.err kernel: [31077.743439] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:51:31 2022 kern.err kernel: [31102.522946] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:51:33 2022 kern.err kernel: [31103.762937] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:51:34 2022 kern.err kernel: [31104.892900] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:51:35 2022 kern.err kernel: [31106.132899] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:51:48 2022 kern.err kernel: [31119.082633] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:51:52 2022 kern.err kernel: [31122.962557] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:51:53 2022 kern.err kernel: [31124.092541] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:51:55 2022 kern.err kernel: [31126.322511] mt76x2e 0000:01:00.0: MAC error detected
Thu Feb  3 15:52:08 2022 kern.err kernel: [31138.892264] mt76x2e 0000:01:00.0: MAC error detected
Very small distance to hdd, many errors

daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 84:c7:ea:8f:cd:bc
Thu Feb  3 16:12:22 2022 daemon.info hostapd: wlan0: STA 84:xx:xx:xx:xx:bc IEEE 802.11: disassociated
[... cut ...]
Thu Feb  3 16:12:27 2022 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) 84:xx:xx:xx:xx:bc
Thu Feb  3 16:12:27 2022 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.245 84:xx:xx:xx:xx:bc
was when iperf was running 
iperf worked on PC

Under load iperf, errors are less or completely absent, depending on the position of the phone.
"Hardware restart was requested" Didn't see it.
Without load and near the hard drive, except for the detected MAC error, I did not see other errors.
MTC WR1201 1200M
Sony XZ
Sorry google translate.

keep watching :)

nbd168, 2022-02-03:
If it only happens occasionally, and the device continues to work afterwards, it should be fine.

DruvsEgo, 2022-02-03:
> If it only happens occasionally, and the device continues to work afterwards, it should be fine.

In my situation "Hardware restart requested" sometimes only appeared next to the hard drive, MAC error detected constantly. When moving further away from the HDD, a clean log.
Thanks for your work!

HiGarfield, 2022-02-05:
> Please try the latest mt76 master version included in my staging tree at https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=summary

The issue still exists.

root@hw24g:~# dmesg | grep mt76x2e
[   17.672474] mt76x2e 0000:02:00.0: ASIC revision: 76120044
[   18.413702] mt76x2e 0000:02:00.0: ROM patch build: 20141115060606a
[   18.423583] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[   18.429129] mt76x2e 0000:02:00.0: Build: 1
[   18.433219] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[   18.457401] mt76x2e 0000:02:00.0: Firmware running!
[24454.006365] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[24454.011849] mt76x2e 0000:02:00.0: Build: 1
[24454.016086] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[24454.045149] mt76x2e 0000:02:00.0: Firmware running!

nbd168, 2022-02-05:
With that much time inbetween, it seems like a rare occurence. Does the device still work after that error? If so, you can safely ignore it.

HiGarfield, 2022-02-06:
> With that much time inbetween, it seems like a rare occurence. Does the device still work after that error? If so, you can safely ignore it.

There was no perceptible network disconnection before and after the firmware reset.

HiGarfield, 2022-02-06:
> > If it only happens occasionally, and the device continues to work afterwards, it should be fine.
> 
> In my situation "Hardware restart requested" sometimes only appeared next to the hard drive, MAC error detected constantly. When moving further away from the HDD, a clean log. Thanks for your work!

A similar case for me. When the  portable HDD attached to the router through  a USB 3.0 port works in extremely heavy load, the mt76x2e firmware reset potentially occurs.

DruvsEgo, 2022-02-09:
> I am also having the same issue. There is no HDD even no usb on M4AG. Chip: MT7612EN Device: Xiaomi MiWiFi 4A gigabit edition
What does not negate the presence of electromagnetic radiation near your router, it does not have to be a hard drive. Try to compile the firmware from here https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=summary with the latest mt76 updates, if you see a lot of "MAC error detected" then there is radiation that interferes with the router.

kyraxyg, 2022-02-10:
This might sound ridiculious but can covering the router with aluminum foil
reduce the restarting/errors?

9 Şub 2022 Çar 19:25 tarihinde DruvsEgo ***@***.***> şunu
yazdı:

weimzh, 2022-02-19:
I built this driver under ubuntu 20.04 (with linux kernel 5.17 rc4) and it happens very frequently if I download/upload a lot when I use the MT7612E card as 5GHz AP:
```
[  300.363012] mt76x2e 0000:02:00.0: MCU message 1f (seq 5) timed out
[  300.410687] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  300.410693] mt76x2e 0000:02:00.0: Build: 1
[  300.410696] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  300.427035] mt76x2e 0000:02:00.0: Firmware running!
[  300.427854] ieee80211 phy0: Hardware restart was requested
[  307.978018] mt76x2e 0000:02:00.0: MCU message 1f (seq 9) timed out
[  308.028702] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  308.028708] mt76x2e 0000:02:00.0: Build: 1
[  308.028710] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  308.049039] mt76x2e 0000:02:00.0: Firmware running!
[  308.049831] ieee80211 phy0: Hardware restart was requested
[  315.594001] mt76x2e 0000:02:00.0: MCU message 1f (seq 13) timed out
[  315.634724] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  315.634730] mt76x2e 0000:02:00.0: Build: 1
[  315.634732] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  315.660144] mt76x2e 0000:02:00.0: Firmware running!
[  315.662036] ieee80211 phy0: Hardware restart was requested
[  325.258997] mt76x2e 0000:02:00.0: MCU message 1f (seq 4) timed out
[  325.309678] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  325.309684] mt76x2e 0000:02:00.0: Build: 1
[  325.309686] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  325.335080] mt76x2e 0000:02:00.0: Firmware running!
[  325.335883] ieee80211 phy0: Hardware restart was requested
[  334.923007] mt76x2e 0000:02:00.0: MCU message 1f (seq 10) timed out
[  334.973698] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  334.973704] mt76x2e 0000:02:00.0: Build: 1
[  334.973706] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  334.993033] mt76x2e 0000:02:00.0: Firmware running!
[  334.993838] ieee80211 phy0: Hardware restart was requested
[  343.562001] mt76x2e 0000:02:00.0: MCU message 1f (seq 15) timed out
[  343.603713] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  343.603720] mt76x2e 0000:02:00.0: Build: 1
[  343.603723] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  343.628999] mt76x2e 0000:02:00.0: Firmware running!
[  343.629804] ieee80211 phy0: Hardware restart was requested
[  354.249998] mt76x2e 0000:02:00.0: MCU message 1f (seq 7) timed out
[  354.300760] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  354.300767] mt76x2e 0000:02:00.0: Build: 1
[  354.300769] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  354.326174] mt76x2e 0000:02:00.0: Firmware running!
[  354.327602] ieee80211 phy0: Hardware restart was requested
[  371.081987] mt76x2e 0000:02:00.0: MCU message 1f (seq 5) timed out
[  371.120739] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  371.120746] mt76x2e 0000:02:00.0: Build: 1
[  371.120748] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  371.145983] mt76x2e 0000:02:00.0: Firmware running!
[  371.146791] ieee80211 phy0: Hardware restart was requested
[  387.913985] mt76x2e 0000:02:00.0: MCU message 1f (seq 3) timed out
[  387.960914] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  387.960920] mt76x2e 0000:02:00.0: Build: 1
[  387.960923] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  387.978001] mt76x2e 0000:02:00.0: Firmware running!
[  387.978807] ieee80211 phy0: Hardware restart was requested
[  441.611003] mt76x2e 0000:02:00.0: MCU message 1f (seq 7) timed out
[  441.658845] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  441.658851] mt76x2e 0000:02:00.0: Build: 1
[  441.658853] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  441.684235] mt76x2e 0000:02:00.0: Firmware running!
[  441.685064] ieee80211 phy0: Hardware restart was requested
```

and the network will lag when this happens.

HiGarfield, 2022-02-19:
A user of mt7612e reported to me that after he cherry-picked the commit https://github.com/openwrt/openwrt/commit/01895c315f0485779a2e312c58708a32b4ea1f9a to his own openwrt repository, the hardware restart occurs more frequently. Will this be related?

HiGarfield, 2022-02-21:
Errors occur on my mt7612.

[1115361.632877] mt76x2e 0000:02:00.0: MAC error detected
[1115379.462616] mt76x2e 0000:02:00.0: MAC error detected
[1115385.692526] mt76x2e 0000:02:00.0: MAC error detected
[1125704.972122] mt76x2e 0000:02:00.0: MAC error detected
[1130542.131878] mt76x2e 0000:02:00.0: MAC error detected
[1142496.507287] mt76x2e 0000:02:00.0: MAC error detected
[1196172.504355] mt76x2e 0000:02:00.0: MAC error detected
[1198073.408306] mt76x2e 0000:02:00.0: MAC error detected
[1201450.892843] mt76x2e 0000:02:00.0: MAC error detected
[1202999.302701] mt76x2e 0000:02:00.0: MAC error detected
[1203540.325349] mt76x2e 0000:02:00.0: MAC error detected
[1204623.940705] mt76x2e 0000:02:00.0: MAC error detected
[1205005.165507] mt76x2e 0000:02:00.0: MAC error detected
[1205807.782983] mt76x2e 0000:02:00.0: MAC error detected
[1207461.188945] mt76x2e 0000:02:00.0: MAC error detected
[1207866.613256] mt76x2e 0000:02:00.0: MAC error detected
[1208236.238007] mt76x2e 0000:02:00.0: MAC error detected
[1208732.460999] mt76x2e 0000:02:00.0: MAC error detected
[1210655.163835] mt76x2e 0000:02:00.0: MAC error detected
[1213067.969856] mt76x2e 0000:02:00.0: MAC error detected
[1213441.594569] mt76x2e 0000:02:00.0: MAC error detected
[1215098.401230] mt76x2e 0000:02:00.0: MAC error detected
[1218085.388776] mt76x2e 0000:02:00.0: MAC error detected
[1218821.408356] mt76x2e 0000:02:00.0: MAC error detected
[1219180.533238] mt76x2e 0000:02:00.0: MAC error detected
[1219336.761022] mt76x2e 0000:02:00.0: MAC error detected
[1219343.090946] mt76x2e 0000:02:00.0: MAC error detected
[1219406.620025] mt76x2e 0000:02:00.0: MAC error detected
[1219758.945002] mt76x2e 0000:02:00.0: MAC error detected
[1219795.674476] mt76x2e 0000:02:00.0: MAC error detected
[1220824.389757] mt76x2e 0000:02:00.0: MAC error detected
[1221204.414403] mt76x2e 0000:02:00.0: MAC error detected
[1221988.833146] mt76x2e 0000:02:00.0: MAC error detected
[1222920.649822] mt76x2e 0000:02:00.0: MAC error detected
[1223020.078406] mt76x2e 0000:02:00.0: MAC error detected
[1223070.707679] mt76x2e 0000:02:00.0: MAC error detected
[1223245.035210] mt76x2e 0000:02:00.0: MAC error detected
[1223450.262294] mt76x2e 0000:02:00.0: MAC error detected
[1223455.092201] mt76x2e 0000:02:00.0: MAC error detected
[1223464.122077] mt76x2e 0000:02:00.0: MAC error detected
[1223475.751915] mt76x2e 0000:02:00.0: MAC error detected
[1223554.880819] mt76x2e 0000:02:00.0: MAC error detected
[1223602.610110] mt76x2e 0000:02:00.0: MAC error detected
[1223668.439169] mt76x2e 0000:02:00.0: MAC error detected
[1223686.568923] mt76x2e 0000:02:00.0: MAC error detected
[1223782.597547] mt76x2e 0000:02:00.0: MAC error detected
[1224290.220397] mt76x2e 0000:02:00.0: MAC error detected
[1225410.334622] mt76x2e 0000:02:00.0: MAC error detected
[1226611.647691] mt76x2e 0000:02:00.0: MAC error detected
[1226632.277435] mt76x2e 0000:02:00.0: MAC error detected
[1226722.906119] mt76x2e 0000:02:00.0: MAC error detected
[1227278.028338] mt76x2e 0000:02:00.0: MAC error detected
[1227716.772096] mt76x2e 0000:02:00.0: MAC error detected
[1227729.951917] mt76x2e 0000:02:00.0: MAC error detected

Then it can not be connected.

HiGarfield, 2022-03-04:
> Errors occur on my mt7612.
>
> [1115361.632877] mt76x2e 0000:02:00.0: MAC error detected [1115379.462616] mt76x2e 0000:02:00.0: MAC error detected [1115385.692526] mt76x2e 0000:02:00.0: MAC error [...]

After several days testing, I find out that commit https://github.com/openwrt/mt76/commit/20d1fed838b958701ed5d030b663ba9227477ab1 leads to this problem. Partly reverting that commit (see https://github.com/HiGarfield/mt76/commit/e0f502019317462fb50cfc5e6ad97d3a61cbed0a) solves the problem.

imago54, 2022-06-15:
Issue is even more frequent on OpenWrt 22.03.0-rc4 (Xiaomi Mi Router 3G). 5GHz (phy1) doesn't survive even one full speedtest.net run.

[  122.054413] mt76x2e 0000:02:00.0: MCU message 1f (seq 1) timed out
[  122.115646] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[  122.121174] mt76x2e 0000:02:00.0: Build: 1
[  122.125620] mt76x2e 0000:02:00.0: Build Time: 201607111443____
[  122.154440] mt76x2e 0000:02:00.0: Firmware running!
[  122.164688] ieee80211 phy1: Hardware restart was requested

newmem12, 2022-06-19:
> try these two commits [HiGarfield/mt76@8b37f81](https://github.com/HiGarfield/mt76/commit/8b37f81861c3b5acc62ecc2f8b0a2795ea3b1a59) [HiGarfield/mt76@786f9ad](https://github.com/HiGarfield/mt76/commit/786f9ad7eb161a9c6f890c83364ac9363d42f8d0)

hi HiGarfield, Could you please guide how to complie your mt76 driver with openwrt?

HiGarfield, 2022-06-19:
> hi HiGarfield, Could you please guide how to complie your mt76 driver with openwrt?

It may not be compatible with latest OpenWrt. It is used in my lede-17.01.4-Mod repository.
https://github.com/HiGarfield/lede-17.01.4-Mod

If you are using the latest OpenWrt, just apply these two commits to the latest mt76.

HiGarfield, 2022-06-19:
> hi HiGarfield, Could you please guide how to complie your mt76 driver with openwrt?

I have made changes on the latest mt76.
You can try the modified mt76 repo on the latest OpenWrt.
https://github.com/HiGarfield/mt76-1

newmem12, 2022-06-19:
> I have made changes on the latest mt76. You can try the modified mt76 repo on the latest OpenWrt. https://github.com/HiGarfield/mt76-1

i'm using latest SDK x-wrt, can you give step by step instructions to apply these 2 commits? or modify the repo on my SDK!
Thank you very much!

HiGarfield, 2022-06-19:
> i'm using latest SDK x-wrt, can you give step by step instructions to apply these 2 commits? or modify the repo on my SDK! Thank you very much!

I applied the changes on the latest x-wrt.
Try this.
https://github.com/HiGarfield/x-wrt

DragonBluep, 2022-06-20:
> try these two commits [HiGarfield/mt76@8b37f81](https://github.com/HiGarfield/mt76/commit/8b37f81861c3b5acc62ecc2f8b0a2795ea3b1a59) [HiGarfield/mt76@786f9ad](https://github.com/HiGarfield/mt76/commit/786f9ad7eb161a9c6f890c83364ac9363d42f8d0)

Seems to be useful for `MAC error detected` on mt7610e.

newmem12, 2022-06-20:
Hi HiGarfield, These two commits only "reduce" the number of times "Hardware Restart" or completely fix this problem?
Thank you!

HiGarfield, 2022-06-20:
> Hi HiGarfield, These two commits only "reduce" the number of times "Hardware Restart" or completely fix this problem? Thank you!

Only reduce. Actually, hardware restart can not be completely avoided, since the hardware really hangs in certain conditions. We have nothing to do with it, and the only solution is resetting it.

In aforementioned commits, I have relaxed the detection of hardware hangs. It is hoped that false alarms in hang detection were reduced.

newmem12, 2022-06-20:
> > Hi HiGarfield, These two commits only "reduce" the number of times "Hardware Restart" or completely fix this problem? Thank you!
> 
> Only reduce. Actually, hardware restart can never be completely avoided, since the hardware really hangs in certain conditions. We have nothing to do with it, and the only solution is resetting it.
> 
> In aforementioned commits, I have relaxed the detection of hardware hangs. It is hoped that false alarms in hang detection were reduced, thus leading to less hardware restarts.

I have problem with hardware restart, after hardware restart wifi 2.4 can't work and i have to restart router or use "wifi" command!
Is there any way to make wifi 2.4 still work when hardware restart happens?
thank you

HiGarfield, 2022-06-20:
> > > Hi HiGarfield, These two commits only "reduce" the number of times "Hardware Restart" or completely fix this problem? Thank you!
> > 
> > 
> > Only reduce. Actually, hardware restart can never be completely avoided, since the hardware really hangs in certain conditions. We have nothing to do with it, and the only solution is resetting it.
> > In aforementioned commits, I have relaxed the detection of hardware hangs. It is hoped that false alarms in hang detection were reduced, thus leading to less hardware restarts.
> 
> I have problem with hardware restart, after hardware restart wifi 2.4 can't work and i have to reboot router or use "wifi" command! Is there any way to make wifi 2.4 still work when hardware restart happens? thank you

2.4GHz is not related to this hardware restart problem. This problem on mt7612e only affects 5GHz.

lukasz1992, 2022-06-25:
Recompile or wait for the next snapshot, to check if problem is resolved - new patch arrived.

lukasz1992, 2022-07-26:
here: https://github.com/openwrt/mt76/commit/ba39ed3b44f1a0b3bd87c72452a261908afba754
OpenWRT 22.03.0-rc5 includes it, so you can try either snapshot or this version.

kofec, 2022-07-26:
I'm sorry to say it but I have:
```
kmod-mt76-core	    5.10.127+2022-07-03-93e3fce9-4
kmod-mt7603	        5.10.127+2022-07-03-93e3fce9-4
kmod-mt76x02-common	5.10.127+2022-07-03-93e3fce9-4
kmod-mt76x2	        5.10.127+2022-07-03-93e3fce9-4
kmod-mt76x2-common	5.10.127+2022-07-03-93e3fce9-4
```
And still see:
```
2022-07-25T17:10:29.117172+02:00 Xiaomi3G kernel: [  416.037332] mt76x2e 0000:02:00.0: MAC error detected
2022-07-25T17:10:35.767475+02:00 Xiaomi3G kernel: [  422.687329] mt76x2e 0000:02:00.0: MAC error detected
2022-07-25T17:10:37.017159+02:00 Xiaomi3G kernel: [  423.937545] mt76x2e 0000:02:00.0: MAC error detected
2022-07-25T17:10:38.497217+02:00 Xiaomi3G kernel: [  425.417304] mt76x2e 0000:02:00.0: MAC error detected
2022-07-25T17:10:39.747138+02:00 Xiaomi3G kernel: [  426.667327] mt76x2e 0000:02:00.0: MAC error detected
2022-07-25T17:10:40.997120+02:00 Xiaomi3G kernel: [  427.917326] mt76x2e 0000:02:00.0: MAC error detected
2022-07-25T17:10:42.257298+02:00 Xiaomi3G kernel: [  429.177332] mt76x2e 0000:02:00.0: MAC error detected
```

lukasz1992, 2022-07-26:
It means that patch does not help and issue is still present

nazar554, 2022-09-18:
Still present on 22.03 stable, running on Xiaomi Mi Router 3G
Getting `mt76x2e 0000:02:00.0: MAC error detected` on 5GHz interface every 2 seconds, even when no clients are connected.

Gingernut1978, 2022-10-16:
No kernel errors as of yet but a hardware reset request is logged.

My R4G running latest master with this pr, [pr10977](https://github.com/openwrt/openwrt/pull/10977) mac80211 update and [pr10945](https://github.com/openwrt/openwrt/pull/10945) kernel 5.15 update.

`52.098386] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
[   52.105494] br-lan: port 4(phy1-ap0) entered blocking state
[   52.111113] br-lan: port 4(phy1-ap0) entered forwarding state
[ 4020.827359] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[ 4020.832869] mt76x2e 0000:01:00.0: Build: 1
[ 4020.837000] mt76x2e 0000:01:00.0: Build Time: 201607111443____
[ 4020.866158] mt76x2e 0000:01:00.0: Firmware running!
[ 4020.876339] ieee80211 phy1: Hardware restart was requested`

Gingernut1978, 2022-10-18:
Got this error also:

```mt76x2e 0000:01:00.0: MAC error detected
[ 5120.081533] mt76x2e 0000:01:00.0: MAC stop failed

1187887012, 2022-11-07:
> 你好。 只是想让你知道并尝试一下。 我对 Newifi 3 D2 也有同样的问题。 我修复了它:) 原因是硬盘连接到我的路由器并位于我的路由器旁边。我使用了更长的电缆并将 HDD 0.5m+（约 70cm，ok）从路由器移开，现在一切正常。 Dmesg 现在很干净 :) 试试看。 或者，如果您没有长电缆，您可以尝试关闭 HDD。 如果您的路由器附近没有 HDD，请寻找其他电磁源。
> 
> 顺便说一句，当我再次将硬盘放在路由器附近时 - 错误又回来了。所以，我认为这是一个硬件问题。
> 
> 希望它会有所帮助。 OpenWrt 19.07.6 r11278-8055e38794 / Newifi-D2 / MediaTek MT7621 ver:1 eco:3 / MT7662E

非常感谢您的问题分享，我的Newifi 3 D2 也有同样的问题，我把机械硬盘放到了路由器顶上，打开盈动硬盘开关就断网！！！看了您的恢复成功解决了这个问题。

DragonBluep, 2023-08-22:
@shown19 Can you try if removing `kmod-usb3` and `kmod-usb-xhci-mtk` packages has some help?

Djfe, 2023-08-22:
are any electronic devices nearby? (did you place your router on a pc/next to a TV/fridge/microwave whatever?)

shown19, 2023-08-23:
> are any electronic devices nearby? (did you place your router on a pc/next to a TV/fridge/microwave whatever?)

it's few inches from my telephone landline, 1ruler from raspberrypi(wireless/bluetooth is off) and my ISP modem(bridged) 1ruler above the router. Is one of them causes interference for the 5ghz to crash or something? or does the telephone landline is the culprit since it's the nearest from the router?

As of this moment(with kmod-usb3 removed as suggested), 22hours uptime and haven't received any 5ghz hardware restart log. It will take 2days atmost to verify if the issue really resolved.

shown19, 2023-08-23:
Update: after 1day and 12hours uptime I got the error:

`[126322.911025] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[126322.916817] mt76x2e 0000:01:00.0: Build: 1
[126322.921369] mt76x2e 0000:01:00.0: Build Time: 201607111443____
[126322.949565] mt76x2e 0000:01:00.0: Firmware running!
[126322.956461] ieee80211 phy1: Hardware restart was requested`

So removing kmod-usb3 or usb3 is not the issue here

shown19, 2023-08-24:
Update: still getting a hardware reset log after removing the telephone landline beside the router.  I guess 5ghz chipset can no longer go for high power (might be faulty or dying already). Setting it back to 12dbm txpower from 20dbm because that's where I no longer received an error log.

shown19, 2023-08-26:
> Update: still getting a hardware reset log after removing the telephone landline beside the router. I guess 5ghz chipset can no longer go for high power (might be faulty or dying already). Setting it back to 12dbm txpower from 20dbm because that's where I no longer received an error log before.

Nevermind, still getting hardware restart requested despite lowering txpower. It seems like software can't do something about this, right? probably a hardware limitation or issue already. Anyway, thanks.

[46 more replies]