openwrt/mt76 #458: Apparently memory leak with an MT7610e

fnarfbargle, 2020-10-07:
G'day All,

I'm using openwrt with kernel 5.4 on a TP-Link TL-WR902AC v3.

I'm seeing a pretty serious and constantly reproducible memory leak whenever I enable the 5G radio.
Investigation to date is here : https://forum.openwrt.org/t/chasing-memory-leak-on-tl-wr902acv3-and-git-head-openwrt/74773

I'm not convinced the bug is in the mt76 code as I've compiled versions of it going back to 2018 and always see the leak. I also see a leak with just the 2.4G radio enabled, but it's much, much smaller and slower.

There is nothing conclusive in /proc/meminfo, no slab usage that points to the memory and kmemleak doesn't identify enough leaks to account for the missing memory.

I have seen a consistent reported leak as follows : 
`unreferenced object 0x83e1d000 (size 176):
  comm "mt76-tx phy1", pid 675, jiffies 75936 (age 7639.236s)
  hex dump (first 32 bytes):
    00 00 00 00 00 00 00 00 00 10 cb 82 00 00 00 00  ................
    00 00 00 00 00 00 00 00 11 00 01 40 01 00 02 00  ...........@....
  backtrace:
    [<ef9e41f5>] kmem_cache_alloc+0x120/0x290
    [<a4a210d5>] __build_skb+0x34/0xd0
    [<a0a51c47>] __netdev_alloc_skb+0x100/0x1a8
    [<c1f86658>] ieee80211_send_bar+0x48/0x1fc [mac80211]
    [<8655c79e>] mt76_tx_complete_skb+0x270/0x4b4 [mt76]
    [<912654ed>] mt76_tx_complete_skb+0x478/0x4b4 [mt76]
`
But I haven't seen anywhere near enough to account for the 10's of megabytes that go missing over a number of hours.
When I say "go missing" I mean "MemFree" and "MemAvailable" in /proc/meminfo gradually work their way from ~25000kB to ~4000kB with no associated increase in any of the other counts, no correlation in an increase in SLAB use and nothing in userspace. 
Eventually the OOM killer pops in and slays hostapd, and then it's all over until I power cycle the unit.

I build images from source, but I've also tried a snapshot built with image-builder just to ensure it's not an artifact of my build environment.

This doesn't happen with the v4 kernels in the 19.07 branch.

So, I'm posting here to ask for advice on how I might further investigate the issue.
I've tried : 
- Periodic compare of slab debug output (nothing noticable or conclusive)
- SLAB tracing (results in a locked up machine with no useful output
- kmemleak (results in lots of the aforementioned reported leaks, but not enough to come even an order of magnitude close to the amount of memory disappearing)

Does anyone have any recommendations on other methods of instrumentation that will work on a machine with ~20M of free RAM?

fnarfbargle, 2020-12-04:
I'm going to close this because I'm not getting anywhere and have abandoned the device. The leak probably lies outside the mt76 driver in any case as there's already been a significant leak patched in mac80211.

closed 2020-12-04 by fnarfbargle

FusionDragon, 2023-10-14:
Currently using R6220 and I think I'm experiencing the same or similar issue, load will slowly get higher then drop suddenly, resulting in an "Unknown event 37" and "CTRL-EVENT-BEACON-LOSS". Can't seem to find a way to prevent it from happening.

> I'm going to close this because I'm not getting anywhere and have abandoned the device. The leak probably lies outside the mt76 driver in any case as there's already been a significant leak patched in mac80211.