openwrt/mt76 #460: cannot connet wifi (client mode) auth failed

jinglei207, 2020-10-09:
At first it worked on channel 26 and 52, but when i switched back to 52, wifi can't connect anymore.
I have check the password many times, it is right.
```
Thu Oct  8 21:51:14 2020 daemon.notice wpa_supplicant[3028]: wlan1: CTRL-EVENT-SSID-REENABLED id=0 ssid="FASTWEB-F4F502"
Thu Oct  8 21:51:14 2020 daemon.notice wpa_supplicant[3028]: wlan1: SME: Trying to authenticate with 88:c3:97:4d:eb:69 (SSID='FASTWEB-F4F502' freq=5260 MHz)
Thu Oct  8 21:51:14 2020 kern.info kernel: [  225.096901] wlan1: authenticate with 88:c3:97:4d:eb:69
Thu Oct  8 21:51:14 2020 kern.info kernel: [  225.109156] wlan1: send auth to 88:c3:97:4d:eb:69 (try 1/3)
Thu Oct  8 21:51:15 2020 kern.info kernel: [  225.889817] wlan1: send auth to 88:c3:97:4d:eb:69 (try 2/3)
Thu Oct  8 21:51:16 2020 kern.info kernel: [  226.881807] wlan1: send auth to 88:c3:97:4d:eb:69 (try 3/3)
Thu Oct  8 21:51:17 2020 kern.info kernel: [  227.873895] wlan1: authentication with 88:c3:97:4d:eb:69 timed out
Thu Oct  8 21:51:17 2020 daemon.notice wpa_supplicant[3028]: wlan1: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="FASTWEB-F4F502" auth_failures=4 duration=60 reason=CONN_FAILED

```

akaztp, 2020-12-19:
I have the same problem on a MT7610E.
OpenWrt 19.07.4 r11208-ce6496d796
I find it interesting that the timeout is of only about 1second!

ryderlee1110, 2021-04-13:
can you try the latest mt76?

akaztp, 2021-04-13:
I can't, sorry. Don't have that router operational anymore. Upgraded to a Netgear R7800 and I never had to worry about it anymore.

FZXizzbKv7jg, 2021-08-02:
Can confirm this error still occurs:

2x Linksys E8450 5ghz ax mode channel 128 (AP -> STA)
Connection breaks after few minutes. Exactly same error messages.