openwrt/mt76 #462: mt7915 in trouble to get it work

ptpt52, 2020-10-13:
the full boot log
https://gist.github.com/ptpt52/6ce37143871c970668d29eec8d4b23fe

sample err log:
```
[   17.463336] mt7621-pci 1e140000.pcie: bus=2 slot=1 irq=24
[   17.468858] pci 0000:00:01.0: enabling device (0006 -> 0007)
[   17.474723] mt7915e 0000:02:00.0: enabling device (0000 -> 0002)
[   17.640930] mt7915e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20200819112315a
[   17.640930] 
[   18.068000] mt7915e 0000:02:00.0: WM Firmware Version: ____000000, Build Time: 20200819112349
[   18.106753] mt7915e 0000:02:00.0: WA Firmware Version: DEV_000000, Build Time: 20200819112407
[   18.223014] mt7915e: probe of 0000:02:00.0 failed with error -22
```

what may cause this?

namidairo, 2020-10-13:
Are you sure you are pointing it at the correct eeprom (presumably) in your factory partition?

That's the first thing that comes to mind that could cause -EINVAL to pop up.

ptpt52, 2020-10-13:
@namidairo  eeprom is correct and pointed correctly

ryderlee1110, 2021-01-09:
Do you use get calibration data from flash? Try this https://github.com/openwrt/mt76/commit/a515727e842304b023344d3a4325cd425d5c3d47

nonyhaha, 2021-11-02:
Hello @hauke 
I managed to rewrite the factory partition, but the error in my case seems to come from the fact that one of the pci devices does not get recognised on boot. 
on the non working router I have:
```
[    1.108740] mt7621-pci 1e140000.pcie: pcie0 no card, disable it (RST & CLK)
[    1.115692] mt7621-pci 1e140000.pcie: pcie2 no card, disable it (RST & CLK)
[    1.122647] mt7621-pci 1e140000.pcie: PCIE1 enabled
[    1.127529] mt7621-pci 1e140000.pcie: PCI coherence region base: 0x60000000, mask/settings: 0xf0000002
[    1.136974] mt7621-pci 1e140000.pcie: PCI host bridge to bus 0000:00
[    1.143341] pci_bus 0000:00: root bus resource [io  0x1e160000-0x1e16ffff]
[    1.150211] pci_bus 0000:00: root bus resource [mem 0x60000000-0x6fffffff]
[    1.157077] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.162602] pci 0000:00:00.0: [0e8d:0801] type 01 class 0x060400
[    1.168635] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[    1.174913] pci 0000:00:00.0: reg 0x14: [mem 0x60300000-0x6030ffff]
[    1.181242] pci 0000:00:00.0: supports D1
[    1.185254] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[    1.192530] pci 0000:01:00.0: [14c3:7915] type 00 class 0x000280
[    1.198618] pci 0000:01:00.0: reg 0x10: [mem 0x00000000-0x000fffff 64bit pref]
[    1.205849] pci 0000:01:00.0: reg 0x18: [mem 0x00000000-0x00003fff 64bit pref]
[    1.213074] pci 0000:01:00.0: reg 0x20: [mem 0x00000000-0x00000fff 64bit pref]
[    1.220412] pci 0000:01:00.0: supports D1 D2
[    1.224684] pci 0000:01:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[    1.231329] pci 0000:01:00.0: 2.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s x1 link at 0000:00:00.0 (capable of 4.000 Gb/s with 5 GT/s x1 link)
[    1.246420] pci 0000:00:00.0: PCI bridge to [bus 01-ff]
[    1.251653] pci 0000:00:00.0:   bridge window [io  0x0000-0x0fff]
[    1.257740] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
[    1.264525] pci 0000:00:00.0:   bridge window [mem 0x60100000-0x602fffff pref]
[    1.271744] pci_bus 0000:01: busn_res: [bus 01-ff] end is updated to 01
[    1.278391] pci 0000:00:00.0: BAR 0: no space for [mem size 0x80000000]
[    1.284997] pci 0000:00:00.0: BAR 0: failed to assign [mem size 0x80000000]
[    1.291956] pci 0000:00:00.0: BAR 8: assigned [mem 0x60000000-0x600fffff]
[    1.298743] pci 0000:00:00.0: BAR 9: assigned [mem 0x60100000-0x602fffff pref]
[    1.305968] pci 0000:00:00.0: BAR 1: assigned [mem 0x60300000-0x6030ffff]
[    1.312754] pci 0000:00:00.0: BAR 7: assigned [io  0x1e160000-0x1e160fff]
[    1.319546] pci 0000:01:00.0: BAR 0: assigned [mem 0x60100000-0x601fffff 64bit pref]
[    1.327294] pci 0000:01:00.0: BAR 2: assigned [mem 0x60200000-0x60203fff 64bit pref]
[    1.335043] pci 0000:01:00.0: BAR 4: assigned [mem 0x60204000-0x60204fff 64bit pref]
[    1.342787] pci 0000:00:00.0: PCI bridge to [bus 01]
[    1.347752] pci 0000:00:00.0:   bridge window [io  0x1e160000-0x1e160fff]
[    1.354536] pci 0000:00:00.0:   bridge window [mem 0x60000000-0x600fffff]
[    1.361316] pci 0000:00:00.0:   bridge window [mem 0x60100000-0x602fffff pref]
[    1.369091] mt7530 mdio-bus:1f: Link is Up - 1Gbps/Full - flow control off
[    1.369396] hctosys: unable to open rtc device (rtc0)
[    1.386360] VFS: Mounted root (squashfs filesystem) readonly on device 31:5.
[    1.397785] Freeing unused kernel memory: 1268K
[    1.402380] This architecture does not have kernel memory protection.
[    1.408800] Run /sbin/init as init process
[    1.858640] init: Console is alive
[    1.862312] init: - watchdog -
[    2.464429] kmodloader: loading kernel modules from /etc/modules-boot.d/*
[    2.553529] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
[    2.562152] init: - preinit -
[    3.228541] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[    3.236886] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[    3.245382] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    3.371163] random: jshn: uninitialized urandom read (4 bytes read)
[    3.435351] random: jshn: uninitialized urandom read (4 bytes read)
[    3.473997] random: jshn: uninitialized urandom read (4 bytes read)
[    3.693371] mt7530 mdio-bus:1f lan1: configuring for phy/gmii link mode
[    3.700434] 8021q: adding VLAN 0 to HW filter on device lan1
[    7.782301] mt7530 mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[    7.789834] IPv6: ADDRCONF(NETDEV_CHANGE): lan1: link becomes ready
[    7.914083] jffs2: notice: (583) jffs2_build_xattr_subsystem: complete building xattr subsystem, 15 of xdatum (13 unchecked, 2 orphan) and 19 of xref (2 dead, 0 orphan) found.
[    7.931803] mount_root: switching to jffs2 overlay
[    7.942497] overlayfs: upper fs does not support tmpfile.
[    7.953108] urandom-seed: Seeding with /etc/urandom.seed
[    8.034055] mt7530 mdio-bus:1f lan1: Link is Down
[    8.050671] procd: - early -
[    8.053646] procd: - watchdog -
[    8.641934] procd: - watchdog -
[    8.645744] procd: - ubus -
[    8.742821] procd: - init -
[    9.341977] kmodloader: loading kernel modules from /etc/modules.d/*
[    9.361292] Loading modules backported from Linux version v5.10.42-0-g65859eca4dff
[    9.368978] Backport generated by backports.git v5.10.42-1-0-gbee5c545
[    9.395227] xt_time: kernel timezone is -0000
[    9.450396] mt7621-pci 1e140000.pcie: bus=1 slot=0 irq=24
[... cut ...]
[    9.471399] urngd: v1.0.2 started.
[    9.593086] mt7915e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20201105222230a
[    9.593086]
[    9.614170] random: crng init done
[    9.617582] random: 7 urandom warning(s) missed due to ratelimiting
[    9.921740] mt7915e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20201105222304
[    9.955518] mt7915e 0000:01:00.0: WA Firmware Version: DEV_000000, Build Time: 20201105222323
[   10.113178] mt7915e: probe of 0000:01:00.0 failed with error -22
```

deece, 2021-11-13:
Just so everything is in 1 place (thanks @hauke @nonyhaha :
```
opkg install kmod-mtd-rw
insmod mtd-rw i_want_a_brick=1
mtd unlock /dev/mtd2
dd if=/dev/mtd2ro bs=131072 skip=0 count=8 of=/tmp/factory-fixed
mtd write /tmp/factory-fixed /dev/mtd2
opkg remove kmod-mtd-rw
reboot
```

aep, 2022-03-10:
@markus289 how did you get the thing into recovery mode in the first place? Mine is answering arp and ping on 192.168.1.20 but ignoring tftp, so i guess thats the wrong mode.

markus289, 2022-03-10:
Yes, it is a different issue. As far as I can tell, the error may happen on rebooted devices. Try to unplug it "for some time" (I left it without power for the night), that helped in my case.
There [is a patch out there](https://patchwork.kernel.org/project/linux-mediatek/patch/70e27cbc652cbdb78277b9c691a3a5ba02653afb.1641540175.git.objelf@gmail.com/), but I did not try it.

Djfe, 2024-02-09:
@ptpt52 is this still an issue? else please close it :)

ptpt52, 2024-02-09:
outdated

closed 2024-02-09 by ptpt52