openwrt/mt76 #465: mt7615e: VHT160 extremely high packet loss

namidairo, 2020-10-27:
As the title says, VHT160 appears to be more or less unusable on the mt7615e, to the point that even DHCP responses don't appear to get through.

It does appear to be running at VHT160 as far as the client and ap are concerned though.

Tested on: https://github.com/openwrt/mt76/commit/b22977c2727db7fea8d487573edb3aa668d8309d

There are a couple casual mentions of this on #432 as well

Not entirely sure what troubleshooting steps I should take here, as I only have 1 readily available 160mhz client to test with.

ryderlee1110, 2020-10-27:
mt7615e does not support vht160

namidairo, 2020-10-27:
I could have sworn the single-band configurations of mt7615 had VHT160 and 80+80, based on the vendor driver users/datasheet.

Perhaps advertising support for it for non-DBDC configurations itself is the bug. 🤔  https://github.com/openwrt/mt76/blob/73301065ac320b691e074b230a05ce24b26794f3/mt7615/init.c#L383-L385

Mushoz, 2021-03-29:
I am fairly sure mt7615 does support 160mhz, right? Has this ever been resolved? I bumped into this issue when searching for 160mhz capabilities of the mt76 driver. While I am able to set up a 160mhz channel (according to the AP at least), I am only able to connect to it with a 80mhz channel Width. The client I am using is the Intel AX200, which AFAIK also supports 160mhz. Any ideas on how to get this working properly?

Mushoz, 2021-03-29:
Actually, both client and router correctly report a 160mhz channel width. However, the link speed is still only 866 Mb/s, despite it being at MSC 9 with a 2x2 configuration on the client. That should result in a 1700 Mb/s link rate, right?

ryderlee1110, 2021-03-29:
1. It is 80+80m at most.
2. need to add 160m in mac80211 RC.

Mushoz, 2021-03-29:
@ryderlee1110 Wouldn´t `iw` show if it was 80+80 though? Both the client and AP reported 160mhz. Is there anywhere where I can read up on the mac80211 status with regards to 160mhz support?

Mushoz, 2021-03-29:
According to this, mac80211 should already support 160mhz: https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=2c14710c54befc69e70fff52240875b2bbf94cd1

ryderlee1110, 2021-03-30:
I meant 'RC' ...

ryderlee1110, 2021-06-15:
@Mushoz Can you please try this https://github.com/ryderlee1110/wireless/commit/a6b7b09d576fab194bca3a58da4e7155efe33ae9 ?

namidairo, 2021-06-15:
Tried it out, still no packet flow to the client that attempt 160mhz, but all the 80mhz clients still work.

The 4-way handshake and assocation would all go through and it would try VHT MCS 9, but then quickly fall back to 20mhz and MCS 0 because no traffic would get through (even at 6mbit).

Funnily enough the same client in linux would stick to 80mhz and work(...it was faster to reboot than to patch the kernel), but the same card in Windows would attempt 160mhz and fall flat as mentioned.

I don't have a spare 7615 or another 160mhz capable client to test on the vendor driver though, so I can't rule out hardware limitation or incompatibility though...