openwrt/mt76 #466: Is there a plan to support easyMesh on MT76xx?

Sean2379607955, 2020-10-31:
There is already image from MTK support easyMesh r1 and r2.
I found openwrt support 802.11s, but not easyMesh.
Is there plan to support easyMesh on openWRT?

dangowrt, 2020-10-31:
This is unrelated to Wifi driver (I know, industry tries to suggest it's a hardware feature -- it is not), hence this is the wrong repository to ask for this.
Maybe this is what you are looking for https://gitlab.com/prpl-foundation/prplmesh/prplMesh

closed 2020-10-31 by dangowrt

Sean2379607955, 2020-11-02:
Thanks for your comment.
But as I known, easyMesh need support both application and driver.
Like WPS client and wifi bridge repeater.

dangowrt, 2020-11-02:
All needed driver features (4-addr mode aka. WDS; WPS for Push-Button authentication) have been available for all mac80211-based Open Source drivers for more than a decade (literally!). multiap support for hostapd/wpa_supplicant has been added to OpenWrt in https://github.com/openwrt/openwrt/commit/2e0f41e73a93e8fa8b74f053e638f05247ee9113 .

xavier83, 2020-11-03:
Is [easyMesh](https://www.wi-fi.org/discover-wi-fi/wi-fi-easymesh) an 802.11 standard and an alternative to proprietary mesh solutions like _OneMesh_ and _AiMesh_? Thanks @Sean2379607955 for the mention.